johnfanv2/LenovoLegionLinux #12: Kernel patch?

GloriousEggroll, 2023-01-23:
Would you be able to provide a kernel patch as an alternative to the kernel module?

I would feel much more comfortable patching this into the Nobara kernel and providing the GUI tool as a package rather than relying on users to try install the kernel module.

johnfanv2, 2023-01-23:
Great idea! I already planned integrating it into the normal kernel source. Do you just need the output of `diff -rupN` between the main line kernel and the patched kernel? Which version of the kernel do you suggest (should not matter)?

I am also experimenting creating a Python package (see source code) for the GUI tool. Maybe this could be a inspiration for creating your package. Currently, I am not sure how to create an entry in the start menu that works on as many distributions as possible.

GloriousEggroll, 2023-01-23:
> Great idea! I already planned integrating it into the normal kernel source. Do you just need the output of `diff -rupN` between the main line kernel and the patched kernel? Which version of the [...]

`diff -rupN` would be perfectly fine but for the sake of adding proper credit and authors here's how I would do it:

download the kernel tarball (faster than git clone)
extract it somewhere

cd kernel-folder
git init
git add --all
git commit -m "initial commit"

from here apply your changes/patch (if you have an existing patch or diff just `patch -Np1 < /path/to/file.patch`

Then 
git add --all
git commit -m "some information about the patch"
git format-patch HEAD~1

That should generate a proper patch with author,subject, etc.

Fedora is currently on kernel 6.1 so that imo would be a good starting point

johnfanv2, 2023-01-24:
Great! I will have a lock at it and will try to generate a patch.

johnfanv2, 2023-01-27:
Could you try this patch? Its autogenerated in Gihtub Actions.

https://github.com/johnfanv2/LenovoLegionLinux/releases/tag/main

johnfanv2, 2023-02-14:
I will soon create another updated patch.

GloriousEggroll, 2023-02-14:
sorry i dropped the ball on a response here, the patch applies and compiles fine on 6.1.9 but i havent heard any feedback from any lenovo users on it

pholostan, 2023-03-10:
I have a Lenovo Legion 5 15ACH6H with an AMD APU and discreet Nvidia GPU. Nobara doesn't work, but I managed to log into the laptop using the kernel boot option systemd.unit=multi-user.target and it looks like all devices are there, running the python scripts seems to work at least. I have  legion_hwmon-isa-0000, /sys/kernel/debug/legion/fancurve etc.

johnfanv2, 2023-03-10:
I have the exact same model (15ACH6H). Could you describe your problems that might relate to this software and how you fixed it? Does now everything works?

pholostan, 2023-03-11:
Everything seems to be working, at least everything I've tried has worked so far with the patched Nobara kernel 6.1.14-201.fsync.fc37.x86_64. Reading and setting fancurve data seems to be working, hwmon seems to be there too. Battery conversation can be set and unset. I have the latest BIOS on my laptop, GKCN59WW.

XenHat, 2023-03-14:
> Could you try this patch? Its autogenerated in Github Actions.
> 
> https://github.com/johnfanv2/LenovoLegionLinux/releases/tag/main

```sh
❯ sudo dmidecode --string=chassis-version      
Legion 5 Pro 16ACH6H

❯ uname -a
Linux LNV22LT001 6.1.19-273-tkg-cfs #1 SMP PREEMPT_DYNAMIC TKG Tue, 14 Mar 2023 18:33:32 +0000 x86_64 GNU/Linux

❯ ls ~/src/linux-tkg/linux61-tkg-userpatches
0001-Add-legion-laptop-v0.1.mypatch

❯ lsmod | grep legion                       
legion_laptop          77824  0
wmi                    45056  4 video,wmi_bmof,legion_laptop,ideapad_laptop
platform_profile       16384  3 amd_pmf,legion_laptop,ideapad_laptop

❯ sudo dmesg | grep legion_laptop
[    3.540276] legion_laptop 0.1 starts loading
[    3.540566] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
[    3.540894] legion PNP0C09:00: legion_laptop loaded for this device

❯ sensors                                   
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.46 V  
SVI2_SoC:    694.00 mV 
Tdie:         +45.4°C  (high = +95.0°C)
Tctl:         +45.4°C  
SVI2_P_Core:  17.12 W  
SVI2_P_SoC:    2.04 W  
SVI2_C_Core:  11.86 A  
SVI2_C_SoC:    2.94 A  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +40.0°C  

nvme-pci-0500
Adapter: PCI adapter
Composite:    +34.9°C  (low  =  -5.2°C, high = +83.8°C)
                       (crit = +87.8°C)

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           2059 RPM
Fan 2:           2147 RPM
CPU Temperature:  +46.0°C  
GPU Temperature:  +36.0°C  
IC Temperature:   +39.0°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +36.9°C  (low  =  -5.2°C, high = +83.8°C)
                       (crit = +87.8°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.04 V  

❯ sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop version: 0.1
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
minifancurve on cool: true
lock fan controller: false
enable maximumfanspeed: false
enable maximumfanspeed status: 0
fan curve current point id: 1
fan curve points size: 9
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 48	 0	 60	 0	 41
1700	 1900	 2	 2	 45	 54	 56	 60	 39	 44
1900	 2000	 2	 2	 51	 58	 56	 60	 42	 50
2200	 2100	 2	 2	 55	 62	 56	 60	 46	 127
2300	 2400	 2	 2	 59	 71	 56	 60	 127	 127
2600	 2700	 2	 2	 68	 76	 56	 65	 127	 127
2900	 3000	 2	 2	 72	 81	 61	 69	 127	 127
3500	 3500	 2	 2	 78	 90	 66	 77	 127	 127
3800	 3800	 2	 2	 87	 127	 72	 127	 127	 127
=====================

```

johnfanv2, 2023-04-02:
@GloriousEggroll I created a new patch that you can use for your new version. It includes some fixes and more supported models.

MrDuartePT, 2023-07-08:
I think we need to create a github ci that make the patch automatic.
I will try get a look on that.

johnfanv2, 2023-07-11:
@MrDuartePT The github action already exists. Whenever you push a tagged commit it will create a kernel patch and create release with the kernel patch (right side) on github.

MrDuartePT, 2023-08-29:
@GloriousEggroll I dont know if you are packaging the python rpm in nobara if not give a look at this:
https://github.com/MrDuartePT/LLL-pkg-repo
The build is failing for now because im tweaking the script to use the repo tagged version, but the rpm posted are from this commit: 644ac89748bc2771ba0e98c2083feaeb6d416c70

closed 2026-09-11 by github-actions[bot]