johnfanv2/LenovoLegionLinux #15: legion_hwmon-isa-0000 CPU temp 0%

BaritskiyIllya, 2023-02-03:
Hi, First of all, I want to thank you for this wonderful tool
After installation everything works as expected except cpu temperature
FAN speed react/trigger only on GPU temp or IC Temp

OS Manjaro
 System: LENOVO 82JF [Legion 5 Pro 16ITH6] (laptop)
 Board: LENOVO LNVNB161216
 Kernel: 5.15.89-1-MANJARO x86_64
 Processor: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz (6/141/1)
**************************************************************
output of $ sensors command

<details>
<summary>Details</summary>
coretemp-isa-0000
Adapter: ISA adapter

Package id 0:  +76.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +60.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +58.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +76.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +61.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +63.0°C  (high = +100.0°C, crit = +100.0°C)
Core 6:        +59.0°C  (high = +100.0°C, crit = +100.0°C)
Core 7:        +60.0°C  (high = +100.0°C, crit = +100.0°C)

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           2081 RPM
Fan 2:           2081 RPM
CPU Temperature:   +0.0°C  
GPU Temperature:  +59.0°C  
IC Temperature:   +55.0°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +52.9°C  (low  =  -0.1°C, high = +82.8°C)
                       (crit = +83.8°C)
Sensor 1:     +45.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +46.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +57.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          17.40 V  

nvme-pci-5700
Adapter: PCI adapter
Composite:    +41.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +43.9°C  (low  = -273.1°C, high = +65261.8°C)
</details>

johnfanv2, 2023-02-03:
This sounds interesting. Could you try the follwing? You can load the CPU with `stress` (see https://www.tecmint.com/linux-cpu-load-stress-test-with-stress-ng-tool/) in the following tests.

- does the CPU temp influence the fan speed when not using the module (after a restart to be sure)? 
- when loading the module (after a restart) but not setting the fan curve, does the CPU temp influence the fan speed? Please output the info with `sudo cat /sys/kernel/debug/legion/fancurve` before changing/setting anything. Are the measured fan speed consistent with an estimated CPU temperature (e.g. from other sensors). Please note that changing from fan curve point to the next point takes time (acceleration. time/deceleration time).
- does the CPU temp influence the fan speed in Windows?

Note: The IC temp (usually heat spreader or DCDCs) does slowly increase with CPU temp, so there is a indirect influence. I am talking about direct influence.

There could be the following reasons:
- CPU temp sensor is not working (electric/mechanical defect; so check with Windows)
- CPU temp sensor somehow deactivated due to energy saving. If the GPU is turned off, it also displays 0 degree
- CPU temp sensor is saved in another place in your firmware in the embedded controller (= hardware chip); But the fan control is implemented in the embedded controller and should work on its own without any driver. So it should react to CPU speed even if we don't read it correctly.
- on some models the "mini fancurve" is enabled after some time (you can disable that) and it changes automatically to another fan curve, which does not consider CPU temperature. But if you read back the fan curve you will see the change.  
- firmware problem: you could try to update/reflash the BIOS

BaritskiyIllya, 2023-02-05:
Yesterday I booted under Windows and updated the bios H1CN51WW -> H1CN52WW and today booted into Linux and found that the temperature sensor is working...

closed 2023-02-05 by BaritskiyIllya