johnfanv2/LenovoLegionLinux #16: Trying to get a legion Y530-15ICH to work

mcodyre, 2023-02-07:
Great project!  I'm trying to get my older gaming laptop working to use as a development machine.  After setting it up I've found the fans to spin way too loudly so I poked around and happily found your project. Thanks for putting this together!

As I'm familiar with all things Linux I figured I'd try to get this working on my machine.
Following your instructions I got to the point where I needed to force the module to load due to this allowlist error:

`[ 1412.449348] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81FV; DMI_BIOS_VERSION:8JCN56WW
[ 1412.449389] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
[ 1412.449390] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81FV; DMI_BIOS_VERSION:8JCN56WW
[ 1412.449393] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[ 1412.449394] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[ 1412.449395] legion PNP0C09:00: legion_laptop not loaded for this device
[ 1412.449428] legion: probe of PNP0C09:00 failed with error -12

I then had to replace the EC chip ID from 0x8227 to  0x8226 for my attempt at getting this working
`
sed -i 's|0x8227|0x8226|g' legion-laptop.c
`
dmesg
`
[ 1827.143935] legion_laptop 0.1 starts loading
[ 1827.143937] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81FV; DMI_BIOS_VERSION:8JCN56WW
[ 1827.143982] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
[ 1827.143983] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81FV; DMI_BIOS_VERSION:8JCN56WW
[ 1827.143985] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[ 1827.143987] legion PNP0C09:00: legion_laptop is forced to load.
[ 1827.143987] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[ 1827.143988] legion PNP0C09:00: Using configuration for system: GKCN
[ 1827.144001] Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0xc400 (in EC) to virtual 0x000000006aaf2a43
**[ 1827.144381] legion PNP0C09:00: Read embedded controller ID 0x8226**
[ 1827.144383] legion PNP0C09:00: Creating debugfs inteface
[ 1827.144389] Creating sysfs inteface
[ 1827.144392] Creating hwmon interface
[ 1827.144457] Creating platform profile support
[ 1827.144460] Init WMI driver support
[ 1827.144482] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[ 1827.144500] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[ 1827.144522] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[ 1827.144537] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[ 1827.144549] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[ 1827.144563] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[ 1827.144582] legion PNP0C09:00: legion_laptop loaded for this device
**[ 1881.094692] Unexpected value in MINIFANCURVE register:0**
`

Module successfully loads although "fan curve points" is 0
`
sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 8226
EC Chip Version: 1a2
legion_laptop version: 0.1
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 1
minifancurve feature enabled: 1
minifancurve on cool: error
lock fan controller: false
enable maximumfanspeed: false
enable maximumfanspeed status: 0
fan curve current point id: 0
**fan curve points size: 0**
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
`

I've followed the instructions from here https://github.com/johnfanv2/LenovoLegionLinux/issues/13 to disassemble the acpi tables and attached them to this email if that helps.

At this point I'm kind of stuck.  Would greatly appreciate some direction to get this working.

Cheers!

**Some system info**

`
00:00.0 Host bridge: Intel Corporation 8th Gen Core 4-core Processor Host Bridge/DRAM Registers [Coffee Lake H] (rev 07)
00:01.0 PCI bridge: Intel Corporation 6th-10th Gen Core Processor PCIe Controller (x16) (rev 07)
00:02.0 VGA compatible controller: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 07)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:12.0 Signal processing controller: Intel Corporation Cannon Lake PCH Thermal Controller (rev 10)
00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10)
00:14.2 RAM memory: Intel Corporation Cannon Lake PCH Shared SRAM (rev 10)
00:14.3 Network controller: Intel Corporation Cannon Lake PCH CNVi WiFi (rev 10)
00:15.0 Serial bus controller: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #0 (rev 10)
00:15.1 Serial bus controller: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #1 (rev 10)
00:16.0 Communication controller: Intel Corporation Cannon Lake PCH HECI Controller (rev 10)
00:17.0 SATA controller: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller (rev 10)
00:1d.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #9 (rev f0)
00:1d.5 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #14 (rev f0)
00:1e.0 Communication controller: Intel Corporation Cannon Lake PCH Serial IO UART Host Controller (rev 10)
00:1f.0 ISA bridge: Intel Corporation HM470 Chipset LPC/eSPI Controller (rev 10)
00:1f.3 Audio device: Intel Corporation Cannon Lake PCH cAVS (rev 10)
00:1f.4 SMBus: Intel Corporation Cannon Lake PCH SMBus Controller (rev 10)
00:1f.5 Serial bus controller: Intel Corporation Cannon Lake PCH SPI Controller (rev 10)
01:00.0 VGA compatible controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP107GL High Definition Audio Controller (rev a1)
06:00.0 Non-Volatile memory controller: Toshiba Corporation BG3 NVMe SSD Controller (rev 01)
07:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
`

johnfanv2, 2023-02-08:
Thanks a lot for the effort! This looks good.

The Y530 seems to have a sligtly different embedded controller. The ACPI tables you have uploaded seem also different after a quick glance. I just had a very quick lock, but could not identify the memory adress where the RAM of the embedded controller (EC) is memory mapped. However, in most model configs this is not used. Instead the EC is accessed by writing to IO ports.

Acessing via memory adress: usually totally safe even if there is a completely different chip/hardware used
Acessing via IO ports: you send commands; if it is the expected chip, this will just read the RAM (when reading values); if it is another chip, it could do anything.

However I saw that the chip ID and version is reported as 8226/1a2 (that were already read via the IO port method after you have modified the code). That looks very promising because it is very similar to the already implemented embedded controller. Therefore, I suggest using the IO port method and not finding the memory address in the ACPI tables.

Now, you update the adresses in the EC memory to adapt it to your version of the embedded controller.

run

cat /sys/kernel/debug/legion/ecmemory | hexdump -C

before and after you change the power mode with Fn+Q and post it here. Try to find which values change and which could represent the fan curve.

Maybe the fan speed or temperatures are already correct? What is outputted when you run lmsensor?

Mit freundlichen Grüßen/ Kind regards

Tobias Friedrich

Mobil:    +49 (0) 177 / 87 3588 5

E-mail:  ***@***.******@***.***>

ELEFANT RACING E.V.
Bayerisches Zentrum für Batterietechnik
Universitätsstr. 30
95447 Bayreuth, Germany

Phone:  +49 (0) 921 / 55 7284

Fax:      +49 (0) 921 / 55 7195

www.elefantracing.de<http://www.elefantracing.de/>

Follow us:

[Facebook]<http://www.facebook.com/elefantracing.bayreuth>[Twitter]<https://twitter.com/elefantracing>[Instagram]<https://www.instagram.com/elefant_racing_bayreuth>[Linkedin]<https://www.linkedin.com/company/elefant-racing-bayreuth-e-v-/>

Creating an electrifying autonomous Startup – every year.

CONFIDENTIAL AND PRIVILEGED

This email message (including any attachments) contains information that may be confidential, be protected by the attorney-client or other applicable privileges, or constitute non-public information.
It is intended to be conveyed only to the designated recipient(s). Any unauthorized review, use, disclosure, or distribution is prohibited. If you are not an intended recipient of this message, please notify the sender by replying to this message and then deleteit from your system immediately.
Thank you.

johnfanv2, 2023-02-08:
Thanks a lot for the effort! This looks good.

The Y530 seems to have a sligtly different embedded controller. The ACPI tables you have uploaded seem also different after a quick glance. I just had a very quick lock, but could not identify the memory adress where the RAM of the embedded controller (EC) is memory mapped. However, in most model configs this is not used. Instead the EC is accessed by writing to IO ports.

Acessing via memory adress: usually totally safe even if there is a completely different chip/hardware used
Acessing via IO ports: you send commands; if it is the expected chip, this will just read the RAM (when reading values); if it is another chip, it could do anything.

However I saw that the chip ID and version is reported as 8226/1a2 (that were already read via the IO port method after you have modified the code). That looks very promising because it is very similar to the already implemented embedded controller. Therefore, I suggest using the IO port method and not finding the memory address in the ACPI tables.

Now, you update the adresses in the EC memory to adapt it to your version of the embedded controller.

run

cat /sys/kernel/debug/legion/ecmemory | hexdump -C

before and after you change the power mode with Fn+Q and post it here. Try to find which values change and which could represent the fan curve.

Maybe the fan speed or temperatures are already correct? What is outputted when you run lmsensor?

Leeo97one, 2023-02-12:
Hi, I think I have the same issue with my Legion Y540-15IRH-PG0.

So I also replaced the value `0x8227` by `0x8226` in `model_v0.embedded_controller_id` before rebuilding and reloading the module (with `forcereloadmodulereadonly`):
```
[ 7060.397550] legion_laptop 0.1 starts loading
[ 7060.397552] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81SY; DMI_BIOS_VERSION:BHCN45WW
[ 7060.397592] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
[ 7060.397593] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81SY; DMI_BIOS_VERSION:BHCN45WW
[ 7060.397595] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[ 7060.397597] legion PNP0C09:00: legion_laptop is forced to load.
[ 7060.397598] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[ 7060.397598] legion PNP0C09:00: Using configuration for system: GKCN
[ 7060.397610] Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0xc400 (in EC) to virtual 0x000000008c12e6b9
[ 7060.397964] legion PNP0C09:00: Read embedded controller ID 0x8226
[ 7060.397965] legion PNP0C09:00: Creating debugfs inteface
[ 7060.397974] Creating sysfs inteface
[ 7060.397977] Creating hwmon interface
[ 7060.398047] Creating platform profile support
[ 7060.398050] Init WMI driver support
[ 7060.398075] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[ 7060.398095] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[ 7060.398106] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[ 7060.398116] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[ 7060.398125] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[ 7060.398135] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[ 7060.398146] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[ 7060.398166] legion PNP0C09:00: legion_laptop loaded for this device
```

`/sys/kernel/debug/legion/fancurve`:
```
EC Chip ID: 8226
EC Chip Version: 1a2
legion_laptop version: 0.1
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 1
minifancurve feature enabled: 1
minifancurve on cool: error
lock fan controller: false
enable maximumfanspeed: false
enable maximumfanspeed status: 0
fan curve current point id: 0
fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
```

Then I inspected `/sys/kernel/debug/legion/ecmemory` while switching the power mode with Fn+Q and noticed that the value at 0x1d cycled between 0 and 2 so I guess I have to set the `ec_register_offsets_v0.ALT_POWERMODE` value to `0xc41d`?
And I'm also not sure what to do with the other values.

Here are 3 dumps (I pressed Fn+Q between each):
```
00000000  00 00 00 00 00 00 00 00  00 00 10 00 00 2b 1e 00  |.............+..|
00000010  00 00 80 00 05 00 75 02  00 01 05 00 00 00 00 00  |......u.........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000040  00 00 00 00 00 00 00 00  00 00 00 00 10 4f 00 00  |.............O..|
00000050  00 00 00 1c 00 00 30 00  00 00 00 00 00 00 00 00  |......0.........|
00000060  00 80 02 03 00 00 00 08  2c b1 04 13 08 00 20 01  |........,..... .|
00000070  00 02 20 01 00 00 00 00  00 00 00 00 00 00 00 00  |.. .............|
00000080  00 00 00 00 08 00 00 00  c0 00 04 00 00 00 00 53  |...............S|
00000090  4d 50 00 31 31 00 00 00  4c 31 37 4d 33 50 47 33  |MP.11...L17M3PG3|
000000a0  00 01 00 86 00 00 00 61  64 00 00 00 64 25 00 00  |.......ad...d%..|
000000b0  30 2b 2c 2a 00 00 36 00  02 11 00 00 00 00 01 00  |0+,*..6.........|
000000c0  70 00 f2 09 38 1d a8 2d  88 2c 82 14 92 10 3c 00  |p...8..-.,....<.|
000000d0  00 00 00 00 00 00 00 1b  7e 07 00 00 00 00 00 00  |........~.......|
000000e0  00 00 1a 00 00 00 00 00  00 00 00 00 00 00 3f 02  |..............?.|
000000f0  01 00 00 40 e6 4e 64 32  00 00 2e 00 00 00 00 00  |...@.Nd2........|
00000100  82 00 00 00 00 00 07 07  67 66 67 66 67 66 13 66  |........gfgfgf.f|
00000110  45 00 ff 03 ff 03 00 00  00 00 00 00 00 00 00 00  |E...............|
[... 88 lines cut ...]
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000160  00 00 00 00 00 00 00 38  50 3c 00 00 00 00 00 00  |.......8P<......|
00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001b0  00 00 00 00 00 00 00 00  32 00 00 00 00 00 00 00  |........2.......|
000001c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001f0  00 00 00 00 f2 09 00 00  00 00 00 00 00 00 00 00  |................|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 02 00 01  |................|
00000240  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000250  35 35 35 00 00 00 00 00  00 00 00 00 00 00 00 00  |555.............|
00000260  00 1e 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  51 f3 64 00 00 00 00 00  00 00 00 00 00 00 00 00  |Q.d.............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002e0  00 00 00 00 00 00 00 64  00 00 00 00 00 00 00 00  |.......d........|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
```

johnfanv2, 2023-02-14:
"0x1d cycled between 0, 1 and 2 so I guess I have to set the ec_register_offsets_v0.ALT_POWERMODE value to 0xc41d?
And I'm also not sure what to do with the other values."

Yes, that would be true. You can try it.
However, I do not see any values that could be the fan speeds or temperatures. How many fan points do you observerse in practice?

MrDuartePT, 2023-10-20:
> Hi, I think I have the same issue with my Legion Y540-15IRH-PG0.
> 
> So I also replaced the value `0x8227` by `0x8226` in `model_v0.embedded_controller_id` before rebuilding and reloading the module (with `forcereloadmodulereadonly`):
> 
> ```
> [ 7060.397550] legion_laptop 0.1 starts loading
> [ 7060.397552] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81SY; DMI_BIOS_VERSION:BHCN45WW
> [ 7060.397592] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
> [ 7060.397593] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81SY; DMI_BIOS_VERSION:BHCN45WW
> [ 7060.397595] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
> [ 7060.397597] legion PNP0C09:00: legion_laptop is forced to load.
> [ 7060.397598] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
> [ 7060.397598] legion PNP0C09:00: Using configuration for system: GKCN
> [ 7060.397610] Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0xc400 (in EC) to virtual 0x000000008c12e6b9
> [ 7060.397964] legion PNP0C09:00: Read embedded controller ID 0x8226
> [ 7060.397965] legion PNP0C09:00: Creating debugfs inteface
> [ 7060.397974] Creating sysfs inteface
> [ 7060.397977] Creating hwmon interface
> [ 7060.398047] Creating platform profile support
> [ 7060.398050] Init WMI driver support
> [ 7060.398075] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
> [ 7060.398095] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
> [ 7060.398106] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
> [ 7060.398116] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
> [ 7060.398125] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
> [ 7060.398135] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
> [ 7060.398146] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
> [ 7060.398166] legion PNP0C09:00: legion_laptop loaded for this device
> ```
> 
> `/sys/kernel/debug/legion/fancurve`:
> 
> ```
> EC Chip ID: 8226
> EC Chip Version: 1a2
> legion_laptop version: 0.1
> legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
> legion_laptop ec_readonly: 1
> minifancurve feature enabled: 1
> minifancurve on cool: error
> lock fan controller: false
> enable maximumfanspeed: false
> enable maximumfanspeed status: 0
> fan curve current point id: 0
> fan curve points size: 0
> Current fan curve in hardware (embedded controller):
> rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
> ```
> 
> Then I inspected `/sys/kernel/debug/legion/ecmemory` while switching the power mode with Fn+Q and noticed that the value at 0x1d cycled between 0, 1 and 2 so I guess I have to set the `ec_register_offsets_v0.ALT_POWERMODE` value to `0xc41d`? And I'm also not sure what to do with the other values.
> 
> Here are 3 dumps (I pressed Fn+Q between each):
> 
> ```
> 00000000  00 00 00 00 00 00 00 00  00 00 10 00 00 2b 1e 00  |.............+..|
> 00000010  00 00 80 00 05 00 75 02  00 01 05 00 00 00 00 00  |......u.........|
> 00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 00000040  00 00 00 00 00 00 00 00  00 00 00 00 10 4f 00 00  |.............O..|
> 00000050  00 00 00 1c 00 00 30 00  00 00 00 00 00 00 00 00  |......0.........|
> 00000060  00 80 02 03 00 00 00 08  2c b1 04 13 08 00 20 01  |........,..... .|
> 00000070  00 02 20 01 00 00 00 00  00 00 00 00 00 00 00 00  |.. .............|
> 00000080  00 00 00 00 08 00 00 00  c0 00 04 00 00 00 00 53  |...............S|
> 00000090  4d 50 00 31 31 00 00 00  4c 31 37 4d 33 50 47 33  |MP.11...L17M3PG3|
> 000000a0  00 01 00 86 00 00 00 61  64 00 00 00 64 25 00 00  |.......ad...d%..|
> 000000b0  30 2b 2c 2a 00 00 36 00  02 11 00 00 00 00 01 00  |0+,*..6.........|
> 000000c0  70 00 f2 09 38 1d a8 2d  88 2c 82 14 92 10 3c 00  |p...8..-.,....<.|
> 000000d0  00 00 00 00 00 00 00 1b  7e 07 00 00 00 00 00 00  |........~.......|
> 000000e0  00 00 1a 00 00 00 00 00  00 00 00 00 00 00 3f 02  |..............?.|
> 000000f0  01 00 00 40 e6 4e 64 32  00 00 2e 00 00 00 00 00  |...@.Nd2........|
[... 94 lines cut ...]
> 00000160  00 00 00 00 00 00 00 38  50 3c 00 00 00 00 00 00  |.......8P<......|
> 00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 000001b0  00 00 00 00 00 00 00 00  32 00 00 00 00 00 00 00  |........2.......|
> 000001c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 000001f0  00 00 00 00 f2 09 00 00  00 00 00 00 00 00 00 00  |................|
> 00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
> 00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 02 00 01  |................|
> 00000240  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 00000250  35 35 35 00 00 00 00 00  00 00 00 00 00 00 00 00  |555.............|
> 00000260  00 1e 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 00000280  51 f3 64 00 00 00 00 00  00 00 00 00 00 00 00 00  |Q.d.............|
> 00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
> 000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 000002e0  00 00 00 00 00 00 00 64  00 00 00 00 00 00 00 00  |.......d........|
> 000002f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> ```

If you can also give me the output of ```sudo dmidecode -t bios``` to add your legion to the driver

stacode123, 2023-11-26:
hey is there plans to fix the problem?

BetThical, 2024-03-13:
Is it going to be fixed?

Cherkah, 2024-05-06:
same question as @BetThical

RockNRollGeek, 2024-08-12:
+1 here with a y540

stacode123, 2024-10-30:
Yes

oliver-joos, 2025-01-04:
+1 with my Lenovo Legion Y540-15IRH

TerminusDeus, 2025-03-25:
+1, could provide some help in investigation if needed regarding Y540.

zerosign, 2025-04-04:
Is there any guides on getting registers correctly for other notebooks, maybe we could bruteforce it for lenovo y530

I use it for homelabs for this old notebook but found the temps are quite troublesome for idle (around 40-50c). 

If there is guide, maybe (not a promise), I can help to debug.

MrDuartePT, 2025-04-10:
@zerosign @TerminusDeus you can try follow this and find what values change [doc/REVERSE_ENGINEER.md](https://github.com/johnfanv2/LenovoLegionLinux/blob/main/doc/REVERSE_ENGINEER.md)

Also @TerminusDeus the project you send say you can enable the max fan speed, I just what to know if the toggle in LLL gui works.

TerminusDeus, 2025-04-10:
@MrDuartePT I managed to work it at least on Fedora, CachyOS, Linux mint.

closed 2026-09-11 by github-actions[bot]