johnfanv2/LenovoLegionLinux #17: Nice work!!!

Hong-gblbcy, 2023-02-13:
this work is excellent!!!
it works good on my computer.
below is computer configuration：
Lenovo Legion R9000K2021H
AMD® Ryzen 9 5900hx with radeon graphics × 16
NVIDIA Corporation GA104M [GeForce RTX 3080 MobileNVIDIA-SMI 525.85.05
Ubuntu 22.04.1 LTS 64-bit

Hong-gblbcy, 2023-02-13:
Bios 59ww

johnfanv2, 2023-02-14:
Thanks, a lot. Can you give the output of dmidecode to identify the model and BIOS better?

Hong-gblbcy, 2023-02-27:
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: GKCN59WW
	Release Date: 11/21/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.59
	Firmware Revision: 1.59

Is this okay？

closed 2023-02-27 by johnfanv2