johnfanv2/LenovoLegionLinux #18: Enable kernel module on Legion 5 15ACH6A

dottokuya, 2023-02-17:
Hi, I just found out about this project from the ArchWiki. Great work!

A quick test on my Legion 5 15ACH6A (the all-AMD variant of 15ACH6) using `sudo make forcereloadmodule` showed that almost everything worked well - I skipped the custom fan curve test though, as I don't want to mess around too much with fan curve; but theoretically it should work as well. I think this mean the kernel module is safe to enable on most 15ACH6A variants.

Output from `dmidecode` - I omitted some irrelevant infos, please let me know if you need full output: 
```
dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
Table at 0xCB707000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: G9CN30WW
        Release Date: 09/06/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.30
        Firmware Revision: 1.30

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82NW
        Version: Legion 5 15ACH6A
        Serial Number: PF38FE75
        UUID: 6c487ae0-00bc-11ec-80f0-902e16ceaceb
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82NW_BU_idea_FM_Legion 5 15ACH6A
        Family: Legion 5 15ACH6A

Handle 0x0002, DMI type 2, 16 bytes
Base Board Information
        Manufacturer: LENOVO
        Product Name: LNVNB161216
        Version: SDK0Q55749 WIN
        Serial Number: PF38FE75
        Asset Tag: NO Asset Tag
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Base Board Chassis Location
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0034, DMI type 133, 5 bytes
OEM-specific Type
        Header and Data:
                85 05 34 00 01
        Strings:
                KHOIHGIUCCHHII

Handle 0x0035, DMI type 200, 16 bytes
OEM-specific Type
        Header and Data:
                C8 10 35 00 01 02 01 27 01 14 01 53 B0 80 01 05
        Strings:
                IdeaPad
                82NW003CVN

Handle 0x0036, DMI type 248, 18 bytes
OEM-specific Type
        Header and Data:
                F8 12 36 00 AC AC AA 17 00 00 00 00 00 00 00 00
                00 00

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table
```

johnfanv2, 2023-02-17:
Great. Thanks a lot for the message! I will add the `G9CN` to the list of compatible models, so that it loads  without forcing it.

Can you give me the output of `sudo cat /sys/kernel/debug/legion/fancurve` so I can make sure it seems to work?

dottokuya, 2023-02-18:
Sure! This output was without changing fan curve with hwmon:
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop version: 0.1
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
minifancurve feature enabled: 1
minifancurve on cool: true
lock fan controller: false
enable maximumfanspeed: false
enable maximumfanspeed status: 0
fan curve current point id: 0
fan curve points size: 4
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       127     0       127     0       37
1600     1600    2       2       127     127     127     127     33      41
1800     1800    2       2       127     127     127     127     38      46
2100     2100    2       2       127     127     127     127     41      127
```

johnfanv2, 2023-02-19:
I added the model/BIOS to the list of compatible models. It now should now work without "forcing" it. If you install the module permanently according to README it should load automatically when booting.

FYI: I saw that the fan curve has only a few entries and the CPU/GPU limits are either 0 or 127 degree Celsius, so it does not take CPU/GPU temp into account and only uses the "IC" temperature (headspreader or something like that). Maybe it is just in the "mini fan curve" mode. It sometimes automatically switches to it. You can disable that according to README if you have some problems.

closed 2023-02-19 by dottokuya