johnfanv2/LenovoLegionLinux #19: Touchpad enabled option doesn't untick when it is disabled

aadish151, 2023-02-19:
Hi,

I'm using this utility with Legion 5 Pro 16ACH6H (AMD 5800H + Nvidia RTX 3070). Everything works as expected. Thanks for this awesome utility! I found this visual bug where the touchpad enabled option doesn't untick when it is disabled. Functionally it works as expected by turning on/off the touchpad.

Screencast of the issue:
[Screencast from 02-19-2023 12:54:47 AM.webm](https://user-images.githubusercontent.com/43163551/219933884-b0079b21-e6a4-4d09-9825-afdf6b8472ae.webm)

johnfanv2, 2023-02-20:
Thanks a lot for testing and the feedback.  I added your model to the list of compatible models. I really appreciate your thorough testing! 

It seems like the effect when setting the touchpad is not immediate after setting it. So the value was read back to early. I hope I have fixed it in the new version that I have pushed. I hope that you can test it.

aadish151, 2023-02-20:
Thanks for the quick fix! Yes, now the touchpad enabled option unticks as expected. Keep up the great work 👍

closed 2023-02-20 by johnfanv2