johnfanv2/LenovoLegionLinux #21: Lenovo Legion 5/Y7000P 2019 support?

KagaJiankui, 2023-03-01:
I'm using a legion 5/lenovo y7000p 2019 running Ubuntu 22.04LTS system. The EC chip id of the y7000p 2019 laptop differs from later model so it is not compatible. Is it possible for the LLL to support the Y7000P 2019 laptop?

dmesg output relating to the EC:
``` log
[ 3071.520260] legion PNP0C09:00: legion_laptop is forced to load.
[ 3071.520263] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[ 3071.520267] legion PNP0C09:00: Using configuration for system: GKCN
[ 3071.520305] Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0xc400 (in EC) to virtual 0x0000000084b647aa
[ 3071.520668] legion PNP0C09:00: Read embedded controller ID 0x8226
[ 3071.520674] legion PNP0C09:00: Expected EC chip id 0x8227 but read 0x8226
[ 3071.520680] Unloading legion ecram
[ 3071.520684] Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0xc400 (in EC) at virtual 0x0000000084b647aa
```

johnfanv2, 2023-03-01:
Please give the output of the command

sudo dmidecoe

For testing you can replace 0x8227 by 0x8226 in the file legion-laptop.c (3 occurences) and compile/load again.

MrDuartePT, 2023-10-19:
@KagaJiankui pls give the dmidecode to be possible to add support

KagaJiankui, 2023-10-20:
`sudo dmidecode` output:

```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.
Table at 0x9DC4D000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: LENOVO
        Version: BHCN42WW
        Release Date: 05/21/2021
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 10 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.42
        Firmware Revision: 1.42

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81Q5
        Version: Legion Y7000P 2019
        Serial Number: PF14M0R8
        UUID: 34c375dd-be65-11e9-80d8-98fa9b935b44
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81Q5_BU_idea_FM_Legion Y7000P 2019
        Family: Legion Y7000P 2019

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: LENOVO
        Product Name: LNVNB161216
        Version: SDK0L77769 WIN
        Serial Number: PF14M0R8
        Asset Tag: NO Asset Tag
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Type2 - Board Chassis Location
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
        Manufacturer: LENOVO
        Type: Notebook
        Lock: Not Present
[... 726 lines cut ...]
                EC FW Version
                GOP Version
                Royal Park Version
                BP1.4.3.0_RP03
                Platform Version
                Client Silicon Version

Handle 0x0044, DMI type 222, 14 bytes
OEM-specific Type
        Header and Data:
                DE 0E 44 00 01 99 00 03 10 01 20 02 30 03
        Strings:
                Memory Init Complete
                End of DXE Phase
                BIOS Boot Complete

Handle 0x0045, DMI type 248, 18 bytes
OEM-specific Type
        Header and Data:
                F8 12 45 00 AC AC AA 17 00 00 00 00 00 00 00 00
                00 00

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table
```

MrDuartePT, 2023-10-20:
@johnfanv2 lenovo is giving diferente ec id i gonna disable the check for now but will need to see how to add multiple ec id, or just forget the check all together

MrDuartePT, 2023-10-20:
@KagaJiankui now that im notice you device is already supported
see this line:  https://github.com/johnfanv2/LenovoLegionLinux/blob/03a2ec7db61aed847780bc3fca9c40b61a992fab/kernel_module/legion-laptop.c#L478
Try testing the lastest commit
just redowload the code or do a ```git pull --rebase```

KagaJiankui, 2023-10-24:
OK. I'll make a rough test since I'm short on time these days.

KagaJiankui, 2026-09-03:
I've moved to Y9000P IAH7H since 2023 and that commit works well. Sorry for this late reply.

closed 2026-09-03 by KagaJiankui