johnfanv2/LenovoLegionLinux #23: Support for Lenovo Legion 7 2022 - Legion 7 16ARHA7

albsen, 2023-03-07:
Hi,

I found your project when looking around for options to tweak power profiles so that I can get more battery life.

From a previous ticket I saw you guiding the reporter to provide you the DSDT.dsl file ( https://github.com/johnfanv2/LenovoLegionLinux/issues/13 ) so I went ahead and already uploaded that.

The current code compiles but the module refuses to load on 6.1, 6.2 and 6.3 the messages are identical.

```
sudo make reloadmodule

...
[ 1741.406538] legion_laptop 0.1 starts unloading
[ 1741.406569] legion_laptop 0.1 unloaded
[ 1741.421131] legion_laptop 0.1 starts loading
[ 1741.421133] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UH; DMI_BIOS_VERSION:K9CN38WW
[ 1741.421193] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
[ 1741.421194] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UH; DMI_BIOS_VERSION:K9CN38WW
[ 1741.421196] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[ 1741.421197] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[ 1741.421198] legion PNP0C09:00: legion_laptop not loaded for this device
[ 1741.421237] legion: probe of PNP0C09:00 failed with error -12
```

When trying to load with forcereload I get the following:

```
sudo make forcereloadmodule

...
[ 1686.086164] legion_laptop 0.1 starts unloading
[ 1686.086203] legion_laptop 0.1 unloaded
[ 1686.097996] legion_laptop 0.1 starts loading
[ 1686.097998] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UH; DMI_BIOS_VERSION:K9CN38WW
[ 1686.098058] legion PNP0C09:00: legion_laptop platform driver 0.1 probing
[ 1686.098060] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UH; DMI_BIOS_VERSION:K9CN38WW
[ 1686.098061] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[ 1686.098063] legion PNP0C09:00: legion_laptop is forced to load.
[ 1686.098063] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[ 1686.098064] legion PNP0C09:00: Using configuration for system: GKCN
[ 1686.098078] Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0xc400 (in EC) to virtual 0x00000000f573450b
[ 1686.098174] legion PNP0C09:00: Read embedded controller ID 0x5507
[ 1686.098175] legion PNP0C09:00: Expected EC chip id 0x8227 but read 0x5507
[ 1686.098176] Unloading legion ecram
[ 1686.098176] Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0xc400 (in EC) at virtual 0x00000000f573450b
[ 1686.098177] Unloading legion ecram done
[ 1686.098177] Unloading legion shared
[ 1686.098178] Unloading legion shared done
[ 1686.098178] legion PNP0C09:00: legion_laptop not loaded for this device
[ 1686.098238] legion: probe of PNP0C09:00 failed with error -12
```

As requested in the other ticket, please find attached the DSDT.dsl file:

[DSDT_16ARHA7.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/10909417/DSDT_16ARHA7.zip)

I took a look around and found the region with most of the fields is at: ```0xFE0B0400```

```
OperationRegion (ERAX, SystemMemory, 0xFE0B0400, 0xFF)
                    Field (ERAX, ByteAcc, Lock, Preserve)
```

The next best list of fields was a bit further down:

```
                    Scope (\)
                    {
                        OperationRegion (LFCN, SystemMemory, 0x92DD2298, 0x0377)
                        Field (LFCN, AnyAcc, Lock, Preserve)
```

I also had a look at your code but couldn't figure out how you know what to map.

Thank you for this project. Let me know if any further information is needed.

johnfanv2, 2023-03-07:
I used the offset from your file (0xFE0B0400) and updated the code to use this offset for your BIOS. Let's try it with it. 

Please pull new version, build with `make` and load with `make forcereloadmodulereadonly`. Then give the output from `dmesg` and (if possible) from `cat /sys/kernel/debug/legion/fancurve` and `/sys/kernel/debug/legion/ecmemory`.

johnfanv2, 2023-03-08:
Hmm. I am a bit surprised that it works. 

Could you please make the following tests and report back:
* Quick Test: Read Sensor Values from Hardware (for details see README): Are the RPM and temperature values plausible and increase/decrease? 
* Change the fancurve. Does the fans react according to the settings of the fan curve and attain approximately the set values.
* Test according to "Change power mode from software" from README. Does `powerprofilesctl list` and `powerprofilesctl set balanced` and `powerprofilesctl set performance` work? Does the LED on the laptop change accordingly?
* Does disabling mini fancurve works described in "Enable or disable the mini fan curve"
* Does "Lock and Unlock the Fan Controller and Fan Speed" works?

closed 2023-04-17 by johnfanv2

johnfanv2, 2023-04-21:
@albsen Greats thanks, alot!

albsen, 2023-04-23:
@johnfanv2 Here are the results you asked for, please let me know if there is something else I can do to help.

sudo dmidecode -t system

<details>
<summary>sudo dmidecode -t system</summary>

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82UH
        Version: Legion 7 16ARHA7
        Serial Number: PF3WZ411
        UUID: 0cc4b303-da6c-11ec-80f2-6c2408849b2e
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82UH_BU_idea_FM_Legion 7 16ARHA7
        Family: Legion 7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

</details>

```sudo dmidecode -t bios```

<details>
<summary>sudo dmidecode -t bios</summary>

```

# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: K9CN41WW
        Release Date: 02/16/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
[... 568 lines cut ...]
EC minifancurve feature enabled: 1
EC minifancurve on cool: false
EC lockfancontroller error: 0
EC lockfancontroller: true
EC fanfullspeed error: error
EC fanfullspeed status: -1
EC fan curve current point id: 2
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       5       0       41      0       65      0       37
1700     1700    5       5       35      45      60      65      33      41
1900     1900    5       5       42      49      60      65      38      46
2100     2000    5       5       46      54      60      65      41      127
2300     2300    2       2       52      64      60      65      127     127
2600     2500    2       2       62      68      60      65      127     127
2900     2800    2       2       65      72      60      65      127     127
3100     3000    2       2       69      76      60      75      127     127
3400     3400    2       2       73      80      72      88      127     127
3600     3600    2       2       77      127     85      127     127     127
=====================

```

</details>