johnfanv2/LenovoLegionLinux #25: windows support

marwan4mr, 2023-03-13:
I would honestly love to see this tool on windows, I know the tools available for windows, but this offers something that others don't, would you consider making a version for windows?

johnfanv2, 2023-04-01:
Unfortunately, it will not be compatible with Windows. Its is programmed to be tightly coupled with Linux because I want to integrate it into the Linux kernel.

closed 2023-04-01 by johnfanv2

russellmm, 2023-09-10:
This is a real bummer. I just bought the Legion Slim 7 (AMD/4060) and the way Lenovo handles the fan control on their Legion laptops is horrid. LegionFanControl is no longer supported (does not work on 2023 model) and LFC only uses exposed API and does not write to the EC.
Will be returning an otherwise greatr laptop and sticking with my ASUS G14.