johnfanv2/LenovoLegionLinux #27: Fan ids and fan speed points, kernel version and models

msmafra, 2023-03-19:
Hello there!

Looking at the setmyfancurve.sh and the python GUI, I got a bit confused about the number of speed points and fans.
In the bash script is 6 points and on the GUI is 10 points. So is 10 points the correct for both and all compatible Lenovo models? Should both point 0 and point 10 be set to 0?

On the case of the fans, in the bash script both times "fan" is referred as "first", and on the GUI both fields are identified as Fan 1. Looking at sensors on my machine, at least, there is Fan 1 and Fan 2, both with distinct speeds at the time I ran sensors. This brings me to another question: Are the the two fans for the whole system where they act on the highest temperature, despite being independently configurable, or one is for the CPU and the other for the GPU?

I'm testing it on a Lenovo Legion 5i (82CF, 5I-15IMH, model: LNVNB161216) with the latest BIOS EFCN58WW (2022-11-15) with a 
Intel Core i7-10750H with GeForce RTX 2060 Mobile [NVIDIA TU106M], using Fedora 37 kernel 6.1.x and 6.2.x.cbl (cachyos kernels). Now I'm actually on Fedora 38 Beta with both Fedora's and CachyOS's kernels being from 6.2. But the question is if it can be used on all kernels, from 6.0 up, or just recommended to use on 6.1 for now?

johnfanv2, 2023-03-25:
I will reply in a week. Currently, I have not much time.

johnfanv2, 2023-04-01:
- I am not really sure if all models use the same exact number of fan points. The last points that are currently not used are displayed with 0. I should make that bettter. I guess the number of points also depends on the mode. just try it with the GUI. it will display them (maybe you have to press the button to read the fan curve again)
- You can try to find out what fan where is by setting one to high speeds and listening it. I am not exactly sure if the fans are exactly the same for each model, so I called them fan 1 or fan 2 to not mislabel them. one each fan point you can set the speed of each fan. but the controller is always at one fan point. there are no two fan points/fan curves for CPU/GPU. but you can set GPU/CPU temp when it should switches from one point to the next.
- I guess it should work with all kernels. If it compiles than it should work. you don't need a new version.

msmafra, 2023-04-01:
It's been actually working with 6.2.x on my system. With both Fedora's and CachyOS' kernels.
I'm messing with your scripts a bit. I hope I can improve them and not break anything.

johnfanv2, 2023-04-02:
Thanks a lot for the feedback. I really appreciate that you are looking into the scripts. It's clear that you have lots of experience with bash!

closed 2023-04-12 by johnfanv2