johnfanv2/LenovoLegionLinux #30: Legion 5 15ARH05 | `Failed to start Apply Fan Curve` boot error message

juba0x00, 2023-03-27:
```
systemd[1]: Failed to start Apply Fan Curve to Lenovo Legion Laptop depending of Power State and Charger Mode (LenovoLegionLinux kernel Module).
```
- this error appears every time I powerup the laptop 
- I'm using Manjaro Linux

[image]

johnfanv2, 2023-04-01:
@MrDuartePT What does this could mean?

MrDuartePT, 2023-04-01:
This error in not warmfull at all in some distro that happen because the service will load before the kernel module. The service will reload when you change the porwer profile.

juba0x00, 2023-04-01:
@MrDuartePT Sorry, I don't know about this project.
I have a new Legion 5 laptop, I used the program for I small time I found some problems but I don't sure if this problem is from the program, One of these is Fn + Q shortcut does not work.
I'll test it again and keep in touch with you.
Thank you for helping.

MrDuartePT, 2023-04-01:
If you update the kernel you need to reinstall the LenovoLegionLinux package I will create a pacman hook to do that. If you reinstall the package pls reload the systemd service manually

juba0x00, 2023-04-04:
@MrDuartePT 
everything is working perfectly, Thank you for this great project
I have a question if you don't mind
The [legion-fan-utils-linux project](https://github.com/MrDuartePT/legion-fan-utils-linux) is not recommended or included in the documentation, so I don't know what I will miss if I don't use it.

MrDuartePT, 2023-04-04:
The only thing the project do is a simple manner of creating fan profiles and apply it and also change between profiles when you change the power mode (queit, balance and performance) both on battery and charger. Me and @Petingoso only made it because we what a more quiet profile on quiet more and a more aggressive curve on peformance and to be persistent across reboot.

closed 2023-04-04 by MrDuartePT