johnfanv2/LenovoLegionLinux #32: Add Legion 5 17ACH6 to supported devices

rozwell, 2023-04-01:
MODEL:
```
# dmidecode -s system-version
Legion 5 17ACH6
```
BIOS:
```
# dmidecode -s bios-version
HHCN31WW
```

@johnfanv2 you asked about these but I totally forgot/missed it: https://github.com/antony-jr/lenovo-legion5-15arh05-scripts/issues/8

Seems to be working fine on `Kubuntu 22.04` with `Kernel 6.2.8-060208-generic` and if you need any output, I'll be glad to provide it.

johnfanv2, 2023-04-02:
Thanks a lot for the feedback! I really appreciate the feedback about whether it works and what exact BIOS it is. I the  I added to the list of compatibly.

closed 2023-04-02 by johnfanv2

rozwell, 2023-05-01:
@johnfanv2 I noticed you've added `Legion 7 17ACH6` instead of `Legion 5 17ACH6`
AFAIK `Legion 7 17ACH6` does not exist :wink:

johnfanv2, 2023-05-01:
Thanks a lot for correcting it! I will change it right now.