johnfanv2/LenovoLegionLinux #44: Lenovo IdeaPad Gaming 3 15ARH05

danayer, 2023-04-12:
Is it possible to get support for my laptop?

**Model and Debug Info**

Model name: Lenovo IdeaPad Gaming 3 15ARH05

AMD Ryzen™ 7 4800H with Radeon™ Graphics × 16
NVIDIA GeForce GTX 1650 Ti / AMD Radeon™ Graphics

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82EY
	Version: IdeaPad Gaming 3 15ARH05
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82EY_BU_idea_FM_IdeaPad Gaming 3 15ARH05
	Family: IdeaPad Gaming 3 15ARH05

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: FCCN19WW
	Release Date: 05/30/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 15488 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.19
	Firmware Revision: 1.19

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

danayer, 2023-04-12:
Output after Make

```
make -C /lib/modules/6.2.9-300.fc38.x86_64/build M=/home/danayer/LenovoLegionLinux/kernel_module modules
make[1]: вход в каталог «/usr/src/kernels/6.2.9-300.fc38.x86_64»
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.0.1 20230318 (Red Hat 13.0.1-0)
  You are using:           gcc (GCC) 13.0.1 20230401 (Red Hat 13.0.1-0)
  CC [M]  /home/danayer/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /home/danayer/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /home/danayer/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /home/danayer/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /home/danayer/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /home/danayer/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: выход из каталога «/usr/src/kernels/6.2.9-300.fc38.x86_64»
```
output after sudo make reloadmodule

```
[    0.000000] Linux version 6.2.9-300.fc38.x86_64 (mockbuild@38f30b3c0c69453fae61718fc43f33bc) (gcc (GCC) 13.0.1 20230318 (Red Hat 13.0.1-0), GNU ld version 2.39-9.fc38) #1 SMP PREEMPT_DYNAMIC Thu Mar 30 22:32:58 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd1,gpt2)/vmlinuz-6.2.9-300.fc38.x86_64 root=UUID=a63d636c-13cb-4d37-8c19-87ab490e4250 ro rhgb quiet rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009f00000-0x0000000009f0cfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000009f0d000-0x00000000bab68fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bab69000-0x00000000bbd68fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bbd69000-0x00000000c9f7efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c9f7f000-0x00000000ccf7efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ccf7f000-0x00000000cdf7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cdf7f000-0x00000000cdffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cdfff000-0x00000000cdffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ce000000-0x00000000ceffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fde00000-0x00000000fdefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000fff1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000a0f33ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000a0f340000-0x0000000a2fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xb534d018-0xb536ce57] usable ==> usable
[    0.000000] e820: update [mem 0xb534d018-0xb536ce57] usable ==> usable
[    0.000000] e820: update [mem 0xb56c7018-0xb56d4657] usable ==> usable
[    0.000000] e820: update [mem 0xb56c7018-0xb56d4657] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009f00000-0x0000000009f0cfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000009f0d000-0x00000000b534d017] usable
[    0.000000] reserve setup_data: [mem 0x00000000b534d018-0x00000000b536ce57] usable
[    0.000000] reserve setup_data: [mem 0x00000000b536ce58-0x00000000b56c7017] usable
[    0.000000] reserve setup_data: [mem 0x00000000b56c7018-0x00000000b56d4657] usable
[    0.000000] reserve setup_data: [mem 0x00000000b56d4658-0x00000000bab68fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000bab69000-0x00000000bbd68fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000bbd69000-0x00000000c9f7efff] usable
[... 2611 lines cut ...]
[    7.238558] systemd[1]: Listening on systemd-userdbd.socket - User Database Manager Socket.
[    7.250098] systemd[1]: Mounting dev-hugepages.mount - Huge Pages File System...
[... cut ...]
[    7.295151] systemd[1]: Stopped systemd-journald.service - Journal Service.
[    7.295593] systemd[1]: Listening on systemd-journald-audit.socket - Journal Audit Socket.
[    7.307273] systemd[1]: Starting systemd-journald.service - Journal Service...
[    7.309393] systemd[1]: Starting systemd-modules-load.service - Load Kernel Modules...
[    7.311091] systemd[1]: Starting systemd-network-generator.service - Generate network units from Kernel command line...
[    7.311215] systemd[1]: systemd-pcrmachine.service - TPM2 PCR Machine ID Measur

johnfanv2, 2023-04-12:
Sorry, I created a bug yesterday. Please download new version I just pushed, compile again, and load module again. Your model is currently not supported but you can try again with `sudo make forcereloadmodule`

danayer, 2023-04-12:
```rmmod legion-laptop.ko || true
insmod legion-laptop.ko
dmesg
[    0.000000] Linux version 6.2.9-300.fc38.x86_64 (mockbuild@38f30b3c0c69453fae61718fc43f33bc) (gcc (GCC) 13.0.1 20230318 (Red Hat 13.0.1-0), GNU ld version 2.39-9.fc38) #1 SMP PREEMPT_DYNAMIC Thu Mar 30 22:32:58 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd1,gpt2)/vmlinuz-6.2.9-300.fc38.x86_64 root=UUID=a63d636c-13cb-4d37-8c19-87ab490e4250 ro rhgb quiet rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009f00000-0x0000000009f0cfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000009f0d000-0x00000000bab68fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bab69000-0x00000000bbd68fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bbd69000-0x00000000c9f7efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c9f7f000-0x00000000ccf7efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ccf7f000-0x00000000cdf7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cdf7f000-0x00000000cdffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cdfff000-0x00000000cdffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ce000000-0x00000000ceffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fde00000-0x00000000fdefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000fff1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000a0f33ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000a0f340000-0x0000000a2fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xb534d018-0xb536ce57] usable ==> usable
[    0.000000] e820: update [mem 0xb534d018-0xb536ce57] usable ==> usable
[    0.000000] e820: update [mem 0xb56c7018-0xb56d4657] usable ==> usable
[    0.000000] e820: update [mem 0xb56c7018-0xb56d4657] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009f00000-0x0000000009f0cfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000009f0d000-0x00000000b534d017] usable
[    0.000000] reserve setup_data: [mem 0x00000000b534d018-0x00000000b536ce57] usable
[    0.000000] reserve setup_data: [mem 0x00000000b536ce58-0x00000000b56c7017] usable
[    0.000000] reserve setup_data: [mem 0x00000000b56c7018-0x00000000b56d4657] usable
[    0.000000] reserve setup_data: [mem 0x00000000b56d4658-0x00000000bab68fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000bab69000-0x00000000bbd68fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000bbd69000-0x00000000c9f7efff] usable
[    0.000000] reserve setup_data: [mem 0x00000000c9f7f000-0x00000000ccf7efff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ccf7f000-0x00000000cdf7efff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000cdf7f000-0x00000000cdffefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000cdfff000-0x00000000cdffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ce000000-0x00000000ceffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fde00000-0x00000000fdefffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000fff1ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x0000000a0f33ffff] usable
[... cut ...]
[    5.669576] amdgpu 0000:05:00.0: amdgpu: ring gfx uses VM inv eng 0 on hub 0
[    5.669578] amdgpu 0000:05:00.0: amdgpu: ring gfx_low uses VM inv eng 1 on hub 0
[    5.669579] amdgpu 0000:05:00.0: amdgpu: ring gfx_high uses VM inv eng 4 on hub 0
[    5.669579] amdgpu 0000:05:00.0: amdgpu: ring comp_1.0.0 uses VM inv eng 5 on hub 0
[    5.669580] amdgpu 0000:05:00.0: amdgpu: ring comp_1.1.0 uses VM inv eng 6 on hub 0
[    5.669581] amdgpu 0000:05:00.0: amdgpu: ring comp_1.2.0 uses VM inv eng 7 on hub 0
[    5.669582] amd

danayer, 2023-04-12:
Output of `sudo cat /sys/kernel/debug/legion/fancurve:`
```EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode: 0
ACPI powermode error: -5
ACPI powermode: 0
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
minifancurve feature enabled: 1
minifancurve on cool: error
lock fan controller: false
enable maximumfanspeed: false
enable maximumfanspeed status: 0
fan curve current point id: 0
fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================```

danayer, 2023-04-12:
If you need more tests and outputs, write please. I will always be happy to help. And I apologize for my English.

johnfanv2, 2023-04-12:
This seems to be a new model. It has the same ID of the embedded controller as models for which it already works. Moreover, the above shows that rapid-charge feature already works. So I am very hopeful. However, existing models are all Legion models and not Ideapad models. So the following information would be useful.

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gaterhing WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```
Which features are the most useful fore you?

danayer, 2023-04-13:
Inspecting WMI entries
```
Results generated by fwts: Version V21.03.00 (2021-03-29 08:56:51).

Some of this work - Copyright (c) 1999 - 2021, Intel Corp. All rights reserved.
Some of this work - Copyright (c) 2010 - 2021, Canonical.
Some of this work - Copyright (c) 2016 - 2021, IBM.
Some of this work - Copyright (c) 2017 - 2021, ARM Ltd.

This test run on 13/04/23 at 15:57:47 on host Linux lenovo 6.2.9-300.fc38.x86_64
#1 SMP PREEMPT_DYNAMIC Thu Mar 30 22:32:58 UTC 2023 x86_64.

Command: "fwts wmi -".
Running tests: wmi.

wmi: Extract and analyse Windows Management Instrumentation (WMI).
--------------------------------------------------------------------------------
Test 1 of 1: Windows Management Instrumentation test.

\_SB_.WMI4._WDG (1 of 2)
  GUID: C3A03776-51AC-49AA-AD0F-F2F7D62C3F3C
  WMI Object:
    Flags          : 0x05 (Expensive | String)
    Object ID      : AD
    Instance       : 0x03

\_SB_.WMI4._WDG (2 of 2)
  GUID: 05901221-D566-11D1-B2F0-00A0C9062910
  WMI Object:
    Flags          : 0x00 (None)
    Object ID      : BD
    Instance       : 0x01

\_SB_.GZFD._WDG (1 of 14)
  GUID: 887B54E3-DDDC-4B2C-8B88-68A26A8835D0
  WMI Method:
    Flags          : 0x02 (Method)
    Object ID      : AA
    Instance       : 0x01
PASSED: Test 1, 887B54E3-DDDC-4B2C-8B88-68A26A8835D0 has associated method
\_SB_.GZFD.WMAA

\_SB_.GZFD._WDG (2 of 14)
  GUID: 887B54E1-DDDC-4B2C-8B88-68A26A8835D0
  WMI Object:
    Flags          : 0x01 (Expensive)
    Object ID      : A0
    Instance       : 0x01

\_SB_.GZFD._WDG (3 of 14)
  GUID: 887B54E2-DDDC-4B2C-8B88-68A26A8835D0
  WMI Object:
    Flags          : 0x01 (Expensive)
    Object ID      : A1
    Instance       : 0x02

\_SB_.GZFD._WDG (4 of 14)
  GUID: BFD42481-AEE3-4501-A107-AFB68425C5F8
  WMI Event:
    Flags          : 0x08 (Event)
    Notification ID: 0xd0
    Reserved       : 0x00
    Instance       : 0x01
FAILED [MEDIUM] WMIUnknownGUID: Test 1, GUID
BFD42481-AEE3-4501-A107-AFB68425C5F8 is unknown to the kernel, a driver may need
to be implemented for this GUID.
A WMI driver probably needs to be written for this WMI event. It can checked for
using: wmi_has_guid("BFD42481-AEE3-4501-A107-AFB68425C5F8"). One can install a
notify handler using
wmi_install_notify_handler("BFD42481-AEE3-4501-A107-AFB68425C5F8", handler,
[... 176 lines cut ...]
Medium failures: 12
 wmi: GUID BFD42481-AEE3-4501-A107-AFB68425C5F8 is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID D062906B-12D4-4510-999D-4831EE80E985 is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID BFD42481-AEE3-4502-A107-AFB68425C5F8 is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID BC72A435-E8C1-4275-B3E2-D8B8074ABA59 is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1 is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID D320289E-8FEA-41E0-86F9-611D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID D320289E-8FEA-41E1-86F9-611D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID D320289E-8FEA-41E0-86F9-711D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID D320289E-8FEA-41E0-86F9-811D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID D320289E-8FEA-41E0-86F9-911D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID 1F649C73-D9C5-475A-A8C9-347306CD3329 is unknown to the kernel, a driver may need to be implemented for this GUID.
 wmi: GUID 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294 is unknown to the kernel, a driver may need to be implemented for this GUID.

Low failures: NONE

Other failures: NONE

Test           |Pass |Fail |Abort|Warn |Skip |Info |
---------------+-----+-----+-----+-----+-----+-----+
wmi            |    3|   12|     |     |     |     |
---------------+-----+-----+-----+-----+-----+-----+
Total:         |    3|   12|    0|    0|    0|    0|
---------------+-----+-----+-----+-----+-----+-----+
```

danayer, 2023-04-13:
Disassembling ACPI tables
[DSDT.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11219706/DSDT.txt)

danayer, 2023-04-13:
Now I will connect another SSD with Windows and send you the data with the latest test

danayer, 2023-04-13:
[Gathering WMI info in Windows.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11220860/Gathering.WMI.info.in.Windows.txt)

danayer, 2023-04-13:
it would be great if gnome power profiles daemon could switch Lenovo power profiles if possible.

danayer, 2023-04-13:
Fn+Q works, but we don't have a multi-colored LED indicator to track the power mode, it only lights blue.

johnfanv2, 2023-04-14:
The files you provided look good. I guess it might work in a few days after I have done some changes to the driver.

 If you press Fn+Q does the LED color change? How are you able to tell that the power mode changes with Fn + Q without the LED changing?

danayer, 2023-04-14:
often you have to switch FN + Q in the game, and there the difference in performance is very visible, but in normal work, I just fly into the BIOS before starting the PC and see what my power profile is: D Then I change the profile as needed. Rarely, but I also use acpi_call 
# echo '\_SB.PCI0.LPC0.EC0.SPMO' > /proc/acpi/call 
# cat /proc/acpi/call; printf '\n' 
As a result you get the value for the bit SPMO (either 0x0 - Intelligent Cooling , 0x1 - Extreme Performance or 0x2 - Battery Saving ).

danayer, 2023-04-14:
The speed of the coolers changes depending on the power profile via FN + Q, and two-level backlighting also works (turn it on, make it brighter and turn it off using FN + Space), but it is displayed as a single-level backlight.

johnfanv2, 2023-04-14:
@danayer I added your model and made some changes. Please download the new version, compile again and load again. 
* Please provide the output when loading the kernel module
* Please change the power mode with Fn + Q and provide the output of `cat /sys/kernel/debug/legion/fancurve` and `cat /sys/firmware/acpi/platform_profile` after each change
* Please change the keyboard backlight with Fn + Space and provide the output of `cat /sys/kernel/debug/legion/fancurve` after each change

You can also try to read or change the power mode using `powerprofilesctl` or `powerprofilesctl set balanced` etc. You have to restart the power-profile-deamon before it works, e.g. with `systemctl restart power-profiles-daemon.service`.

danayer, 2023-04-15:
```cat /sys/kernel/debug/legion/fancurve```
**Battery Saving**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode access method: 3
powermode error: 0
powermode: 1
EC powermode error: 0
EC powermode: 2
ACPI powermode error: -5
ACPI powermode: 0
WMI powermode error: 0
WMI powermode: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight on/off: 0
WMI backlight: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
**Intelligent Cooling**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode access method: 3
powermode error: 0
powermode: 2
EC powermode error: 0
EC powermode: 0
ACPI powermode error: -5
ACPI powermode: 0
WMI powermode error: 0
WMI powermode: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight on/off: 0
WMI backlight: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
**Extreme Performance** 
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode access method: 3
powermode error: 0
powermode: 3
EC powermode error: 0
EC powermode: 1
ACPI powermode error: -5
ACPI powermode: 0
WMI powermode error: 0
WMI powermode: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight on/off: 0
WMI backlight: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

danayer, 2023-04-15:
```cat /sys/firmware/acpi/platform_profile```
**Battery Saving**
```quiet```
**Intelligent Cooling**
```balanced```
**Extreme Performance** 
```performance```

danayer, 2023-04-15:
```cat /sys/kernel/debug/legion/fancurve```
**The backlight is off.**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode access method: 3
powermode error: 0
powermode: 1
EC powermode error: 0
EC powermode: 2
ACPI powermode error: -5
ACPI powermode: 0
WMI powermode error: 0
WMI powermode: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight on/off: 0
WMI backlight: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
**First backlight level.**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode access method: 3
powermode error: 0
powermode: 1
EC powermode error: 0
EC powermode: 2
ACPI powermode error: -5
ACPI powermode: 0
WMI powermode error: 0
WMI powermode: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight on/off: 0
WMI backlight: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
**Second level of backlight.**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
powermode access method: 3
powermode error: 0
powermode: 1
EC powermode error: 0
EC powermode: 2
ACPI powermode error: -5
ACPI powermode: 0
WMI powermode error: 0
WMI powermode: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight on/off: 0
WMI backlight: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

danayer, 2023-04-15:
I'm confused by errors in dmesg
```[ 1672.859407] legion_laptop: Error for WMI method call to get keyboard brightness```
output with ```sudo dmesg```
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11238880/dmesg.txt)

danayer, 2023-04-15:
when trying to run ```sudo python/legion_linux/legion_gui.py``` an error occurs now
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Traceback (most recent call last):
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion_gui.py", line 451, in <module>
    main()
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion_gui.py", line 420, in main
    contr.init(read_from_hw=not do_not_excpect_hwmon)
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion_gui.py", line 56, in init
    sync_checkbox_from_feature(
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion_gui.py", line 16, in sync_checkbox_from_feature
    hw_value = feature.get()
               ^^^^^^^^^^^^^
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion.py", line 156, in get
    invalue = self._read_file_int(self.filename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion.py", line 68, in _read_file_int
    return int(FileFeature._read_file_str(file_path))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/danayer/LenovoLegionLinux/python/legion_linux/legion.py", line 64, in _read_file_str
    return str(filepointer.read()).strip()
               ^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument
```

danayer, 2023-04-15:
after the reboot, the output changed slightly in ```sudo dmesg```
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11238974/dmesg.txt)

danayer, 2023-04-15:
through the command ```sensors```, it now sees the speed of the cooler, it depends on the mode, although it's strange, on a balanced mode, the speed is higher than on a productive one.
**Battery Saving**
```
[danayer@lenovo ~]$ sensors
ath10k_hwmon-pci-0400
Adapter: PCI adapter
temp1:        +75.0°C  

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           7936 RPM
Fan 2:           7967 RPM
CPU Temperature:  +39.0°C  
GPU Temperature:  +42.0°C  
IC Temperature:    +0.0°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +36.9°C  (low  = -273.1°C, high = +74.8°C)
                       (crit = +84.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +68.0°C  

amdgpu-pci-0500
Adapter: PCI adapter
vddgfx:        1.27 V  
vddnb:       899.00 mV 
edge:         +52.0°C  
PPT:          19.00 W  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.10 V  
```

**Intelligent Cooling**
```
[danayer@lenovo ~]$ sensors
ath10k_hwmon-pci-0400
Adapter: PCI adapter
temp1:        +73.0°C  

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           9216 RPM
Fan 2:           9252 RPM
CPU Temperature:  +39.0°C  
GPU Temperature:  +46.0°C  
IC Temperature:    +0.0°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +36.9°C  (low  = -273.1°C, high = +74.8°C)
                       (crit = +84.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +71.8°C  

amdgpu-pci-0500
Adapter: PCI adapter
vddgfx:        1.26 V  
vddnb:       874.00 mV 
edge:         +51.0°C  
PPT:          16.00 W  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.10 V  
```

**Extreme Performance**
```
[danayer@lenovo ~]$ sensors
ath10k_hwmon-pci-0400
Adapter: PCI adapter
temp1:        +73.0°C  

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           8960 RPM
Fan 2:           8995 RPM
CPU Temperature:  +39.0°C  
GPU Temperature:  +46.0°C  
IC Temperature:    +0.0°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +36.9°C  (low  = -273.1°C, high = +74.8°C)
                       (crit = +84.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +75.8°C  

amdgpu-pci-0500
Adapter: PCI adapter
vddgfx:        1.30 V  
vddnb:       874.00 mV 
edge:         +51.0°C  
PPT:          31.00 W  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.10 V  
```

danayer, 2023-04-15:
despite the fact that changes in the speed of the coolers have begun, there are no differences from the "cat/sys/kernel/debug /legion/fancurve" that I gave you above. Lines 1 in 1, like there.

danayer, 2023-04-16:
noticed that the sensors fall off from time to time

johnfanv2, 2023-04-16:
1. It seems like power mode is working. Can you also change the power mode with the gnome GUI tools? If this works power profiles are done.
2. Fan speed: Are you sure that the sensor values (fan speed and temperature) are correct? Fan speeds of 7000-9000 rpm are very high. The fan speed is never exactly constant but changes 1-5 rpm each second. Is it constant? I guess it takes the wrong values. I will prepare a new version of the driver to use another function to get them.
3. Temperature: Do you think these are correct? Does the CPU temperature increase if you put the CPU under load. Try with `sudo stress --cpu 10 --timeout 200` to load the CPU.  Otherweise, I will prepare a new version of the driver to use another function to get the temperatures.
4. GUI Exception: Currently, I have not disabled features that do not work for your model to find all features that do work. I uploaded a new version that should mark them red instead of throwing an error. Please test it.

danayer, 2023-04-16:
1. yes, the power profiles work perfectly, and without a power supply everything is correct, productive is not available without a power supply.
2. 3. yes, the speed of the coolers is indicated incorrectly, but it is not static, but the temperature is static.
[image]

danayer, 2023-04-16:
3. yes, the program is starting now, thank you very much)
[image]

johnfanv2, 2023-04-16:
@danayer Could you please try the new version and check if fan speeds and temperatures are now ok? I just updated it. Please also give the output of `sudo cat /sys/kernel/debug/legion/fancurve` (once).

danayer, 2023-04-16:
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 3
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 90
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 56
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 4600
1 fanspeed EC error: 0
1 fanspeed EC: 8960
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 4600
1 fanspeed WMI error: 0
1 fanspeed WMI: 4600
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 4600
2 fanspeed error: 0
2 fanspeed: 4400
2 fanspeed EC error: 0
2 fanspeed EC: 8995
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 4400
2 fanspeed WMI error: 0
2 fanspeed WMI: 4400
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 4400
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

danayer, 2023-04-16:
the speed of the cooler seems to be changing correctly now, I put it on a stress test, it is growing, and after the stress test, it began to fall, which I heard. so it's working now)
[image]

danayer, 2023-04-16:
but the temperature now shows 0 :(

danayer, 2023-04-16:
[image]
although errors seem to be issued in ```sudo cat /sys/kernel/debug/legion/fancurve```, it reads the rotation speed seems to be correct. (this is in idle mode on power saving mode)

danayer, 2023-04-16:
```sudo dmesg``` just in case)
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11243267/dmesg.txt)

johnfanv2, 2023-04-16:
@danayer  Please try a new version and check if temperatures are now correct.

danayer, 2023-04-16:
now everything works perfectly)
[image]
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 2
CPU temperature error: 0
CPU temperature: 85
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 85
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 55
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 55
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 4600
1 fanspeed EC error: 0
1 fanspeed EC: 8960
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 4600
1 fanspeed WMI error: 0
1 fanspeed WMI: 4600
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 4600
2 fanspeed error: 0
2 fanspeed: 4400
2 fanspeed EC error: 0
2 fanspeed EC: 8995
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 4400
2 fanspeed WMI error: 0
2 fanspeed WMI: 4400
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 4400
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

danayer, 2023-04-16:
the only thing that scares errors in ```sudo dmesg```, is it okay?
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11243301/dmesg.txt)

johnfanv2, 2023-04-16:
The errors should only occur when you call `sudo cat /sys/kernel/debug/legion/fancurve` and initially the the error `legion_laptop: Error reading keyboard brighntess` etc. are ok. Here, it checks different methods. Some methods are not available on your model. Then nothing happens and it just displays an error.

johnfanv2, 2023-04-16:
@danayer Could you also provide the output of `sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C`? Please change the power mode and provide the output after each change. Then I can check if setting the fan curve is possible on your model.

danayer, 2023-04-16:
**Extreme Performance**
```
00000000  00 01 00 00 00 00 00 01  00 00 10 00 00 20 20 20  |.............   |
00000010  00 00 80 00 07 00 35 01  00 01 01 00 00 01 00 00  |......5.........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000040  00 00 00 00 00 00 00 00  00 00 00 00 83 51 00 00  |.............Q..|
00000050  00 00 00 26 00 00 20 00  00 00 00 00 00 00 00 00  |...&.. .........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000080  00 00 00 00 08 00 00 00  c0 00 13 00 00 01 00 53  |...............S|
00000090  75 6e 77 6f 64 61 00 32  4c 31 39 44 33 50 46 34  |unwoda.2L19D3PF4|
000000a0  00 01 00 86 00 00 00 66  68 00 00 00 68 25 00 00  |.......fh...h%..|
000000b0  29 21 20 22 26 20 00 22  02 11 00 00 00 80 01 00  |)! "& ."........|
000000c0  00 00 00 0b ab 0f d2 2e  88 2c c6 11 2b 0f 49 00  |.........,..+.I.|
000000d0  00 00 00 00 00 00 00 19  4b 06 00 00 00 00 00 00  |........K.......|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 8e 00  |................|
000000f0  01 00 00 40 0f 51 64 32  00 00 00 00 00 00 00 00  |...@.Qd2........|
00000100  82 00 00 00 00 00 06 06  13 13 1e 13 1e 13 00 00  |................|
00000110  00 00 ff ff 9b 00 00 00  00 00 41 2a 2a 00 00 00  |..........A**...|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000150  2c 32 3e 7f 7f 7f 7f 7f  2c 32 3e 7f 7f 7f 7f 7f  |,2>.....,2>.....|
00000160  00 00 00 00 7f 3a 1c 7f  3a 1a 7f 7f 00 7f 7f 00  |.....:..:.......|
00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000190  00 29 2f 3a 3a 3a 3a 3a  00 29 2f 3a 3a 3a 3a 3a  |.)/:::::.)/:::::|
000001a0  00 10 1a 1c 1c 1c 1c 1c  00 0e 16 1a 1a 1a 1a 1a  |................|
000001b0  00 00 00 00 00 00 00 00  29 26 00 00 00 00 00 00  |........)&......|
000001c0  7f 7f 7f 7f 7f 7f 7f 7f  7f 7f 7f 7f 7f 7f 7f 7f  |................|
000001d0  00 7f 7f 7f 7f 7f 7f 7f  00 7f 7f 7f 7f 7f 7f 7f  |................|
000001e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000001f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 47 47  |..............GG|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000250  00 00 00 00 00 00 00 00  aa 00 00 00 00 00 00 00  |................|
00000260  57 32 53 5a 30 38 45 30  46 41 42 00 00 00 00 00  |W2SZ08E0FAB.....|
00000270  88 13 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000300
```
**Intelligent Cooling**
```
00000000  00 00 00 00 00 00 00 01  00 00 10 00 00 20 20 20  |.............   |
00000010  00 00 80 00 07 00 35 01  00 01 01 00 00 00 00 00  |......5.........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000040  00 00 00 00 00 00 00 00  00 00 00 00 83 51 00 00  |.............Q..|
00000050  00 00 00 28 00 00 20 00  00 00 00 00 00 00 00 00  |...(.. .........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000080  00 00 00 00 08 00 00 00  c0 00 13 00 00 01 00 53  |...............S|
00000090  75 6e 77 6f 64 61 00 32  4c 31 39 44 33 50 46 34  |unwoda.2L19D3PF4|
000000a0  00 01 00 86 00 00 00 66  68 00 00 00 68 25 00 00  |.......fh...h%..|
000000b0  29 24 20 25 28 20 00 25  02 11 00 00 00 80 01 00  |)$ %( .%........|
000000c0  00 00 00 0b ab 0f d3 2e  88 2c c6 11 2b 0f 49 00  |.........,..+.I.|
000000d0  00 00 00 00 00 00 00 1a  4b 06 00 00 00 00 00 00  |........K.......|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 8e 00  |................|
000000f0  01 00 00 40 0f 51 64 32  00 00 00 00 00 00 00 00  |...@.Qd2........|
00000100  82 00 00 00 00 00 05 05  44 13 1e 13 1e 13 44 44  |........D.....DD|
00000110  00 00 ff ff 9a 00 00 00  00 00 41 2a 2a 00 00 00  |..........A**...|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 01  |................|
00000150  2c 32 38 42 4e 51 54 7f  2c 32 38 42 4e 51 54 7f  |,28BNQT.,28BNQT.|
00000160  00 00 00 00 7f 50 2e 7f  50 2c 7f 3a 2e 7f 3a 2c  |.....P..P,.:..:,|
00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
[... 39 lines cut ...]
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000150  2c 32 3e 7f 7f 7f 7f 7f  2c 32 3e 7f 7f 7f 7f 7f  |,2>.....,2>.....|
00000160  00 00 00 00 7f 3a 1c 7f  3a 1a 7f 7f 00 7f 7f 00  |.....:..:.......|
00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000190  00 29 2f 3a 3a 3a 3a 3a  00 29 2f 3a 3a 3a 3a 3a  |.)/:::::.)/:::::|
000001a0  00 10 1a 1c 1c 1c 1c 1c  00 0e 16 1a 1a 1a 1a 1a  |................|
000001b0  d1 08 60 07 10 0e 00 00  28 26 00 00 00 00 00 00  |..`.....(&......|
000001c0  7f 7f 7f 7f 7f 7f 7f 7f  7f 7f 7f 7f 7f 7f 7f 7f  |................|
000001d0  00 7f 7f 7f 7f 7f 7f 7f  00 7f 7f 7f 7f 7f 7f 7f  |................|
000001e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 01 00 00  |................|
00000240  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000250  00 00 00 00 00 00 00 00  aa 00 00 00 00 00 00 00  |................|
00000260  57 32 53 5a 30 38 45 30  46 41 42 00 00 00 00 00  |W2SZ08E0FAB.....|
00000270  88 13 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000300
```

johnfanv2, 2023-04-17:
@danayer I hope I made changes to allow you to change the fan curve, i.e. set the fan speed at each temperature. Could you use the new version and give me the output of `sudo cat /sys/kernel/debug/legion/fancurve` (for each power mode). 

Could you check if the measured fan speed is the fan speed that corresponds the the temperatures in the fan curve table? You can also try changing the fan speeds or temperatures of the fan curve in the GUI and verify if the fan is controlled according to these values. When you change something the changes will not apply immediately but only if the temperature of the CPU or GPU changes to the next level.

danayer, 2023-04-17:
**Extreme Performance**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 2
CPU temperature error: 0
CPU temperature: 41
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 41
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 39
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 39
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
[... 161 lines cut ...]
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 8
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 44	 0	 127	 0	 0
1600	 1400	 0	 0	 41	 50	 127	 127	 0	 0
2600	 2200	 0	 0	 47	 62	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
=====================
```

it reads correctly, but when you try to change it through the application, when you click "apply to hw", it simply resets everything to default.

[image]

johnfanv2, 2023-04-17:
Can you change the fan speed only in the GUI or does it reset to default?

danayer, 2023-04-17:
it reverts to the default value. I also tried following the instructions 
```
# Go to folder LenovoLegionLinux and run it. It should output "Writing fan curve successfully!" if it finished successfully
sudo ./setmyfancurve.sh
# And check new fan curve
sudo cat/sys/kernel/debug/legion/fancurve
```
it is also wrapped by default

danayer, 2023-04-17:
at the bottom there is an inscription in the application
[image]

danayer, 2023-04-17:
when you try to change this through the app they change visually, but when you click "apply to hw" it just returns everything to default. 
a more accurate description)

johnfanv2, 2023-04-17:
@danayer  It almost works. It created a new version with more logging. Please test the new version with `sudo cat/sys/kernel/debug/legion/fancurve`. Then provide the output of `sudo dmesg`.

danayer, 2023-04-17:
**Extreme Performance**
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 2
CPU temperature error: 0
CPU temperature: 49
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 49
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 41
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 41
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 2700
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2700
1 fanspeed WMI error: 0
1 fanspeed WMI: 2700
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2700
2 fanspeed error: 0
2 fanspeed: 2500
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2500
2 fanspeed WMI error: 0
2 fanspeed WMI: 2500
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 2500
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
[... 156 lines cut ...]
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 8
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 44	 0	 127	 0	 0
1600	 1400	 0	 0	 41	 50	 127	 127	 0	 0
2600	 2200	 0	 0	 47	 62	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
2800	 2600	 0	 0	 58	 127	 127	 127	 0	 0
=====================
```
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11245723/dmesg.txt)

johnfanv2, 2023-04-17:
Sorry, I meant to say `sudo ./setmyfancurve.sh` instead of `sudo cat/sys/kernel/debug/legion/fancurve`.

danayer, 2023-04-17:
`sudo ./setmyfancurve.sh`
```
MODEL
IdeaPad Gaming 3 15ARH05
BIOS
FCCN19WW

Using hwmon directory: /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon5
Writing fancurve succesful!
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 2
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 52
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 41
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 41
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 2600
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2600
1 fanspeed WMI error: 0
1 fanspeed WMI: 2600
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2600
2 fanspeed error: 0
2 fanspeed: 2200
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2200
2 fanspeed WMI error: 0
2 fanspeed WMI: 2200
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 2200
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 8
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 44	 0	 56	 0	 0
1600	 1400	 0	 0	 41	 50	 52	 56	 0	 0
2600	 2200	 0	 0	 47	 56	 52	 56	 0	 0
2800	 2600	 0	 0	 52	 66	 52	 56	 0	 0
3100	 2900	 0	 0	 62	 78	 52	 56	 0	 0
3500	 3300	 0	 0	 74	 81	 52	 59	 0	 0
3900	 3700	 0	 0	 77	 84	 55	 62	 0	 0
4600	 4400	 0	 0	 80	 93	 58	 71	 0	 0
=====================
Writing fancurve succesful!
MODEL
IdeaPad Gaming 3 15ARH05
BIOS
FCCN19WW
`sudo dmesg`
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11245888/dmesg.txt)

johnfanv2, 2023-04-17:
@danayer Hmm. Writing works but the value is overwritten a few milliseconds later again. Please try my new version again, run ./setmyfancurve.sh instead and provide the same outputs again.

danayer, 2023-04-17:
```
MODEL
IdeaPad Gaming 3 15ARH05
BIOS
FCCN19WW

Using hwmon directory: /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon5
Writing fancurve succesful!
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 2
CPU temperature error: 0
CPU temperature: 43
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 43
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 41
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 41
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 2500
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2500
1 fanspeed WMI error: 0
1 fanspeed WMI: 2500
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2500
2 fanspeed error: 0
2 fanspeed: 2200
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2200
2 fanspeed WMI error: 0
2 fanspeed WMI: 2200
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 2200
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 8
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 44	 0	 56	 0	 0
1600	 1400	 0	 0	 41	 50	 52	 56	 0	 0
2600	 2200	 0	 0	 47	 56	 52	 56	 0	 0
2800	 2600	 0	 0	 52	 66	 52	 56	 0	 0
3100	 2300	 0	 0	 62	 78	 52	 56	 0	 0
3500	 2500	 0	 0	 74	 81	 52	 59	 0	 0
3900	 3700	 0	 0	 77	 84	 55	 62	 0	 0
4600	 4400	 0	 0	 80	 93	 58	 71	 0	 0
=====================
Writing fancurve succesful!
MODEL
IdeaPad Gaming 3 15ARH05
BIOS
FCCN19WW
```
[dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11246043/dmesg.txt)

johnfanv2, 2023-04-17:
There might be a way to prevent the the firmware in the embedded controller from overwriting it again with the default values. I might be disassemble it the the future but not now. So I guess setting the fan curve will not be implemented now.

Is there any other features needed by you?

There are some additional features implemented that are not in the GUI yet. They are in folder `/sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/`
```bash
ls /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00

aslcodeversion            gpu_ctgp2_powerlimit  leds               rapidcharge
cpu_longterm_powerlimit   gpu_ctgp_powerlimit   lockfancontroller  subsystem
cpumaxfrequency           gpu_ppab_powerlimit   modalias           thermalmode
cpu_shortterm_powerlimit  gsync                 overdrive          touchpad
driver                    hwmon                 PNP0C0A:00         uevent
driver_override           isacfitforoc          power              VPC2004:00
fan_fullspeed             issupportcpuoc        powerchargemode    winkey
firmware_node             issupportgpuoc        powermode
```
You can read and write to most of these files to read and set properties, e.g.
```bash
# read rapidcharge status
cat /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/rapidcharge
# enable rapidcharge status (as root)
echo 1 > /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/rapidcharge
```
If it returns an error, then it is not available for your model. Otherwise, you can test if it has an effect. You can check all these features if they might work for you model.

closed 2023-04-17 by johnfanv2