johnfanv2/LenovoLegionLinux #45: Lenovo Ideapad Gaming 3 15ACH6 Power plan in Gnome

kleidiss, 2023-04-13:
Fan control and other Legion stuff will never be supported for this laptop but when i was testing i noticed that power profiles also didn't work

Is there any way to make them work with gnome power profile daemon so we get better integration , would make Linux much nicer to use
IDK if it's possible tho

kleidiss, 2023-04-13:
Will add the command outputs later since i dont have access to Linux rn , just wanted to know if this was even worth considering on Ideapad laptops

kleidiss, 2023-04-14:
Trying to build the modules gives me this long error on Ubuntu 22.10 live usb 

[errorlog.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11227804/errorlog.txt)

kleidiss, 2023-04-14:
System Information
	Manufacturer: LENOVO
	Product Name: 82K2
	Version: IdeaPad Gaming 3 15ACH6
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82K2_BU_idea_FM_IdeaPad Gaming 3 15ACH6
	Family: IdeaPad Gaming 3 15ACH6

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
Table at 0xCB703000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: H3CN37WW(V2.07)
	Release Date: 12/22/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.37
	Firmware Revision: 1.37

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82K2
	Version: IdeaPad Gaming 3 15ACH6
	Serial Number: MP27A03J
	UUID: 1e1c717a-9f52-11ec-810f-e4a8dfba67a7
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82K2_BU_idea_FM_IdeaPad Gaming 3 15ACH6
	Family: IdeaPad Gaming 3 15ACH6

Handle 0x0002, DMI type 2, 16 bytes
Base Board Information
	Manufacturer: LENOVO
	Product Name: LNVNB161216
[... 592 lines cut ...]
	OEM-specific Information: 0x00000000

Handle 0x0035, DMI type 133, 5 bytes
OEM-specific Type
	Header and Data:
		85 05 35 00 01
	Strings:
		KHOIHGIUCCHHII

Handle 0x0036, DMI type 200, 16 bytes
OEM-specific Type
	Header and Data:
		C8 10 36 00 01 02 01 27 01 14 01 52 B0 C1 01 05
	Strings:
		IdeaPad
		82K2007FRM

Handle 0x0037, DMI type 248, 18 bytes
OEM-specific Type
	Header and Data:
		F8 12 37 00 AC AC AA 17 00 00 00 00 00 00 00 00
		00 00

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table

johnfanv2, 2023-04-14:
I am currently working on the driver for another Ideapad Gaming 3 15. There it seems like it will be possible. The driver does not load for your model because an ACPI method is not available or renamed for your model. To evaluate it I would need the following information described in the sections below. 

Probably power mode change might work. Can you switch it with Fn + Q?
Keyboard backlight might also work? Does it work already? How many different levels do you have? Only on and off?
A few other functions might also work. 
Custom fan control will probably not work as it is implemented now but might work after modification. How many different speed levels are currently used? Three or more? Does the fan control change when you change power mode with Fn + Q (e.g. in Windows)?

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```

kleidiss, 2023-04-14:
> Probably power mode change might work with changes to the driver. Can you switch it with Fn + Q? Keyboard backlight might also work with changes to the driver. Does it work already? How many different levels do you have? Only on and off? A few other functions might also work with the driver. Custom fan control will probably not work as it is implemented now but might work after modification. How many different speed levels are currently used? Three or more? Does the fan control change when you change power mode with Fn + Q (e.g. in Windows)?

Yes i can switch the power mode with fn+Q and even the nvidia-powerd.service recognizes the high performance mode (after i restart the device) and gives the full 85W to the machine (On the latest 530 drivers at least)

Keyboard backlight switching also already works
It has 4 states , 

State 1: Wave Left Speed 2 Brightness 1 
State 2: Static Multiple Colors 
State 3: Blue Static Color
State 4: Off

So this is the same as Windows with no drivers i suppose

kleidiss, 2023-04-14:
> ## Gathering WMI info in Windows

[powershelloutput.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11234109/powershelloutput.txt)

kleidiss, 2023-04-14:
> ## Inspecting WMI entries
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11234462/fwts_wmi.log)

kleidiss, 2023-04-14:
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11234518/acpi_re.zip)

kleidiss, 2023-04-14:
Tell me if i missed something but this should be it

johnfanv2, 2023-04-15:
@kleidiss I pushed a new version that you can try. Please download the new version, compile again and load again.

- Please provide the output when loading the kernel module
- Please change the power mode with Fn + Q and provide the output of cat /sys/kernel/debug/legion/fancurve and cat /sys/firmware/acpi/platform_profile after each change

You can also try to read or change the power mode using powerprofilesctl or powerprofilesctl set balanced etc. You have to restart the power-profile-deamon before it works, e.g. with systemctl restart power-profiles-daemon.service.

kleidiss, 2023-04-15:
@johnfanv2  I still cant seem to load the modules i get this error 
[anothererror.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11237967/anothererror.txt)

johnfanv2, 2023-04-15:
@kleidiss Sorry, I had to make another change for your model. Try the new version.

kleidiss, 2023-04-15:
Ok now the gnome profile switcher works 

FN+Q changes it on the GUI perfectly 

However i saw some errors for keyboard rgb so that wont work  

[  295.251383] legion_laptop: Init keyboard backlight LED driver
[  295.251384] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[  295.251385] legion_laptop: Error for WMI method call to get keyboard brightness
[  295.251385] legion_laptop: Error reading keyboard brighntess
[  295.251385] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...

kleidiss, 2023-04-15:
Also on a unrelated note i'm failing to compile this on fedora

Make gives error 2

kleidiss, 2023-04-15:
Wait i found a small bug which is probably a GNOME bug but its worth mentioning anyway

If you click the performance mode on the GNOME MEnu via a mouse it doesnt change 
Balanced and Power Saver Does

johnfanv2, 2023-04-15:
@kleidiss  Thanks for the bug report. I hopefully fixed it. Can you please check the new version?

kleidiss, 2023-04-15:
> @kleidiss Thanks for the bug report. I hopefully fixed it. Can you please check the new version?

Everything working fine now 
Thanks

kleidiss, 2023-04-15:
But in battery mode  i noticed that performance is still available  and checking the command it seems that the laptop still uses performance mode even in battery power
Now idk if you can limit that to only balanced and quiet mode in battery power but you should look into it becuase it will destroy battery life

Even on battery cat /sys/firmware/acpi/platform_profile
performance

johnfanv2, 2023-04-16:
The powermode should have the following behaviour:
- if you are in performance mode and plug out the charger it should go to balanced or quiet automatically
- when you are running on battery you should be able to all modes (also performance) manually if you want. In the files you provided I see 6 modes: quiet-AC, balanced-AC, performance-AC and uiet-DC, balanced-DC, performance-DC. I guess AC stands for charger and DC for battery.

Now you can test keyboard backlight support. As far as I have seen from the files your provided access to the hardware is not implemented in the WMI layer. Nevertheless, please set the backlight to off, medium, full and provide the output of 

```bash
cat /sys/kernel/debug/legion/fancurve 
cat  /sys/class/leds/platform::kbd_backlight/brightness
```
after each setting. Also provide the output once of
```bash
ls -lah  /sys/class/leds/*
```

It might also be supported by another driver in the kernel. To test it, please uninstall the lenovo_legion driver (this module) and make a restart. Do not load it again. Then do the test above again and provide the same output for the three states. After that you can install/load the legion driver it again.

kleidiss, 2023-04-16:
My keyboard has full rgb so it has 5 states including off

kleidiss, 2023-04-16:
[wave (1).zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11241027/wave.1.zip)
[pluse (2).zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11241028/pluse.2.zip)
[rgb_pulse (3).zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11241032/rgb_pulse.3.zip)
[static blue  (4).zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11241033/static.blue.4.zip)
[off (5).zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11241034/off.5.zip)

kleidiss, 2023-04-16:
I had the compress the txt files because the txt uploading wast working 
And one of the commands just gave a not found error but since i was copy-pasting i ran it anyway so just ignore it

In a day ill try these without the kernel modules if that's needed

johnfanv2, 2023-04-16:
ok. testing without the kernel module is not needed anymore. the keyboard backlight is neither supported by the WMI layer used by legion nor by the ideapad driver as far as I see. I guess it is done by USB. Can you control it in Windows with LegionToolkit or Vantage? If this is important for you, you can try maybe this https://github.com/InstinctEx/lenovo-ideapad-legion-keyboard-led.

johnfanv2, 2023-04-16:
Backlight control seems to be done by USB and can be done by OpenRGB: https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2868

kleidiss, 2023-04-16:
> Backlight control seems to be done by USB and can be done by OpenRGB: https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2868

Is it already supported?
I'll try tonight if so

johnfanv2, 2023-04-16:
> If you want, we could try getting CPU and GPU temperature. From the files you provided it seems like it would work. I would have to make small changes to the driver.

johnfanv2, 2023-04-16:
@kleidiss Could you please try the new version and check if fan speeds and temperatures are now ok? I just updated it. Please also give the output of sudo cat /sys/kernel/debug/legion/fancurve (once).

kleidiss, 2023-04-16:
> sudo cat /sys/kernel/debug/legion/fancurve

EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228186128
temperature access method: 3
CPU temperature error: 0
CPU temperature: 55
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: 0
CPU temperature WMI: 55
CPU temperature WMI2 error: -14
CPU temperature WMI2: 55
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: -14
GPU temperature WMI2: 35
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 0
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
EC fanfullspeed error: error
EC fanfullspeed status: -1
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
=====================

kleidiss, 2023-04-17:
Hey unrelated question but i was thinking 
Ik this is way too much to ask considering how god awful nvidia is on linux but could it be possible to also port the overclock features for vantage

Yes there is a way to already do OC but it requires to render the screen through the nvidia gpu all the time and run xorg as root and also it doesn't care about what mode you're on

johnfanv2, 2023-04-17:
The temperatures for CPU and GPU seem to be correct. Please get the newest version. You can check them with `sensors` or the GUI `psensor`. Do you think they are correct? Does the CPU temperature increase if you put the CPU under load?
The fan speed seem to not correct. I have not found a way to access them via WMI or ACPI. Access to the EC via port IO does not seem to be possible. Accessing the EC via memory mapping might be possible but I will not implement it right now. So fan speed access is not possible right now.

I might implement GPU overclocking:
1. Can you describe the exact steps how it is done in Windows?
2. Can you describe the exact steps and commands+programs how it is done in Linux?

kleidiss, 2023-04-17:
@johnfanv2 

Even on Windows i cannot access the fan speed neither modfiy it so i think it's hardware locked by Lenovo
Even the geforce app on windows cannot access it , same with MSI Afterburner 

And about the temperatures they seem to be correct as far as i saw but i'll double check anyway

kleidiss, 2023-04-17:
>     1. Can you describe the exact steps how it is done in Windows?
> 
>     2. Can you describe the exact steps and commands+programs how it is done in Linux?

On Windows it just works out of the box as far as i know or maybe you need vantage services to be able to k=modify it i am not sure

But on Linux it's a pain in the ass 
https://download.nvidia.com/XFree86/Linux-x86_64/430.14/README/xconfigoptions.html#Coolbits
https://wiki.archlinux.org/title/NVIDIA/Tips_and_tricks#Overclocking_and_cooling

But keep in mind first you need to run Xorg as root for nvidia-settings to apply the offsets
Second they tend to reset after a system restart so you need to apply it each time
Creating a xorg.conf on this laptop completely breaks the xorg server so you need to modify the 10-nvidia.conf on etc/x11/xorg.conf.d or usr/share/x11/xorg.conf.d

kleidiss, 2023-04-17:
I can confirm the GPU temps are accurate when using sensors

johnfanv2, 2023-04-17:
Thanks a lot for testing all the stuff. I might do the NVIDIA stuff in the future but it seems to be a rather complicated task that I cannot do quickly.

kleidiss, 2023-04-17:
> Thanks a lot for testing all the stuff.

No problem 

>  I might do the NVIDIA stuff in the future but it seems to be a rather complicated task that I cannot do quickly.

It's a pain the ass i wouldn't fault you if you straight up ignored it
The shit nvidia uses on linux is years out of date and doesn;t even work on Wayland

johnfanv2, 2023-04-17:
There are also some features in `/sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/`
```bash
ls /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00

aslcodeversion            gpu_ctgp2_powerlimit  leds               rapidcharge
cpu_longterm_powerlimit   gpu_ctgp_powerlimit   lockfancontroller  subsystem
cpumaxfrequency           gpu_ppab_powerlimit   modalias           thermalmode
cpu_shortterm_powerlimit  gsync                 overdrive          touchpad
driver                    hwmon                 PNP0C0A:00         uevent
driver_override           isacfitforoc          power              VPC2004:00
fan_fullspeed             issupportcpuoc        powerchargemode    winkey
firmware_node             issupportgpuoc        powermode
```
You can read and write to most of these files to read and set properties, e.g.
```bash
# read rapidcharge status
cat /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/rapidcharge
# enable rapidcharge status (as root)
echo 1 > /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/rapidcharge
```
If it returns an error than it is not available for your model. Otherwise, you can test if it has an effect.

closed 2023-04-17 by johnfanv2