johnfanv2/LenovoLegionLinux #46: Testing new features on 2022 models

johnfanv2, 2023-04-18:
This is a issue to test the new features on 2022 models.

**Model and Debug Info**

Model name: ???
CPU model: ??? (e.g. AMD Ryzen 7 5800H)
GPU model: ??? (e.g NVIDIA RTX 3700)
Keyboard backlight: ??? (RGB, single color with off/on, single color with off/medium/bright)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
???
```

Output of `sudo dmidecode -t bios`:
```text
???
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
???
```

The following information would be useful.

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```

javaarchive, 2023-04-19:
Model name: Legion 5 Pro
CPU model: AMD Ryzen 9 6900HX with Radeon Graphics (16)
GPU model: NVIDIA RTX 3070TI
Keyboard backlight: RGB, has I think like 4 separate color controllable lights. 

Hang tight while I get the data from linux and then windows. 

Linux (nobara project) commands were run under performance mode:
```
[raymond@localhost-0 ~]$ sudo dmidecode -t system
[sudo] password for raymond:         
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RG
	Version: Legion 5 Pro 16ARH7H
	Serial Number: redacted
	UUID: redacted
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RG_BU_idea_FM_Legion 5 Pro 16ARH7H
	Family: Legion 5 Pro 16ARH7H

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

[raymond@localhost-0 ~]$ sudo dmidecode -t bios
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JUCN55WW
	Release Date: 10/12/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.55
	Firmware Revision: 1.55

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

[raymond@localhost-0 ~]$ sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
minifancurve feature enabled: 1
minifancurve on cool: true
lock fan controller: false
enable maximumfanspeed: false
enable maximumfanspeed status: 0
fan curve current point id: 5
fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 5	 5	 0	 58	 0	 63	 0	 34
1600	 1600	 5	 5	 54	 58	 60	 63	 31	 36
1800	 1800	 5	 5	 54	 58	 60	 63	 33	 43
2000	 2000	 5	 5	 54	 58	 60	 63	 41	 127
2200	 2200	 2	 2	 54	 64	 60	 63	 127	 127
2500	 2600	 2	 2	 62	 67	 60	 63	 127	 127
2800	 2800	 2	 2	 65	 70	 60	 67	 127	 127
3000	 3100	 2	 2	 68	 73	 64	 71	 127	 127
3400	 3500	 2	 2	 70	 80	 68	 74	 127	 127
3700	 3700	 2	 2	 74	 86	 72	 79	 127	 127
=====================
```

johnfanv2, 2023-04-19:
@javaarchive Thanks for your help! I really appreciate it. I am looking forward to the other information and will then inspect it to verify my assumptions that I used for the new models.

vasubawa, 2023-04-20:
Model and Debug Info

Model name: Legion 5 15ARH7H
CPU model: AMD Ryzen 7 6800H
GPU model: NVIDIA RTX 3070TI
Keyboard backlight: 4-Zone RGB

This is on pretty fresh install of EndeavourOS

<details><summary>sudo dmidecode -t system</summary>
<p>
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RD
	Version: Legion 5 15ARH7H
	Serial Number: [Redacted]
	UUID: [Redacted]
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RD_BU_idea_FM_Legion 5 15ARH7H
	Family: Legion 5 15ARH7H

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
</p>
</details> 

<details><summary>sudo dmidecode -t bios</summary>
<p>
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JUCN58WW
	Release Date: 02/09/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
[... 1815 lines cut ...]
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: 3
Methods:
Name: MEM_Get_OC_Status
WmiMethodId: 1
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Get Memory OC Status
Implemented: True

Name: MEM_Set_OC_Status
WmiMethodId: 2
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Set Memory OC Status
Implemented: True

Name: MEM_Set_OC_Data
WmiMethodId: 3
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Set Memory OC Data
Implemented: True
</p>
</details>

Frankiscophil, 2023-04-22:
**Model and Debug Info**

**Model name:** Lenovo Legion 5i Pro Y9000P IAH7H
**CPU model:** 12th Gen Intel Core i7-12700H
**GPU model:** NVIDIA RTX 3060
**Keyboard backlight:** Single color with off/medium/bright

Running on Pop!_OS 22.04

<details>
<summary><b>Output of </b><code>sudo dmidecode -t system</code><b>:</b></summary>
<br>
<pre>
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RF
	Version: Legion Y9000P IAH7H
	Serial Number: ********
	UUID: ********
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RF_BU_idea_FM_Legion Y9000P IAH7H
	Family: Legion Y9000P IAH7H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
</pre>
</details>

<details>
<summary><b>Output of </b><code>sudo dmidecode -t bios</code><b>:</b></summary>
<br>
<pre>
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: J2CN49WW
	Release Date: 11/25/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.49
	Firmware Revision: 1.49

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
</pre>
</details>

<details>
<summary><b>Output of </b><code>sudo cat /sys/kernel/debug/legion/fancurve</code><b>:</b></summary>
<br>
<pre>
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 50
CPU temperature EC error: 0
CPU temperature EC: 50
CPU temperature ACPI error: 0
CPU temperature ACPI: 50
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 50
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2077
1 fanspeed EC error: 0
1 fanspeed EC: 2077
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2000
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 2000
2 fanspeed error: 0
2 fanspeed: 2184
2 fanspeed EC error: 0
2 fanspeed EC: 2184
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2100
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 2100
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 3
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 5	 5	 0	 58	 0	 64	 0	 37
1700	 1600	 5	 5	 54	 58	 59	 64	 33	 42
1800	 1800	 5	 5	 54	 58	 59	 64	 38	 48
2000	 2100	 5	 5	 54	 58	 59	 64	 43	 127
2200	 2300	 2	 2	 54	 62	 59	 64	 127	 127
2500	 2600	 2	 2	 58	 66	 59	 64	 127	 127
2800	 2900	 2	 2	 62	 70	 59	 64	 127	 127
3000	 3100	 2	 2	 66	 74	 59	 69	 127	 127
3400	 3500	 2	 2	 70	 78	 64	 75	 127	 127
3700	 3700	 2	 2	 74	 127	 71	 127	 127	 127
=====================
</pre>
</details>

<h2>Inspecting WMI Entries</h2>
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11300415/fwts_wmi.log)

<h2>Disassembling ACPI tables</h2>
[DSDT.dsl.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11300411/DSDT.dsl.zip)

wondersalmon, 2023-04-23:
Model and Debug Info

Model name: **Legion 5 Pro 16ARH7H** September 2022

CPU model: **AMD Ryzen 7 6800H**

GPU model: **NVIDIA RTX 3060**

Keyboard backlight: **RGB**, 4 separate color controllable lights.

**Output of sudo dmidecode -t system**
 

    dmidecode 3.5
    Getting SMBIOS data from sysfs.
    SMBIOS 3.4.0 present.
    
    Handle 0x0001, DMI type 1, 27 bytes
    System Information
    	Manufacturer: LENOVO
    	Product Name: 82RG
    	Version: Legion 5 Pro 16ARH7H
    	Wake-up Type: Power Switch
    	SKU Number: LENOVO_MT_82RG_BU_idea_FM_Legion 5 Pro 16ARH7H
    	Family: Legion 5 Pro 16ARH7H
    
    Handle 0x001B, DMI type 12, 5 bytes
    System Configuration Options
    	Option 1: String1 for Type12 Equipment Manufacturer
    	Option 2: String2 for Type12 Equipment Manufacturer
    	Option 3: String3 for Type12 Equipment Manufacturer
    	Option 4: String4 for Type12 Equipment Manufacturer
    
    Handle 0x0029, DMI type 32, 20 bytes
    System Boot Information
    	Status: No errors detected

**Output of sudo dmidecode -t bios**

    # dmidecode 3.5
    Getting SMBIOS data from sysfs.
    SMBIOS 3.4.0 present.
    
    Handle 0x0000, DMI type 0, 26 bytes
    BIOS Information
    	Vendor: LENOVO
    	Version: JUCN59WW
    	Release Date: 03/15/2023
    	Address: 0xE0000
    	Runtime Size: 128 kB
    	ROM Size: 32 MB
    	Characteristics:
    		PCI is supported
    		BIOS is upgradeable
    		BIOS shadowing is allowed
    		Boot from CD is supported
    		Selectable boot is supported
    		EDD is supported
    		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
    		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
    		5.25"/360 kB floppy services are supported (int 13h)
    		5.25"/1.2 MB floppy services are supported (int 13h)
    		3.5"/720 kB floppy services are supported (int 13h)
    		3.5"/2.88 MB floppy services are supported (int 13h)
    		8042 keyboard services are supported (int 9h)
    		CGA/mono video services are supported (int 10h)
    		ACPI is supported
    		USB legacy is supported
    		BIOS boot specification is supported
    		Targeted content distribution is supported
[... 1385 lines cut ...]
Methods:
Name: MEM_Get_OC_Status
WmiMethodId: 1
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Get Memory OC Status
Implemented: True

Name: MEM_Set_OC_Status
WmiMethodId: 2
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Set Memory OC Status
Implemented: True

Name: MEM_Set_OC_Data
WmiMethodId: 3
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Set Memory OC Data
Implemented: True
</details>

If you need more information, I'm happy to provide. You are doing a good job
[legion.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11304352/legion.tar.gz)

RzVa317, 2023-04-26:
Model name: Legion S7
CPU model: AMD Ryzen 7 6800H
GPU model: Radeon RX 6600S
Keyboard backlight: RGB

<details>

<summary>sudo dmidecode -t system</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
Manufacturer: LENOVO
Product Name: 82UG
Version: Legion S7 16ARHA7
Wake-up Type: Power Switch
SKU Number: LENOVO_MT_82UG_BU_idea_FM_Legion S7 16ARHA7
Family: Legion S7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
Option 1: String1 for Type12 Equipment Manufacturer
Option 2: String2 for Type12 Equipment Manufacturer
Option 3: String3 for Type12 Equipment Manufacturer
Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
Status: No errors detected
```

</details>

<details>

<summary>sudo dmidecode -t bios</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
Vendor: LENOVO
Version: KFCN32WW
Release Date: 09/29/2022
Address: 0xE0000
Runtime Size: 128 kB
ROM Size: 32 MB
Characteristics:
PCI is supported
BIOS is upgradeable
BIOS shadowing is allowed
Boot from CD is supported
Selectable boot is supported
EDD is supported
Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
5.25"/360 kB floppy services are supported (int 13h)
5.25"/1.2 MB floppy services are supported (int 13h)
3.5"/720 kB floppy services are supported (int 13h)
3.5"/2.88 MB floppy services are supported (int 13h)
8042 keyboard services are supported (int 9h)
CGA/mono video services are supported (int 10h)
ACPI is supported
USB legacy is supported
BIOS boot specification is supported
Targeted content distribution is supported
UEFI is supported
BIOS Revision: 1.32
Firmware Revision: 1.32

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
Language Description Format: Long
Installable Languages: 4
en|US|iso8859-1
fr|FR|iso8859-1
ja|JP|unicode
zh|TW|unicode
Currently Installed Language: en|US|iso8859-1
```

</details>

<details>

<summary>sudo cat /sys/kernel/debug/legion/fancurve</summary>

```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 48
CPU temperature EC error: 0
CPU temperature EC: 48
CPU temperature ACPI error: 0
CPU temperature ACPI: 48
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 48
GPU temperature error: 0
GPU temperature: 44
GPU temperature EC error: 0
GPU temperature EC: 44
GPU temperature ACPI error: 0
GPU temperature ACPI: 45
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 45
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2063
1 fanspeed EC error: 0
1 fanspeed EC: 2063
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2000
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 2000
2 fanspeed error: 0
2 fanspeed: 2020
2 fanspeed EC error: 0
2 fanspeed EC: 2020
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2000
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 2000
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 3
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       60      80      70      76      0       32
1500     1600    2       2       60      80      70      76      31      34
1700     1800    2       2       60      80      70      76      33      40
2000     2000    2       2       60      80      70      76      39      46
2200     2200    2       2       60      80      70      76      45      50
2500     2500    2       2       60      80      70      76      49      53
2800     2800    2       2       60      80      70      76      52      56
3200     3200    2       2       60      80      70      76      55      58
3500     3500    2       2       60      80      70      76      57      61
3500     3500    2       2       60      80      70      76      60      64

```
</details>

[acpi.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11328438/acpi.tar.gz)
[fwts.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11328442/fwts.tar.gz)
[windows_wmi.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11328463/windows_wmi.txt)

Thank you for your great work on this!

johnfanv2, 2023-04-29:
@javaarchive @vasubawa @Frankiscophil @wondersalmon @RzVa317  Thanks for the provided data. I created a new version based on that. Please download the new python GUI and the new kernel module (and load it). I

In the other options tab you should have many new features. Please test the following and upload the log from the new "log tab" or outputted to the console (should be same).

# 1. Test: Custom Power Mode
Please select the custom power mode. The LED for the power mode on the laptop should turn pink(ish).
Does it work?

# 2. Test: Max Fan Speed in Custom Mode
Please enable the max fan speed on the fan curve tab while being in custom power mode. This should set the speed of the fans to maximum. Then you can disable it again.
Does it work?

# 3. Test: Check CPU Power Limits in Custom Mode
Do you have reasonable values for the all the different CPU power limits? Nothing should be marked red or be unreasonable.

# 3. Test: Modify CPU Power Limit in Custom Mode
While in custom mode, change the power limit for the CPU (e.g. 1 or 2 W; sometimes only 5W steps are possible). Are the values accepted when you click "Apply to HW"? You can check if this really changes something if you use ryzendadj (for AMD) or similar for Intel.

# 4. Test: Check GPU Power Limits in Custom Mode
Do you have reasonable values for the all the different GPU power limits/temperature limits? Nothing should be marked red or be unreasonable.

# 5. Test: Modify GPU Power Limit in Custom Mode
While in custom mode, change the power limit for the GPU (e.g. 1 or 2 W; sometimes only 5W steps are possible). Are the values accepted when you click "Apply to HW"?

# 6. Test: Check CPU or GPU overclock
Can you enable CPU and GPU overclock? Is it marked red (error)? Unfortunately, this is not available in my older model so I am not sure what it does exactly. 

# 7. Test: Rapid Charge and Battery conservation mode
- Please enable rapid charge while the battery is charging. Does it increase charging speed or decreased time until fully chaged?
- Does enabling rapid charge disable conservation mode?
- Does enabling conservation mode stops charging?
- Does enabling conservation mode disables rapid charging?

The GSync option might be used to disable hybrid mode after reboot but I have not fully debugged it. In particular, it will not change immediately (only after reboot).

javaarchive, 2023-05-01:
This is some very early data but here's some output after running `sudo make reloadmodule`
```
191607.424372] Unloading legion WMI
[191607.424458] Unloading legion WMI done
[191607.424459] Unloading legion platform profile
[191607.424461] Unloading legion platform profile done
[191608.926113] Unloading legion hwon
[191608.926238] Unloading legion hwon done
[191608.926240] Unloading legion sysfs
[191608.926244] Unloading legion sysfs done
[191608.926245] Unloading legion dubugfs
[191608.926260] Unloading legion dubugfs done
[191608.926261] Unloading legion ecram
[191608.926263] Unloading legion ecram done
[191608.926264] Unloading legion shared
[191608.926264] Unloading legion shared done
[191608.926265] Legion platform unloaded
[191608.926344] legion_laptop exit
[191608.944346] legion_laptop: legion_laptop starts loading
[191608.944414] legion PNP0C09:00: legion_laptop platform driver probing
[191608.944416] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[191608.944417] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[191608.944419] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[191608.944420] legion PNP0C09:00: Using configuration for system: JUCN
[191608.944452] legion PNP0C09:00: ACPI CFG: 2081289482
[191608.944536] legion PNP0C09:00: Read embedded controller ID 0x8227
[191608.944537] legion PNP0C09:00: Creating debugfs inteface
[191608.944542] legion_laptop: Creating sysfs inteface
[191608.944563] legion_laptop: Creating hwmon interface
[191608.944621] legion_laptop: Creating platform profile support
[191608.944623] legion_laptop: Init WMI driver support
[191608.944638] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[191608.944658] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[191608.944666] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[191608.944678] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[191608.944686] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[191608.944692] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[191608.944699] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[191608.944713] legion_wmi 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294: Register after probing for WMI.
[191608.944728] legion_laptop: Init keyboard backlight LED driver
[191609.266011] legion_laptop: Error ACPI call for reading keyboard brightness: expected a value between 1 and 3, but got 0
[191609.266020] legion_laptop: Error reading keyboard brighntess
[191609.266024] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[191609.266028] legion PNP0C09:00: legion_laptop loaded for this device
[191609.706835] legion_laptop: Fan event: legion type: 2;  acpi type: 3 (1=integer)
[191648.996093] legion_laptop: Unloading legion WMI
[191648.996176] legion_laptop: Unloading legion WMI done
[191648.996178] legion_laptop: Unloading legion platform profile
[191648.996180] legion_laptop: Unloading legion platform profile done
[191648.996273] legion_laptop: Set powermode
[191648.996275] legion_laptop: Unexpected power mode value ignored: 0
[191650.497268] legion_laptop: Set powermode
[191651.573748] legion_laptop: Unloading legion hwon
[191651.573914] legion_laptop: Unloading legion hwon done
[191651.573916] legion_laptop: Unloading legion sysfs
[191651.573932] legion_laptop: Unloading legion sysfs done
[191651.573933] legion_laptop: Unloading legion dubugfs
[191651.573948] legion_laptop: Unloading legion dubugfs done
[191651.573949] legion_laptop: Unloading legion ecram
[191651.573951] legion_laptop: Unloading legion ecram done
[191651.573952] legion_laptop: Unloading legion shared
[191651.573952] legion_laptop: Unloading legion shared done
[191651.573953] legion_laptop: Legion platform unloaded
[191651.574037] legion_laptop: legion_laptop exit
[191651.587141] legion_laptop: legion_laptop starts loading
[191651.587212] legion PNP0C09:00: legion_laptop platform driver probing
[191651.587214] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[191651.587215] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[191651.587217] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[191651.587218] legion PNP0C09:00: Using configuration for system: JUCN
[191651.587249] legion PNP0C09:00: ACPI CFG: 2081289482
[191651.587332] legion PNP0C09:00: Read embedded controller ID 0x8227
[191651.587333] legion PNP0C09:00: Creating debugfs inteface
[191651.587337] legion_laptop: Creating sysfs inteface
[191651.587346] legion_laptop: Creating hwmon interface
[191651.587391] legion_laptop: Creating platform profile support
[191651.587399] legion_laptop: Init WMI driver support
[191651.587419] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[191651.587442] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[191651.587449] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[... cut ...]
[191651.587561] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[191651.587561] legion PNP0C09:00: legion_laptop loaded for this device
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
```
(I'm not sure if this unloaded the nobara builtin module). 
Keyboard backlight being 0 might be expected, I had some other program turn off the backlight so it's actually not doing any rgb. I'll get to reinstalling the gui in another comment.

javaarchive, 2023-05-01:
Log from gui "log". 
```
2023-04-30 21:55:30,700 - INFO: [    6.738828] legion_laptop starts loading
[    6.740837] legion PNP0C09:00: legion_laptop platform driver probing
[    6.741738] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[    6.744504] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[    6.745463] legion PNP0C09:00: Using configuration for system: JUCN
[    6.746437] legion PNP0C09:00: Read embedded controller ID 0x8227
[    6.747177] legion PNP0C09:00: Creating debugfs inteface
[    6.758240] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[    6.759705] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[    6.761238] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[    6.762480] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[    6.764150] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[    6.765170] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[    6.766328] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[    6.768075] legion_wmi 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294: Register after probing for WMI.
[    6.770841] legion PNP0C09:00: legion_laptop loaded for this device
[26515.366341] legion PNP0C09:00: Resumed PM in legion-laptop
[27442.358763] legion PNP0C09:00: Resumed PM in legion-laptop
[28294.176630] legion PNP0C09:00: Resumed PM in legion-laptop
[28644.655138] legion PNP0C09:00: Resumed PM in legion-laptop
[28720.098537] legion PNP0C09:00: Resumed PM in legion-laptop
[60411.768114] legion PNP0C09:00: Resumed PM in legion-laptop
[60701.959051] legion PNP0C09:00: Resumed PM in legion-laptop
[60914.222674] legion PNP0C09:00: Resumed PM in legion-laptop
[71461.107681] legion PNP0C09:00: Resumed PM in legion-laptop
[147099.007286] legion PNP0C09:00: Resumed PM in legion-laptop
[148606.423354] legion PNP0C09:00: Resumed PM in legion-laptop
[191607.424372] Unloading legion WMI
[191607.424458] Unloading legion WMI done
[191607.424459] Unloading legion platform profile
[191607.424461] Unloading legion platform profile done
[191608.926113] Unloading legion hwon
[191608.926238] Unloading legion hwon done
[191608.926240] Unloading legion sysfs
[191608.926244] Unloading legion sysfs done
[191608.926245] Unloading legion dubugfs
[191608.926260] Unloading legion dubugfs done
[191608.926261] Unloading legion ecram
[191608.926263] Unloading legion ecram done
[191608.926264] Unloading legion shared
[191608.926264] Unloading legion shared done
[191608.926344] legion_laptop exit
[191608.944346] legion_laptop: legion_laptop starts loading
[191608.944414] legion PNP0C09:00: legion_laptop platform driver probing
[191608.944416] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[191608.944417] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82RG; DMI_BIOS_VERSION:JUCN55WW
[191608.944419] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[191608.944420] legion PNP0C09:00: Using configuration for system: JUCN
[191608.944452] legion PNP0C09:00: ACPI CFG: 2081289482
[191608.944536] legion PNP0C09:00: Read embedded controller ID 0x8227
[191608.944537] legion PNP0C09:00: Creating debugfs inteface
[191608.944542] legion_laptop: Creating sysfs inteface
[191608.944563] legion_laptop: Creating hwmon interface
[191608.944621] legion_laptop: Creating platform profile support
[191608.944623] legion_laptop: Init WMI driver support
[191608.944638] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[191608.944658] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[191608.944666] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[191608.944678] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[191608.944686] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[191608.944692] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[191608.944699] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[191608.944713] legion_wmi 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294: Register after probing for WMI.
[191608.944728] legion_laptop: Init keyboard backlight LED driver
[191609.266011] legion_laptop: Error ACPI call for reading keyboard brightness: expected a value between 1 and 3, but got 0
[191609.266020] legion_laptop: Error reading keyboard brighntess
[191609.266024] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[191609.266028] legion PNP0C09:00: legion_laptop loaded for this device
[... 208 lines cut ...]
            ^^^^^^^^^^^^^^^^^^
  File "/home/raymond/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 191, in get
    return self._read_file_int(self.filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raymond/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 113, in _read_file_int
    return int(self._read_file_str(file_path))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
INFO:legion_linux.legion:Feature StrFileFeature reading: quiet balanced balanced-performance performance
Set to value: balanced-performance
INFO:legion_linux.legion:Feature PlatformProfileFeature writing: balanced-performance
update_view_from_feature 0
INFO:legion_linux.legion:Feature StrFileFeature reading
INFO:legion_linux.legion:Feature StrFileFeature reading: quiet balanced balanced-performance performance
INFO:legion_linux.legion:Feature PlatformProfileFeature reading
INFO:legion_linux.legion:Feature PlatformProfileFeature reading: balanced-performance
```

arabcian, 2023-05-04:
Hi im Cihan from discord

# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: GKCN60WW
	Release Date: 03/07/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.60
	Firmware Revision: 1.60

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

-------------------------------------------------------------------------------------------------------------------------------------------------------------------------

EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 50
CPU temperature EC error: 0
CPU temperature EC: 50
CPU temperature ACPI error: 0
CPU temperature ACPI: 50
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 50
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 42
[... 151 lines cut ...]
Parsing completed

Found 10 external control methods, reparsing with new information
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed
Disassembly completed
ASL Output:    DSDT.dsl - 1213822 bytes

 iASL Warning: There were 10 external control methods found during
 disassembly, but only 7 were resolved (3 unresolved). Additional
 ACPI tables may be required to properly disassemble the code. This
 resulting disassembler output file may not compile because the
 disassembler did not know how many arguments to assign to the
 unresolved methods. Note: SSDTs can be dynamically loaded at
 runtime and may or may not be available via the host OS.

 In addition, the -fe option can be used to specify a file containing
 control method external declarations with the associated method
 argument counts. Each line of the file must be of the form:
     External (<method pathname>, MethodObj, <argument count>)
 Invocation:
     iasl -fe refs.txt -d dsdt.aml

arabcian, 2023-05-05:
1-Custom power mode works
2-Max fan speed greyed out, havent tested the cli
3-Cpu power limits changes accordingly. Everything in order.
4-cTGP and Dynamic boost limit could be changed.
5-Doestn affect GPU power limits. But should be cared in Windows.
6-RED
7-Both battery conservation and rapid charge works but cause a crash on GUI.

2023-05-05 14:55:54,126 - INFO: [    0.920101] legion_laptop: loading out-of-tree module taints kernel.
[    0.920654] legion_laptop: legion_laptop starts loading
[    0.920677] legion PNP0C09:00: legion_laptop platform driver probing
[    0.920678] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82JQ; DMI_BIOS_VERSION:GKCN60WW
[    0.920679] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82JQ; DMI_BIOS_VERSION:GKCN60WW
[    0.920680] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[    0.920681] legion PNP0C09:00: Using configuration for system: GKCN
[    0.920735] legion PNP0C09:00: ACPI CFG: 2081289482
[    0.920736] legion_laptop: Creating RAM access to embedded controller
[    0.920745] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x00000000a91bc04e
[    0.920801] legion PNP0C09:00: Read embedded controller ID 0x8227
[    0.920802] legion PNP0C09:00: Creating debugfs inteface
[    0.920807] legion_laptop: Creating sysfs inteface
[    0.920820] legion_laptop: Creating hwmon interface
[    0.920885] legion_laptop: Creating platform profile support
[    0.920887] legion_laptop: Init WMI driver support
[    0.920894] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[    0.920911] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[    0.920917] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[    0.920922] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[    0.920927] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[    0.920932] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[    0.920936] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[    0.920952] legion_wmi 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294: Register after probing for WMI.
[    0.920961] legion_laptop: Init keyboard backlight LED driver
[    0.921021] legion_laptop: Init Y-Logo LED driver
[    0.921069] legion_laptop: Init IO-Port LED driver
[    0.921091] legion_laptop: Error ACPI call for reading keyboard brightness: expected a value between 1 and 2, but got 0
[    0.921092] legion_laptop: Error reading brighntess for light: 5
[    0.921092] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[    0.921093] legion PNP0C09:00: legion_laptop loaded for this device
[    2.470421] legion_laptop: Set fancurve
[    2.475750] legion_laptop: Set fancurve
[    2.481049] legion_laptop: Set fancurve
[    2.486531] legion_laptop: Set fancurve
[    2.492073] legion_laptop: Set fancurve
[    2.497562] legion_laptop: Set fancurve
[    2.503022] legion_laptop: Set fancurve
[    2.508285] legion_laptop: Set fancurve
[    2.513539] legion_laptop: Set fancurve
[    2.518759] legion_laptop: Set fancurve
[    2.524129] legion_laptop: Set fancurve
[    2.529636] legion_laptop: Set fancurve
[    2.535285] legion_laptop: Set fancurve
[    2.542265] legion_laptop: Set fancurve
[    2.547320] legion_laptop: Set fancurve
[    2.552601] legion_laptop: Set fancurve
[    2.557923] legion_laptop: Set fancurve
[    2.563474] legion_laptop: Set fancurve
[    2.569222] legion_laptop: Set fancurve
[    2.574884] legion_laptop: Set fancurve
[    2.580552] legion_laptop: Set fancurve
[    2.586262] legion_laptop: Set fancurve
[    2.591887] legion_laptop: Set fancurve
[    2.597603] legion_laptop: Set fancurve
[    2.603271] legion_laptop: Set fancurve
[    2.608906] legion_laptop: Set fancurve
[    2.614562] legion_laptop: Set fancurve
[    2.620264] legion_laptop: Set fancurve
[    2.625899] legion_laptop: Set fancurve
[    2.631695] legion_laptop: Set fancurve
[    2.637328] legion_laptop: Set fancurve
[... 172 lines cut ...]

2023-05-05 14:55:54,152 - INFO: Feature OverdriveFeature reading
2023-05-05 14:55:54,152 - INFO: Feature OverdriveFeature reading: 0
2023-05-05 14:55:54,152 - INFO: Feature GsyncFeature reading
2023-05-05 14:55:54,152 - INFO: Feature GsyncFeature reading: 0
2023-05-05 14:55:54,152 - INFO: Feature StrFileFeature reading
2023-05-05 14:55:54,152 - INFO: Feature StrFileFeature reading: quiet balanced balanced-performance performance
2023-05-05 14:55:54,152 - INFO: Feature PlatformProfileFeature reading
2023-05-05 14:55:54,152 - INFO: Feature PlatformProfileFeature reading: balanced-performance
2023-05-05 14:55:54,152 - INFO: Feature CPULongtermPowerLimit reading
2023-05-05 14:55:54,153 - INFO: Feature CPULongtermPowerLimit reading: 80
2023-05-05 14:55:54,153 - INFO: Feature CPUShorttermPowerLimit reading
2023-05-05 14:55:54,153 - INFO: Feature CPUShorttermPowerLimit reading: 88
2023-05-05 14:55:54,153 - INFO: Feature CPUPeakPowerLimit reading
2023-05-05 14:55:54,153 - INFO: Feature CPUPeakPowerLimit reading: 0
2023-05-05 14:55:54,153 - INFO: Feature CPUCrossLoadingPowerLimit reading
2023-05-05 14:55:54,153 - INFO: Feature CPUCrossLoadingPowerLimit reading: 0
2023-05-05 14:55:54,153 - INFO: Feature CPUAPUSPPTPowerLimit reading
2023-05-05 14:55:54,153 - INFO: Feature CPUAPUSPPTPowerLimit reading: 0
2023-05-05 14:55:54,154 - INFO: Feature GPUCTGPPowerLimit reading
2023-05-05 14:55:54,154 - INFO: Feature GPUCTGPPowerLimit reading: 35
2023-05-05 14:55:54,154 - INFO: Feature GPUPPABPowerLimit reading
2023-05-05 14:55:54,154 - INFO: Feature GPUPPABPowerLimit reading: 15
2023-05-05 14:55:54,154 - INFO: Feature GPUTemperatureLimit reading
2023-05-05 14:55:54,154 - INFO: Feature GPUTemperatureLimit reading: 0

foobar101118, 2023-05-05:
Model name: Legion 5i Pro Gen 7 16IAH7H
CPU model: Intel i7-12700h
GPU model: NVIDIA RTX 3070ti
Keyboard backlight: RGB

<details>
<summary>sudo dmidecode -t system</summary>

```bash   
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RF
	Version: Legion 5 Pro 16IAH7H
	Serial Number: XXXXXXXX
	UUID: XXXXXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RF_BU_idea_FM_Legion 5 Pro 16IAH7H
	Family: Legion 5 Pro 16IAH7H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```
</details>
<details>
<summary>sudo dmidecode -t bios</summary>

```bash
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: J2CN49WW
	Release Date: 11/25/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
[... 122 lines cut ...]
2800	 2900	 2	 2	 62	 70	 60	 69	 127	 127
3000	 3100	 2	 2	 66	 75	 65	 74	 127	 127
3400	 3500	 2	 2	 70	 79	 69	 82	 127	 127
3700	 3700	 2	 2	 75	 127	 78	 127	 127	 127
=====================

```
</details>

#### WMI entries
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11406908/fwts_wmi.log)

#### ACPI tables
[dsdt.dsl.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11406927/dsdt.dsl.tar.gz)

## New features test

1. Test: Custom Power Mode: Yes, it turns pinkish
2. Test: Max Fan Speed in Custom Mode: Option greyed-out even in custom power mode
3. Test: Check CPU Power Limits in Custom Mode: Options more or less reasonable - long term always stays at 60w, short term changes depending on the mode. They seem to be capped to 100 (due to [this](https://github.com/johnfanv2/LenovoLegionLinux/blob/752dd84ee0bb55d2b7a740ccf1a11f219e087a56/python/legion_linux/legion_linux/legion.py#L366)) even though my CPU can go higher (up to 135)
4. Test: Modify CPU Power Limit in Custom Mode: Changed short term PL from 90 to 95 in custom mode - can confirm that it worked
5. Test: Check GPU Power Limits in Custom Mode: cTGP is always 0, PPAB is 15, they don't seem reasonable to me
6. Test: Modify GPU Power Limit in Custom Mode: Didn't test due to the above
7. Test: Check CPU or GPU overclock: Unable to enable - they are red
8. Test: Rapid Charge and Battery conservation mode: Battery conservation mode has always worked; couldn't test rapid charge

RzVa317, 2023-05-08:
Model name: Legion S7
CPU model: AMD Ryzen 7 6800H
GPU model: Radeon RX 6600S
Keyboard backlight: RGB
# New tests
1. Custom power mode doesn't turn LED pink for me (on battery power)
	- Going to quiet mode turns light blue and turns off keyboard backlight
	- Selecting balanced turns light white and turns on keyboard backlight and then the GUI automatically switches to Custom Mode
	(Later for test 7 when I plugged my laptop in the light did turn pink)
2. Checkbox for max speed greyed out in custom power mode
3. - Only CPU Long Term Power Limit (85) and Short Term Power Limit (100) have values, all others are 0
	- Unable to verify any changes with ryzenadj. I'm not familiar with ryzenadj and will need to do some more googling
4. All GPU values are 0
	- cTGP Power Limit - 0
	- PPAB Power Limit - 0
	- Temperature Limit - 0
5. Same as above
6. CPU Overclock is marked red, GPU overclock is visible but when trying to select it, the box unchecks itself
7.
	- Disabled rapid charging and plugged laptop in (54 mins to fully charged), enabled again after a few minutes it went to 32 mins to fully charged
	- Enabling rapid charging disables conservation
	- Enabled conservation at 42% -- time to charged went back up -- charging didn't stop at 50% but did stop at 80%
	- Enabling conservation disables rapid charging

<details>
<summary>Log Output</summary>

2023-05-08 17:23:47,975 - INFO: [30432.078093]  snd_hda_scodec_cs35l41_spi snd_acp_pci drm_buddy snd_intel_sdw_acpi mt76 snd_pci_acp6x snd_hda_scodec_cs35l41_i2c gpu_sched bluetooth snd_hda_codec kvm snd_pci_acp5x snd_hda_scodec_cs35l41 snd_hda_core i2c_algo_bit snd_rn_pci_acp3x ecdh_generic snd_hwdep drm_ttm_helper irqbypass snd_hda_cs_dsp_ctls snd_acp_config mac80211 snd_pcm ttm rapl cs_dsp snd_soc_acpi ideapad_laptop think_lmi vfat snd_soc_cs35l41_lib snd_timer firmware_attributes_class pcspkr sparse_keymap wmi_bmof drm_display_helper fat k10temp i2c_piix4 snd_pci_acp3x libarc4 legion_laptop(OE) cec platform_profile snd soundcore joydev amd_pmc serial_multi_instantiate acpi_tad acpi_cpufreq mousedev psmouse mac_hid vmmon(OE) vmw_vmci ryzen_smu(OE) fuse loop ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee usbhid wl(POE) serio_raw atkbd crct10dif_pclmul crc32_pclmul crc32c_intel libps2 polyval_clmulni vivaldi_fmap hid_multitouch nvme sdhci_pci polyval_generic cfg80211 gf128mul nvme_core
[31822.613127]  snd_hda_scodec_cs35l41_spi snd_acp_pci drm_buddy snd_intel_sdw_acpi mt76 snd_pci_acp6x snd_hda_scodec_cs35l41_i2c gpu_sched bluetooth snd_hda_codec kvm snd_pci_acp5x snd_hda_scodec_cs35l41 snd_hda_core i2c_algo_bit snd_rn_pci_acp3x ecdh_generic snd_hwdep drm_ttm_helper irqbypass snd_hda_cs_dsp_ctls snd_acp_config mac80211 snd_pcm ttm rapl cs_dsp snd_soc_acpi ideapad_laptop think_lmi vfat snd_soc_cs35l41_lib snd_timer firmware_attributes_class pcspkr sparse_keymap wmi_bmof drm_display_helper fat k10temp i2c_piix4 snd_pci_acp3x libarc4 legion_laptop(OE) cec platform_profile snd soundcore joydev amd_pmc serial_multi_instantiate acpi_tad acpi_cpufreq mousedev psmouse mac_hid vmmon(OE) vmw_vmci ryzen_smu(OE) fuse loop ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee usbhid wl(POE) serio_raw atkbd crct10dif_pclmul crc32_pclmul crc32c_intel libps2 polyval_clmulni vivaldi_fmap hid_multitouch nvme sdhci_pci polyval_generic cfg80211 gf128mul nvme_core
[31824.644485]  snd_hda_scodec_cs35l41_spi snd_acp_pci drm_buddy snd_intel_sdw_acpi mt76 snd_pci_acp6x snd_hda_scodec_cs35l41_i2c gpu_sched bluetooth snd_hda_codec kvm snd_pci_acp5x snd_hda_scodec_cs35l41 snd_hda_core i2c_algo_bit snd_rn_pci_acp3x ecdh_generic snd_hwdep drm_ttm_helper irqbypass snd_hda_cs_dsp_ctls snd_acp_config mac80211 snd_pcm ttm rapl cs_dsp snd_soc_acpi ideapad_laptop think_lmi vfat snd_soc_cs35l41_lib snd_timer firmware_attributes_class pcspkr sparse_keymap wmi_bmof drm_display_helper fat k10temp i2c_piix4 snd_pci_acp3x libarc4 legion_laptop(OE) cec platform_profile snd soundcore joydev amd_pmc serial_multi_instantiate acpi_tad acpi_cpufreq mousedev psmouse mac_hid vmmon(OE) vmw_vmci ryzen_smu(OE) fuse loop ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee usbhid wl(POE) serio_raw atkbd crct10dif_pclmul crc32_pclmul crc32c_intel libps2 polyval_clmulni vivaldi_fmap hid_multitouch nvme sdhci_pci polyval_generic cfg80211 gf128mul nvme_core
[31972.990317] legion PNP0C09:00: Resumed PM in legion-laptop
[31979.765117]  snd_rpl_pci_acp6x btmtk mt76_connac_lib snd_intel_dspcfg kvm_amd snd_hda_scodec_cs35l41_spi snd_acp_pci drm_buddy snd_intel_sdw_acpi mt76 snd_pci_acp6x snd_hda_scodec_cs35l41_i2c gpu_sched bluetooth snd_hda_codec kvm snd_pci_acp5x snd_hda_scodec_cs35l41 snd_hda_core i2c_algo_bit snd_rn_pci_acp3x ecdh_generic snd_hwdep drm_ttm_helper irqbypass snd_hda_cs_dsp_ctls snd_acp_config mac80211 snd_pcm ttm rapl cs_dsp snd_soc_acpi ideapad_laptop think_lmi vfat snd_soc_cs35l41_lib snd_timer firmware_attributes_class pcspkr sparse_keymap wmi_bmof drm_display_helper fat k10temp i2c_piix4 snd_pci_acp3x libarc4 legion_laptop(OE) cec platform_profile snd soundcore joydev amd_pmc serial_multi_instantiate acpi_tad acpi_cpufreq mousedev psmouse mac_hid vmmon(OE) vmw_vmci ryzen_smu(OE) fuse loop ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee usbhid wl(POE) serio_raw a [... cut ...]
2023-05-08 17:44:48,243 - INFO: Feature BatteryConservation reading: 0
2023-05-08 17:47:09,102 - INFO: Feature RapidChargingFeature writing: 0
2023-05-08 17:47:09,103 - INFO: Feature BatteryConservation writing: 1
2023-05-08 17:47:09,203 - INFO: Feature BatteryConservation reading
2023-05-08 17:47:09,204 - INFO: Feature BatteryConservation reading: 1
2023-05-08 17:47:09,305 - INFO: Feature RapidChargingFeature reading
2023-05-08 17:47:09,306 - INFO: Feature RapidChargingFeature reading: 0

</details>

Peter-Levine, 2023-05-09:
Model name: Lenovo Legion 5 Pro-16ACH6H Laptop - Type 82JQ
CPU model: AMD Ryzen 7 5800H
GPU model: Nvidia GeForce RTX 3070 Mobile
Keyboard backlight: off
<details>

<summary>sudo dmidecode -t system</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JQ
        Version: Legion 5 Pro 16ACH6H
        Serial Number:
        UUID:
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JQ_BU_idea_FM_Legion 5 Pro 16ACH6H
        Family: Legion 5 Pro 16ACH6H

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```
</details>

<details>
<summary>sudo dmidecode -t bios</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN60WW
        Release Date: 03/07/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.60
        Firmware Revision: 1.60

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```
</details>

<details>
<summary>sudo cat /sys/kernel/debug/legion/fancurve</summary>

```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 46
CPU temperature EC error: 0
CPU temperature EC: 46
CPU temperature ACPI error: 0
CPU temperature ACPI: 46
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 46
GPU temperature error: 0
GPU temperature: 39
GPU temperature EC error: 0
GPU temperature EC: 39
GPU temperature ACPI error: 0
GPU temperature ACPI: 39
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 39
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1868
1 fanspeed EC error: 0
1 fanspeed EC: 1868
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1800
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1800
2 fanspeed error: 0
2 fanspeed: 1755
2 fanspeed EC error: 0
2 fanspeed EC: 1755
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1700
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 16
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 1
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       63      0       66      0       43
1800     1700    2       2       58      63      62      66      38      46
1800     1900    2       2       58      63      62      66      42      51
2100     2100    2       2       58      63      62      66      46      127
2300     2400    2       2       58      69      62      66      127     127
2600     2600    2       2       63      76      62      66      127     127
2800     2900    2       2       71      80      62      70      127     127
3500     3500    2       2       75      88      65      75      127     127
3800     3800    2       2       83      94      70      78      127     127
4200     4200    2       2       90      127     74      127     127     127
=====================
```
</details>

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11426594/fwts_wmi.log)
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11426630/acpi_re.tar.gz)

sunkwei, 2023-05-10:
Model name: Legion Y9000P IRX8 (2023)
CPU model: 13th Gen Intel(R) Core(TM) i9-13900HX
GPU model: Nvidia GeForce RTX 4060 Mobile
Keyboard blacklght: off

sudo dmidecode -t system
```
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82WK
        Version: Legion Y9000P IRX8
        Serial Number: PF430F5V
        UUID: cce4366d-a858-11ed-80f3-e880886396a5
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WK_BU_idea_FM_Legion Y9000P IRX8
        Family: Legion Y9000P IRX8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

```

sudo dmidecode -t bios
```
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: KWCN31WW
        Release Date: 02/09/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.31
        Firmware Revision: 1.31

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```

no pth  /sys/kernel/debug/logion

electronarchy, 2023-05-14:
Model name: Lenovo Legion 7 Gen 7 (AMD Advantage Edition)
CPU model: AMD Ryzen 9 6900HX
GPU model: AMD ATI Radeon RX 6850 XT
Keyboard backlight: Per Key RGB, 6 modes -- customizable via Lenovo Vantage in Windows

Output of sudo dmidecode -t system. Please remove Serial Number and UUID for privacy:

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82UH
        Version: Legion 7 16ARHA7
        Serial Number: XXXXXXXX
        UUID: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82UH_BU_idea_FM_Legion 7 16ARHA7
        Family: Legion 7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of sudo dmidecode -t bios:

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: K9CN42WW
        Release Date: 03/10/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.42
        Firmware Revision: 1.42

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of sudo cat /sys/kernel/debug/legion/fancurve:

```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 39
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 39
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 39
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1687
1 fanspeed EC error: 0
1 fanspeed EC: 1687
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1600
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1600
2 fanspeed error: 0
2 fanspeed: 1687
2 fanspeed EC error: 0
2 fanspeed EC: 1687
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1600
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1600
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 1
EC fan curve points size: 4
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       127     0       127     0       37
1600     1600    2       2       127     127     127     127     33      41
1700     1700    2       2       127     127     127     127     38      46
1900     1900    2       2       127     127     127     127     41      127
```

chybz, 2023-06-01:
Hello, here's my contribution. Thanks !
[dmidecode.bios.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11623586/dmidecode.bios.log)
[dmidecode.system.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11623587/dmidecode.system.log)
[fancurve.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11623588/fancurve.log)
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11623589/fwts_wmi.log)

ronyclau, 2023-06-03:
**Model and Debug Info**

Model name: Legion Pro 7 (Intel, Gen 8)
CPU model: Intel i9-13900HX
GPU model: NVIDIA GeForce RTX 4090
Keyboard backlight: Per-key RGB

<details>
<summary>Output of `sudo dmidecode -t system`</summary>

```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
	Serial Number: ********
	UUID: ********-****-****-****-************
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
	Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

</details>

<details>
<summary>Output of `sudo dmidecode -t bios`</summary>

```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
[... 623 lines cut ...]
Pass 2 parse of [SSDT]
External object resolution file           SSDT11
Input file SSDT11, Length 0x53F (1343) bytes
ACPI: SSDT 0x0000000000000000 00053F (v01 LENOVO CB-01    00000001 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT10
Input file SSDT10, Length 0x2016 (8214) bytes
ACPI: SSDT 0x0000000000000000 002016 (v02 LENOVO CB-01    00000001 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file            SSDT1
Input file SSDT1, Length 0xBB3C (47932) bytes
ACPI: SSDT 0x0000000000000000 00BB3C (v02 LENOVO CB-01    00000001 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
Pass 1 parse of [DSDT]
Firmware Error (ACPI): Failure creating named object [\_SB.PC00.XHCI.RHUB.HS14.SADX], AE_ALREADY_EXISTS (20200925/dswload-387)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20200925/psobject-264)
Could not parse ACPI tables, AE_ALREADY_EXISTS
```

</details>

[WMI info in Windows](https://gist.github.com/ronyclau/75214f51072ebd5fd1f85ff179b3e0fd#file-windows_wmi-txt)

404useryv35notfound, 2023-06-06:
hi there
i just got a 2022 legion S7 (16IAH7) and it errors when trying to load the module (errors at the end of post if needed).

Model and Debug Info:

Model name: lenovo legion S7 16IAH7
CPU model: Intel i7 12700H
GPU model: NVIDIA RTX 3060
Keyboard backlight: RGB

**Output of sudo dmidecode -t system:**

╰─λ sudo dmidecode -t system
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x001F, DMI type 32, 11 bytes
System Boot Information
Status: No errors detected

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
Option 1: J6H1:1-X Boot with Default; J8H1:1-X BIOS RECOVERY

Handle 0x001B, DMI type 1, 27 bytes
System Information
Manufacturer: LENOVO
Product Name: 82TF
Version: Legion S7 16IAH7

Wake-up Type: Other
SKU Number: LENOVO_MT_82TF_BU_idea_FM_Legion S7 16IAH7
Family: Legion S7 16IAH7

Handle 0x0060, DMI type 15, 31 bytes
System Event Log
Area Length: 82 bytes
Header Start Offset: 0x0000
Header Length: 16 bytes
Data Start Offset: 0x0010
Access Method: General-purpose non-volatile data functions
Access Address: 0x00F0
Status: Valid, Not Full
Change Token: 0x00000004
Header Format: Type 1
Supported Log Type Descriptors: 4
Descriptor 1: POST error
Data Format 1: POST results bitmap
Descriptor 2: PCI system error
Data Format 2: None
Descriptor 3: System reconfigured
Data Format 3: None
Descriptor 4: Log area reset/cleared
Data Format 4: None

**Output of sudo dmidecode -t bios:**

λ sudo dmidecode -t bios
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
Language Description Format: Abbreviated
Installable Languages: 1
enUS
Currently Installed Language: enUS

[... 1450 lines cut ...]
[Di,  6. Jun 2023, 11:42:58] legion_laptop: legion_laptop exit
[Di,  6. Jun 2023, 11:42:58] legion_laptop: legion_laptop starts loading
[Di,  6. Jun 2023, 11:42:58] legion PNP0C09:00: legion_laptop platform driver probing
[Di,  6. Jun 2023, 11:42:58] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TF; DMI_BIOS_VERSION:JYCN39WW
[Di,  6. Jun 2023, 11:42:58] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TF; DMI_BIOS_VERSION:JYCN39WW
[Di,  6. Jun 2023, 11:42:58] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[Di,  6. Jun 2023, 11:42:58] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[Di,  6. Jun 2023, 11:42:58] legion PNP0C09:00: legion_laptop not loaded for this device
[Di,  6. Jun 2023, 11:42:58] legion: probe of PNP0C09:00 failed with error -12
[Di,  6. Jun 2023, 12:06:59] legion_laptop: legion_laptop exit
[Di,  6. Jun 2023, 12:06:59] legion_laptop: legion_laptop starts loading
[Di,  6. Jun 2023, 12:06:59] legion PNP0C09:00: legion_laptop platform driver probing
[Di,  6. Jun 2023, 12:06:59] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TF; DMI_BIOS_VERSION:JYCN39WW
[Di,  6. Jun 2023, 12:06:59] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TF; DMI_BIOS_VERSION:JYCN39WW
[Di,  6. Jun 2023, 12:06:59] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[Di,  6. Jun 2023, 12:06:59] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[Di,  6. Jun 2023, 12:06:59] legion PNP0C09:00: legion_laptop not loaded for this device
[Di,  6. Jun 2023, 12:06:59] legion: probe of PNP0C09:00 failed with error -12
bash -c "./issue-warning.sh"
If you have a 2022 or 2023 model, please help testing the new features 'here'
If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here'

if you need aynthing elese, let me know.

cheers and thanks in advance

SirRFI, 2023-06-07:
Model name: **Lenovo Legion 7 Gen 7 AMD** (`16ARHA7`)
CPU model: **AMD Ryzen 7 6800H**
GPU model: **AMD Radeon RX 6700M**
Keyboard backlight: Per key RGB, 4 brightness levels - they are controlled and saved with firmware
Distro: Fedora 38 GNOME

[Lenovo Legion 7 Gen 7 AMD 16ARHA7.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11678857/Lenovo.Legion.7.Gen.7.AMD.16ARHA7.zip)

404useryv35notfound, 2023-06-07:
hi there :)thanks! sorry for the stupid question, but since this is not my model of notebook, should i still test it?thanks so much and have a great day!cheersyvesVon meinem iPhone gesendet

clockforrest, 2023-06-17:
Model name: Lenovo Legion 5 15ach6
CPU model: AMD Ryzen 5 5600H
GPU model: NVIDIA RTX 3050 Mobile
Keyboard backlight: Single color with medium
Distro: Ubuntu 22.04

dmidecode -t system
<details><summary>Details</summary>
<p>

# dmidecode 3.3

Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82JW
	Version: Legion 5 15ACH6
	Serial Number: PF3DAB3C
	UUID: a29853ab-3dd0-11ec-80f0-84a9389b2e21
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82JW_BU_idea_FM_Legion 5 15ACH6
	Family: Legion 5 15ACH6

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
</p>
</details> 

dmidecode -t bios
<details><summary>Details</summary>
<p>

# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: HHCN30WW
	Release Date: 09/15/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.30
	Firmware Revision: 1.30

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

</p>
</details> 

cat /sys/kernel/debug/legion/fancurve
<details><summary>Details</summary>
<p>

EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 59
CPU temperature EC error: 0
CPU temperature EC: 59
CPU temperature ACPI error: 0
CPU temperature ACPI: 59
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 59
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 39
GPU temperature EC error: 0
GPU temperature EC: 39
GPU temperature ACPI error: 0
GPU temperature ACPI: 39
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 39
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2173
1 fanspeed EC error: 0
1 fanspeed EC: 2173
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2100
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 2100
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 2173
2 fanspeed EC error: 0
2 fanspeed EC: 2173
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2100
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 2100
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 16
powermode ACPI error: -5
powermode ACPI: 1190236672
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 3
EC fan curve points size: 9
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 59	 0	 60	 0	 35
1700	 1700	 2	 2	 54	 59	 56	 60	 32	 41
1900	 1900	 2	 2	 54	 59	 56	 60	 35	 48
2100	 2100	 2	 2	 54	 62	 56	 60	 43	 127
2300	 2300	 2	 2	 58	 68	 56	 60	 127	 127
2600	 2600	 2	 2	 64	 72	 56	 60	 127	 127
2900	 2900	 2	 2	 68	 78	 56	 64	 127	 127
3500	 3500	 2	 2	 73	 85	 60	 70	 127	 127
3700	 3700	 2	 2	 80	 127	 64	 127	 127	 127
=====================
Current fan curve in hardware (WMI) (might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp

</p>
</details> 

fwts_re
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11778289/fwts_wmi.log)

acpi_re
[DSDT.dsl.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11778294/DSDT.dsl.txt)

BehradJi, 2023-06-25:
Model name: Legion 5 Pro 16IAH7H
CPU model: 12th Gen Intel® Core™ i7-12700H
GPU model: NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
Keyboard backlight: RGB, 3 zone color
OS: Archlinux (latest packages)

<details>
<summary>sudo dmidecode -t system</summary>
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RF
	Version: Legion 5 Pro 16IAH7H
	Serial Number: ****
	UUID: ****
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RF_BU_idea_FM_Legion 5 Pro 16IAH7H
	Family: Legion 5 Pro 16IAH7H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

</details>

<details>
<summary>sudo dmidecode -t bios</summary>
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: J2CN49WW
	Release Date: 11/25/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.49
	Firmware Revision: 1.49

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

</details>

<details>
<summary>sudo cat /sys/kernel/debug/legion/fancurve</summary>
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
</details>

[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11858956/acpi_re.zip)

brett-owen, 2023-06-26:
Model name: Lenovo Legion Slim 5 16APH8
CPU model: AMD Ryzen 7 7840HS
GPU model: NVIDIA RTX 4060 Mobile
Keyboard backlight: RGB
OS: Archlinux

Output of sudo dmidecode -t system:
```
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82Y9
        Version: Legion Slim 5 16APH8
        Serial Number: ******
        UUID: ******
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82Y9_BU_idea_FM_Legion Slim 5 16APH8
        Family: Legion Slim 5 16APH8

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002A, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of sudo dmidecode -t bios:
```
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: M3CN31WW
        Release Date: 04/21/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.31
        Firmware Revision: 1.31

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of sudo cat /sys/kernel/debug/legion/fancurve:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11870258/fwts_wmi.log)
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11870256/acpi_re.tar.gz)
[wmi_windows.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11870269/wmi_windows.txt)

realsnick, 2023-07-06:
Hi, I just got the Legion 7i Pro 2023...i'm using nixos , i'm not sure how to load this kernel module but will be happy to test. 

these are the outputs i got

$ sudo dmidecode -t system

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
	Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```
$ sudo dmidecode -t bios# dmidecode 3.5
```
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN36WW
	Release Date: 04/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.36
	Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

znahas, 2023-07-06:
Model name: Lenovo Legion Slim 5 16IRH8
CPU model: 13th Gen Intel® Core™ i7-13700H × 20
GPU model: NVIDIA GeForce RTX™ 4060 Laptop GPU / Intel® Graphics (RPL-P)

# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82YA
	Version: Legion Slim 5 16IRH8
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82YA_BU_idea_FM_Legion Slim 5 16IRH8
	Family: Legion Slim 5 16IRH8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: M2CN24WW
	Release Date: 04/27/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.24
	Firmware Revision: 2.24

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

 sudo cat /sys/kernel/debug/legion/fancurve
**cat: /sys/kernel/debug/legion/fancurve: No such file or directory**

tubiakou, 2023-07-11:
## Model Summary
- OS:  `Fedora Linux 38 (Thirty Eight) x86_64`
- Kernel: `6.4.3-0.rc2.250.vanilla.fc38.x86_64`
- Model Name: `82UF Yoga 7 16IAH7`
- CPU Model:    `12th Gen Intel i7-12700H (20) @ 4.600GHz`
- GPU Model:    `GPU 0:  Intel Alder Lake-P integrated Graphics   GPU 1: Intel DG2 Arc A370M`
- Keyboard Backlight:   `Single Colour (white) with off/low/high brightness`
---

## Build Log with Errors
Any guidance on how to get the `sudo make forcereloadmodule` to complete successfully would be greatly appreciated.
Note that the build was done against a 6.4.3 RC2 Kernel.  However similiar build issues occurred against a Fedora stock 6.3.12 kernel.

```
$ make
make -C /lib/modules/6.4.3-0.rc2.250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.3-0.rc2.250.vanilla.fc38.x86_64'
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.3-0.rc2.250.vanilla.fc38.x86_64'
 

$ sudo make forcereloadmodule
rmmod legion-laptop.ko || true
insmod legion-laptop.ko force=1
dmesg --ctime

   <whole bunch of dmesg output>

[Tue Jul 11 01:50:03 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:03 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:03 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:03 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:03 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:03 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:04 2023] ACPI BIOS Error (bug):
[acpi_dsl.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010350/acpi_dsl.zip)
[fwts_wmi.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010351/fwts_wmi.zip)
[windows_wmi.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010352/windows_wmi.zip)
 Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:04 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:04 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:05 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:05 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:05 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:07 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:07 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:07 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:08 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:08 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:08 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:09 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:09 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:09 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:10 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:10 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:10 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:10 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:10 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:10 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:12 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:12 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:12 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[... cut ...]
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```
---

## Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```
---
## Uploaded Files
[acpi_dsl.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010367/acpi_dsl.zip)
[fwts_wmi.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010368/fwts_wmi.zip)
[windows_wmi.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010369/windows_wmi.zip)

YisuiDenghua, 2023-07-14:
Model name: Lenovo Legion Y9000X IAH7 (2022)
CPU model: 12th Gen Intel i7-12700H (20) @ 4.600GHz 
GPU model: NVIDIA GeForce RTX 3060 Mobile / Max-Q 
Keyboard backlight:  single color with off/medium/bright

output of `sudo dmidecode -t system`

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x001F, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: J6H1:1-X Boot with Default; J8H1:1-X BIOS RECOVERY

Handle 0x001B, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82TF
	Version: Legion Y9000X IAH7

	Wake-up Type: Other
	SKU Number: LENOVO_MT_82TF_BU_idea_FM_Legion Y9000X IAH7
	Family: Legion Y9000X IAH7

Handle 0x0060, DMI type 15, 31 bytes
System Event Log
	Area Length: 98 bytes
	Header Start Offset: 0x0000
	Header Length: 16 bytes
	Data Start Offset: 0x0010
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x00F0
	Status: Valid, Not Full
	Change Token: 0x00000005
	Header Format: Type 1
	Supported Log Type Descriptors: 4
	Descriptor 1: POST error
	Data Format 1: POST results bitmap
	Descriptor 2: PCI system error
	Data Format 2: None
	Descriptor 3: System reconfigured
	Data Format 3: None
	Descriptor 4: Log area reset/cleared
	Data Format 4: None

```

Output of `dmidecode -t bios` :

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Abbreviated
	Installable Languages: 1
		enUS
	Currently Installed Language: enUS

Handle 0x001A, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
[... 129 lines cut ...]
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT12
Input file SSDT12, Length 0x2A83 (10883) bytes
ACPI: SSDT 0x0000000000000000 002A83 (v02 LENOVO PtidDevc 00001000 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT11
Input file SSDT11, Length 0x1F1A (7962) bytes
ACPI: SSDT 0x0000000000000000 001F1A (v02 LENOVO Ther_Rvp 00001000 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT10
Input file SSDT10, Length 0x6C21 (27681) bytes
ACPI: SSDT 0x0000000000000000 006C21 (v02 LENOVO AdlP_2MR 00001000 INTL 20200717)
Pass 1 parse of [SSDT]
Firmware Error (ACPI): Failure creating named object [\_SB.PC00.PEG2.PEGP._S0W], AE_ALREADY_EXISTS (20220331/dswload-495)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20220331/psobject-372)
Could not parse external ACPI tables, AE_ALREADY_EXISTS

```

[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12046737/acpi_re.zip)

## Gathering WMI info in Windows

johnfanv2, 2023-07-14:
@tubiakou Obviously, the driver does not load on your model. Can you please create a new issue so we can troubleshoot and try to fix step-by-step it?

johnfanv2, 2023-07-14:
@ YisuiDenghua Obviously, the driver does not load on your model. Can you please create a new issue so we can troubleshoot and try to fix step-by-step it?

YisuiDenghua, 2023-07-16:
> @YisuiDenghua Obviously, the driver does not load on your model. Can you please create a new issue so we can troubleshoot and try to fix step-by-step it?

Referenced in a new issue #86 . Thank you for developing the driver!

WllCt, 2023-07-23:
Lenovo legion 5 pro AMD

Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WM
	Version: Legion Pro 5 16ARX8
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WM_BU_idea_FM_Legion Pro 5 16ARX8
	Family: Legion Pro 5 16ARX8

Handle 0x0017, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: LPCN39WW
	Release Date: 04/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.39
	Firmware Revision: 1.39

Handle 0x0018, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12139065/fwts_wmi.log)
[DSDT.dsl.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12139067/DSDT.dsl.log)

luhma666, 2023-07-24:
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WK
	Version: Legion Pro 5 16IRX8
	Serial Number: XXXXXX
	UUID: XXXXXXXXXXXXXXXXXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WK_BU_idea_FM_Legion Pro 5 16IRX8
	Family: Legion Pro 5 16IRX8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN36WW
	Release Date: 04/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.36
	Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

cat: /sys/kernel/debug/legion/fancurve: No such file or directory

[DSDT.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12147813/DSDT.zip)
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12147815/fwts_wmi.log)

johnfanv2, 2023-07-24:
@luhma666 It seems like it does not work on your model because you got the error "cat: /sys/kernel/debug/legion/fancurve: No such file or directory". Have you tried loading it according to the README (see parts for `make` and `make reloadmodule` in README)? Have you used the code from github?

If you cannot make it load, then please open a new issue with your model name where we can try to troubleshoot it step by step.

adityaparik, 2023-07-30:
Model and Debug Info

Model name: Lenovo Legion Pro 7i (16IRX8H)
CPU model: Intel i9-13900HX
GPU model: Nvidia RTX 4090
Keyboard backlight: Per-key RGB

**Output of sudo dmidecode -t system:**

# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

```
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
	Serial Number:
	UUID:
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
	Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

# Output of sudo dmidecode -t bios:

```
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN36WW
	Release Date: 04/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
[... 134 lines cut ...]
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1500	 0	 0	 0	 0	 0	 0	 0	 0	 0
1700	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
```

Note the 1500 and 1700 in the fan curve numbers. I had run the test before commenting here, and this is the result of the commands to modify the fan curve.

# Inspecting WMI entries
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12208623/fwts_wmi.log)

# Disassembling ACPI tables
I get this error when running the final iasl command:
```
Firmware Error (ACPI): Failure creating named object [\_SB.PC00.XHCI.RHUB.HS14.SADX], AE_ALREADY_EXISTS (20200925/dswload-387)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20200925/psobject-264)
Could not parse ACPI tables, AE_ALREADY_EXISTS
```

johnfanv2, 2023-07-30:
@adityaparik It seems like setting the fan curve settings does not work? Please open a new issue with this bug, where we can debug it. 

* In the new issue please also report the output of the two commands:
```bash
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
```
* If you get an error in "Disassembling ACPI tables", then please just upload all the files in the created folder `acpi_re`, e.g. in a zip file. Then I will try to disassemble it.

adityaparik, 2023-07-30:
Hello @johnfanv2 , thanks for all your work. I have edited my original comment with the zip file in it.

Just an observation, I noticed lot of ACPI errors in the `sudo make reloadmodule` command when I ran it this time. The module did load ultimately. I do not remember if there were so many errors in the last boot.

A question: all the commands you mention, I can run them from any working directory, right?

johnfanv2, 2023-07-30:
@adityaparik The acpi_re in the zip folder seems to be empty. Maybe there is somethings missing. Maybe you can try again the following (I added some comments where the folder is changed)
```bash
# Create folder for all the new files
# Can be run from any folder, e.g. in home folder or /tmp
mkdir acpi_re
# This goes into the newly created folder acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
# If you run this you should copy some files with SDT in its name to the current acpi_re folder
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# You should find some files with DST now in the current folder acpi_re

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

The commands mentioned in my last message can be run from anywhere.

adityaparik, 2023-07-30:
@johnfanv2 , Done! I uploaded a new zip file. This should work as I downloaded it and checked for its contents.

JerrettDavis, 2023-08-05:
Model and Debug Info

Model name: Legion Pro 5 16IRX8
CPU model: Intel i9-13900HX
GPU model: Nvidia RTX 4070
Keyboard backlight: Per-key RGB

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WK
	Version: Legion Pro 5 16IRX8
	Serial Number: XXXXXXX
        UUID: 00000000-0000-0000-0000-0000000000000
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WK_BU_idea_FM_Legion Pro 5 16IRX8
	Family: Legion Pro 5 16IRX8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN33WW
	Release Date: 02/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.33
	Firmware Revision: 1.33

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:

```
sudo cat /sys/kernel/debug/legion/fancurve
```

johnfanv2, 2023-08-05:
@JerrettDavis You should be able to set a custom fan curve (ony the first entry 0-1000) in custom mode in the GUI.

JerrettDavis, 2023-08-05:
> @JerrettDavis You should be able to set a custom fan curve (ony the first entry 0-1000) in custom mode in the GUI.

I believe you're correct:

[image]

<details>
<summary>Output of `sudo cat /sys/kernel/debug/legion/fancurve`:</summary>
<p>
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 49
CPU temperature EC error: 0
CPU temperature EC: 55
CPU temperature ACPI error: 0
CPU temperature ACPI: 49
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 49
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1700
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 1700
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 255
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 255
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1500	 0	 0	 0	 0	 0	 0	 0	 0	 0
1900	 0	 0	 0	 0	 0	 0	 0	 0	 0
2100	 0	 0	 0	 0	 0	 0	 0	 0	 0
2200	 0	 0	 0	 0	 0	 0	 0	 0	 0
2500	 0	 0	 0	 0	 0	 0	 0	 0	 0
2900	 0	 0	 0	 0	 0	 0	 0	 0	 0
3500	 0	 0	 0	 0	 0	 0	 0	 0	 0
3700	 0	 0	 0	 0	 0	 0	 0	 0	 0
3700	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1500	 0	 0	 0	 0	 0	 0	 0	 0	 0
1900	 0	 0	 0	 0	 0	 0	 0	 0	 0
2100	 0	 0	 0	 0	 0	 0	 0	 0	 0
2200	 0	 0	 0	 0	 0	 0	 0	 0	 0
2500	 0	 0	 0	 0	 0	 0	 0	 0	 0
2900	 0	 0	 0	 0	 0	 0	 0	 0	 0
3500	 0	 0	 0	 0	 0	 0	 0	 0	 0
3700	 0	 0	 0	 0	 0	 0	 0	 0	 0
3700	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
</p>
</details>

[log.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12268072/log.txt)

themightymak, 2023-08-06:
Model name: LENOVO Legion 7 16ARHA7 (82UH)
CPU model: 16 × AMD Ryzen 7 6800H with Radeon Graphics
GPU model: AMD Radeon RX 6700M
Keyboard backlight: RGB
`[🧱] × sudo dmidecode -t system
[sudo] Passwort für _redacted_:
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
Manufacturer: LENOVO
Product Name: 82UH
Version: Legion 7 16ARHA7
Serial Number: _redacted_
UUID: _redacted_
Wake-up Type: Power Switch
SKU Number: LENOVO_MT_82UH_BU_idea_FM_Legion 7 16ARHA7
Family: Legion 7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
Option 1: String1 for Type12 Equipment Manufacturer
Option 2: String2 for Type12 Equipment Manufacturer
Option 3: String3 for Type12 Equipment Manufacturer
Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
Status: No errors detected`

`sudo dmidecode -t bios
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
Vendor: LENOVO
Version: K9CN30WW
Release Date: 06/01/2022
Address: 0xE0000
Runtime Size: 128 kB
ROM Size: 32 MB
Characteristics:
PCI is supported
BIOS is upgradeable
BIOS shadowing is allowed
Boot from CD is supported
Selectable boot is supported
EDD is supported
Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
5.25"/360 kB floppy services are supported (int 13h)
5.25"/1.2 MB floppy services are supported (int 13h)
3.5"/720 kB floppy services are supported (int 13h)
3.5"/2.88 MB floppy services are supported (int 13h)
8042 keyboard services are supported (int 9h)
CGA/mono video services are supported (int 10h)
ACPI is supported
USB legacy is supported
BIOS boot specification is supported
Targeted content distribution is supported
UEFI is supported
BIOS Revision: 1.30
Firmware Revision: 1.30

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
Language Description Format: Long
Installable Languages: 4
en|US|iso8859-1
fr|FR|iso8859-1
ja|JP|unicode
zh|TW|unicode
Currently Installed Language: en|US|iso8859-1`

`sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 49
CPU temperature EC error: 0
CPU temperature EC: 49
CPU temperature ACPI error: 0
CPU temperature ACPI: 49
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 49
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 46
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12270484/fwts_wmi.log)

GPU temperature ACPI: 46
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 46
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2182
1 fanspeed EC error: 0
1 fanspeed EC: 2182
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2100
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 2100
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12270485/fwts_wmi.log)
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12270488/fwts_wmi.log)

1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 2073
2 fanspeed EC error: 0
2 fanspeed EC: 2073
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2000
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 2000
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 16
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 3
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       5       0       41      0       65      0       37
1700     1700    5       5       35      45      60      65      33      41
1900     1900    5       5       42      49      60      65      38      46
2100     2000    5       5       46      54      60      65      41      127
2300     2300    2       2       52      64      60      65      127     127
2600     2500    2       2       62      68      60      65      127     127
2900     2800    2       2       65      72      60      65      127     127
3100     3000    2       2       69      76      60      75      127     127
3400     3400    2       2       73      80      72      88      127     127
3600     3600    2       2       77      127     85      127     127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================`

linuxazo, 2023-08-11:
I guess this is the same laptop as in previous post from @themightymak ...  
Either way, I am posting more data @johnfanv2 

Model name:  **Lenovo Legion 7 16ARHA7 (AMD Advantage Edition)**
CPU Model:  AMD Ryzen 7 6800H
GPU Model:  **Integrated: APU AMD R680M Rembrandt / Discrete: AMD RX 6700M**
Keyboard backlight:  RGB per key backlight with off/low/medium/bright 

Output of `sudo dmidecode -t system`:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82UH
        Version: Legion 7 16ARHA7
        Serial Number: removed for privacy
        UUID: removed for privacy
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82UH_BU_idea_FM_Legion 7 16ARHA7
        Family: Legion 7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```
Output of `sudo dmidecode -t bios`:
```
BIOS Information
        Vendor: LENOVO
        Version: K9CN42WW
        Release Date: 03/10/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.42
        Firmware Revision: 1.42

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
[... 89 lines cut ...]
EC fan curve current point id: 1
EC fan curve points size: 4
Current fan curve in hardware:

rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       127     0       127     0       37
1600     1600    2       2       127     127     127     127     33      41
1700     1700    2       2       127     127     127     127     38      46
1900     1900    2       2       127     127     127     127     41      127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
**Inspection of WMI entries:**
Output of `sudo fwts wmi - > fwts_wmi.log`:
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12318170/fwts_wmi.log)
**Disassembling ACPI tables:**
Output of `iasl -e SSDT* -d DSDT` (the whole acpi_re folder):
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12318214/acpi_re.zip)
**Gathering WMI info in Windows:**
[Legion7_16ARHA7_WMI_Info_Windows.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12318222/Legion7_16ARHA7_WMI_Info_Windows.txt)

Thank You so much for all your effort!
In case You need something else, feel free to mention it.

johnfanv2, 2023-08-15:
@bee344 This is a new model/BIOS not added yet. Please create a new issue with your model name, copy the info above to the new issue, and upload all the files/logs from the three steps above. When you get an error for the ACPI tables, just upload all the files before you get the error. Then we can troubleshoot and add all features step by step in the new issue.

I added your model to the code to test it. So please get the newest code, compile again, and provide the output of `cat: /sys/kernel/debug/legion/fancurve` again.

lavavex, 2023-08-21:
Here is My info, will file dump the rest later
`# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82UG
	Version: Legion S7 16ARH7
	Serial Number: XXXXXXXX (censored this)
	UUID: 015e6017-ac34-0b5b-a4e5-XXXXXXXXXXXX (censored this)
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82UG_BU_idea_FM_Legion S7 16ARH7
	Family: Legion S7 16ARH7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

╭─lavavex@Lenowovo ~/LenovoLegionLinux ‹heads/main› 
╰─$ sudo dmidecode -t bios  
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KFCN33WW
	Release Date: 11/21/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.33
	Firmware Revision: 1.33

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

╭─lavavex@Lenowovo ~/LenovoLegionLinux ‹heads/main› 
╰─$ sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 34
CPU temperature EC error: 0
CPU temperature EC: 34
CPU temperature ACPI error: 0
CPU temperature ACPI: 34
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 34
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1768
1 fanspeed EC error: 0
1 fanspeed EC: 1768
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1700
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 1748
2 fanspeed EC error: 0
2 fanspeed EC: 1748
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1700
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 1
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 2
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	0	2	2	60	80	70	76	0	32
1600	1600	2	2	60	80	70	76	31	34
1700	1700	2	2	60	80	70	76	33	40
2000	2000	2	2	60	80	70	76	39	46
2200	2200	2	2	60	80	70	76	45	50
2500	2500	2	2	60	80	70	76	49	53
2800	2800	2	2	60	80	70	76	52	56
3200	3200	2	2	60	80	70	76	55	58
3500	3500	2	2	60	80	70	76	57	61
3500	3500	2	2	60	80	70	76	60	64
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
╭─lavavex@Lenowovo ~/LenovoLegionLinux ‹heads/main› 
╰─$ `

ChatCG3, 2023-09-28:
ChatCG3
sudo dmidecode -t system
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
	Serial Number:
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
	Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

cg3@LenovoLegionCG3:~$ sudo dmidecode -t bios
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN38WW
	Release Date: 06/20/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.38
	Firmware Revision: 1.38

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

cg3@LenovoLegionCG3:~$ sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 50
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 50
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 50
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 42
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 42
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1700
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 1700
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 16
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================

byebye, 2023-10-03:
### Model and Debug Info

Model name: Legion Pro 5 16ITH6H
CPU model: Intel i7-11800H
GPU model: Nvidia RTX 3060
Keyboard backlight: white
Distro: Arch Linux

<details>
<summary>sudo dmidecode -t system </summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82JD
	Version: Legion 5 Pro 16ITH6H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82JD_BU_idea_FM_Legion 5 Pro 16ITH6H
	Family: Legion 5 Pro 16ITH6H

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

</details>

<details>
<summary>sudo dmidecode -t bios</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: H1CN51WW
	Release Date: 10/08/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.51
	Firmware Revision: 1.51

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

</details>

<details>
<summary>sudo cat /sys/kernel/debug/legion/fancurve</summary>

```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 44
CPU temperature EC error: 0
CPU temperature EC: 44
CPU temperature ACPI error: 0
CPU temperature ACPI: 44
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 44
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 42
GPU temperature ACPI error: 0
GPU temperature ACPI: 42
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 42
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2736
1 fanspeed EC error: 0
1 fanspeed EC: 2736
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1800
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1800
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 2566
2 fanspeed EC error: 0
2 fanspeed EC: 2566
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1700
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 2
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 1
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	0	5	5	0	51	0	62	0	37
1800	1700	5	5	48	51	59	62	34	41
1800	1900	5	5	48	51	59	62	38	44
2100	2100	5	5	48	51	59	62	41	127
2400	2300	2	2	48	61	59	62	127	127
2600	2600	2	2	54	70	59	67	127	127
2900	2900	2	2	64	78	63	70	127	127
3500	3500	2	2	73	83	67	78	127	127
3800	3800	2	2	78	88	74	82	127	127
4200	4200	2	2	84	127	79	127	127	127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

</details>

Output from  commands:

[fwts_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12791267/fwts_re.zip)
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12791265/acpi_re.zip)

SirRFI, 2023-10-03:
@johnfanv2 Do you need anything else to support `16ARHA7`? (I provided what you asked for here](https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-1580988755). Admittedly most stuff work on firmware level, so it just works out of the box on Linux. However, it would be nice to see some additional support regarding battery life, ability to switch between battery charging modes and MUX switch support (hybrid/discrete graphics card) on system level, rather than going to BIOS/UEFI first.

trivoldus28, 2023-10-13:
**Model and Debug Info**

Model name: Lenovo Legion Slim 7 16APH8
CPU model: AMD Ryzen 7 7840HS
GPU model: NVIDIA RTX 4600
Keyboard backlight: RGB per key with off/medium/bright

Output of `sudo dmidecode -t system`:
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82Y4
        Version: Legion Slim 7 16APH8
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82Y4_BU_idea_FM_Legion Slim 7 16APH8
        Family: Legion Slim 7 16APH8

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

`-t bios`
```
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: M1CN34WW
        Release Date: 08/07/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.34
        Firmware Revision: 1.34

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Fan curve: 
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12889007/fwts_wmi.log)

[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/12889066/acpi_re.tar.gz)

Thanks for all the work!

paralaks, 2023-10-15:
Hi,
I have executed the Powershell script output with  device and windows info. Once I boot into Linux I will add the rest. Thanks so much for working on this!

[Lenovo-Legion-5-Slim-2023-Powershell-script-output.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12908519/Lenovo-Legion-5-Slim-2023-Powershell-script-output.txt)

MrDuartePT, 2023-10-21:
For what i can conclude is like the 2021 model the 2022 everything is support.
The 2023 devices is a different story since lenovo change how the fan control works and other things

lcizzle, 2023-10-24:
Legion Pro 7 16IRX8H

appsforartists, 2023-11-16:
Model name: Legion Go
CPU model: AMD Ryzen Z1 Extreme (7840U)
GPU model: AMD Radeon 780M
Power button backlight: controlled by firmware to show the power throttling of the device

```
[nix-shell:~]$ sudo dmidecode -t system
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
  Manufacturer: LENOVO
  Product Name: 83E1
  Version: Legion Go 8APU1
  Wake-up Type: Power Switch
  SKU Number: LENOVO_MT_83E1_BU_idea_FM_Legion Go 8APU1
  Family: Legion Go 8APU1

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
  Option 1: String1 for Type12 Equipment Manufacturer
  Option 2: String2 for Type12 Equipment Manufacturer
  Option 3: String3 for Type12 Equipment Manufacturer
  Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
  Status: No errors detected

[nix-shell:~]$ sudo dmidecode -t bios
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
  Vendor: LENOVO
  Version: N3CN23WW
  Release Date: 10/13/2023
  Address: 0xE0000
  Runtime Size: 128 kB
  ROM Size: 32 MB
  Characteristics:
    PCI is supported
    BIOS is upgradeable
    BIOS shadowing is allowed
    Boot from CD is supported
    Selectable boot is supported
    EDD is supported
    Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
    Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
    5.25"/360 kB floppy services are supported (int 13h)
    5.25"/1.2 MB floppy services are supported (int 13h)
    3.5"/720 kB floppy services are supported (int 13h)
    3.5"/2.88 MB floppy services are supported (int 13h)
    8042 keyboard services are supported (int 9h)
    CGA/mono video services are supported (int 10h)
    ACPI is supported
    USB legacy is supported
    BIOS boot specification is supported
    Targeted content distribution is supported
    UEFI is supported
  BIOS Revision: 1.23
  Firmware Revision: 1.23

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
  Language Description Format: Long
[... 590 lines cut ...]
Pass 2 parse of [SSDT]
External object resolution file           SSDT11
Input file SSDT11, Length 0x5EB (1515) bytes
ACPI: SSDT 0x0000000000000000 0005EB (v02 LENOVO CB-01    00000001 ACPI 00040000)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT10
Input file SSDT10, Length 0x77A (1914) bytes
ACPI: SSDT 0x0000000000000000 00077A (v02 LENOVO CB-01    00000001 ACPI 00040000)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file            SSDT1
Input file SSDT1, Length 0x8416 (33814) bytes
ACPI: SSDT 0x0000000000000000 008416 (v02 LENOVO CB-01    00000001 ACPI 00040000)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
Pass 1 parse of [DSDT]
Firmware Error (ACPI): Failure creating named object [\_SB.PCI0.GPP6.WLAN._S0W], AE_ALREADY_EXISTS (20230331/dswload-495)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-372)
Could not parse ACPI tables, AE_ALREADY_EXISTS

[nix-shell:~/acpi_re]$ ls
DSDT      SSDT1   SSDT11  SSDT13  SSDT15  SSDT17  SSDT19  SSDT20  SSDT22  SSDT24  SSDT26  SSDT28  SSDT4  SSDT6  SSDT8
DSDT.dsl  SSDT10  SSDT12  SSDT14  SSDT16  SSDT18  SSDT2   SSDT21  SSDT23  SSDT25  SSDT27  SSDT3   SSDT5  SSDT7  SSDT9
```

kafkalainen, 2023-11-27:
Model name: Lenovo Legion Slim 5 16APH8
CPU model: AMD Ryzen 7 7840HS
GPU model: NVIDIA RTX 4070 Mobile
Keyboard backlight: RGB
OS: Archlinux

Output of sudo dmidecode -t system
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82Y9
        Version: Legion Slim 5 16APH8
        Serial Number: ******
        UUID: ******
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82Y9_BU_idea_FM_Legion Slim 5 16APH8
        Family: Legion Slim 5 16APH8

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002A, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of sudo dmidecode -t bios:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: M3CN34WW
        Release Date: 06/16/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.34
        Firmware Revision: 1.34

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of sudo cat /sys/kernel/debug/legion/fancurve:
```
/sys/kernel/debug/legion/fancurve: No such file or directory
```

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13480629/fwts_wmi.log)
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/13480690/acpi_re.tar.gz)

Thank you for all your work! :pray:

eode, 2023-12-01:
### Model and Debug Info

Model name: Lenovo Legion Slim 7 Gen 7
CPU model: AMD Ryzen 9 6900HX with Radeon Graphics
GPU Integrated: REMBRANDT (rembrandt, LLVM 15.0.7, DRM 3.54, 6.5.6-76060506-generic) (0x1681) (from glxinfo)
GPU Discrete: AMD Radeon RX 6800S (navi23, LLVM 15.0.7, DRM 3.54, 6.5.6-76060506-generic) (0x73ef) (from glxinfo)
Keyboard backlight: RGB 4-Zone

### Output of sudo dmidecode -t system:

```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82UG
	Version: Legion S7 16ARHA7
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82UG_BU_idea_FM_Legion S7 16ARHA7
	Family: Legion S7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

### Output of sudo dmidecode -t bios:

```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KFCN32WW
	Release Date: 09/29/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.32
	Firmware Revision: 1.32

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

### Output of sudo cat /sys/kernel/debug/legion/fancurve:

```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 52
CPU temperature ACPI error: 0
CPU temperature ACPI: 52
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 52
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2051
1 fanspeed EC error: 0
1 fanspeed EC: 2051
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2000
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 2000
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 2038
2 fanspeed EC error: 0
2 fanspeed EC: 2038
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2000
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 2000
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 1
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 3
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 60	 80	 70	 76	 0	 32
1600	 1600	 2	 2	 60	 80	 70	 76	 31	 34
1700	 1700	 2	 2	 60	 80	 70	 76	 33	 40
2000	 2000	 2	 2	 60	 80	 70	 76	 39	 46
2200	 2200	 2	 2	 60	 80	 70	 76	 45	 50
2500	 2500	 2	 2	 60	 80	 70	 76	 49	 53
2800	 2800	 2	 2	 60	 80	 70	 76	 52	 56
3200	 3200	 2	 2	 60	 80	 70	 76	 55	 58
3500	 3500	 2	 2	 60	 80	 70	 76	 57	 61
3500	 3500	 2	 2	 60	 80	 70	 76	 60	 64
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

### Inspecting WMI entries

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13520612/fwts_wmi.log)

### Disassembling ACPI tables

[DSDT.dsl.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/13520633/DSDT.dsl.txt)

Thanks for making this!

jul1u5, 2023-12-08:
### Model and Debug Info

Model name: Lenovo Legion Slim 5 14APH8
CPU model: AMD Ryzen 7 7840HS
GPU model: NVIDIA RTX 4060
Keyboard backlight: single color with off/medium/bright

Output of `sudo dmidecode -t system`:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82Y5
	Version: Legion Slim 5 14APH8
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82Y5_BU_idea_FM_Legion Slim 5 14APH8
	Family: Legion Slim 5 14APH8

Handle 0x001E, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0037, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```

### Output of sudo cat /sys/kernel/debug/legion/fancurve:

???

Maiksu619, 2023-12-10:
Fresh install of Pop_OS on a clean hard drive with disk encryption on model 16ARHA7 AMD certified.  I experienced some issues on install and don't seem to be able to get everything running correctly.  I would very much like some community help to get these features working and this model should be supported.

SomeUserName1, 2023-12-14:
**Model and Debug Info**
 
Model name: Legion Pro 7 16IRX8H
CPU model:  Intel 13th Gen Core i9-13980HX
 GPU model: NVIDIA RTX 4090
Keyboard backlight: single color with off/on

Output of `sudo dmidecode -t bios`:

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
	Serial Number: XXXXXXXX
	UUID: XXXXXXXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
	Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

 Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 52
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 52
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 42
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 42
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1700
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 1700
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 18
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1500	 0	 0	 0	 0	 0	 0	 0	 0	 0
1700	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1500	 0	 0	 0	 0	 0	 0	 0	 0	 0
1700	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================

```
Note: I ran the initial test, fan 2 changes didn't work

## Inspecting WMI entries

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13677546/fwts_wmi.log)

## Disassembling ACPI tables
Errored.
[iasl_error.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13677547/iasl_error.log)

[sdt.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/13677628/sdt.zip)

## Reload Module & dmesg

[dmesg.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13677642/dmesg.log)
[make_reloadmodule.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13677644/make_reloadmodule.log)

## sensors
[sensors.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13677681/sensors.log)

Side notes: After hibernation the screen doesn't turn back on and the webcam doesn't work. Maybe I forgot to install sth, so probably a missing driver. If you have a clue what, I'd appreciate evey hint.

Thanks for your work!

lvskng, 2023-12-15:
Pretty fresh install of Fedora 39

Battery conservation, rapid charging, touchpad and charge output USB and hybrid mode settings seem to be working. Fan curves, overclock etc. don't work.

Model and Debug Info

Model name: 14APH8
CPU model: AMD Ryzen™ 7 7840HS with Radeon™ 780M Graphics × 16
GPU model: NVIDIA GeForce RTX™ 4060
Keyboard backlight: single color off/medium/bright

Output of sudo dmidecode -t system. Please remove Serial Number and UUID for privacy:

# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82Y5
	Version: Legion Slim 5 14APH8
	Serial Number: 
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82Y5_BU_idea_FM_Legion Slim 5 14APH8
	Family: Legion Slim 5 14APH8

Handle 0x001E, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0037, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

Output of sudo dmidecode -t bios:

# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: MACN23WW
	Release Date: 08/18/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.23
	Firmware Revision: 1.23

Handle 0x001F, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

Output of sudo cat /sys/kernel/debug/legion/fancurve:

EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 35
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 35
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 48
1 fanspeed EC error: 0
1 fanspeed EC: 48
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2100
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2100
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2200
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2200
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================

The following information would be useful.
Inspecting WMI entries

# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log

Then upload files.
Disassembling ACPI tables
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/13684965/fwts_wmi.log)

# Install requiered tools
sudo apt install acpica-tools

# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
[DSTD.dsl.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/13684991/DSTD.dsl.tar.gz)
Error: Could not parse ACPI tables, AE_ALREADY_EXISTS

[50 more replies]