johnfanv2/LenovoLegionLinux #48: Support for Legion 7 15IMH05 81YT005HMH (2020)

tukanoidd, 2023-04-25:
Is there any way to make this kernel module work on this device? When I try to test it, it's not in any of the lists, so I'm not sure if I should just install it with dkms and pray? 

**Model and Debug Info**

Model name: Lenovo Legion 7 15IMH05 81YT005HMH
CPU model: Intel Core i7-10750H
GPU model: GeForce RTX 2070 SUPER Mobile / Max-Q
Keyboard backlight: CORSAIR iCUE RGB

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81YT
        Version: Lenovo Legion 7 15IMH05
        Serial Number: XXXXXXXX
        UUID: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81YT_BU_idea_FM_Legion 7 15IMH05
        Family: Legion 7 15IMH05

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: E9CN66WW(V4.11)
        Release Date: 09/30/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 12 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                UEFI is supported
        BIOS Revision: 4.66
        Firmware Revision: 4.66

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```

tukanoidd, 2023-04-25:
Ok, so I decided to be risky and installed it via `dkms`. System loaded fine, `dkms status` shows the modules as installed, but I'm not sure how to get logs for it.

## Tests 
### **Quick Test: Reading Current Fancurve from Hardware**
No `/sys/kernel/debug/legion` directory

### **Quick Test: Read Sensor Values from Hardware**
Output of `sensors`:
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +47.0°C  (high = +100.0°C, crit = +100.0°C)
```
No legion_hwmon, but coretemp an no Fan info.

### **Quick Test: Change Current Fan Curve from Hardware with hwmon**
Directory nonexistent
Output of `sudo ls -la /sys/module/legion_laptop/drivers/platform:legion/`:
```
total 0
drwxr-xr-x  2 root root    0 Apr 25 16:40 .
drwxr-xr-x 41 root root    0 Apr 25 16:39 ..
--w-------  1 root root 4096 Apr 25 16:52 bind
lrwxrwxrwx  1 root root    0 Apr 25 16:52 module -> ../../../../module/legion_laptop
--w-------  1 root root 4096 Apr 25 16:40 uevent
--w-------  1 root root 4096 Apr 25 16:52 unbind
```

##  **Normal Usage**
### **Temperature and Fan Monitoring**
`psensor` output:
[image]
No fans, same as sensors (figures)

### **Changing and Setting your own Fan Curve with the Python GUI**
[image]
Everything is greyed out (of course), the only usable options are these:
[image]
But I see they are not legion-specific soooo

I hope I provided enough information for you to work with. Or if you have any pointers on how I could try implementing support myself (since I own the hardware). But I don't really have any kernel-dev experience, so I would appreciate any useful info or links to that info.

tukanoidd, 2023-04-25:
Hi, thanks for the quick reply!

Here's the output of `sudo make forcereloadmodule` (legion portion):
```
[  104.594895] legion_laptop: legion_laptop starts loading
[  104.594936] legion PNP0C09:00: legion_laptop platform driver probing
[  104.594938] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81YT; DMI_BIOS_VERSION:E9CN66WW(V4.11)
[  104.594939] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81YT; DMI_BIOS_VERSION:E9CN66WW(V4.11)
[  104.594941] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[  104.594942] legion PNP0C09:00: legion_laptop is forced to load.
[  104.594942] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[  104.594943] legion PNP0C09:00: Using configuration for system: GKCN
[  104.594952] legion PNP0C09:00: Could not evaluate ACPI _STA
[  104.594952] legion PNP0C09:00: Could not init ACPI access
[  104.594952] legion_laptop: Unloading legion shared
[  104.594953] legion_laptop: Unloading legion shared done
[  104.594953] legion PNP0C09:00: legion_laptop not loaded for this device
[  104.594981] legion: probe of PNP0C09:00 failed with error -5
[  157.714009] legion_laptop: legion_laptop exit
[  157.736785] legion_laptop: legion_laptop starts loading
[  157.736857] legion PNP0C09:00: legion_laptop platform driver probing
[  157.736859] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81YT; DMI_BIOS_VERSION:E9CN66WW(V4.11)
[  157.736863] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81YT; DMI_BIOS_VERSION:E9CN66WW(V4.11)
[  157.736870] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[  157.736874] legion PNP0C09:00: legion_laptop is forced to load.
[  157.736876] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[  157.736878] legion PNP0C09:00: Using configuration for system: GKCN
[  157.736890] legion PNP0C09:00: Could not evaluate ACPI _STA
[  157.736892] legion PNP0C09:00: Could not init ACPI access
[  157.736894] legion_laptop: Unloading legion shared
[  157.736895] legion_laptop: Unloading legion shared done
[  157.736896] legion PNP0C09:00: legion_laptop not loaded for this device
[  157.736934] legion: probe of PNP0C09:00 failed with error -5
```

Unfortunately, it won't load even in this state, seems to be a problem with ACPI for some reason. I'm not sure if it's a hardware issue, or Void Linux, and it's linux6.2 package (I updated the kernel version in the meantime and rebooted multiple times since then, so I don't think that is causing the issue).

Understandably, `/sys/kernel/legion` doesn't exist still. (So can't get memory view either).

Couldn't get wmi info, fwts is not available on Void and I couldn't find an alternative (unless someone could reference me to one) either, sadly. And I don't have windows installed, I moved to Linux fully about 2 years ago now

Acpi tables I was able to disassemble though, here are all the files
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11326681/acpi_re.zip)

johnfanv2, 2023-04-26:
@dimesizehole Thanks for the help with the WMI tables! This really helped. 

@tukanoidd This ACPI methods in this BIOS version seems different. Getting the output of `fwts` described above would really help. Maybe you could boot from a pen drive with a live Ubuntu system. In the meantime I added the model to the kernel module and tried to guess the parameters. Please get the new version from git and try the steps above again.

tukanoidd, 2023-04-26:
> Maybe you could boot from a pen drive with a live Ubuntu system. 

Oh ye, idk why i didn't think of it myself, i already have a ventoy usb with multiple ios, adding Ubuntu won't be a problem😅 not at my laptop rn, but will send the info asap

tukanoidd, 2023-04-26:
Got the Ubuntu Live USB working, ran fwts, here's the output. I did see some errors pop up, but im not sure if those are relevant. Here's the output
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11332852/fwts_wmi.log)

Gonna try pulling master now and trying again

tukanoidd, 2023-04-26:
Anything else I should provide?

johnfanv2, 2023-04-26:
Memory access and fan control does not seem to work. We probably need some changes. Let's test the other things before we do that.

# Test power mode
- Test reading
    - Please change the power mode with Fn + Q. After each change, provide the output of `cat /sys/firmware/acpi/platform_profile` and `sudo cat /sys/kernel/debug/legion/fancurve`. 
- Test writing
    - Do you have a LED on your laptop that indicates the power mode? Please change the power mode with software with the follwoing commands. The LED should change. Please provide the output of the commands.
```bash
# run as root (sudo su)

echo quiet > /sys/firmware/acpi/platform_profile
cat /sys/firmware/acpi/platform_profile
# or balanced
echo balanced > /sys/firmware/acpi/platform_profile
cat /sys/firmware/acpi/platform_profile
# or performance
echo performance > /sys/firmware/acpi/platform_profile
cat /sys/firmware/acpi/platform_profile
```
- What do you mean by Ghz boost? Is this a feature on your model?

# Test Fan Speed and Temperature
Open psensor. You should see newly values for CPU and GPU temperature and two fan speed. Run something CPU intensive that does not use the GPU much (https://www.tecmint.com/linux-cpu-load-stress-test-with-stress-ng-tool/). This should increase mainly CPU temperature. The fan speed should also increase. Make sure the speed value makes sense, e.g. by listening to the fan noise. 
Run something GPU intensive that does not use the CPU much. The GPU temperature should mainly increase.

johnfanv2, 2023-04-26:
# Test rapid charging
You can also test rapid charging be enabling/disabling it in the GUI and checking if it makes a differences, e.g. by checking the charging current (should be larger with rapid charging) or the estimated time until it is full (should be smaller if rapid charging is enabled).

tukanoidd, 2023-04-27:
> Please change the power mode with Fn + Q.
Oh yeah, sorry about that, idk why I didn't think of it in the first place:

## Test Reading
Here are the outputs of `cat /sys/firmware/acpi/platform_profile`:
**Balanced**: `balanced`
**Performance**: `performance`
**Quiet**: `quiet`

And here are the outputs for `sudo cat /sys/kernel/debug/legion/fancurve`:
**Balanced**
```
EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228186132
temperature access method: 3
CPU temperature error: 0
CPU temperature: 93
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: 0
CPU temperature WMI: 93
CPU temperature WMI2 error: -14
CPU temperature WMI2: 93
GPU temperature error: 0
GPU temperature: 47
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: 0
GPU temperature WMI: 47
GPU temperature WMI2 error: -14
GPU temperature WMI2: 47
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 3200
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: 0
1 fanspeed WMI: 3200
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 3200
2 fanspeed error: 0
2 fanspeed: 3300
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: 0
2 fanspeed WMI: 3300
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 3300
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: 1
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 0
[... 312 lines cut ...]
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: 1
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 0
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
EC fanfullspeed error: error
EC fanfullspeed status: -1
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

tukanoidd, 2023-04-27:
And sorry for a late response, got sick, just got well enough to be able to do this properly :)

johnfanv2, 2023-04-28:
@tukanoidd Ok. So everything except fan speed seems to work. Are you sure that the fan speed wasn't just constant? Unfortunately, on your model one has only an accuracy of 100.

tukanoidd, 2023-04-28:
> @tukanoidd Ok. So everything except fan speed seems to work. Are you sure that the fan speed wasn't just constant? Unfortunately, on your model one has only an accuracy of 100.

Honestly, not sure if it was just constant, but it feels strange to actually not change at all in 5-10 min

johnfanv2, 2023-04-28:
@tukanoidd I see that values 1800, 1600, 3200, and 3300 were in the outputs of `sudo cat /sys/kernel/debug/legion/fancurve`. From your image I see that 2300 was measured (first value), 1600 (second) was minimum, and 3500 (last) was maximum. Could you provide the the image of the complete window of psensor while you stress the CPU?

tukanoidd, 2023-04-28:
> @tukanoidd I see that values 1800, 1600, 3200, and 3300 were in the outputs of `sudo cat /sys/kernel/debug/legion/fancurve`. From your image I see that 2300 was measured (first value), 1600 (second) [...]

Not at home rn, but will do when I'm back

tukanoidd, 2023-04-28:
Ok, no, my bad, I didn't realize it was actually working, the values were changing as the fans sped up. 
Psensor under stress: 
[image]

Psensor without stress:
[image]

I don't know why I didn't recognize that change, I might've paid more attention to CPU, sorry for that

johnfanv2, 2023-04-28:
1. Great! So temps, fan speed, and power mode works. 
2. Have you checked that enabling/disabling rapid charge actually has an effect on the charging speed.
3. To get fan control working we have to inspect the memory. Does `sudo busybox devmem 0xFC7E0800` works on your distribution? You probably first have to install the busybox package.

tukanoidd, 2023-04-29:
1. Ye, seems like it :)
2. Honestly, not sure. I usually use my laptop as a desktop (I just like the portability of laptops and that I can just take it with me anywhere if needed), so I have power plugged in all the time, so the battery could've degraded quite a bit at this point. I did try the setting though, and Gnome power settings did seem to skip from "14 minutes remaining" to 10 (I had 80% battery at that point) in about a minute (I'm assuming it just takes time for the settings to reload the info) but I'm not sure if that's a reliable source of information.
3. Ye, had to install [busybox-huge](https://voidlinux.org/packages/?arch=x86_64&q=busybox-huge) (for anyone who also uses Void) for devmem to work, but the output is `0x8E000000`

johnfanv2, 2023-05-03:
@tukanoidd 
1. ok, good!
2. I guess it works.
3. It seems like the busybox tool can only access at most 128bit. I changed the kernel module to read the whole memory of the embedded controller via memory mapped access to find the memory addresses where the fan control values might be stored. Please get the new code, compile, load the kernel module. Please change the power mode and provide the output of `cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C` after each change.

tukanoidd, 2023-05-03:
# Balance
```
00000000  00 00 00 8f 00 00 00 00  00 52 c0 00 00 a2 56 00  |.........R....V.|
00000010  00 00 30 40 30 01 1a 1a  26 00 00 00 40 62 00 00  |..0@0...&...@b..|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 24 00 00 00  00 00 00 00 00 00 00 00  |....$...........|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00  |................|
00000090  43 65 6c 78 70 65 72 00  00 4c 31 39 43 34 50 43  |Celxper..L19C4PC|
000000a0  00 00 00 c7 41 00 00 4e  00 00 00 08 64 2a 00 78  |....A..N....d*.x|
000000b0  4d 4d 2d 00 37 00 35 45  00 00 00 0f 00 5a 00 00  |MM-.7.5E.....Z..|
000000c0  00 02 4f 12 f2 05 20 3f  28 3c 40 1f a0 1d 3e 24  |..O... ?(<@...>$|
000000d0  7c 09 7d 09 8c 09 a0 1d  f9 02 59 40 00 00 2c 52  ||.}.......Y@..,R|
000000e0  00 00 00 c4 0f cb 0f cf  0f c0 0f 00 00 00 00 00  |................|
000000f0  00 00 00 00 00 00 f8 43  60 00 00 46 00 00 00 00  |.......C`..F....|
00000100  01 65 26 55 1a 09 05 00  46 00 41 5c 15 50 5c 20  |.e&U....F.A\.P\ |
00000110  50 5c 20 50 5c 20 50 5c  20 50 5c 20 50 5c 20 52  |P\ P\ P\ P\ P\ R|
00000120  5f 23 5e 64 23 1a 1a 01  65 24 6d 1a 08 00 00 46  |_#^d#...e$m....F|
00000130  00 41 5c 15 50 5c 20 50  5c 20 50 5c 20 50 5c 20  |.A\.P\ P\ P\ P\ |
00000140  50 5c 20 50 5c 20 52 5f  23 5e 64 23 1a 1a 00 00  |P\ P\ R_#^d#....|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000160  00 00 00 00 00 01 01 00  00 00 00 00 00 00 00 00  |................|
00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  00 00 00 64 00 00 00 00  00 00 00 00 00 00 00 00  |...d............|
000001e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000300  00 00 00 01 14 5a e6 e6  00 08 00 10 02 00 00 32  |.....Z.........2|
00000310  00 00 00 12 00 00 22 00  00 00 00 00 03 00 0a 00  |......".........|
00000320  00 00 00 1a 00 10 02 1a  00 10 02 1a 00 10 02 1a  |................|
00000330  00 10 02 1a 00 10 02 1a  00 10 02 1a 00 10 02 1a  |................|
00000340  00 00 00 00 00 00 00 00  00 09 80 00 00 06 00 00  |................|
00000350  00 00 00 00 10 00 00 00  00 00 00 00 00 00 00 00  |................|
00000360  00 00 00 00 00 00 00 00  06 00 00 00 00 1b 00 00  |................|
00000370  00 00 00 00 00 2a 00 80  40 00 00 00 d5 08 cd 38  |.....*..@......8|
00000380  cb 44 00 00 00 00 00 00  00 00 00 00 2b 00 00 0b  |.D..........+...|
00000390  00 00 01 00 07 00 00 00  00 00 00 02 01 80 00 00  |................|
000003a0  3e 3e 88 0b 00 00 00 00  00 00 00 00 00 00 00 00  |>>..............|
000003b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000003e0  00 00 00 00 00 00 00 00  00 01 00 00 6c 5b 00 a8  |............l[..|
000003f0  0b 00 00 00 00 00 00 00  00 00 00 00 00 01 00 33  |...............3|
00000400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000500  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000510  00 02 20 ff ff 00 00 f0  30 10 00 c0 00 00 83 82  |.. .....0.......|
00000520  1c 0e 05 00 00 00 00 f2  30 fd 22 2e 06 00 64 31  |........0."...d1|
00000530  20 70 00 00 00 00 50 f0  b5 00 40 11 08 00 9b 00  | p....P...@.....|
00000540  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000560  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 20  |............... |
00000570  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000580  5a a5 01 80 98 0a 75 5b  00 00 00 00 00 00 00 00  |Z.....u[........|
00000590  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000005a0  08 00 00 00 00 00 04 04  04 04 25 04 37 25 00 00  |..........%.7%..|
000005b0  00 60 00 00 00 84 01 00  00 00 00 00 00 00 00 00  |.`..............|
000005c0  00 04 00 00 00 00 00 00  ff 00 00 01 00 00 00 00  |................|
000005d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000600
```

# Performance
```
00000000  00 00 00 8f 00 00 00 00  00 52 c0 00 00 a2 56 00  |.........R....V.|
00000010  00 00 31 41 31 00 23 23  26 00 01 00 40 62 00 00  |..1A1.##&...@b..|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... 99 lines cut ...]
00000390  00 00 01 00 07 00 00 00  00 00 00 02 01 80 00 00  |................|
000003a0  3f 3f 88 0b 00 00 00 00  00 00 00 00 00 00 00 00  |??..............|
000003b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000003e0  00 00 00 00 00 00 00 00  00 01 00 00 6c 5b 00 a8  |............l[..|
000003f0  0b 00 00 00 00 00 00 00  00 00 00 00 00 01 00 33  |...............3|
00000400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000500  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000510  00 02 20 ff ff 00 00 f0  30 10 00 c0 00 00 83 82  |.. .....0.......|
00000520  1c 0e 05 00 00 00 00 f2  30 fd 22 2e 06 00 64 31  |........0."...d1|
00000530  20 70 00 00 00 00 50 f0  b5 00 40 11 08 00 9b 00  | p....P...@.....|
00000540  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000560  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 20  |............... |
00000570  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000580  5a a5 01 80 98 0a 75 5b  00 00 00 00 00 00 00 00  |Z.....u[........|
00000590  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000005a0  08 00 00 80 00 00 20 0c  0c 04 25 04 37 25 03 03  |...... ...%.7%..|
000005b0  00 60 00 00 00 84 09 00  00 00 00 00 00 00 00 00  |.`..............|
000005c0  00 04 00 00 00 00 00 00  ff 00 00 01 00 00 00 00  |................|
000005d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000600
```

johnfanv2, 2023-05-04:
@tukanoidd I don't see any values that could be interpreted as fan speeds or temperature limits for each step. If you find the number/addresses I can add them to the driver. You might want to look for these numbers in the dumps yourself because it is a little harder to see them from only three dumps:
- They are usually numbers that increase from byte to byte and that change between the three modes. 
- There are usually at least 3 steps, so one expects three increasing numbers. In other legions models, there are 8 to 10 steps.
- There are numbers for fan speed and temperature limit.
- The numbers do not change if you do not change power mode but just output the hexdump again.
Therefore, I can not activate fan control right now. Are there any other features that your are looking for?

tukanoidd, 2023-05-13:
Oh shoot, sorry for such a late response, I got busy at a job and completely forgot to check up on this! 

I'm not sure I have enough knowledge to understand the hexdumps, but it's fine, no worries! I've managed for 2 years without fan controls, so I don't mind. You've been a great help, thank you!

Closing the issue. Thanks again!

closed 2023-05-13 by tukanoidd