johnfanv2/LenovoLegionLinux #50: [BUG] Can't disable G-Sync on Legion 5 17ACH6

Armynator, 2023-04-28:
**Problem Descripition**
I have a Legion 5 17ACH6. The display has no G-Sync support.
After enabling G-Sync through the UI, it can't be disabled anymore. The checkbox stays checked.

**Model and Debug Info**

Model name: Legion 5 17ACH6
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3070
Keyboard backlight: single color with off/medium/bright

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82JY
	Version: Legion 5 17ACH6H
	Serial Number: xxx
	UUID: xxx
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82JY_BU_idea_FM_Legion 5 17ACH6H
	Family: Legion 5 17ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: GKCN58WW
	Release Date: 09/15/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.58
	Firmware Revision: 1.58

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 63
CPU temperature EC error: 0
CPU temperature EC: 63
CPU temperature ACPI error: 0
CPU temperature ACPI: 63
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 63
GPU temperature error: 0
GPU temperature: 47
GPU temperature EC error: 0
GPU temperature EC: 47
GPU temperature ACPI error: 0
GPU temperature ACPI: 47
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 47
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 5458
1 fanspeed EC error: 0
1 fanspeed EC: 5458
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 5400
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 5400
2 fanspeed error: 0
2 fanspeed: 5528
2 fanspeed EC error: 0
2 fanspeed EC: 5528
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 5500
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 5500
powermode access method: 3
powermode error: 0
powermode: 255
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: -1750041368
powermode WMI error: 0
powermode WMI: 255
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 2
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 4
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 48	 0	 60	 0	 42
1700	 1900	 2	 2	 45	 54	 56	 60	 39	 44
1900	 2000	 2	 2	 51	 58	 56	 60	 42	 50
2200	 2100	 2	 2	 55	 62	 56	 60	 46	 127
2300	 2400	 2	 2	 59	 71	 56	 60	 127	 127
2600	 2700	 2	 2	 68	 76	 56	 65	 127	 127
2900	 3000	 2	 2	 72	 81	 61	 69	 127	 127
3500	 3500	 2	 2	 78	 90	 66	 77	 127	 127
3800	 3800	 2	 2	 87	 95	 72	 79	 127	 127
4500	 4500	 2	 2	 91	 127	 78	 127	 127	 127
=====================

```

johnfanv2, 2023-04-29:
@Armynator Thanks for the bug report. I guess the GSync Feature should better be called Hybrid mode. It was called gsync method because the WMI method in the hardware firmware is called like it. I updated the GUI. You should enable/disable hybrid mode with it (after a restart). Please test the new GUI.

Armynator, 2023-04-30:
Yup it's indeed hybrid mode. After disabling hybrid mode in BIOS the checkbox is unchecked again.
However, unchecking/disabling it through the GUI doesn't seem to work.

johnfanv2, 2023-04-30:
Could you please try the following?
- Enable hybrid/dynamic graphics in BIOS
- Then output `cat /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/gsync` and copy it here.
- Disable hybrid/dynamic graphics in BIOS
- Then output `cat /sys/module/legion_laptop/drivers/platform\:legion/PNP0C09\:00/gsync` and copy it here.

After that (please do it after the above) get the new code, build and reload the new kernel module, and open the the new GUI. Run it with `sudo python/legion_linux/legion_linux/legion_gui.py` There should not be a checkbox anymore.  
- Then enable hybrid mode in the GUI, restart and check in BIOS if enabled.
- Then disable hybrid mode in the GUI, restart and check in BIOS if disabled.
I assume that either writing or reading it might not work with the new adaptions.

kleidiss, 2023-06-02:
What does this do on laptops that don't support nvidia only mode/Gsync?

Like i saw that i can change this setting but i haven't touched it becuase it either does nothing or can brick the BIOS  

Mu device is the Ideapad Gaming 3 15ACH6

johnfanv2, 2023-06-03:
It should enable/disable the hybrid/dynamic graphics in BIOS. It only takes effect after a restart. Please test it like described in the post above: https://github.com/johnfanv2/LenovoLegionLinux/issues/50#issuecomment-1528987626.

It does not seem to be Gsync. This was just the name of the WMI method.

kleidiss, 2023-06-03:
As expected this just resets back to 1 after a restart because lenovo thought it would be funny to completly fuck the 2021 ideapad line

Maybe graying the option out on ideapad line would be a good idea

closed 2023-08-20 by Armynator