johnfanv2/LenovoLegionLinux #51: Lenovo legion 7i 16ithg6

ulrikstrid, 2023-04-28:
I don't know how to post this correctly, but I have the Legion 7i from 2021 and have packaged it for [nixos](https://github.com/NixOS/nixpkgs/pull/228506). From some initial testing **everything seems to work**, even the battery conservation setting which I've been looking for since switching over to Linux.

If you want more info let me know and I'll post whatever.

**Model and Debug Info**

Model name: Lenovo legion 7i 16ithg6
CPU model: Core i9-11980HK
GPU model: NVIDIA GeForce RTX 3080 Laptop GPU
Keyboard backlight: RGB

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
???
```

Output of `sudo dmidecode -t bios`:
```text
???
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
???
```

johnfanv2, 2023-04-28:
Great! Thanks for creating the nixos package. You might want to update it in a month because right now new features are being added and tested.

Can you give me the output of `sudo dmidecode -t bios` so I can get the BIOS version. Then I can add your model and bios to the list of compatible models.

ulrikstrid, 2023-05-02:
I updated the initial issue description with the outputs requested.

When my nixpkgs PR is merged, packaging Linux modules and even Python code is new to me so I've been asking for some thorough reviews, it will be available to search for on https://search.nixos.org as you mention. But the install instructions will be as simple as:
`boot.extraModulePackages = [ config.boot.kernelPackages.lenovo-legion-module ];`

johnfanv2, 2023-06-11:
Great, I added a link to the NixOS packages.

closed 2023-06-11 by johnfanv2