johnfanv2/LenovoLegionLinux #53: Secure Boot key signing

Reaversword, 2023-04-30:
Hello!, first of all, thank you for create LenovoLegionLinux!!!

I'm trying to follow your guide, but no idea how to sign -whatever... file?- to get secure boot enabled and have everything working.

From here:
https://github.com/dell/dkms#secure-boot

"With the appropriate key material on the system, enroll the public key:"

What's "appropiate key material" of LLL? LenovoLegionLinux/kernel_module/legion-laptop.ko?. If I do: mokutil --import /LenovoLegionLinux/kernel_module/legion-laptop.ko
it tells to me:
Abort!!! legion-laptop.ko is not a valid x509 certificate in DER format

Sorry, I've never in my life signed anything, I have no idea what to do or what I need to sign or why if I use 

# mokutil --import /var/lib/dkms/mok.pub" (with than end " and without it)

I don't receive any requesting for put a secure boot password.

I'm absolutely blind about this issue.

Anybody can point so lose people as myself in the right way?

johnfanv2, 2023-04-30:
I guess @MrDuartePT knows this better!

MrDuartePT, 2023-04-30:
Well i dont know what distro you are using but i recommed using sbctl is a easy process and allow to use the Microsoft key to be possible to use discrete mode with secure boot enable on linux (if you remove microsoft key you get black screen in discrete mode and in hybrid the nvidia module dosent load on boot)
 
 the guide i will post helps just install sbctl on your distro
 secure boot guide: https://wiki.archlinux.org/title/User:Krin/Secure_Boot,_full_disk_encryption,_and_TPM2_unlocking_install#Secure_boot

pm4rcin, 2023-05-11:
> @Reaversword Well i dont know what distro you are using but i recommed using sbctl is a easy process and allow to use the Microsoft key to be possible to use discrete mode with secure boot enable on linux (if you remove microsoft key you get black screen in discrete mode and in hybrid the nvidia module dosent load on boot [i know that because i try to use custom keys one time])
> 
> the guide i will post helps just install sbctl on your distro secure boot guide: https://wiki.archlinux.org/title/User:Krin/Secure_Boot,_full_disk_encryption,_and_TPM2_unlocking_install#Secure_boot

Do you mean that proprietary/nouveau just won't work at all in hybrid without those M$ keys? I thought that the arch wiki warning was mostly for some old laptops with broken BIOSes. Regarding nvidia is there some error that complaints about it or it just some generic error or smth?

MrDuartePT, 2023-05-12:
If you remove the microsoft keys and activate secure boot you will get a black screen if you boot in discrete mode and in hybrid mode the nvidia/nouveau modules will load but the card cant be used. The problem is how nvidia vbios is sign on laptop and because lenovo legion laptop dont have optirom in the tpm you need to use the microsoft keys

This issues on sbctl project can help you understand:
https://github.com/Foxboron/sbctl/issues/85#issuecomment-886539689
https://github.com/Foxboron/sbctl/issues/40

MrDuartePT, 2023-05-12:
@pm4rcin @Reaversword @johnfanv2 in the latest bios Option Rom is working:
➜  ~ doas sbctl status
Installed:    ✓ sbctl is installed
Owner GUID:
Setup Mode:    ✓ Disabled
Secure Boot:    ✓ Enabled
Vendor Keys:    tpm-eventlog
This means if you use sbctl to create or import your custom keys (or you can do everthing manually) and include the Option Rom the mux will be intact (you need to have AMD PSP or intel equivalent enable on the bios for the TPM to work after that i think you can disable it)

pm4rcin, 2023-05-12:
@MrDuartePT that's great to hear. Could you tell which laptop model do you have? I also have read the links you've provided and some more and found one example of hard-brick because of buggy firmware. But it was thinkpad t14s gen 1 AMD so I guess it shouldn't be a problem but who knows... I'll probably try it sooner or later but first I need to figure out how to do that on Fedora Sericea since it uses Windows keys. Or rI'll return again to Arch or smth :D

MrDuartePT, 2023-05-13:
My laptop is a Legion 5 gen 6 AMD variant. 
Fedora use MS key by default to provide compatible but i think you can change to Option Rom if possible. Try updating your bios version and then try to see if possible to change before distro hopping. 
I think i will close this issue if you need any help go to the discord server and ping me.

mercifulboss, 2023-05-25:
Will this allow the preservation of secureboot keys after kernel updates? I heard that you need to manually recreate keys after every update (or something like that) and that sounds like a pain on arch

MrDuartePT, 2023-06-16:
I allways use the same keys and my system automatic sign the kernel file after building the initramfs

MrDuartePT, 2023-07-08:
Since this dosent have any activity for the past 1 month and the kernel module work with secure boot (at least on my setup using sbctl) I  will close this issue.

closed 2023-07-08 by MrDuartePT