johnfanv2/LenovoLegionLinux #54: Testing controlling light of Y-Logo in lid and light at IO-Ports [looking for testers]

johnfanv2, 2023-05-01:
If you have a model with a light at the IO-ports at the back, a light in the Y-Logo in the lid, or another light in the lid please help to test this feature. Please provide the following information. It must be tested **on multiple different** models since I assume they have slightly different methods. I will first gather information from different models (see below), then implement it, and then ask you to test if it works.

**Model and Debug Info**

Model name: ???
CPU model: ??? (e.g. AMD Ryzen 7 5800H)
GPU model: ??? (e.g NVIDIA RTX 3700)
Keyboard backlight: ??? (RGB, single color with off/on, single color with off/medium/bright)
Light in lid or logo: ??? (yes/no)
Light at IO-Ports at back: ??? (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
???
```

Output of `sudo dmidecode -t bios`:
```text
???
```

The following information would be useful.

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```

Guanran928, 2023-05-04:
[image]
Are you looking for this?

Lenovo Legion Y7000 (china version)
Intel i7-8750H + Nvidia 1050ti

johnfanv2, 2023-05-04:
@Guanran928 Yes, but I am not sure if it will work on this older model. It would be great if you can do the tests.

Guanran928, 2023-05-05:
[image]

[image]

Oops. I think disabling Y-logo light feature isn't available for my model, but here's my system information anyway

johnfanv2, 2023-05-05:
@Guanran928 Sorry, I forgot to mention that you also have to pull and load the kernel module (see above and README) before it might work :smiley: . Could you upload the file output1.txt and output2.txt. I guess I have to add your model to the kernel module before it work and I need this information to do it.

Guanran928, 2023-05-07:
Here you go :D.

[output1.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11415267/output1.txt)
[output2.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11415268/output2.txt)

Guanran928, 2023-05-07:
Also, I think the command `sudo make reloadmodule` isn't working for me.
The log says:
`[  648.065732] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.`
Currently I have no idea how to force load XD

Full log here:
[1.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11415316/1.txt)

johnfanv2, 2023-05-07:
Thanks a lot for the logs. You can force loading it with `sudo make forcereloadmodule`.

Guanran928, 2023-05-07:
Looks like it is still not working with `sudo make forcereloadmodule`

Here is the log with `sudo make forcereloadmodule > 2.txt`
[2.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11415387/2.txt)

johnfanv2, 2023-05-07:
@Guanran928 I added your model/BIOS version to the the code. You can try it now by getting the new code (`git pull` or just downloading again), compiling again (`make`), and loading again (`make reloadmodule`) it should now work without forcing.

Guanran928, 2023-05-07:
`sudo make reloadmodule` worked, but FN+L didn't work for me, after some researching, I found that disabling Y-Logo light is not a thing on my model with this specific GPU ([according to Lenovo](https://iknow.lenovo.com.cn/home/detail/code/181742)), I didn't know this and I am really sorry. I apologize for the effort that you spent. Is there anything else I can provide?

johnfanv2, 2023-05-07:
@Guanran928 Thanks for the feedback on controlling the light!

If you want, you can test if the other features, e.g. in the python GUI, work on your laptop in Linux. If something does not work, then feel free to report back I can try to adapt the software to make it work with your model.

Peter-Levine, 2023-05-09:
Model name: Lenovo Legion 5 Pro-16ACH6H Laptop - Type 82JQ
CPU model: AMD Ryzen 7 5800H
GPU model: Nvidia GeForce RTX 3070 Mobile
Keyboard backlight: off
Light in lid or logo: no

<details>

<summary>sudo dmidecode -t system</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JQ
        Version: Legion 5 Pro 16ACH6H
        Serial Number:
        UUID:
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JQ_BU_idea_FM_Legion 5 Pro 16ACH6H
        Family: Legion 5 Pro 16ACH6H

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```
</details>

<details>
<summary>sudo dmidecode -t bios</summary>

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN60WW
        Release Date: 03/07/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
[... 197 lines cut ...]
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       63      0       66      0       43
1800     1700    2       2       58      63      62      66      38      46
1800     1900    2       2       58      63      62      66      42      51
2100     2100    2       2       58      63      62      66      46      127
2300     2400    2       2       58      69      62      66      127     127
2600     2600    2       2       63      76      62      66      127     127
2800     2900    2       2       71      80      62      70      127     127
3500     3500    2       2       75      88      65      75      127     127
3800     3800    2       2       83      94      70      78      127     127
4200     4200    2       2       90      127     74      127     127     127
=====================
```
</details>

<details>

<summary>cat /sys/class/leds/platform\:\:ylogo/brightness (with light on)</summary>
0

</details>

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11426594/fwts_wmi.log)
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11426630/acpi_re.tar.gz)

RzVa317, 2023-05-09:
Model and Debug Info

Model name: Legion S7
CPU model: AMD Ryzen 7 6800H
GPU model: Radeon RX 6600S
Keyboard backlight: RGB
Light in lid or logo: No
Light at IO-Ports at back: Yes

<details>
<summary>sudo dmidecode -t system</summary>

```bash
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82UG
	Version: Legion S7 16ARHA7
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82UG_BU_idea_FM_Legion S7 16ARHA7
	Family: Legion S7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

</details>

<details>
<summary>sudo dmidecode -t bios</summary>

```bash
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KFCN32WW
	Release Date: 09/29/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
[... 237 lines cut ...]
2800	 2800	 2	 2	 60	 80	 70	 76	 52	 56
3200	 3200	 2	 2	 60	 80	 70	 76	 55	 58
3500	 3500	 2	 2	 60	 80	 70	 76	 57	 61
3500	 3500	 2	 2	 60	 80	 70	 76	 60	 64
=====================
```

</details>

<details>
<summary>cat /sys/class/leds/platform\:\:ioport/brightness</summary>

Same error as above.

Running: `cat /sys/class/leds/platform\:\:ylogo/brightness` still returns `0`

</details>

</details>

The attached files were obtained a couple of weeks ago, let me know if you need more recent data.

[windows_wmi.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11435178/windows_wmi.txt)
[fwts.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11435179/fwts.tar.gz)
[acpi.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11435180/acpi.tar.gz)

potatoalienof13, 2023-06-02:
Model name: Lenovo Pro 7i Gen 8 16IRX8
CPU model: i9-13900hx
GPU model:rtx 4070
Keyboard backlight: RGB 
Light in lid or logo: No
Light at IO-Ports at back: Yes, not user-controllable (yes/no)

`sudo dmidecode -t system` 
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WR
	Version: Legion Pro 7 16IRX8
	Serial Number: ############
	UUID: #############
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WR_BU_idea_FM_Legion Pro 7 16IRX8
	Family: Legion Pro 7 16IRX8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

`sudo dmidecode -t bios`
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN35WW
	Release Date: 04/07/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.35
	Firmware Revision: 1.35

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```
[fmts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11638679/fmts_wmi.log)
[results.log](ht
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11638702/acpi_re.tar.gz)
tps://github.com/johnfanv2/LenovoLegionLinux/files/11638680/results.log)

XenHat, 2023-06-03:
Model name: Lenovo Legion 5 Pro 16ach6h (Model 82JQ)
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3060
Keyboard backlight: RGB (firmware control, no iCUE required)
Light in lid or logo: Yes
Light at IO-Ports at back: No (?)

```
$ sudo dmidecode -t system | grep -v UUID | grep -v "Serial Number"
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JQ
        Version: Legion 5 Pro 16ACH6H
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JQ_BU_idea_FM_Legion 5 Pro 16ACH6H
        Family: Legion 5 Pro 16ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected

```
```
$ sudo dmidecode -t bios
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN59WW
        Release Date: 11/21/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.59
        Firmware Revision: 1.59

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
````

Will update the post shortly after I pull and compile the latest code

XenHat, 2023-06-04:
A friend with a similar laptop also kindly ran the windows command for me.

Model name: Lenovo Legion 5 Pro 6th Gen
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3070
Keyboard backlight: RGB (firmware control, no iCUE required)
Light in lid or logo: Yes
Light at IO-Ports at back: No (?)

[Windows PowerShell.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11646261/Windows.PowerShell.txt)

halfling81, 2023-06-05:
Model name: Legion 7 16ACHg6
CPU model: AMD Ryzen™ 9 5900HX Processor(Ryzen™ 9 5900HX)
GPU model:  NVIDIA® GeForce RTX™ 3080 16GB
Keyboard backlight: RGB (firmware control, no iCUE required)
Light in lid or logo: Yes
Light at IO-Ports at back: Yes

### sudo dmidecode -t system 
```shell
 $ sudo dmidecode -t system | grep -v "Serial Number" | grep -v UUID
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82N6
        Version: Legion 7 16ACHg6
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82N6_BU_idea_FM_Legion 7 16ACHg6
        Family: Legion 7 16ACHg6

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

### sudo dmidecode -t bios
```shell
$ sudo dmidecode -t bios
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN54WW
        Release Date: 05/05/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.54
        Firmware Revision: 1.54

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

#### fwts

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11649895/fwts_wmi.log)

#### ACPI Tables

[ACPI_tables.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11649936/ACPI_tables.tar.gz)

SirRFI, 2023-06-07:
I would like to contribute, but `ioport/brightness` is missing. I have **Lenovo Legion 7 Gen 7 AMD** (`16ARHA7`).

```$ ls /sys/class/leds/
input2::capslock     input33::capslock    input4::capslock
input2::numlock      input33::numlock     input4::compose
input2::scrolllock   input33::scrolllock  input4::kana
input31::capslock    input3::capslock     input4::numlock
input31::compose     input3::compose      input4::scrolllock
input31::kana        input3::kana         platform::kbd_backlight
input31::numlock     input3::numlock      platform::ylogo
input31::scrolllock  input3::scrolllock```

SirRFI, 2023-06-08:
@XenHat I followed instructions from README. Latest commit in local directory matches current latest: https://github.com/johnfanv2/LenovoLegionLinux/commit/f3f363fc1c24469dfc7683003b9ba91dec38f5af

SirRFI, 2023-06-22:
It's been a while, so I removed old directory, updated the system, rebooted and repeated steps:

`/sys/class/leds/` before `sudo make reloadmodule`
```
input2::capslock/   input2::scrolllock/ input3::compose/    input3::numlock/    input6::capslock/   input6::kana/       input6::scrolllock/
input2::numlock/    input3::capslock/   input3::kana/       input3::scrolllock/ input6::compose/    input6::numlock/
```

`/sys/class/leds/` after `sudo make reloadmodule`
```
input2::capslock/        input3::capslock/        input3::numlock/         input6::compose/         input6::scrolllock/      
input2::numlock/         input3::compose/         input3::scrolllock/      input6::kana/            platform::kbd_backlight/ 
input2::scrolllock/      input3::kana/            input6::capslock/        input6::numlock/         platform::ylogo/
```

`platform::ioport/brightness` is still missing.

Perhaps the instruction/code isn't adjusted to my model. It has toggle backlight for ports on the back, but no "Y-Logo" - instead it's "LEGION" text logo with RGB backlight.

daemonofchaos, 2023-06-28:
Model name: Legion 7-15IMH05
CPU model: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz
GPU model: TU104BM [GeForce RTX 2070 SUPER Mobile / Max-Q] (rev a1)
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: yes

Output of sudo dmidecode -t system
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81YT
        Version: Lenovo Legion 7 15IMH05
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81YT_BU_idea_FM_Legion 7 15IMH05
        Family: Legion 7 15IMH05

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

Output of sudo dmidecode -t bios
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: E9CN66WW(V4.11)
        Release Date: 09/30/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 12 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
[... 100 lines cut ...]
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
EC fanfullspeed error: error
EC fanfullspeed status: -1
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

output of cat   /sys/class/leds/platform\:\:ioport/brightness
cat: '/sys/class/leds/platform::ioport/brightness': No such file or directory

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11888365/fwts_wmi.log)

[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11888380/acpi_re.tar.gz)

sakgoyal, 2023-07-14:
# Model name: Legion Y7000p-1060
## Feedora 38 w/ KDE

CPU model: 8750h
GPU model: gtx1060-6gb
Keyboard backlight: white (off, med, bright)
Light in lid or logo: Big white Y logo
Light at IO-Ports at back: No (except small red power charging indicator)
<details> 
  <summary>Output of `sudo dmidecode -t system` </summary>

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81LF
        Version: Lenovo Legion Y7000P-1060
        Serial Number: [redacted]
        UUID: d56f1dfa-b1a5-11e8-b5f5-e86a640e2769
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81LF_BU_idea_FM_Legion Y7000P-1060
        Family: Legion Y7000P-1060

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0024, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0031, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

```
</details>

<details>
<summary>Output of sudo dmidecode -t bios: </summary>

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: LENOVO
        Version: 9VCN22WW
        Release Date: 11/30/2020
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 10 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.22
        Firmware Revision: 1.22

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```
</details>

- Fn+L does nothing
- `/sys/kernel/debug/legion` directory does not exist 
- `/sys/class/leds/platform\:\:ylogo/` directory does not exist

Log file: [fwts.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12045257/fwts.log)

[image]

johnfanv2, 2023-07-14:
@sakgoyal Can you somehow control the light of the logo in the lid, e.g. with Ctrl+U or Ctrl+Y? Otherwise, I think controlling the lights is not possible.

sakgoyal, 2023-07-14:
I was not able to control it no matter what. I tried `Fn+{all letters}`.
I just want to be able to control the fan behavior (Fn+Q). I am ok with the light staying on. 
I am also able to manually control the conservation mode for the battery using this command when inside `sudo su`:
`echo 1 > /sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode`
I didnt know this model even supported that option until I tried it. I never saw that option in vantage when I was on windows.

johnfanv2, 2023-07-15:
@sakgoyal Then I guess it is not possible to control the lights by software when it is not possible to control it by some key combo. I see that you use a older model. I have to add it to the kernel driver module. Please create a new issue, where we can test it and try to test all the features step-by-step.

linuxazo, 2023-08-11:
I guess this is the same laptop as in previous post from @themightymak ...  
Either way, I am posting more data @johnfanv2 

Model name:  **2022 Lenovo Legion 7 16ARHA7 (82UH) - AMD Advantage Edition**
CPU Model:  **AMD Ryzen 7 6800H**
GPU Model:  **Integrated: APU AMD R680M Rembrandt / Discrete: AMD RX 6700M**
Keyboard backlight:  **RGB per key backlight with off/low/medium/bright** 

Output of `sudo dmidecode -t system`:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82UH
        Version: Legion 7 16ARHA7
        Serial Number: removed for privacy
        UUID: removed for privacy
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82UH_BU_idea_FM_Legion
 7 16ARHA7
        Family: Legion 7 16ARHA7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```
Output of `sudo dmidecode -t bios`:
```
BIOS Information
        Vendor: LENOVO
        Version: K9CN42WW
        Release Date: 03/10/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.42
        Firmware Revision: 1.42

Handle 0x001C, DMI type 13, 22 bytes
[... 125 lines cut ...]
`cat: '/sys/class/leds/platform::ioport/brightness': No such file or directory`

- Lights ON (Fn+U)
Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
[fn+u_on_fancurve.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12324738/fn%2Bu_on_fancurve.txt)
Output of `sudo cat /sys/class/leds/platform\:\:ioport/brightness`:
`cat: '/sys/class/leds/platform::ioport/brightness': No such file or directory`

**Lid LEGION light**

- LEGION Light OFF (Fn+L)
Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
[fn+l_off_fancurve.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12324705/fn%2Bl_off_fancurve.txt)
Output of `sudo cat /sys/class/leds/platform\:\:ylogo/brightness`:
`0`

- LEGION Light ON (Fn+L)
Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
[fn+l_on_fancurve.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12324711/fn%2Bl_on_fancurve.txt)
Output of `sudo cat /sys/class/leds/platform\:\:ylogo/brightness`:
`0`

**Please let me know how you would like me to test it and where and how you would like me to post the results and logs and eventual bugs or issue for further development for this laptop.**

Thank You so much for all your effort!

johnfanv2, 2023-08-11:
@linuxazo Did I understand that correctly that you can control the light at the IO-Ports and the light in the logo with the keys Fn+U and Fn+L? This is interesting because I want to get this into the driver and software but I cannot control it on my own. I will look at the ACPI dissambly and try to find a way to control it on the newer models. Currently, you logs looks like it does not work. Stay tuned!

linuxazo, 2023-08-11:
Yes. You understood correctly.  I can control those lights without issues.

linuxazo, 2023-08-11:
> @linuxazo Did I understand that correctly that you can control the light at the IO-Ports and the light in the logo with the keys Fn+U and Fn+L? This is interesting because I want to get this into the driver and software but I cannot control it on my own model. I will look at the ACPI dissambly and try to find a way to control it on the newer models. Currently, you logs looks like it does not work. Stay tuned!

@johnfanv2 Yes. You understood correctly. I can control those lights with those keyboard shortcuts without issues.

linuxazo, 2023-08-11:
This is the output of  `sudo dmesg | grep legion`:

```
[   13.906686] legion_laptop: legion_laptop starts loading
[   13.906721] legion PNP0C09:00: legion_laptop platform driver probing
[   13.906723] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UH; DMI_BIOS_VERSION:K9CN42WW
[   13.906724] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UH; DMI_BIOS_VERSION:K9CN42WW
[   13.906726] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[   13.906727] legion PNP0C09:00: Using configuration for system: K9CN
[   13.908930] legion PNP0C09:00: ACPI CFG: 2081289482
[   13.908932] legion_laptop: Creating RAM access to embedded controller
[   13.908942] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x00000000e1a3b3a8
[   13.909033] legion PNP0C09:00: Read embedded controller ID 0x5507
[   13.909034] legion PNP0C09:00: Skipped checking embedded controller id
[   13.909035] legion PNP0C09:00: Creating debugfs inteface
[   13.909041] legion_laptop: Creating sysfs inteface
[   13.909381] legion_laptop: Creating hwmon interface
[   13.909679] legion_laptop: Creating platform profile support
[   13.909682] legion_laptop: Init WMI driver support
[   13.909816] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[   13.910710] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[   13.910863] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[   13.910936] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[   13.910956] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[   13.911113] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[   13.911289] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[   13.911309] legion_laptop: Init keyboard backlight LED driver
[   13.931251] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[   13.931254] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 3, but got 0
[   13.931255] legion_laptop: Error reading keyboard brighntess
[   13.931256] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[   13.931257] legion_laptop: Init Y-Logo LED driver
[   13.933558] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[   13.936085] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[   13.936088] legion_laptop: Init IO-Port LED driver
[   13.936528] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[   13.936530] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[   13.936531] legion_laptop: Error reading brighntess for light: 5
[   13.936532] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[   13.936533] legion PNP0C09:00: legion_laptop loaded for this device
```
Hope it helps...

SirRFI, 2023-08-12:
> @linuxazo Did I understand that correctly that you can control the light at the IO-Ports and the light in the logo with the keys Fn+U and Fn+L? This is interesting because I want to get this into [...]

Yes. Legion 7 Gen 7 AMD (`16ARHA7`, 2022) controls most stuff on firmware level, so it works out of the both on both Windows and Linux, even before loading the OS actually. This applies to toggling highlighting icons for the ports on the back, LEGION logo on the lid and switching keyboard RGB profiles and brightness on the keyboard. [I posted my notes about the device, including function keys and Linux compatibility here](https://www.reddit.com/r/LenovoLegion/comments/143j91n/research_notes_about_lenovo_legion_7_gen_7_amd/). What didn't work last time I tested is `Fn` + `R` for 60/165Hz switch. [There's open issue to support the device in OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2825) if that's any use to you.

linuxazo, 2023-08-12:
@SirRFI I saw in your reddit post that you are searching for proprietary fingerprint drivers:
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-edge-laptops/thinkpad-e14-gen-4-type-21eb-and-21ec/downloads/driver-list/component?name=Fingerprint%20Reader&id=A4F7592E-3C73-4CCF-ABB9-09549219DFC8

Also, for Debian: 
https://wiki.debian.org/SecurityManagement/fingerprint%20authentication

I can confirm it works as well...

gudata, 2024-05-04:
Model name: LENOVO Legion 5 Pro - 82JD004LBM

CPU model: 11th Gen Intel(R) Core(TM) i7-11800H
GPU model: NVIDIA Corporation GA104M [GeForce RTX 3070 Mobile / Max-Q] (rev a1)
Keyboard backlight: RGB

Light in lid or logo: yes = Y
Light at IO-Ports at back: no

```
# sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 41
CPU temperature EC error: 0
CPU temperature EC: 41
CPU temperature ACPI error: 0
CPU temperature ACPI: 41
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2736
1 fanspeed EC error: 0
1 fanspeed EC: 2736
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1800
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 2681
2 fanspeed EC error: 0
2 fanspeed EC: 2681
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
[... 147 lines cut ...]
fanfullspeed EC: 0
EC fan curve current point id: 1
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       5       0       48      0       67      0       36
1800     1700    5       5       45      48      63      67      32      40
1800     1900    5       5       45      48      63      67      37      43
2100     2100    5       5       45      48      63      67      40      127
2400     2300    2       2       45      58      63      67      127     127
2600     2600    2       2       52      66      63      73      127     127
2900     2900    2       2       60      72      69      76      127     127
3500     3500    2       2       68      77      73      79      127     127
3800     3800    2       2       73      80      75      82      127     127
4200     4200    2       2       76      127     78      127     127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```
 /sys/class/leds/platform\:\:ylogo/brightness
..no platform..

[lenovo-5-pro.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/15210745/lenovo-5-pro.zip)

unknown10777, 2024-06-11:
Model and Debug Info

Model name: Y7000 2020
CPU model: Intel I5 10300H (e.g. AMD Ryzen 7 5800H)
GPU model: Nvidia GTX 1650Ti (e.g NVIDIA RTX 3700)
Keyboard backlight: `single with off/medium/bright`
Light in lid or logo: `yes`
Light at IO-Ports at back: `no?`
Keyboard backlight driver in kernel is working fine, but only in off/bright, not medium

Output of sudo dmidecode -t system. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82AV
        Version: Lenovo Legion Y7000 2020
        Serial Number: wiped
        UUID: wiped
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82AV_BU_idea_FM_Legion Y7000 2020
        Family: Legion Y7000 2020

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0024, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0031, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```
Output of sudo dmidecode -t bios:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: EFCN46WW
        Release Date: 12/04/2020
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 11 MB
[... 223 lines cut ...]
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 2
EC fan curve points size: 9
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       53      0       52      0       36
0        0       2       2       49      53      48      52      33      45
0        0       2       2       49      53      48      52      41      127
0        0       2       2       49      64      48      52      127     127
4500     4500    2       2       60      68      48      52      127     127
4500     4500    2       2       64      75      48      52      127     127
4500     4500    2       2       71      80      48      56      127     127
4500     4500    2       2       76      84      52      65      127     127
4500     4500    2       2       80      127     61      71      127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
    Get output of cat   /sys/class/leds/platform\:\:ylogo/brightness and copy here,
```
not found
```

tomasvalek, 2024-06-13:
**Model and Debug Info**

Model name: Lenovo Legion Pro 7 16ARX8H
CPU model: AMD Ryzen 9 7945HX
GPU model: NVIDIA GeForce RTX 4080
Keyboard backlight: RGB
Light in lid or logo: No
Light at IO-Ports at back: Yes. Led keyboard and led line under notebook case.

`sudo dmidecode -t system`

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WS
	Version: Legion Pro 7 16ARX8H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WS_BU_idea_FM_Legion Pro 7 16ARX8H
	Family: Legion Pro 7 16ARX8H

Handle 0x0017, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```  

`sudo dmidecode -t bios`    

    
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: LPCN45WW
	Release Date: 08/02/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.45
	Firmware Revision: 1.45

Handle 0x0018, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

**Inspecting WMI entries**
[fwts_wmi.log](https://github.com/user-attachments/files/15828515/fwts_wmi.log)

**ACPI_RE**
[acpi_re.zip](https://github.com/user-attachments/files/15828546/acpi_re.zip)

CCF100, 2024-06-16:
# fastfetch output:
```
❯ fastfetch
                  -`                     caleb@caleb-gaming-laptop-archlinux
                 .o+`                    -----------------------------------
                `ooo/                    OS: Arch Linux x86_64
               `+oooo:                   Host: 81SX (Legion Y540-15IRH)
              `+oooooo:                  Kernel: Linux 6.9.4-zen1-1-zen
              -+oooooo+:                 Uptime: 33 mins
            `/:-:++oooo+:                Packages: 4726 (pacman), 13 (flatpak), 6 (snap)
           `/++++/+++++++:               Shell: zsh 5.9
          `/++++++++++++++:              Display (BOE06FB): 1920x1080 @ 60Hz [Built-in]
         `/+++ooooooooooooo/`            DE: KDE Plasma 6.0.5
        ./ooosssso++osssssso+`           WM: KWin (Wayland)
       .oossssso-````/ossssss+`          WM Theme: Breeze
      -osssssso.      :ssssssso.         Theme: Breeze (Dark) [QT], Breeze-Dark [GTK2], Breeze [GTK3/4]
     :osssssss/        osssso+++.        Icons: breeze-dark [QT], breeze-dark [GTK2/3/4]
    /ossssssss/        +ssssooo/-        Font: Noto Sans (10pt) [QT], Noto Sans (10pt) [GTK2/3/4]
  `/ossssso+/:-        -:/+osssso+-      Cursor: DeppinDark (24px)
 `+sso+:-`                 `.-/+oso:     Terminal: konsole 24.5.0
`++:.                           `-/+/    Terminal Font: SpaceMono Nerd Font (11pt)
.`                                 `/    CPU: Intel(R) Core(TM) i5-9300H (8) @ 4.10 GHz
                                         GPU: NVIDIA GeForce GTX 1660 Ti Mobile [Discrete]
                                         Memory: 5.23 GiB / 15.42 GiB (34%)
                                         Swap: 4.04 GiB / 27.56 GiB (15%)
                                         Disk (/): 233.79 GiB / 238.38 GiB (98%) - btrfs
                                         Disk (/media/DataEXT4): 1.78 TiB / 1.82 TiB (98%) - btrfs
                                         Local IP (wlp0s20f3): 192.168.1.89/24 *
                                         Battery: 55% [Charging]
                                         Locale: C.UTF-8

                                         ████████████████████████
                                         ████████████████████████
    ~                                                                     ✔  at 03:59:01 PM  
```
# dmidecode -t -system
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81SX
        Version: Legion Y540-15IRH
        Serial Number: <REDACTED>
        UUID: <REDACTED>
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81SX_BU_idea_FM_Legion Y540-15IRH
        Family: Legion Y540-15IRH

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0024, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
[... 320 lines cut ...]
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light Y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
# couldn't get fwts to build, but I tried... 😭
# ACPI tables:
[acpi_re.zip](https://github.com/user-attachments/files/15859873/acpi_re.zip)

Sandrdr, 2024-10-03:
> Если у вас есть модель с подсветкой на портах ввода-вывода сзади, подсветкой в ​​Y-Logo на крышке или еще одной подсветкой на крышке, пожалуйста, помогите протестировать эту функцию. Пожалуйста, предоставьте следующую информацию. Ее необходимо протестировать **на нескольких разных** моделях, поскольку я предполагаю, что у них немного разные методы. Сначала я соберу информацию с разных моделей (см. ниже), затем реализую ее, а затем попрошу вас проверить, работает ли она.
> 
> **Модель и отладочная информация**
> 
> Название модели: ??? Модель ЦП: ??? (например, AMD Ryzen 7 5800H) Модель ГП: ??? (например, NVIDIA RTX 3700) Подсветка клавиатуры: ??? (RGB, один цвет с выключением/включением, один цвет с выключением/средней/яркой) Подсветка крышки или логотипа: ??? (да/нет) Подсветка портов ввода-вывода сзади: ??? (да/нет)
> 
> Вывод `sudo dmidecode -t system`. Пожалуйста, удалите серийный номер и UUID для конфиденциальности:
> 
> ```
> ???
> ```
> 
> Вывод `sudo dmidecode -t bios`:
> 
> ```
> ???
> ```
> 
> # Тест состояния света для чтения
> Пожалуйста, извлеките новейший код, скомпилируйте и загрузите модуль ядра, как описано в файле README, т. е. до шага `sudo make reloadmodule`.
> 
> ## Свет IO-порта
> Подсветку на портах ввода-вывода можно включить/выключить нажатием Fn+U (может быть, как-то иначе?).
> 
> * Выключи свет.
> * Получите вывод `sudo cat /sys/kernel/debug/legion/fancurve`и скопируйте сюда.
> * Получите вывод `cat   /sys/class/leds/platform\:\:ioport/brightness`и скопируйте сюда.
> * Включи свет.
> * Получите вывод `sudo cat /sys/kernel/debug/legion/fancurve`и скопируйте сюда.
> * Получите вывод `cat   /sys/class/leds/platform\:\:ioport/brightness`и скопируйте сюда.
> 
> ## Y-Логотип/Подсветка крышки
> Подсветку можно включить/выключить нажатием Fn+L (может быть, как-то иначе?).
> 
> * Выключи свет.
> * Получите вывод `sudo cat /sys/kernel/debug/legion/fancurve`и скопируйте сюда.
> * Получите вывод `cat   /sys/class/leds/platform\:\:ylogo/brightness`и скопируйте сюда,
> * Включи свет.
> * Получите вывод `sudo cat /sys/kernel/debug/legion/fancurve`и скопируйте сюда.
> * Получите вывод `cat   /sys/class/leds/platform\:\:ylogo/brightness`и скопируйте сюда,
> 
> # Следующая информация будет полезна, но необязательна
> ## Проверка записей WMI
> ```shell
> # Install required tools
> sudo apt install fwts
> 
> # Create folder for all the new files
> mkdir fwts_re
> cd fwts_re
> sudo fwts wmi - > fwts_wmi.log
> ```
> 
> Затем загрузите файлы.
> 
> ## Разборка таблиц ACPI
> ```shell
> # Install requiered tools
> sudo apt install acpica-tools
> ```
> 
> ```shell
> # Create folder for all the new files
> mkdir acpi_re
> cd acpi_re
> 
> # List ACPI tables and copy them
> ls /sys/firmware/acpi/tables/
> sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .
> 
[... 112 lines cut ...]
> ```powershell
> $wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
> foreach ($wmi_class in $wmi_classes){
>   Write-Host "########################################"
>   Write-Host "########################################"
>   Write-Host "########################################"
>   Write-Host "Name:" $wmi_class.Name
>   Write-Host "Class Name:" $wmi_class.Name 
>   Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
>   Write-Host "Description:" $wmi_class.Methods.Count
>   Write-Host "Methods:"
>   foreach ($method in $wmi_class.Methods){
>     Write-Host "Name:" $method.Name
>     Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
>     Write-Host "Class Name:" $wmi_class.Name 
>     Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
>     Write-Host "Description:" $method.Qualifiers["Description"].Value
>     Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
>     Write-Host ""
>   }
>   Write-Host ""
> }
> ```

I have Legion 7 16 archa7 (r5 6800H & rx 6850m xt), , but I can't install the script. I will be grateful if someone helps + it will be possible to test everything you want. fedora silverblue 41

IndoMK, 2024-10-09:
#### Lenovo Legion 7 testing
Model Name: Legion 7 16IAX7 - Type 82TD
CPU model: 12th Gen Intel® Core™ i9-12900HX
GPU model: Integrated: Mesa Intel® UHD Graphics / Discrete: Nvidia GeForce RTX 3080 Ti
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: yes, plus bar around case

output `sudo dmidecode -t system`
```
# dmidecode 3.6  
Getting SMBIOS data from sysfs.  
SMBIOS 3.4 present.  
  
Handle 0x0001, DMI type 1, 27 bytes  
System Information  
       Manufacturer: LENOVO  
       Product Name: 82TD  
       Version: Legion 7 16IAX7  
       Serial Number: redacted  
       UUID: redacted  
       Wake-up Type: Power Switch  
       SKU Number: LENOVO_MT_82TD_BU_idea_FM_Legion 7 16IAX7  
       Family: Legion 7 16IAX7  
  
Handle 0x0024, DMI type 12, 5 bytes  
System Configuration Options  
       Option 1: ConfigOptions1  
       Option 2: ConfigOptions2  
       Option 3: ConfigOptions3  
  
Handle 0x0027, DMI type 15, 29 bytes  
System Event Log  
       Area Length: 0 bytes  
       Header Start Offset: 0x0000  
       Header Length: 8192 bytes  
       Data Start Offset: 0x2000  
       Access Method: General-purpose non-volatile data functions  
       Access Address: 0x0000  
       Status: Valid, Not Full  
       Change Token: 0x12345678  
       Header Format: OEM-specific  
       Supported Log Type Descriptors: 3  
       Descriptor 1: POST memory resize  
       Data Format 1: None  
       Descriptor 2: POST error  
       Data Format 2: POST results bitmap  
       Descriptor 3: Log area reset/cleared  
       Data Format 3: None  
  
Handle 0x0034, DMI type 32, 11 bytes  
System Boot Information  
       Status: No errors detected
```

output `sudo dmidecode -t bios`
```
# dmidecode 3.6  
Getting SMBIOS data from sysfs.  
SMBIOS 3.4 present.  
  
Handle 0x0000, DMI type 0, 26 bytes  
BIOS Information  
       Vendor: LENOVO  
       Version: K1CN48WW.  
       Release Date: 07/10/2024  
       Address: 0xE0000  
       Runtime Size: 128 kB  
       ROM Size: 16 MB  
       Characteristics:  
[... 513 lines cut ...]
External object resolution file           SSDT12  
Input file SSDT12, Length 0x1D1C (7452) bytes  
ACPI: SSDT 0x0000000000000000 001D1C (v02 LENOVO CB-01    00000001      01000013)  
Pass 1 parse of [SSDT]  
Pass 2 parse of [SSDT]  
External object resolution file           SSDT11  
Input file SSDT11, Length 0xA1D6 (41430) bytes  
ACPI: SSDT 0x0000000000000000 00A1D6 (v02 LENOVO CB-01    00000001      01000013)  
Pass 1 parse of [SSDT]  
Pass 2 parse of [SSDT]  
External object resolution file           SSDT10  
Input file SSDT10, Length 0x9B6 (2486) bytes  
ACPI: SSDT 0x0000000000000000 0009B6 (v02 LENOVO CB-01    00000001 INTL 20200717)  
Pass 1 parse of [SSDT]  
Pass 2 parse of [SSDT]  
External object resolution file            SSDT1  
Input file SSDT1, Length 0x50D9 (20697) bytes  
ACPI: SSDT 0x0000000000000000 0050D9 (v02 LENOVO CB-01    00000001 INTL 20200717)  
Pass 1 parse of [SSDT]  
Pass 2 parse of [SSDT]  
Pass 1 parse of [DSDT]  
Firmware Error (ACPI): Failure creating named object [\_SB.PC00.RP13.PXSX._RST], AE_ALREADY_EXISTS (20230628/dswload-495)  
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230628/psobject-372)  
Could not parse ACPI tables, AE_ALREADY_EXISTS
```

bugparty, 2024-11-11:
Model and Debug Info

Model name: Y9000K 2021
CPU model: Intel 11800H
GPU model:  NVIDIA RTX 3080
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: yes
---
# Gathering WMI info in Windows
----
```
########################################
########################################
########################################
Name: LENOVO_UTILITY_EVENT
Class Name: LENOVO_UTILITY_EVENT
Class GUID: {8fc0de0c-b4e4-43fd-b0f3-8871711c1294}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_LIGHTING_EVENT
Class Name: LENOVO_LIGHTING_EVENT
Class GUID: {1e3391a1-2c89-464d-95d9-3028b72e7a33}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_GAMEZONE_SMART_FAN_SETTING_EVENT
Class Name: LENOVO_GAMEZONE_SMART_FAN_SETTING_EVENT
Class GUID: {D320289E-8FEA-41E1-86F9-611D83151B5F}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_GAMEZONE_KEYLOCK_STATUS_EVENT
Class Name: LENOVO_GAMEZONE_KEYLOCK_STATUS_EVENT
Class GUID: {10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_GAMEZONE_SMART_FAN_MODE_EVENT
Class Name: LENOVO_GAMEZONE_SMART_FAN_MODE_EVENT
Class GUID: {D320289E-8FEA-41E0-86F9-611D83151B5F}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_SR_EVENT
Class Name: LENOVO_SR_EVENT
Class GUID: {95D1DF76-D6C0-4E16-9193-7B2A849F3DF2}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_GAMEZONE_TEMP_EVENT
[... 1081 lines cut ...]
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: 3
Methods:
Name: MEM_Get_OC_Status
WmiMethodId: 1
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Get Memory OC Status
Implemented: True

Name: MEM_Set_OC_Status
WmiMethodId: 2
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Set Memory OC Status
Implemented: True

Name: MEM_Set_OC_Data
WmiMethodId: 3
Class Name: LENOVO_MEMORY_METHOD
Class GUID: {03607fce-0d83-4612-8a6e-4a4ef0415ea9}
Description: Set Memory OC Data
Implemented: True
```

joppino, 2025-05-27:
Model name: Lenovo Legion 7 PRO 16IAX10H
CPU model: Intel(R) Core(TM) Ultra 9 275HX
GPU model: Nvidia RTX 5080
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: yes

`
dmidecode -t system: 
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.6 present.
# SMBIOS implementations newer than version 3.5.0 are not
# fully supported by this version of dmidecode.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83F5
	Version: Legion Pro 7 16IAX10H
	Serial Number: XXXX
	UUID: XXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83F5_BU_idea_FM_Legion Pro 7 16IAX10H
	Family: Legion Pro 7 16IAX10H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
`

dmidecode -t bios:
`
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.6 present.
# SMBIOS implementations newer than version 3.5.0 are not
# fully supported by this version of dmidecode.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: Q7CN26WW
	Release Date: 03/04/2025
	ROM Size: 48 MB
[... 230 lines cut ...]
Max speed for fancurve: 10000
Current fan curve in hardware:
Fan curve current point id: 0
Fan curve points size: 0
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
Fan curve current point id: 0
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
`

cat /sys/class/leds/platform\:\:ylogo/brightness
0