johnfanv2/LenovoLegionLinux #55: [BUG] Module loaded and sensors found but no "/sys/kernel/debug/legion/fancurve"

VaclavMasin, 2023-05-01:
**Problem Descripition**
As per the title - kernel module seems to load properly:
```text
[ 1374.716146] legion_laptop starts loading
[ 1374.716199] legion PNP0C09:00: legion_laptop platform driver probing
[ 1374.716201] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82JY; DMI_BIOS_VERSION:GKCN59WW
[ 1374.716203] Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82JY; DMI_BIOS_VERSION:GKCN59WW
[ 1374.716204] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[ 1374.716206] legion PNP0C09:00: Using configuration for system: GKCN
[ 1374.716254] legion PNP0C09:00: Read embedded controller ID 0x8227
[ 1374.716255] legion PNP0C09:00: Creating debugfs inteface
[ 1374.716260] Creating sysfs inteface
[ 1374.716262] Creating hwmon interface
[ 1374.716309] Creating platform profile support
[ 1374.716312] Init WMI driver support
[ 1374.716330] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[ 1374.716356] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[ 1374.716366] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[ 1374.716374] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[ 1374.716381] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[ 1374.716389] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[ 1374.716396] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[ 1374.716420] legion PNP0C09:00: legion_laptop loaded for this device
```
 and sensors are also found:
```text
$ sensors
legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           1760 RPM
Fan 2:           1972 RPM
CPU Temperature:  +40.0°C  
GPU Temperature:  +37.0°C  
IC Temperature:   +40.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          17.12 V  

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +40.4°C  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  
```

but:
```text
# cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

**Model and Debug Info**

Model name: Legion 5
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA GeForce RTX 3070 Laptop GPU
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: not sure - just a small white LED where the power plug goes (if that's what's meant here)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode -t system
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JY
        Version: Legion 5 17ACH6H
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JY_BU_idea_FM_Legion 5 17ACH6H
        Family: Legion 5 17ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode -t bios
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN59WW
        Release Date: 11/21/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.59
        Firmware Revision: 1.59

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
# cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

johnfanv2, 2023-05-01:
@VaclavMasin What distribution are you using? Maybe you have disabled mounting the debugfs filesystem? What is the output of `sudo ls /sys/kernel/debug`?

johnfanv2, 2023-05-01:
Thanks a lot for the info! Maybe I add this to the README.

closed 2023-05-01 by johnfanv2