johnfanv2/LenovoLegionLinux #58: [BUG] Keyboard leds on 2021 model

lucasmc64, 2023-05-11:
**Problem Descripition**
Running `sudo dmesg` everything seems fine except for the keyboard LEDs apparently. I received the following lines from the command output:

```bash
[47680.799426] legion_laptop: Error ACPI call for reading keyboard brightness: expected a value between 1 and 2, but got 0
[47680.799427] legion_laptop: Error reading brighntess for light: 5
[47680.799428] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
```

But I didn't notice any apparent impact, I can still change the intensity of the keyboard's leds with FN+Q.
In all other tests listed in the README there were no errors.

**Model and Debug Info**

Distribution: Fedora 37
Model name: Lenovo Legion 5 15ACH6
CPU model: AMD Ryzen 7 5800H
GPU model:  NVIDIA RTX 3050
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: yes

Note: All information below was taken with the module not permanently installed.

Output of `sudo dmidecode -t system`.
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82QJ
	Version: Legion 5 15ACH6
	Serial Number: xxxxxxxx
	UUID: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82QJ_BU_idea_FM_Legion 5 15ACH6
	Family: Legion 5 15ACH6

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: HHCN29WW
	Release Date: 06/16/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.29
	Firmware Revision: 1.29

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 50
CPU temperature EC error: 0
CPU temperature EC: 50
CPU temperature ACPI error: 0
CPU temperature ACPI: 50
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 50
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1761
1 fanspeed EC error: 0
1 fanspeed EC: 1761
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1700
2 fanspeed error: 0
2 fanspeed: 1767
2 fanspeed EC error: 0
2 fanspeed EC: 1767
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1700
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 1
EC fan curve points size: 9
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 59	 0	 60	 0	 35
1700	 1700	 2	 2	 54	 59	 56	 60	 32	 41
1900	 1900	 2	 2	 54	 59	 56	 60	 35	 48
2100	 2100	 2	 2	 54	 62	 56	 60	 43	 127
2300	 2300	 2	 2	 58	 68	 56	 60	 127	 127
2600	 2600	 2	 2	 64	 72	 56	 60	 127	 127
2900	 2900	 2	 2	 68	 78	 56	 64	 127	 127
3500	 3500	 2	 2	 73	 85	 60	 70	 127	 127
3700	 3700	 2	 2	 80	 127	 64	 127	 127	 127
=====================
```

johnfanv2, 2023-05-12:
@lucasmc64 Thanks for the bug report! Could you pleas turn turn the backlight to different levels with Fn + Space (or is it Fn + Q on your model; Fn + Q changes power mode on my model) and provide the output of `sudo cat /sys/kernel/debug/legion/fancurve` on each level.

lucasmc64, 2023-05-13:
Oh sorry, it's really FN + Space 🤦‍♂️.

Note: I permanently installed the kernel module via DKMS, everything is apparently working fine.

Output with keyboard leds off:

```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 39
CPU temperature EC error: 0
CPU temperature EC: 39
CPU temperature ACPI error: 0
CPU temperature ACPI: 39
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 39
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1711
1 fanspeed EC error: 0
1 fanspeed EC: 1711
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1700
2 fanspeed error: 0
2 fanspeed: 1705
2 fanspeed EC error: 0
2 fanspeed EC: 1705
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1700
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
[... 177 lines cut ...]
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 1
EC fan curve points size: 9
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 127	 0	 60	 0	 35
1700	 1700	 2	 2	 127	 127	 56	 60	 32	 41
1900	 1900	 2	 2	 127	 127	 56	 60	 35	 48
2100	 2100	 2	 2	 127	 127	 56	 60	 43	 127
2300	 2300	 2	 2	 127	 127	 56	 60	 127	 127
2600	 2600	 2	 2	 127	 127	 56	 60	 127	 127
2900	 2900	 2	 2	 127	 127	 56	 64	 127	 127
3500	 3500	 2	 2	 127	 127	 60	 70	 127	 127
3700	 3700	 2	 2	 127	 127	 64	 127	 127	 127
=====================
```

Thank you for your help :)

johnfanv2, 2023-05-14:
From the output I see that keyboard backlight seems to work, at least reading its current state. The error message seems to be from the "IO-Port LED" at the back of the laptop. If they can be controlled you can turn it off with Fn + U and it also should work with this software but there is an error. However, this is not possible on all models.

lucasmc64, 2023-05-14:
Fn + U has no apparent effect. On the back of the laptop there is only the led indicating that the power cable is connected.
And the system seems to be working well even after more than 24 hours of operation.
I just wanted to make sure it wasn't a serious problem. Thanks a lot for your help and for making my legion a little more usable :)

closed 2023-06-11 by johnfanv2