johnfanv2/LenovoLegionLinux #59: Automatic recompilation on kernel update [Arch]

mercifulboss, 2023-05-13:
Arch updates the kernel quite frequently and this is becoming a hassle to constantly recompile (I am also worried it'll break the kernel or something). I am wondering if its possible to have it in the AUR (I looked around but couldn't find any AUR packages) so this recompiles automatically when the kernel updates? 

I understand that you eventually want to mainline this into the kernel, but in the meantime can this be a stopgap measure? Its less of an issue on other distros I imagine as they don't update the kernel as often.

mercifulboss, 2023-05-14:
Three things I'd like to mention in addition to the above:

1. Also, my backlight is broken. It doesn't turn on if I use the function keys (though all the function keys work) and it doesn't turn on through the software

2. Is it possible to have the GUI not tied to the folder (ie it is part of the system and I can open it from start menu/krunner)?

3. KDE does not offer the power performance options or at least I can't find them anywhere. Please help!

I am on a 81Y6 Lenovo Legion 5 15IMH05H

johnfanv2, 2023-05-14:
0. There is a Arch package that installs it with DKMS so it recompiles automatically. It is linked at the top of the README. Though it is not the newest version.
1. Which backlight? Screen or keyboard?
2.  There is a Arch package that installs the GUI It is linked at the top of the README. Though it is not the newest version.
3. You can also change power mode in the new GUI if do not find a way with KDE.

MrDuartePT, 2023-05-14:
@johnfanv2 besides the version listed in the aur is not the latest i change the aur PKGBUILD in the way is listed in the Arch Wiki and will push allways the latest version

mercifulboss, 2023-05-14:
> 0. There is a Arch package that installs it with DKMS so it recompiles automatically. It is linked at the top of the README. Though it is not the newest version.
> 
>     1. Which backlight? Screen or keyboard? Please also fill out the info from a bug report.
> 
>     2. There is a Arch package that installs the GUI It is linked at the top of the README. Though it is not the newest version.
> 
>     3. You can also change power mode in the new GUI if do not find a way with KDE.

Thank you for taking the time to reply! Hope I didn't come off as rude, wasn't my intention if that was so. I am appreciate of the work you did as I have been waiting for something like this for ages.

0. I wasn't aware that DKMS recompiles on every update. Thanks!
1. Keyboard backlight, screen backlight works fine. Keyboard backlight used to work, but it stopped working a while ago. While testing this program, I noticed that I get errors for keyboard backlight
2. It's this one right? `sudo python/legion_linux/legion_linux/legion_gui.py` The thing I was trying to say its a hassle to run it from the folder every time, would it be possible to somehow make it run from start menu? Or is that a limitation of the program? Maybe even make it launch as a tray application so we can quickly change battery modes (turn conservation mode on/off for example) or if running on the integrated chip, turn off fans entirely).
3. Yes, I found it. Thanks! Have finally been able to reach feature parity with windows. Thogh would be nice if it was a tray app. Would this be possible?

MrDuartePT, 2023-05-14:
@mercifulboss if you install lenovolegionlinux-dkms-git it also install lenovolegionlinux-git and that package will create a desktop file to open the gui app and also add the legion_cli command you not need to run from the folder :)

johnfanv2, 2023-05-14:
>     1. Keyboard backlight, screen backlight works fine. Keyboard backlight used to work, but it stopped working a while ago. While testing this program, I noticed that I get errors for keyboard backlight
Pleases create a new bug report (https://github.com/johnfanv2/LenovoLegionLinux/issues/new?assignees=&labels=&projects=&template=bug-report.md&title=%5BBUG%5D) and include the requested information so I can try to fix it again.
> 2. It's this one right? `sudo python/legion_linux/legion_linux/legion_gui.py` The thing I was trying to say its a hassle to run it from the folder every time, would it be possible to somehow make it [...]
You can try to install it such that it runs globally and from the start menu. It is a little bit different on each distribution. You could do that by trying the linked AUR package or trying to copy the files to the right locations in Arch if this does not work. The latter is often different in each distribution and sometimes a little tricky (see https://github.com/johnfanv2/LenovoLegionLinux/blob/main/deploy/python_install_installer_pkg.sh and https://github.com/johnfanv2/LenovoLegionLinux/blob/main/deploy/python_install_pip_pkg.sh). That's why there is no still no install script or package for each distribution.
>     3. Yes, I found it. Thanks! Have finally been able to reach feature parity with windows. Thogh would be nice if it was a tray app. Would this be possible?
I already worked on it a bit (the version is not uploaded yet). Which features (power mode, battery?, ...) should be configurable or displayed in the tray?

mercifulboss, 2023-05-14:
> @mercifulboss if you install lenovolegionlinux-dkms-git it also install lenovolegionlinux-git. That package will create a desktop file to open the gui app and also add the legion_cli command you not need to run from the folder :)
> 
> About the tray app i think KDE have the option to change the mode in the battery icon (but i don't remember i don't use KDE for quite a while)

Thank you for your reply! I do see the app in the start menu, but clicking it does nothing. It only works if I start the program with the command mentioned above

johnfanv2, 2023-05-14:
@mercifulboss Have you used the AUR package or another method? If you user the former, this might be problem of the AUR package @MrDuartePT.

Could you also file the bug report and tell me your wishes for the tray features (see above)?

mercifulboss, 2023-05-14:
> @mercifulboss if you install lenovolegionlinux-dkms-git it also install lenovolegionlinux-git. That package will create a desktop file to open the gui app and also add the legion_cli command you not [...]

Thank you for your reply! I do see the app in the start menu, but clicking it does nothing. It only works if I start the program with the command mentioned above

mercifulboss, 2023-05-14:
> @mercifulboss Have you used the AUR package or another method? If you user the former, this might be problem of the AUR package @MrDuartePT.
> 
> Could you also file the bug report and tell me your wishes for the tray features (see above)?

I filed a bug report here: 
https://github.com/johnfanv2/LenovoLegionLinux/issues/60

Apologies for the late reply. I was filling it out.

I'd like to see an option for "Power profiles (since KDE doesn't have a native option for it like GNOME - maybe the program can do a check if its KDE/GNOME and show power-profiles or not depending on DE? Not sure if that's too hard to do), Toggle for Battery Conservation mode, Rapid Charge and Settings (to change fan curves manually)"

Those are just my wishes, maybe you have another idea :)

Thank you for being so responsive :)

MrDuartePT, 2023-05-14:
Try ```pkexec legion_gui```
Record the process of both pls.

mercifulboss, 2023-05-14:
> Try `pkexec legion_gui` in the terminal Record the process of both pls.

Here is what I get:

```
Traceback (most recent call last):
  File "/usr/local/bin/legion_gui", line 33, in <module>
    sys.exit(load_entry_point('legion-linux==1.0.0', 'console_scripts', 'legion_gui')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/legion_gui", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1142, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'legion_gui'
```

MrDuartePT, 2023-05-14:
> Here is what I get:
>
> ```
> Traceback (most recent call last):
>   File "/usr/local/bin/legion_gui", line 33, in <module>
>     sys.exit(load_entry_point('legion-linux==1.0.0', 'console_scripts', [...]

And ```legion_cli``` works

mercifulboss, 2023-05-14:
> 

Here is what I get:

```
Traceback (most recent call last):
  File "/usr/local/bin/legion_cli", line 33, in <module>
    sys.exit(load_entry_point('legion-linux==1.0.0', 'console_scripts', 'legion_cli')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/legion_cli", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1142, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'legion_cli'

```

MrDuartePT, 2023-05-14:
Ok this a problem with the PKGBUILD i know how to solve it.

mercifulboss, 2023-05-14:
> Ok this a problem with the PKGBUILD i know how to solve it.

Let me know if there is anything else you need :)

MrDuartePT, 2023-05-14:
Try now

MrDuartePT, 2023-05-14:
By the way for the power profile you need to install powerdevil (i think plasma should work like gnome and after that should be available in the menu below) :)

https://pointieststick.files.wordpress.com/2021/07/screenshot_20210723_122713.png?w=1090

https://wiki.archlinux.org/title/KDE#Power_management

mercifulboss, 2023-05-14:
> Try now (update the package)

I updated from the links in the github repo (package doesn't auto-update using `yay -Syu`), it still doesn't work when I click the gui tool. I get same problem as last time - login prompt appears and then closes after I enter my password.

mercifulboss, 2023-05-14:
> 

Here is what I see on my "Battery and Brightness" indicator

[image]

I have powerdevil installed too:
[image]

Maybe I'm doing something wrong - I'm still a linux noob :)

MrDuartePT, 2023-05-15:
Ok i try the PKGBUILD in manjaro vm and is working, well try to dowload the PKGBUILD and install manually if yay dosent updated automatic

kleidiss, 2023-06-02:
> 
> Edit: I think I don't have `powerprofiles daemon` installed because I am using AutoCPUFreq 

auto-cpufreq disables powerprofile daemon once you install it so remove that if you want to change from KDE panel

MrDuartePT, 2023-07-08:
A hook was made in the aur package a 1 month ago this issue will be close

closed 2023-07-08 by MrDuartePT