johnfanv2/LenovoLegionLinux #60: [BUG] 81Y6 Lenovo Legion 5 15IMH05H Backlight stopped working

mercifulboss, 2023-05-14:
**Problem Descripition**
Keyboard backlight stopped working on 81Y6 Lenovo Legion 5 15IMH05H

Briefly turns on at system reboot before the Lenovo Legion logo appears then turns off and cannot be activated by buttons or software

**Model and Debug Info**

Distribution: Archlinux 6.3.1-arch2-1 kernel
Model name: 81Y6 Lenovo Legion 5 15IMH05H
CPU model: Intel i7-10750H (12) @ 5.000GHz
GPU model: NVIDIA GeForce RTX 2060 Mobile 
Keyboard backlight: Single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81Y6
        Version: Lenovo Legion 5 15IMH05H
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81Y6_BU_idea_FM_Legion 5 15IMH05H
        Family: Legion 5 15IMH05H

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0024, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0031, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: EFCN46WW
        Release Date: 12/04/2020
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 11 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.46
        Firmware Revision: 1.46

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 52
CPU temperature ACPI error: 0
CPU temperature ACPI: 57
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 46
GPU temperature EC error: 0
GPU temperature EC: 46
GPU temperature ACPI error: 0
GPU temperature ACPI: 46
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2015
1 fanspeed EC error: 0
1 fanspeed EC: 2015
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2000
1 fanspeed WMI error: 0
1 fanspeed WMI: 2000
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2000
2 fanspeed error: 0
2 fanspeed: 1921
2 fanspeed EC error: 0
2 fanspeed EC: 1921
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1900
2 fanspeed WMI error: 0
2 fanspeed WMI: 1900
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 1900
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 2
EC fan curve points size: 9
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       7       0       127     0       52      0       38
1700     1600    5       7       127     127     48      52      35      45
2000     1900    5       7       127     127     48      52      41      50
2100     2100    5       7       127     127     48      52      47      127
2300     2300    5       7       127     127     48      56      127     127
2600     2600    5       7       127     127     53      60      127     127
2900     2900    5       7       127     127     57      64      127     127
3500     3500    5       7       127     127     61      73      127     127
3800     3800    5       7       127     127     64      127     127     127
=====================

```

johnfanv2, 2023-05-15:
Can you switch the brightness state with Fn + Space?

If this works: Please uninstall any AUR package and DKMS. Then use the newest version of the code, recompile the kernel module and reload it (see README). Then please provide the output after loading. 
Then change the brighness and run `sudo cat /sys/kernel/debug/legion/fancurve`. Please provide the output after switching to each state (off, medium, bright).

mercifulboss, 2023-05-25:
> Can you switch the brightness state with Fn + Space?
>
> If this works: Please uninstall any AUR package and DKMS. Then use the newest version of the code, recompile the kernel module and reload it [...]

I just tried Fn+Space on newest kernel (that removed this patch) and keyboard brightness works but only with this key combo, not with the standard one.

Sorry for disappearing, I got a bit busy.

johnfanv2, 2023-05-26:
Thanks for the reply. I assume that I just have to set one setting/switch to make it work again. If it works with Fn+Space, please try this test:

If this works: Please uninstall any AUR package and DKMS. Then use the newest version of the code, recompile the kernel module and reload it (see README). Then please provide the output after loading. Then change the brighness and run sudo cat /sys/kernel/debug/legion/fancurve. Please provide the output after switching to each state (off, medium, bright).

What is the standard key combo? Is this something special of your distribution or desktop environment?

MrDuartePT, 2023-10-19:
@mercifulboss pls provide that output @johnfanv2 ask you if not i will need to close this issue as uncomplete

MrDuartePT, 2023-10-19:
ok by the way the aur now use the latest git independent of the version reported in the aur website if works from source you can revert to the aur

closed 2026-09-11 by github-actions[bot]