johnfanv2/LenovoLegionLinux #61: Run GUI as non-root

ulrikstrid, 2023-05-22:
**Is your feature request related to a problem? Please describe.**

Running a GUI application as root is frowned upon since it has a much larger attack surface than CLI applications.

**Describe the solution you'd like**

It would be ideal if the GUI shelled out to the CLI and ran just that part in a privileged mode.

**Describe alternatives you've considered**

The alternative is to run as root which works, but is dangerous.

More information here: https://github.com/NixOS/nixpkgs/pull/228506#issuecomment-1555425778

johnfanv2, 2023-05-22:
Good idea. This would also make it possible to create working a system tray menu. 

Do you have an idea how to implement it? Do you know other programs that have to deal with the same problem and already found a solution?

How to run the root program?
1. Start a background process running as root and a GUI process running as user. One might have to start the program with root which spans the GUI process with user privileges. Or the GUI program is started with user privileges and asks for a password to start the root program.
2. Start a root process each time a file with root access is written. That is almost any time a attribute of the laptop if changed. The user might be asked each time.
3. Other possibilities?

How to do communication between GUI process and root process?
1. DBUS between GUI process and root process. Security? Then every other program might talk to root process.
2. XMLRPC between GUI process and root process. Security? Then every other program might talk to root process.
3. Other possibilities?

MrDuartePT, 2023-09-14:
The gui now a day can be run as root and non root the package can alter the python file to not install the files needed to run as root. 
@ulrikstrid @johnfanv2 you think this issue can be close?

ulrikstrid, 2023-09-14:
@MrDuartePT is it documented somewhere how to package it without needing root? If so I can try to do that for NixOS

MrDuartePT, 2023-10-19:
@ulrikstrid you can run the gui without root buy using this command
```legion_gui --use_legion_cli_to_write```
And also just include this desktop file in the nix package:
https://github.com/johnfanv2/LenovoLegionLinux/blob/main/python/legion_linux/legion_linux/legion_gui_user.desktop

ulrikstrid, 2023-10-27:
For anyone on NixOS there is a open PR that adds the changes here: https://github.com/NixOS/nixpkgs/pull/263760

closed 2023-10-27 by ulrikstrid