johnfanv2/LenovoLegionLinux #62: [BUG] Is  Legion 7i Pro 2023 supported?

lithces, 2023-05-31:
**Problem Descripition**
I am running latest Arch Linux and I successfully installed the kernel dkms.

The modules `legion_laptop` are loaded, however in dmesg I found that legion PNP0C09:00: legion_laptop not loaded for this device.

Just wondering if we can use the tool for the latest Legion 7i Pro 2023, in China Y9000P 至尊版 2023

**Model and Debug Info**

Distribution: 6.3.4-2-MANJARO (64-bit) (including version, e.g. Ubuntu 24.04)
Model name: Legion Y9000P IRX8H
CPU model: Intel 13900HX (e.g. AMD Ryzen 7 5800H)
GPU model: NVIDIA RTX 4090 (e.g NVIDIA RTX 3700)
Keyboard backlight: single color with off/medium/bright (RGB, single color with off/on, single color with off/medium/bright)
Light in lid or logo: ??? (yes/no)
Light at IO-Ports at back: ??? (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82WQ
        Version: Legion Y9000P IRX8H
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Y9000P IRX8H
        Family: Legion Y9000P IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: KWCN36WW
        Release Date: 04/28/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.36
        Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```

Output of  `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

johnfanv2, 2023-05-31:
This seems is one of the newest model, so I am very hopeful that it will work. Please provide the following information so that I can configure the driver:

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. If you do not have Windows skip it. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```
Which features are the most useful for you?

ronyclau, 2023-06-03:
This is the exact same modal of my laptop, but mine is with NVIDIA RTX 4080. [Dumps here.](https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-1575163650)

johnfanv2, 2023-06-11:
@ronyclau @lithces  I updated the kernel module. 
* Please remove/uninstall any DKMS modules or other old installed modules.
* Then please get the new code, compile the kernel module and load it again. 
* Please report back, what works and what works not. It is very likely that it will need some adaptions and fixes.

lithces, 2023-06-12:
Thankyou @johnfanv2

I tried the latest Arch AUR and still the similar problem. The kernel module is loaded but something appears wrong in dmesg output
`[    3.441768] legion_laptop: legion_laptop starts loading
[    3.443547] legion PNP0C09:00: legion_laptop platform driver probing
[    3.443551] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82WQ; DMI_BIOS_VERSION:KWCN36WW
[    3.443552] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82WQ; DMI_BIOS_VERSION:KWCN36WW
[    3.443555] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[    3.443557] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[    3.443558] legion PNP0C09:00: legion_laptop not loaded for this device`

I was unable to install fwts-git in AUR now, however I attached the output of ACPI tables and the output from the windows system,

[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11716827/acpi_re.tar.gz)
[windows_dump.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11716828/windows_dump.tar.gz)

Note that when I was dumping the ACPI table on linux i got error messages as follows,

`Firmware Error (ACPI): Failure creating named object [\_SB.PC00.XHCI.RHUB.HS14.SADX], AE_ALREADY_EXISTS (20221020/dswload-495)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20221020/psobject-372)
Could not parse ACPI tables, AE_ALREADY_EXISTS
`

Similar errors appear when I am trying to detach the discrete GPU (for better battery life) using acpi_call as described in [Arch Wiki](https://wiki.archlinux.org/title/hybrid_graphics)

Your insights are highly appreciated.

johnfanv2, 2023-06-12:
@lithces Sorry, for my incomprehensibly message. 

Please uninstall any package, DKMS module, ARCH package etc. and compile and load the new version from github as described in the README on github (using make etc.)

ronyclau, 2023-06-12:
@johnfanv2 Thanks for the prompt fix. The module can be loaded now, however, tests revealed that both temperature and fan speed were not reported correctly. Please see the logs below:

<details>
<summary>Log output of <code>sudo reloadmodule</code></summary>

```
[Tue Jun 13 03:19:41 2023] legion_laptop: legion_laptop starts loading
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82WQ; DMI_BIOS_VERSION:KWCN36WW
[Tue Jun 13 03:19:41 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82WQ; DMI_BIOS_VERSION:KWCN36WW
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: Using configuration for system: KWCN
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: ACPI CFG: 2081289494
[Tue Jun 13 03:19:41 2023] legion_laptop: Creating RAM access to embedded controller
[Tue Jun 13 03:19:41 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x00000000ff72c65d
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: Read embedded controller ID 0x5507
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: Creating debugfs inteface
[Tue Jun 13 03:19:41 2023] legion_laptop: Creating sysfs inteface
[Tue Jun 13 03:19:41 2023] legion_laptop: Creating hwmon interface
[Tue Jun 13 03:19:41 2023] legion_laptop: Creating platform profile support
[Tue Jun 13 03:19:41 2023] legion_laptop: Init WMI driver support
[Tue Jun 13 03:19:41 2023] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[Tue Jun 13 03:19:41 2023] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[Tue Jun 13 03:19:41 2023] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[Tue Jun 13 03:19:41 2023] legion_laptop: Init keyboard backlight LED driver
[Tue Jun 13 03:19:41 2023] legion_laptop: Init Y-Logo LED driver
[Tue Jun 13 03:19:41 2023] legion_laptop: Init IO-Port LED driver
[Tue Jun 13 03:19:41 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[Tue Jun 13 03:19:41 2023] legion_laptop: Error reading brighntess for light: 5
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[Tue Jun 13 03:19:41 2023] legion PNP0C09:00: legion_laptop loaded for this device
```

</details>

<details>
<summary>Quick Test: Reading Current Fancurve from Hardware</summary>

```shell
> sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 80
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 58
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
GPU temperature error: 0
GPU temperature: 87
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 45
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 18039
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2000
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
2 fanspeed error: 0
2 fanspeed: 14083
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2100
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

</details>

<details>
<summary>Quick Test: Read Sensor Values from Hardware</summary>

None of the figures changed regardless of CPU/GPU load. The fan speeds were obviously wrong. Only relevant part is shown.

```shell
> sensors

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           18039 RPM
Fan 2:           14083 RPM
CPU Temperature:  +80.0°C
GPU Temperature:  +87.0°C
IC Temperature:   +55.0°C
```

</details>

johnfanv2, 2023-06-12:
@ronyclau This seems to be a new software on the embedded controller. I disabled everything using specifics of the embedded controller and just use generic WMI methods for now. 

Please uninstall old versions and compile and load the new version from github as described in the README on github (using make etc.), again.

Most stuff should work except setting the fancurve. You should be able to proceed testing except fan curve settings. To make the fan curve work, please change the power mode with Fn+Q and provide the output of the following commands in **each** mode. 

```bash
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve
```

lithces, 2023-06-12:
@johnfanv2 
I cloned the latest repo 
```
commit c2eb040de864da0a30f7e0b3b0f91ec2dcd43cf6 (HEAD -> main, origin/main, origin/HEAD)
Author: johnfan <johndoe@example.com>
Date:   Mon Jun 12 23:50:49 2023 +0200

    Use WMI methods for bios KWCN and disable fancurve access

```
and tried the 'make' method. Seems that the kernel module was loaded. I attached  some outputs as follows,

<details>
<summary>
`make reloadmodule`, note that there are TONS of similar 'ACPI Error's before unloading legion WMI.
</summary>

```

[二 6月 13 07:25:37 2023] ACPI Error: No handler for Region [RTCM] (0000000002c55d70) [SystemCMOS] (20230331/evregion-130)
[二 6月 13 07:25:37 2023] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
[二 6月 13 07:25:37 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:37 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:38 2023] legion_laptop: Unloading legion WMI
[二 6月 13 07:25:38 2023] legion_laptop: Unloading legion WMI done
[二 6月 13 07:25:38 2023] legion_laptop: Unloading legion platform profile
[二 6月 13 07:25:38 2023] legion_laptop: Unloading legion platform profile done
[二 6月 13 07:25:38 2023] legion_laptop: Set powermode
[二 6月 13 07:25:38 2023] legion_laptop: Unexpected power mode value ignored: 0
[二 6月 13 07:25:38 2023] ACPI Error: No handler for Region [RTCM] (0000000002c55d70) [SystemCMOS] (20230331/evregion-130)
[二 6月 13 07:25:38 2023] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
[二 6月 13 07:25:38 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:38 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:39 2023] ACPI Error: No handler for Region [RTCM] (0000000002c55d70) [SystemCMOS] (20230331/evregion-130)
[二 6月 13 07:25:39 2023] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
[二 6月 13 07:25:39 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:39 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:39 2023] legion_laptop: Set powermode
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion hwon
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion hwon done
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion sysfs
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion sysfs done
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion dubugfs
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion dubugfs done
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion ecram
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion ecram done
[二 6月 13 07:25:39 2023] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x000000002b4d3965
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion shared
[二 6月 13 07:25:39 2023] legion_laptop: Unloading legion shared done
[二 6月 13 07:25:39 2023] legion_laptop: Legion platform unloaded
[二 6月 13 07:25:40 2023] ACPI Error: No handler for Region [RTCM] (0000000002c55d70) [SystemCMOS] (20230331/evregion-130)
[二 6月 13 07:25:40 2023] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
[二 6月 13 07:25:40 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:40 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[二 6月 13 07:25:41 2023] legion_laptop: legion_laptop exit
[二 6月 13 07:25:41 2023] legion_laptop: legion_laptop starts loading
[二 6月 13 07:25:41 2023] legion PNP0C09:00: legion_laptop platform driver probing
[二 6月 13 07:25:41 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82WQ; DMI_BIOS_VERSION:KWCN36WW
[二 6月 13 07:25:41 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82WQ; DMI_BIOS_VERSION:KWCN36WW
[二 6月 13 07:25:41 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[二 6月 13 07:25:41 2023] legion PNP0C09:00: Using configuration for system: KWCN
[二 6月 13 07:25:41 2023] legion PNP0C09:00: ACPI CFG: 2081289494
[二 6月 13 07:25:41 2023] legion_laptop: Creating RAM access to embedded controller
[二 6月 13 07:25:41 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x0000000043e44218
[二 6月 13 07:25:41 2023] legion PNP0C09:00: Read embedded controller ID 0x5507
[二 6月 13 07:25:41 2023] legion PNP0C09:00: Creating debugfs inteface
[二 6月 13 07:25:41 2023] legion_laptop: Creating sysfs inteface
[二 6月 13 07:25:41 2023] legion_laptop: Creating hwmon interface
[二 6月 13 07:25:41 2023] legion_laptop: Creating platform profile support
[二 6月 13 07:25:41 2023] legion_laptop: Init WMI driver support
[... 137 lines cut ...]
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

</details>

potatoalienof13, 2023-06-13:
Setting the fancurve does not work, nor does reading it. It complains about `/sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon6/pwm1_auto_point1_pwm` being missing. 
Setting the power mode works, but not reading the power limits. If you want any logs or anything, please ask. I would really be happy to get LenovoLegionLinux working on my laptop.

potatoalienof13, 2023-06-13:
It is reading and setting the power mode correctly with software.  The LED changes correctly. Here are the outputs of the commands. 

Quiet mode:
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
```
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000600
```
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
```
00000000  00 00 00 02 00 00 00 00  00 00 10 00 00 00 1e 28  |...............(|
00000010  00 00 80 00 05 00 10 06  00 01 06 00 00 00 20 00  |.............. .|
00000020  02 00 00 0c 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 40 82  |..............@.|
00000040  11 01 02 00 50 01 00 00  00 00 00 00 bb 56 00 00  |....P........V..|
00000050  2c 01 e6 00 00 00 c0 81  00 07 00 09 00 00 00 08  |,...............|
00000060  00 80 02 03 00 00 00 08  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 04 00 00 00  c0 00 42 00 80 08 01 53  |..........B....S|
00000090  4d 50 00 32 30 32 32 00  4c 32 32 4d 34 50 43 31  |MP.2022.L22M4PC1|
000000a0  00 09 00 8e 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000b0  33 25 00 26 27 27 00 27  02 11 00 00 00 80 01 00  |3%.&''.'........|
000000c0  70 00 45 20 a7 09 bb 41  a0 3c 06 27 55 28 50 00  |p.E ...A.<.'U(P.|
000000d0  00 00 00 00 00 00 00 1c  77 11 00 00 00 00 00 00  |........w.......|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 0e 00  |................|
000000f0  01 00 00 40 98 56 88 45  00 00 00 00 00 00 00 00  |...@.V.E........|
00000100  80 00 00 00 00 00 06 06  77 77 77 77 77 77 66 77  |........wwwwwwfw|
00000110  00 00 ff ff 00 00 00 00  00 00 35 7f 2f 00 00 00  |..........5./...|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000130  00 00 03 01 00 00 00 00  92 00 00 00 00 00 00 00  |................|
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  10 0a 00 50 00 00 00 00  00 00 00 00 00 00 00 00  |...P............|
000001e0  77 46 03 37 6f 19 37 57  50 00 37 00 00 00 00 00  |wF.7o.7WP.7.....|
000001f0  01 02 03 04 05 06 07 08  08 08 02 01 08 00 00 00  |................|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000220  00 00 00 00 00 00 00 00  01 00 01 00 45 20 00 00  |............E ..|
00000230  30 00 00 00 00 00 00 31  31 00 00 00 00 00 00 01  |0......11.......|
00000240  00 00 00 00 00 00 00 1e  00 00 03 00 00 00 00 00  |................|
00000250  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000260  58 33 58 50 33 34 52 30  32 44 37 00 00 00 00 00  |X3XP34R02D7.....|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  a5 f3 64 03 00 00 00 00  00 00 00 00 00 00 00 00  |..d.............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002e0  00 00 00 00 00 00 00 64  00 00 00 01 01 00 00 00  |.......d........|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 25 27 00 00  |............%'..|
00000300
```
sudo cat /sys/kernel/debug/legion/fancurve
```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 3
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 55
CPU temperature ACPI error: 0
CPU temperature ACPI: 51
CPU temperature WMI error: 0
CPU temperature WMI: 0
[... 458 lines cut ...]
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 255
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

johnfanv2, 2023-06-13:
Great. Now that power mode works. Please get the modified code again, recompile, and test sensor values (see README). It should display correct temperature values (except for IC/chip) and correct fan speeds. Please put some load onto your CPU and GPU to check that everything is labelled correctly. Note that GPU temp might be 0 if the GPU is off or not used.

potatoalienof13, 2023-06-14:
I can confirm that sensors work, including fan speeds. Everything is labeled correctly.  
Here are the command outputs:
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
```
00000000  00 01 00 02 00 00 11 00  33 17 10 00 00 00 1f 2a  |........3......*|
00000010  00 00 80 00 05 00 10 06  00 01 06 00 00 00 00 00  |................|
00000020  01 00 00 0c 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 40 82  |..............@.|
00000040  11 01 02 00 50 01 00 00  00 00 00 00 bb 56 00 00  |....P........V..|
00000050  2c 01 e6 00 00 00 c0 81  00 07 00 09 00 00 00 08  |,...............|
00000060  00 80 02 03 00 00 00 08  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 04 00 00 00  c0 00 42 00 80 08 01 53  |..........B....S|
00000090  4d 50 00 32 30 32 32 00  4c 32 32 4d 34 50 43 31  |MP.2022.L22M4PC1|
000000a0  00 09 00 8e 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000b0  32 27 00 28 27 29 39 29  02 11 00 00 00 80 01 00  |2'.(')9)........|
000000c0  70 00 dd 1f a7 09 b5 41  a0 3c 06 27 55 28 4f 00  |p......A.<.'U(O.|
000000d0  00 00 00 00 00 00 00 1b  77 11 00 00 00 00 00 00  |........w.......|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 0e 00  |................|
000000f0  01 00 00 40 98 56 88 45  00 00 00 00 01 00 11 00  |...@.V.E........|
00000100  80 00 00 00 00 00 07 07  77 77 77 77 77 77 77 77  |........wwwwwwww|
00000110  00 00 ff ff 00 00 00 00  00 00 3f 45 40 00 00 00  |..........?E@...|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000130  00 00 32 71 00 00 00 00  92 00 00 00 00 00 00 00  |..2q............|
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  10 19 00 50 00 00 00 00  00 00 00 00 00 00 00 00  |...P............|
000001e0  77 46 03 37 6f 19 37 57  50 00 37 00 00 00 00 00  |wF.7o.7WP.7.....|
000001f0  01 02 03 04 05 06 07 08  08 08 02 01 08 00 00 00  |................|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000220  00 00 00 00 00 00 00 00  01 00 01 00 dd 1f 00 00  |................|
00000230  30 00 00 00 00 00 00 31  31 00 00 00 00 00 00 01  |0......11.......|
00000240  00 00 00 00 00 00 00 1e  00 00 03 00 00 00 00 00  |................|
00000250  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000260  58 33 58 50 33 34 52 30  32 44 37 00 00 00 00 00  |X3XP34R02D7.....|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  76 f3 64 03 00 00 00 00  00 00 00 00 00 00 00 00  |v.d.............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002e0  00 00 00 00 00 00 00 64  00 00 00 01 01 00 00 00  |.......d........|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 27 27 00 00  |............''..|
00000300  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000320  00 20 33 00 00 3a 20 33  00 00 3d 20 3f 00 ff ff  |. 3..: 3..= ?...|
*
00000340  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000380  ff 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000390  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000003e0  ff 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000003f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000400  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000480  40 eb bf 4c 1e 9b 11 b2  a8 5c 84 f0 82 a6 ef 3f  |@..L.....\.....?|
00000490  00 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000004a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000500  00 00 00 00 00 00 00 00  00 00 32 d2 00 00 00 00  |..........2.....|
00000510  80 02 00 00 01 0b 00 00  00 00 00 00 00 02 0a 01  |................|
00000520  80 02 01 00 01 09 00 00  00 00 00 00 00 03 0a 01  |................|
00000530  00 00 00 00 00 00 00 00  02 00 00 00 00 00 00 00  |................|
00000540  00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00  |................|
00000550  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000570  02 00 00 40 00 00 00 00  00 00 00 00 02 00 00 40  |...@...........@|
00000580  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... 25 lines cut ...]
00000110  00 00 ff ff 00 00 00 00  00 00 3f 45 40 00 00 00  |..........?E@...|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000130  00 00 61 71 00 00 00 00  92 00 00 00 00 00 00 00  |..aq............|
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  10 19 00 50 00 00 00 00  00 00 00 00 00 00 00 00  |...P............|
000001e0  77 46 03 37 6f 19 37 57  50 00 37 00 00 00 00 00  |wF.7o.7WP.7.....|
000001f0  01 02 03 04 05 06 07 08  08 08 02 01 08 00 00 00  |................|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... cut ...]
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  73 f3 64 03 00 00 00 00  00 00 00 00 00 00 00 00  |s.d.............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002e0  00 00 00 00 00 00 00 64  00 00 00 01 01 00 00 00  |.......d........|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 27 27 00 00  |............''..|
00000300
```

johnfanv2, 2023-06-14:
Thanks for the feedback. Now, that these basic features work. Let's test some of the miscellaneous features before we come to fan control.

Please get the new code for kernel module and GUI. Recompile kernel module and reload and perform the following tests with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software. Provide the log output from the GUI.

# Fn Lock
- Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
- Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn 

# Win Key
- Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
- Disable Win Key: Win Key has no function anymore

# Touchpad
- Enable Touchpad: Touchpad works
- Disable Touchpad: Touchpad does not work

# Battery Conservation
- Enabled: charging stops if above 60 % (or 80% depending on model)
- Disabled: charging works up to 100%
Note: Sometimes it takes a 10-20 seconds before it takes effect

# Rapid Charging
- Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
- Disabled: Charging is slower
Note: Sometimes it takes a 10-20 seconds before it takes effect .

# Charge Output Always On
- Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
- Disabled: USB outputs are off when laptop is off

# Display Overdrive
- Enabled: display somehow faster
- Disabled: display not faster
Just check if checkbox can be checked or not if you cannot verify that it has an effect.

In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:
```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```

ronyclau, 2023-06-15:
@johnfanv2 

# Test Checklist 

- [x] Fn Lock
- [x] Win Key
- [ ] Touchpad
  - Grayed out. Related log:
```
INFO:legion_linux.legion:Feature TouchpadFeature with path None
WARNING:legion_linux.legion:Feature TouchpadFeature exist not. exits: 0
```
- [x] Battery Conservation
- [x] Rapid Charging
  - Can activate the rapid charging, but it takes a long time (couple of minutes) after disabling this for the charge rate and voltage to drop 
- [x] Charge Output Always On
- [x] Display Overdrive

# Backlight

This model has 4 levels of backlight: off, low, medium, and bright.

Commands output at each level:

<details>
<summary>Off</summary>

```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 56
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 56
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 56
GPU temperature error: 0
GPU temperature: 44
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 44
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 44
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2100
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2100
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
[... 424 lines cut ...]

[ 1532.689992] No Arguments are initialized for method [SNTM]

[ 1532.689998] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[ 1532.690014] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[ 1533.024547] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[ 1533.687768] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[ 1533.687785] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[ 1533.688187] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[ 1533.688306] legion_laptop: Unexpected value in MINIFANCURVE register:0
[ 1533.688511] legion_laptop: No access method for fancurve:0
[ 1533.688512] legion_laptop: Size of object: 48
[ 1533.688611] legion_laptop: Unexpected ACPI result for 92549549-4bde-4f06-ac04-ce8bf898dbaa:5: expected type 3 but got 3; expected length 200 but got 88;
[ 1533.693779] ACPI Error: No handler for Region [RTCM] (00000000e2aadff2) [SystemCMOS] (20221020/evregion-130)
[ 1533.693791] ACPI Error: Region SystemCMOS (ID=5) has no handler (20221020/exfldio-261)

[ 1533.693801] No Local Variables are initialized for Method [SNTM]

[ 1533.693803] No Arguments are initialized for method [SNTM]

[ 1533.693806] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[ 1533.693815] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
```

</details>

johnfanv2, 2023-06-15:
Please get the new code for kernel module and GUI. Recompile kernel module and reload and perform the following tests with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software. Provide the log output from the GUI.

# Touchpad
Please try the updated version of the GUI and kernel module.

# Rapid Charging
So I guess, changing the setting has an effect. It is just delayed.

# Backlight
@ronyclau Is it correct that you have a RGB backlight? This might be controlled by USB or another method. You could try https://openrgb.org/ . Can you control it in Windows with Vantage **and** Legion Toolkit?

# Fan curve
Please give again the output of 
```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```

ronyclau, 2023-06-17:
@johnfanv2 Thanks for the update. Touchpad control (both read and write) is working now.

My model has per-key RGB. I can control it with Lenovo Vantage and Legion Toolkit in Windows, but OpenRGB doesn't detect my keyboard in Linux (didn't try in Windows though).

Below is the fan curve dump. I have included a bit more of `dmesg` output, as it includes a backtrack of an array OOB access:
<details>

```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 63
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 63
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 63
GPU temperature error: 0
GPU temperature: 51
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 51
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 51
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2200
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2200
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2200
2 fanspeed error: 0
2 fanspeed: 2400
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2400
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2400
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
[... 61 lines cut ...]
[14172.097853]  __x64_sys_read+0x19/0x30
[14172.097856]  do_syscall_64+0x58/0x90
[14172.097860]  ? handle_mm_fault+0x119/0x330
[14172.097865]  ? do_user_addr_fault+0x1e8/0x720
[14172.097871]  ? syscall_exit_to_user_mode+0x29/0x50
[14172.097874]  ? exit_to_user_mode_prepare+0x30/0xb0
[14172.097881]  ? irqentry_exit_to_user_mode+0x9/0x20
[14172.097884]  ? irqentry_exit+0x43/0x50
[14172.097886]  ? exc_page_fault+0x91/0x1b0
[14172.097891]  entry_SYSCALL_64_after_hwframe+0x72/0xdc
[14172.097897] RIP: 0033:0x7f133b514992
[14172.097904] Code: c0 e9 b2 fe ff ff 50 48 8d 3d fa b2 0c 00 e8 c5 1d 02 00 0f 1f 44 00 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 0f 05 <48> 3d 00 f0 ff ff 77 56 c3 0f 1f 44 00 00 48 83 ec 28 48 89 54 24
[14172.097907] RSP: 002b:00007ffd9b39f748 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[14172.097911] RAX: ffffffffffffffda RBX: 0000000000020000 RCX: 00007f133b514992
[14172.097913] RDX: 0000000000020000 RSI: 00007f133b718000 RDI: 0000000000000003
[14172.097914] RBP: 00007f133b718000 R08: 00007f133b717010 R09: 00007f133b717010
[14172.097916] R10: 0000000000000022 R11: 0000000000000246 R12: 0000000000022000
[14172.097918] R13: 0000000000000003 R14: 0000000000020000 R15: 0000000000020000
[14172.097921]  </TASK>
[14172.097922] ================================================================================
[14172.600947] ACPI Error: No handler for Region [RTCM] (00000000054d3996) [SystemCMOS] (20221020/evregion-130)
[14172.600970] ACPI Error: Region SystemCMOS (ID=5) has no handler (20221020/exfldio-261)
```

</details>

johnfanv2, 2023-06-17:
I will have to look into the backlight stuff. 
Sorry, for the out of bounds access of the array and the crash. I suggest restarting. I fixed it in the code just uploaded. Please try it and report the output from above.

ronyclau, 2023-06-17:
@johnfanv2 Rebuilt and reloaded the module. Looks like the OOB is fixed now (saw your fix, straight to the point :smile:)

<details>
<summary>New dump here</summary>

```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 61
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 61
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 61
GPU temperature error: 0
GPU temperature: 49
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 49
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 49
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2200
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2200
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2200
2 fanspeed error: 0
2 fanspeed: 2400
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2400
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2400
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI) (might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================

[22482.854234] No Arguments are initialized for method [SNTM]

[22482.854239] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[22482.854258] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[22483.866208] ACPI Error: No handler for Region [RTCM] (00000000054d3996) [SystemCMOS] (20221020/evregion-130)
[22483.866233] ACPI Error: Region SystemCMOS (ID=5) has no handler (20221020/exfldio-261)

[22483.866254] No Local Variables are initialized for Method [SNTM]

[22483.866258] No Arguments are initialized for method [SNTM]

[22483.866264] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[22483.866282] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[22484.866050] ACPI Error: No handler for Region [RTCM] (00000000054d3996) [SystemCMOS] (20221020/evregion-130)
[22484.866060] ACPI Error: Region SystemCMOS (ID=5) has no handler (20221020/exfldio-261)

[22484.866069] No Local Variables are initialized for Method [SNTM]

[22484.866071] No Arguments are initialized for method [SNTM]

[22484.866073] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[22484.866081] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20221020/psparse-529)
[22485.098213] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[22485.098265] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[22485.098743] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[22485.098860] legion_laptop: Unexpected value in MINIFANCURVE register:0
[22485.099107] legion_laptop: No access method for fancurve:0
[22485.099109] legion_laptop: Size of object: 48
[22485.099280] legion_laptop: ACPI result for 92549549-4bde-4f06-ac04-ce8bf898dbaa:5: ACPI buffer length: 88
```

</details>

lithces, 2023-06-18:
@johnfanv2 I pulled the latest code "commit 611f97b4d83ceb0c45aea415ac771a0173e89853"

Seems that there are still problem with FanCurve. See below for the output,

```    ~/Downloads/legion/LenovoLegionLinux    main    sudo cat /sys/kernel/debug/legion/fancurve                                                                                                                      ✔  2m 9s  
sudo dmesg | tail -n 30
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 44
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 44
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 44
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1700
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 1700
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
[... 39 lines cut ...]
[  751.177513] ideapad_acpi VPC2004:00: Unknown event: 12
[  751.561461] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  751.561495] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[  751.572308] ideapad_acpi VPC2004:00: Unknown event: 12
[  751.831263] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  751.831297] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[  751.841533] ideapad_acpi VPC2004:00: Unknown event: 12
[  752.381159] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  752.381188] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[  752.391392] ideapad_acpi VPC2004:00: Unknown event: 12
[  752.640835] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  752.640848] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[  752.651834] ideapad_acpi VPC2004:00: Unknown event: 12
[  752.914553] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  752.914587] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[  752.925223] ideapad_acpi VPC2004:00: Unknown event: 12
[  827.888453] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[  827.888496] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[  827.888904] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[  827.889019] legion_laptop: Unexpected value in MINIFANCURVE register:0
[  827.889250] legion_laptop: No access method for fancurve:0
[  827.889252] legion_laptop: Size of object: 48
[  827.889410] legion_laptop: ACPI result for 92549549-4bde-4f06-ac04-ce8bf898dbaa:5: ACPI buffer length: 88

```

camberkenpas, 2023-06-20:
@johnfanv2 I don't have my Legion 7i Pro Gen 8 yet, but it seems a common issue that people are seeing messages like these pretty regularly:
```
[  587.416970] No Local Variables are initialized for Method [SNTM] 

[  587.416971] No Arguments are initialized for method [SNTM] 

[  587.416972] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529) 
[  587.416975] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529) 
[  588.424882] ACPI Error: No handler for Region [RTCM] (0000000007a2f8f3) [SystemCMOS] (20230331/evregion-130) 
[  588.424889] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
```

You'll see a few on the initial boot, and it seems that's all... But after resuming from sleep, messages relating to SNTM and RTCM occur pretty regularly (once per second it sounds like). Is this also something you can help with?

Let me know if you need a new issue. I've posted here since you can see some of these messages showing up under this issue already.

lithces, 2023-06-21:
@johnfanv2 Seems the fan curve for RTX4080 is working now, but for RTX4090 there is still some work to do.
Do you need some information from me (with RTX4090)?

johnfanv2, 2023-06-22:
It looks like reading of the fan curve in custom mode works. It displays 100, 200, 300, ... in the output of @lithces . In the output of @ronyclau it shows 0, 0, 0, ... . I am not sure why it is 0. Maybe this is the default value or was set with Windows tools.

I added the feature to set the fan curve. Please use the GUI and enter custom mode. Then change the fan curve (currently only the first row for fan 1), write it to hardware, and read it back. Please report back if it works and provide the log output from the GUI and dmesg. Note that this is a new model and I am not sure what the unit for fan speed is. I guess currently, 1000 = 100%.

camberkenpas, 2023-06-22:
@johnfanv2 Unfortunately, I won't have my Legion 7i Pro Gen 8 until July at some point (and perhaps not until mid July or later as the shipping date has already been pushed back a bit). I'm just being proactive at trying to resolve any issues. :smile:  

I've asked other Gen 8 owners to see if any of them can open the issue. Failing that, I'll open one myself once I have my Gen 8 in the next month or so.

I suspect it's a problem in the ACPI firmware itself, but I'm unsure how well that could be worked around. 

Thank you so much for your time!

lithces, 2023-06-23:
@johnfanv2 
I (rtx4090) tried to set 250 to the first point (default to 100), and second point (default to 200), both attempts lead to immediate hang of the system (Screen stops updating, no response from mouse and keyboard)

Is there any way we could see the logs?

johnfanv2, 2023-06-23:
Sorry for the crash. It is a little hard to develop for the newer models without having one. I heavily rely on the ACPI disassembly and the provided memory dumps.

Please try the new version.

lithces, 2023-07-01:
@johnfanv2 Seems that we are able to adjust the rpm for points ID other the first. I changed the value for the second, third point and Apply to HW, then I read the curve, I got this
```
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100      0       0       0       0       0       0       0       0       0
300      0       0       0       0       0       0       0       0       0
400      0       0       0       0       0       0       0       0       0
400      0       0       0       0       0       0       0       0       0
500      0       0       0       0       0       0       0       0       0
600      0       0       0       0       0       0       0       0       0
700      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100      0       0       0       0       0       0       0       0       0
300      0       0       0       0       0       0       0       0       0
400      0       0       0       0       0       0       0       0       0
400      0       0       0       0       0       0       0       0       0
500      0       0       0       0       0       0       0       0       0
600      0       0       0       0       0       0       0       0       0
700      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
```

Any way to apply the pressure so that it could exactly reach the 2rd or 3rd level?

lithces, 2023-07-01:
@johnfanv2 I have another question regarding to the GPU power settings. Currently is limited to 80W from nvidia-smi, on windows it is set to 175W..

I do find the following errors 
'''
2023-07-01 08:19:04,023 - ERROR: Feature GPUCTGPPowerLimit reading error [Errno 22] Invalid argument
'''

Any insight that we could increase the GPU power?

MrDuartePT, 2023-07-31:
> @johnfanv2 I have another question regarding to the GPU power settings. Currently is limited to 80W from nvidia-smi, on windows it is set to 175W..
>
> I do find the following errors ''' 2023-07-01 [...]

I know this a bit old but the gpu power is not controller by the LLL module but by the nvidia driver, for getting the boost you need to enable nvidia-powerd.service, nvidia powerd is a bit buggy on older driver i recommend 535 onwards.
For exemple i can get the full 130w in my RTX 3070 with powerd enable

MrDuartePT, 2023-10-19:
By the way for people using the aur
@lithces @ronyclau @potatoalienof13 @camberkenpas 
The aur now use the latest git every time independent of version reported in the aur website.
You can use without problems I gonna close this as complete any problems with support for your laptop pls reopen.

closed 2023-10-19 by MrDuartePT