johnfanv2/LenovoLegionLinux #63: [BUG] Nobara 38 module keeps unlaoding after restart

kleidiss, 2023-06-05:
**Problem Descripition**
Usually i do not have problems installing this module but on nobara its a diffrent sotry
The module doesn't load after install , i have to type "sudo make reloadmodule" to properly load it and it goes away after
 a restart 
 This does not happen on fedora either
 
 Sorry if this a very obvious fix , i am a novice in linux 

**Model and Debug Info**

Lenovo Ideapad Gaming 3 15aCH6

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
/sys/kernel/debug/legion/fancurve: No such file or directory

```

kleidiss, 2023-06-05:
And no the included patch from GE doesn't work either

kleidiss, 2023-06-06:
Oh i thought he converted it into a patch and kept on updating it 
Didn't know he was using the release which is outdated

Guess it makes sens now , yeah it goes away after a restart
Anyway to force it permanently

johnfanv2, 2023-06-06:
The patch is in each release. Support for your model will be in the next release/patch. I will have to test all the new features for the the new models for the next release.

kleidiss, 2023-06-06:
Alright thanks

Before i close the issue i have a question 

Would it be possible to integrate amd_pstate_epp control to the power platofrm driver

Since amd_pstate_epp is getting enabled by default on future kernel versions it will conflict with the current platofrm profile driver in power profile daemon  (amd_pstate_epp has its own driver for power profile daemon)

johnfanv2, 2023-06-06:
Could you give link to more details:
I just found that amd_pstate_epp will come with a scaling_driver, which is different from the platform_profile (driver). Do you think that amd_pstate_epp will create its own platform_profile driver that is in conflict with the platform_profile driver by the legion module?
Note that if you switch power mode with Fn + Q , without any driver or the legion module, will change some parameters of the CPU.

kleidiss, 2023-06-06:
I am not sure to be honest 
After loading the legion module it seems that p-state goes back to balended mode

And it seems like power-profile-daemon will add support for multiple drivers soon
https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/merge_requests/123

closed 2023-06-08 by kleidiss

GloriousEggroll, 2023-07-28:
fwiw if you just let me know on discord or via https://gitlab.com/GloriousEggroll/nobara-images/-/issues I can get it updated. I came across this issue by chance when checking for updates on my own. Without someone notifying me I don't know that it needs an update.