johnfanv2/LenovoLegionLinux #66: Legion 5 17ARH05H

MGCreator, 2023-06-10:
Hello, I am about to migrate from Windows 10 to Fedora 38 or POP_OS and I want to be sure to have a support for some of the most used features from Lenovo Vantage. I saw that there is no support for Legion 5 17ARH05H (Current Bios: FSCN26WW).
Thank you in advance!

Features (if not added):
1. Conservation Mode
2. Battery Details window
3. Rapid Charge

**Model and Debug Info**

Distribution: Fedora 38, POP_OS
Model name: Legion 5 17ARH05H, Current Bios: FSCN26WW
CPU model: AMD Ryzen 7 4800H
GPU model: NVIDIA RTX 2600
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

johnfanv2, 2023-06-11:
It will probably work since it is confirmed on the a model with the bios FSCN14WW, which is a successor of FSCN26WW. Conservation mode and rapid charge will most likely work since these are "simple and old features".

Battery details and statistics are supported by other tools in the distribution, e.g. [https://vitux.com/view-detailed-laptop-battery-report-on-ubuntu/](https://vitux.com/view-detailed-laptop-battery-report-on-ubuntu/) in Ubuntu/POP_OS.

MrDuartePT, 2023-10-19:
@MGCreator Please test the kernel driver and report back
I will close the issue since for what i know gen5 work well from some feadback i receive from Discord Legion server
For package instalation on Fedora and Ubuntu you can use this
https://github.com/johnfanv2/LenovoLegionLinux/tree/main/package_repo

closed 2023-10-19 by MrDuartePT