johnfanv2/LenovoLegionLinux #67: Support fn+r refresh rate toggle

st0nie, 2023-06-14:
now I use a script to toggle refresh rate
```sh
#!/bin/bash

HI_CMD="kscreen-doctor output.eDP-1.mode.0"
LOW_CMD="kscreen-doctor output.eDP-1.mode.1"
if [ ! -f ~/.cache/refreshed ]; then
        echo "HI" >~/.cache/refreshed
fi

STATUS="$(cat ~/.cache/refreshed)"

if [[ $1 == "toggle" ]]; then
        if [[ $STATUS = "HI" ]]; then
                $LOW_CMD
                echo "LOW" >~/.cache/refreshed
        else
                $HI_CMD
                echo "HI" >~/.cache/refreshed
        fi
elif [[ $1 == "init" ]]; then
        if [[ $STATUS = "LOW" ]]; then
                $LOW_CMD
                echo "LOW" >~/.cache/refreshed
        else
                $HI_CMD
                echo "HI" >~/.cache/refreshed
        fi
elif [[ $1 == "low" ]]; then
        $LOW_CMD
        echo "LOW" >~/.cache/refreshed
elif [[ $1 == "high" ]]; then
        $HI_CMD
        echo "HI" >~/.cache/refreshed
else
        echo "$STATUS"
fi
```

and acpid event file
```conf
event=1E3391A1-2C89- 000000e8 00000000
action=sudo -u ston HOME=/home/ston/ DIPLAY=":1" XDG_RUNTIME_DIR=/run/user/1000 QT_QPA_PLATFORM=wayland /home/ston/.local/bin/refreshed toggle
```

But there is a problem with this. I use any fn+letter key combination to trigger this switch. Is there a way to use fn+q?

MrDuartePT, 2023-06-17:
I also have the same issue I also create a script for more DE
https://github.com/MrDuartePT/hyprland-dotfiles/blob/master/refresh-rate-fn-r.sh

MrDuartePT, 2023-06-17:
dmesg output when pressing the key:
<@1059490124917903390> well when i press the special key that dosent work your drive output this 
```[Sat Jun 17 00:33:24 2023] legion_laptop: Event: legion type: 4;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:25 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:26 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:27 2023] legion_laptop: Event: legion type: 4;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:27 2023] legion_laptop: Event: legion type: 4;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:41 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:41 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:49 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:50 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:51 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:33:51 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:02 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:05 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:05 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:06 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:06 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:07 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:15 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:15 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:35:16 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:36:23 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:36:29 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)
[Sat Jun 17 00:36:31 2023] legion_laptop: Event: legion type: 1;  acpi type: 1 (1=integer)```
acpi_listen:
```8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000
 8FC0DE0C-B4E4- 000000d0 00000000
 1E3391A1-2C89- 000000e8 00000000```
 
 Also evtest dosent detect keypress:
 ```No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Lid Switch
/dev/input/event10:	HD-Audio Generic Mic
/dev/input/event11:	Razer Razer Basilisk
/dev/input/event12:	Razer Razer Basilisk
/dev/input/event14:	Ideapad extra buttons
/dev/input/event15:	PC Speaker
/dev/input/event16:	HDA NVidia HDMI/DP,pcm=3
/dev/input/event17:	HDA NVidia HDMI/DP,pcm=7
/dev/input/event18:	HDA NVidia HDMI/DP,pcm=8
/dev/input/event19:	HDA NVidia HDMI/DP,pcm=9
/dev/input/event2:	Power Button
/dev/input/event20:	HD-Audio Generic Headphone
/dev/input/event21:	Razer Razer Basilisk
/dev/input/event3:	AT Raw Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	Video Bus
/dev/input/event6:	ITE Tech. Inc. ITE Device(8910) Keyboard
/dev/input/event7:	MSFT0001:00 04F3:31DD Mouse
/dev/input/event8:	ITE Tech. Inc. ITE Device(8910) Wireless Radio Control
/dev/input/event9:	MSFT0001:00 04F3:31DD Touchpad
Select the device event number [0-21]: 14
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Ideapad extra buttons"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 138 (KEY_HELP)
    Event code 148 (KEY_PROG1)
    Event code 149 (KEY_PROG2)
    Event code 186 (KEY_F16)
    Event code 190 (KEY_F20)
    Event code 192 (KEY_F22)
    Event code 193 (KEY_F23)
    Event code 202 (KEY_PROG3)
    Event code 203 (KEY_PROG4)
    Event code 212 (KEY_CAMERA)
    Event code 227 (KEY_SWITCHVIDEOMODE)
    Event code 240 (KEY_UNKNOWN)
    Event code 247 (KEY_RFKILL)
    Event code 364 (KEY_FAVORITES)
    Event code 431 (KEY_DISPLAYTOGGLE)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)```

MrDuartePT, 2023-06-21:
@st0nie you can discuss your ideias here.

darthf1, 2024-05-16:
Am I correct that this will be resolved with the latest ubuntu kernel? Havent tested myself yet:

https://www.omgubuntu.co.uk/2024/05/linux-kernel-6-9-features

> There’s also support for the Lenovo IdeaPad ‘refresh rate’ key (which you’ll be shocked to hear us used to change display refresh rate); fan speed control for Microsoft Surface laptops, and the Lenovo ThinkPad ACPI driver now works with more Fn keys.

MrDuartePT, 2024-05-16:
> Am I correct that this will be resolved with the linux kernel 6.9? It mentions IdeaPad specifically, but?
> 
> https://www.omgubuntu.co.uk/2024/05/linux-kernel-6-9-features
> 
> > There’s also support for the Lenovo IdeaPad ‘refresh rate’ key (which you’ll be shocked to hear us used to change display refresh rate); fan speed control for Microsoft Surface laptops, and the Lenovo ThinkPad ACPI driver now works with more Fn keys.

Well I have kernel 6.9 install, I could test it. But if dosen't work I also planning on creating a patch to add fn+star (legion star symbol) and fn+PrtSC buttons

darthf1, 2024-05-17:
> @st0nie @darthf1 dosent work on my model, the key is not recognised:
> ```
> Input driver version is 1.0.1
> Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
> Input device name: "Ideapad extra buttons"
> Supported events:
>   Event type 0 (EV_SYN)
>   Event type 1 (EV_KEY)
>     Event code 1 (KEY_ESC)
>     Event code 138 (KEY_HELP)
>     Event code 148 (KEY_PROG1)
>     Event code 149 (KEY_PROG2)
>     Event code 186 (KEY_F16)
>     Event code 190 (KEY_F20)
>     Event code 192 (KEY_F22)
>     Event code 193 (KEY_F23)
>     Event code 202 (KEY_PROG3)
>     Event code 203 (KEY_PROG4)
>     Event code 212 (KEY_CAMERA)
>     Event code 227 (KEY_SWITCHVIDEOMODE)
>     Event code 240 (KEY_UNKNOWN)
>     Event code 247 (KEY_RFKILL)
>     Event code 364 (KEY_FAVORITES)
>     Event code 445 (?)
>     Event code 446 (?)
>     Event code 562 (?)
>     Event code 634 (?)
>   Event type 4 (EV_MSC)
>     Event code 4 (MSC_SCAN)
>   ```
> 

Too bad, thanks for checking!

tingao, 2026-11-06:
Update: I think this can work without acpid, and without the other Fn keys
setting it off.

On a Legion 7 16IRX9 (83FD) the Fn+R event is identifiable. The gamezone WMI
device delivers scancode 0x10, and the in-tree ideapad driver already maps
that to KEY_REFRESH_RATE_TOGGLE since 6.9. The legion_laptop driver gets the
event but drops the payload, so I opened #652 to report it as a real key.

In my test evtest shows KEY_REFRESH_RATE_TOGGLE for Fn+R, and Fn+Q arrives on
a different device and value, so nothing else fires the toggle.