johnfanv2/LenovoLegionLinux #68: Lenovo Legion 5i/Y7000 2019 PG0 support?

chinese-wzq, 2023-06-16:
**Model and Debug Info**

Model name: `Legion 5i(not sure)/Y7000 2019 PG0`
CPU model: `12 × Intel® Core™ i7-9750H CPU @ 2.60GHz`
GPU model: `NVIDIA GeForce GTX 1650/PCIe/SSE2`
Keyboard backlight: `single color with off/medium/bright`
Light in lid or logo: yes
Light at IO-Ports at back: no 

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: INVALID
        Version: INVALID
        Serial Number: INVALID
        UUID: --deleted--
        Wake-up Type: Power Switch
        SKU Number: LENOVO_BI_IDEAPADBH_BU_idea_FM_IDEAPAD
        Family: IDEAPAD

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0025, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0032, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

```

Output of `sudo dmidecode -t bios`:

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: LENOVO
        Version: BHCN45WW
        Release Date: 05/24/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 10 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.45
        Firmware Revision: 1.45

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`(by replacing 0x8227 by 0x8226,I force load the module):

```
EC Chip ID: 8226
EC Chip Version: 1a2
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228773126
temperature access method: 1
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 50
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 46
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 1700
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 1700
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1400
2 fanspeed WMI error: 0
2 fanspeed WMI: 1400
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 1400
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 1400
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI) (might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

The following information would be useful.
[DSDT.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/11776711/DSDT.zip)

johnfanv2, 2023-06-17:
@chinese-wzq 
I added your model/BIOS to the kernel module. Please get the new version, recompile, and reload the kernel module and test it. 

Then please test the following features.

# Temperatures and Fan Speeds
Please read the fan speeds and temperatures (see README) and confirm that they are sensible and change accordingly, e.g. when you put load onto the CPU the CPU temp and the fan speeds should increase, but GPU temp should stay roughly the same.

# Powermode

We will start by confirming that power mode change work **and to get me more debug info**. Please change the power mode with Fn+Q and provide the output of the following commands in **each** mode. 

```bash
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve
```

Then please also check if the power mode is read correctly in the OS (see README). Then please check if you can set the power mode with the GUI (see README) and the LED changes accordingly.

Tests the following features with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software. 

Please provide the log output from the GUI from the log tab for all the tests.

# Fn Lock
- Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
- Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn 

# Win Key
- Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
- Disable Win Key: Win Key has no function anymore

# Touchpad
- Enable Touchpad: Touchpad works
- Disable Touchpad: Touchpad does not work

# Battery Conservation
- Enabled: charging stops if above 60 % (or 80% depending on model)
- Disabled: charging works up to 100%
Note: Sometimes it takes a 10-20 seconds before it takes effect

# Rapid Charging
- Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
- Disabled: Charging is slower
Note: Sometimes it takes a 10-20 seconds before it takes effect .

# Charge Output Always On
- Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
- Disabled: USB outputs are off when laptop is off

# Display Overdrive
- Enabled: display somehow faster
- Disabled: display not faster
Just check if checkbox can be checked or not if you cannot verify that it has an effect.

# Keyboard Backlight

In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:

sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30

chinese-wzq, 2023-06-17:
# Temperatures and Fan Speeds [x]
It works.

# Powermode [x]
<details><summary>Quiet Mode output</summary>
<p>

```
 >>>>>>>>>> sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
00000000  00 00 00 00 00 00 0e 00  00 00 10 00 00 2f 21 00  |............./!.|
00000010  00 00 80 00 01 00 4e 04  02 01 0a 30 00 12 00 00  |......N....0....|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000040  00 00 00 00 00 00 00 00  00 00 00 00 3e 4f 00 00  |............>O..|
00000050  00 00 00 2d 00 00 2f 00  00 00 00 00 00 00 00 00  |...-../.........|
00000060  00 80 02 03 00 00 00 08  00 00 00 00 08 00 20 01  |.............. .|
00000070  00 02 20 01 00 00 00 00  00 00 00 00 00 00 00 00  |.. .............|
00000080  00 00 00 00 01 00 00 00  c0 00 43 00 00 00 00 4c  |..........C....L|
00000090  47 43 00 31 31 00 00 00  4c 31 37 4c 33 50 47 31  |GC.11...L17L3PG1|
000000a0  00 01 00 86 00 00 00 61  64 00 00 00 64 25 00 00  |.......ad...d%..|
000000b0  31 2c 2e 2b 2d 00 36 00  02 11 00 00 00 80 01 00  |1,.+-.6.........|
000000c0  30 00 1e 0c ee 00 5a 2d  4c 2c 82 14 88 14 3b 00  |0.....Z-L,....;.|
000000d0  00 00 00 00 00 00 00 22  26 08 00 00 00 00 00 00  |......."&.......|
000000e0  00 00 19 00 00 00 00 00  00 00 00 00 00 00 42 01  |..............B.|
000000f0  01 00 00 40 f7 4e 64 32  00 00 00 00 00 00 0c 00  |...@.Nd2........|
00000100  82 00 00 00 00 00 03 03  25 25 13 6b 44 6b 12 12  |........%%.kDk..|
00000110  5e 00 ff 03 ff 03 00 00  00 00 00 00 00 00 00 00  |^...............|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000160  00 00 00 00 00 00 00 38  50 3c 00 00 00 00 00 00  |.......8P<......|
00000170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000180  00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
00000190  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001b0  af 05 e8 04 0e 0c 00 00  2f 2d 00 00 00 00 00 00  |......../-......|
000001c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001f0  00 00 00 00 1e 0c 00 00  00 00 00 00 0e 0c 61 61  |..............aa|
00000200  0e 0c 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 01  |................|
00000240  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000250  36 36 36 00 00 00 00 00  00 00 00 00 00 00 00 00  |666.............|
00000260  00 1e 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  15 f3 64 06 00 00 00 00  00 00 00 00 00 00 00 00  |..d.............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002e0  00 00 00 00 00 00 00 64  00 00 00 00 00 00 00 00  |.......d........|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000300  ee 1e ee 1e ee 1e 0a 00  ee 1e ee 1e ee 1e ee 1e  |................|
00000310  ee 1e ee 1e ee 1e ee 1e  ee 1e ee 1e ee 1e ee 1e  |................|
*
00000350  0a 00 ee 1e ee 1e ee 1e  ee 1e ee 1e ee 1e ee 1e  |................|
00000360  ee 1e ee 1e ee 1e 0a ee  1e ee 1e ee 1e ee 1e 0a  |................|
00000370  00 0a ee 1e ee 1e ee 1e  ee 1e ee 1e ee 1e ee 1e  |................|
00000380  ee 1e ee 1e ee 1e ee 1e  ee 1e ee 1e ee 1e ee 1e  |................|
*
000003b0  0a 0a 00 05 00 05 00 03  00 04 00 03 00 06 06 00  |................|
000003c0  07 07 08 00 08 00 04 00  04 00 25 26 0a 00 00 00  |..........%&....|
000003d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000003f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 1e ce  |................|
00000400  00 ac 1d ac 1d ac 1e ac  1d ac 1e ac 1d ac 10 00  |................|
00000410  ac 1a 20 ac 15 ac 17 ac  10 00 ac 1a 20 ac 15 ac  |.. ......... ...|
00000420  17 ac 10 10 ac 1a 00 ac  12 ac 10 10 ac 1a 00 ac  |................|
00000430  12 ac 10 10 ac 1a 00 ac  12 ac 10 80 ac 1a 00 ac  |................|
[... 1934 lines cut ...]
2023-06-18 00:47:00,831 - INFO: Feature StrFileFeature reading: quiet balanced performance
2023-06-18 00:47:00,832 - INFO: Feature PlatformProfileFeature writing: performance
2023-06-18 00:47:01,049 - INFO: Feature StrFileFeature reading
2023-06-18 00:47:01,050 - INFO: Feature StrFileFeature reading: quiet balanced performance
2023-06-18 00:47:01,051 - INFO: Feature PlatformProfileFeature reading
2023-06-18 00:47:01,053 - INFO: Feature PlatformProfileFeature reading: performance
2023-06-18 00:47:08,224 - INFO: Click on UI 'Quiet Mode' element for PlatformProfileFeature.
2023-06-18 00:47:08,226 - INFO: Feature StrFileFeature reading
2023-06-18 00:47:08,227 - INFO: Feature StrFileFeature reading: quiet balanced performance
2023-06-18 00:47:08,228 - INFO: Feature PlatformProfileFeature writing: quiet
2023-06-18 00:47:08,442 - INFO: Feature StrFileFeature reading
2023-06-18 00:47:08,443 - INFO: Feature StrFileFeature reading: quiet balanced performance
2023-06-18 00:47:08,444 - INFO: Feature PlatformProfileFeature reading
2023-06-18 00:47:08,446 - INFO: Feature PlatformProfileFeature reading: quiet
2023-06-18 00:47:22,207 - INFO: Click on UI 'Y-Logo/Lid LED light' element for YLogoLight.
```

</p>
</details>

# Others
After I closed the GUI, Python on the terminal was still running, and Ctrl+C couldn't stop either.It can only be killed through htop
I use translation software.

# Very good project, love from China💗

chinese-wzq, 2023-07-18:
Can you close this issue if the support is completed?

johnfanv2, 2023-07-18:
Sorry, for the late response. I can help you later to test, enable, and fix more features but currently I have lots of other stuff to do :sob:

chinese-wzq, 2023-08-25:
I notice that I can't set fan curve at all.
```
EC Chip ID: 8226
EC Chip Version: 1a2
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228773126
temperature access method: 2
CPU temperature error: 0
CPU temperature: 94
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 94
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 49
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 49
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 2100
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2100
1 fanspeed WMI error: 0
1 fanspeed WMI: 2100
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2100
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 2100
2 fanspeed error: 0
2 fanspeed: 1800
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1800
2 fanspeed WMI error: 0
2 fanspeed WMI: 1800
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 1800
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 1800
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 0
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
As you see,  the value of "EC fan curve points size" is 0.

Interestingly,`sensors` provides fan speed:
```
legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           2100 RPM
Fan 2:           1800 RPM
CPU Temperature:  +91.0°C  
GPU Temperature:  +49.0°C  
IC Temperature:    +0.0°C
```

And I wonder whether the GPU power draw depends on power mode?On my computer,the GPU max power draw is always 50W.But (Legion WMI project)[https://github.com/adnanpri/legion-wmi] tell me that GPU max power draw is different in different power mode.

MrDuartePT, 2023-10-19:
> I notice that I can't set fan curve at all.
>
> ```
> EC Chip ID: 8226
> EC Chip Version: 1a2
> legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
> legion_laptop [...]

@chinese-wzq 
You can still not set fan cuves with the latest commits?
About the GPU power draw you should check with nvidia-smi (If you have nvidia card) in last driver the power limit at least change when on battery (but you can´t edit the value since nvidia remove the feature).

chinese-wzq, 2024-02-14:
> > I notice that I can't set fan curve at all.
> > ```
> > EC Chip ID: 8226
> > EC Chip Version: 1a2
> > legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
> > legion_laptop ec_readonly: 0
> > ACPI CFG error: 0
> > ACPI CFG: 4228773126
> > temperature access method: 2
> > CPU temperature error: 0
> > CPU temperature: 94
> > CPU temperature EC error: 0
> > CPU temperature EC: 0
> > CPU temperature ACPI error: 0
> > CPU temperature ACPI: 94
> > CPU temperature WMI error: 0
> > CPU temperature WMI: 0
> > CPU temperature WMI2 error: -14
> > CPU temperature WMI2: 0
> > CPU temperature WMI3 error: -14
> > CPU temperature WMI3: 0
> > GPU temperature error: 0
> > GPU temperature: 49
> > GPU temperature EC error: 0
> > GPU temperature EC: 0
> > GPU temperature ACPI error: 0
> > GPU temperature ACPI: 49
> > GPU temperature WMI error: 0
> > GPU temperature WMI: 0
> > GPU temperature WMI2 error: -14
> > GPU temperature WMI2: 0
> > GPU temperature WMI3 error: -14
> > GPU temperature WMI3: 0
> > fan speed access method: 3
> > 1 fanspeed error: 0
> > 1 fanspeed: 2100
> > 1 fanspeed EC error: 0
> > 1 fanspeed EC: 0
> > 1 fanspeed ACPI error: 0
> > 1 fanspeed ACPI: 2100
> > 1 fanspeed WMI error: 0
> > 1 fanspeed WMI: 2100
> > 1 fanspeed WMI2 error: -14
> > 1 fanspeed WMI2: 2100
> > 1 fanspeed WMI3 error: -14
> > 1 fanspeed WMI3: 2100
> > 2 fanspeed error: 0
> > 2 fanspeed: 1800
> > 2 fanspeed EC error: 0
> > 2 fanspeed EC: 0
> > 2 fanspeed ACPI error: 0
> > 2 fanspeed ACPI: 1800
> > 2 fanspeed WMI error: 0
> > 2 fanspeed WMI: 1800
> > 2 fanspeed WMI2 error: -14
> > 2 fanspeed WMI2: 1800
> > 2 fanspeed WMI3 error: -14
> > 2 fanspeed WMI3: 1800
> > powermode access method: 3
> > powermode error: 0
> > powermode: 1
> > powermode EC error: 0
> > powermode EC: 0
> > powermode ACPI error: -5
> > powermode ACPI: 0
> > powermode WMI error: 0
> > powermode WMI: 1
> > has custom powermode: 0
> > ACPI rapidcharge error: 0
> > ACPI rapidcharge: 0
[... 122 lines cut ...]
powermode WMI: 3
has custom powermode: 0
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light Y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: false
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

closed 2026-09-11 by github-actions[bot]