johnfanv2/LenovoLegionLinux #70: Lenovo 7 16IAX7 support

FY0u11, 2023-06-17:
Hi there. 

I have this model. And stuck at first step of tests. 

[image]

As I can understand my laptop is not support. Any things I can do to fix it?

johnfanv2, 2023-06-18:
This seems to be a newer model from 2022. Please provide the following information so that I can configure the driver:

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. If you do not have Windows skip it. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```
Which features are the most useful for you?

FY0u11, 2023-06-19:
## FWTS

Results generated by fwts: Version V23.05.00 (2023-05-23 04:00:48).

Some of this work - Copyright (c) 1999 - 2021, Intel Corp. All rights reserved.
Some of this work - Copyright (c) 2010 - 2021, Canonical.
Some of this work - Copyright (c) 2016 - 2021, IBM.
Some of this work - Copyright (c) 2017 - 2021, ARM Ltd.

This test run on 19/06/23 at 14:52:59 on host Linux fy0u11 6.3.8-arch1-1 #1 SMP
PREEMPT_DYNAMIC Wed, 14 Jun 2023 20:10:31 +0000 x86_64.

Command: "fwts wmi -".
Running tests: wmi.

wmi: Extract and analyse Windows Management Instrumentation (WMI).
--------------------------------------------------------------------------------
Test 1 of 1: Windows Management Instrumentation test.

\_SB_.PC00.AWMI._WDG (1 of 2)
  GUID: 603E9613-EF25-4338-A3D0-C46177516DB7
  WMI Method:
    Flags          : 0x02 (Method)
    Object ID      : AA
    Instance       : 0x01
FAILED [LOW] WMIMultipleMethod: Test 1, GUID
603E9613-EF25-4338-A3D0-C46177516DB7 has multiple associated methods WMAA
defined, this is a firmware bug that leads to ambiguous behaviour.

\_SB_.PC00.AWMI._WDG (2 of 2)
  GUID: 05901221-D566-11D1-B2F0-00A0C9062910
  WMI Object:
    Flags          : 0x00 (None)
    Object ID      : 00
    Instance       : 0x01

\_SB_.WMIS._WDG (1 of 3)
  GUID: 77E614ED-F19E-46D6-A613-A8669FEE1FF0
  WMI Method:
    Flags          : 0x02 (Method)
    Object ID      : SR
    Instance       : 0x01
PASSED: Test 1, 77E614ED-F19E-46D6-A613-A8669FEE1FF0 has associated method
\_SB_.WMIS.WMSR

\_SB_.WMIS._WDG (2 of 3)
  GUID: 05901221-D566-11D1-B2F0-00A0C9062910
  WMI Object:
    Flags          : 0x00 (None)
    Object ID      : DF
    Instance       : 0x01

\_SB_.WMIS._WDG (3 of 3)
  GUID: 95D1DF76-D6C0-4E16-9193-7B2A849F3DF2
  WMI Event:
    Flags          : 0x08 (Event)
    Notification ID: 0xd0
    Reserved       : 0x00
    Instance       : 0x01
FAILED [MEDIUM] WMIUnknownGUID: Test 1, GUID
95D1DF76-D6C0-4E16-9193-7B2A849F3DF2 is unknown to the kernel, a driver may need
to be implemented for this GUID.
A WMI driver probably needs to be written for this WMI event. It can checked for
using: wmi_has_guid("95D1DF76-D6C0-4E16-9193-7B2A849F3DF2"). One can install a
notify handler using
wmi_install_notify_handler("95D1DF76-D6C0-4E16-9193-7B2A849F3DF2", handler,
NULL). http://lwn.net/Articles/391230 describes how to write an appropriate
driver.

\_SB_.WMIU._WDG (1 of 3)
[... 422 lines cut ...]
Pass 2 parse of [SSDT]
External object resolution file           SSDT12
Input file SSDT12, Length 0x1D1C (7452) bytes
ACPI: SSDT 0x0000000000000000 001D1C (v02 LENOVO CB-01    00000001      01000013)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT11
Input file SSDT11, Length 0xA131 (41265) bytes
ACPI: SSDT 0x0000000000000000 00A131 (v02 LENOVO CB-01    00000001      01000013)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT10
Input file SSDT10, Length 0x9B6 (2486) bytes
ACPI: SSDT 0x0000000000000000 0009B6 (v02 LENOVO CB-01    00000001 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file            SSDT1
Input file SSDT1, Length 0x50D9 (20697) bytes
ACPI: SSDT 0x0000000000000000 0050D9 (v02 LENOVO CB-01    00000001 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
Pass 1 parse of [DSDT]
Firmware Error (ACPI): Failure creating named object [\_SB.PC00.RP13.PXSX._RST], AE_ALREADY_EXISTS (20221020/dswload-495)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20221020/psobject-372)
Could not parse ACPI tables, AE_ALREADY_EXISTS

FY0u11, 2023-06-19:
About the most useful features for me. I just want to control my fans speed. Sometimes they are spinning too loud and I have no opportunity to make them quite (it creates uncomfortable situations when I'm working away from home). On Windows I can enable quiet mode, so I would like the same feature on Linux. Thank you in advance.

FY0u11, 2023-06-22:
Now the output of make reloadmodule is following:

[21641.242671] legion_laptop: loading out-of-tree module taints kernel.
[21641.242804] legion_laptop: module verification failed: signature and/or required key missing - tainting kernel
[21641.246206] legion_laptop: legion_laptop starts loading
[21641.246276] legion PNP0C09:00: legion_laptop platform driver probing
[21641.246278] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TD; DMI_BIOS_VERSION:K1CN28WW
[21641.246279] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TD; DMI_BIOS_VERSION:K1CN28WW
[21641.246282] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[21641.246283] legion PNP0C09:00: Using configuration for system: K1CN
[21641.246363] legion PNP0C09:00: ACPI CFG: 3925729556
[21641.246364] legion_laptop: Creating RAM access to embedded controller
[21641.246383] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x0000000070079181
[21641.246726] legion PNP0C09:00: Read embedded controller ID 0x5263
[21641.246727] legion PNP0C09:00: Expected EC chip id 0x8227 but read 0x5263
[21641.246728] legion_laptop: Unloading legion ecram
[21641.246729] legion_laptop: Unloading legion ecram done
[21641.246729] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x0000000070079181
[21641.246731] legion_laptop: Unloading legion shared
[21641.246732] legion_laptop: Unloading legion shared done
[21641.246733] legion PNP0C09:00: legion_laptop not loaded for this device
[21641.247979] legion: probe of PNP0C09:00 failed with error -12

Here is all the files from the "Disassembling ACPI tables"
[acpi.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11835960/acpi.tar.gz)

johnfanv2, 2023-06-22:
Without the ACPI I assumed it is an older model. Is this a 2022 or 2023 model? 

I updated the driver to use a newer config. Please get the new version, recompile, and reload the module and try again.

FY0u11, 2023-06-23:
Still the error. I bought the laptop in december, 2022. It should be 2022 model.

[image]

FY0u11, 2023-06-23:
It can help. https://psref.lenovo.com/Detail/Legion/Legion_7_16IAX7?M=82TD0075GE

MrDuartePT, 2023-10-19:
@FY0u11 you can report if the latest version works?

closed 2026-09-11 by github-actions[bot]