johnfanv2/LenovoLegionLinux #72: Support for older Legion y720

lilongueti, 2023-06-22:
I know it's an older model, but it always had overheating problems (99 C while gaming) and i was hoping this would help
[image]

Distribution: Pop!_OS 22.04 LTS
Model name: Legion Y720
CPU model: Intel i7 7700HQ
GPU model: Nvidia GTX 1060
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: yes, but i dont think its configurable, at least not on windows
Light at IO-Ports at back: no 

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 80VR
	Version: Lenovo Y720-15IKB
	Serial Number: -
	UUID: -
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_80VR_BU_idea_FM_Y720-15IKB
	Family: Y720-15IKB

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: 4GCN38WW
	Release Date: 07/18/2018
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 6 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.38
	Firmware Revision: 1.38

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
doesn't exist
```

johnfanv2, 2023-06-23:
We can try but I am not sure if all features will work. Which features are the most useful for you? 

To this end, please provide the following information so that I can configure the driver:

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. If you do not have Windows skip it. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```

### First Debug 
I added your model to the driver to get more information. Please get the newest version and try to load it. Then, please provide the output of the following commands:
```
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | grep legion
```

lilongueti, 2023-06-23:
here are the files on my nextcloud, let me know if you would rather have them in some other way
https://nextcloud.quasarmexico.com/s/bJNDd3eSrBFW5Sp
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
```
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000600
```
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
```
00000000  00 00 00 00 00 00 22 00  00 00 00 00 00 49 56 00  |......"......IV.|
00000010  00 00 80 00 05 00 09 02  00 01 07 00 00 00 00 00  |................|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000040  00 00 00 00 00 00 00 00  00 00 00 00 71 4d 00 00  |............qM..|
00000050  00 00 00 0f 00 00 56 00  00 00 00 00 00 00 00 00  |......V.........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000080  00 00 00 00 08 00 00 00  80 00 c2 01 00 00 00 53  |...............S|
00000090  4d 50 00 31 31 00 00 00  4c 31 36 4d 34 50 42 30  |MP.11...L16M4PB0|
000000a0  00 08 00 86 00 00 00 61  64 00 00 00 64 25 02 00  |.......ad...d%..|
000000b0  55 00 00 00 00 00 00 00  02 11 00 00 00 00 11 00  |U...............|
000000c0  70 02 cd 07 f7 02 bd 3c  00 3c 70 17 ff 13 27 13  |p......<.<p...'.|
000000d0  fc 05 a7 05 00 00 00 29  3a 06 0b 00 00 00 00 00  |.......):.......|
000000e0  00 00 19 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000f0  01 00 00 00 75 4b 30 43  00 00 00 00 00 00 24 00  |....uK0C......$.|
00000100  82 00 00 00 00 00 03 03  1d 1d 37 1d 1d 1d 25 1d  |..........7...%.|
00000110  fe 00 16 03 00 00 ff 03  00 00 00 00 00 00 00 00  |................|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000150  21 29 37 42 48 50 5a 5a  21 29 37 42 48 50 5a 5a  |!)7BHPZZ!)7BHPZZ|
00000160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000190  00 1c 24 32 3d 43 4b 55  00 1c 24 32 3d 43 4b 55  |..$2=CKU..$2=CKU|
000001a0  00 13 17 19 1d 1f 21 23  00 10 15 17 1f 21 23 25  |......!#.....!#%|
000001b0  83 0d 46 0e 23 25 07 03  56 00 00 00 00 00 00 00  |..F.#%..V.......|
000001c0  00 00 00 3c 43 50 55 55  00 00 00 3c 43 50 55 55  |...<CPUU...<CPUU|
000001d0  00 00 00 00 37 3e 4b 50  00 00 00 00 37 3e 4b 50  |....7>KP....7>KP|
000001e0  00 00 00 00 1d 1f 21 23  00 00 00 00 1f 21 23 25  |......!#.....!#%|
000001f0  00 00 00 00 cd 07 00 00  03 07 00 00 23 25 5a 5a  |............#%ZZ|
00000200  23 25 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |#%..............|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 02 00 00  |................|
00000240  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000280  32 fc 64 06 00 00 00 00  00 00 00 00 00 00 00 00  |2.d.............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  00 40 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |.@..............|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002e0  00 00 00 00 00 00 00 64  00 00 00 00 00 00 00 00  |.......d........|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000300
```
sudo cat /sys/kernel/debug/legion/fancurve
```
EC Chip ID: 8226
EC Chip Version: 1a2
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 3155031318
temperature access method: 1
CPU temperature error: 0
CPU temperature: 33
CPU temperature EC error: 0
CPU temperature EC: 33
CPU temperature ACPI error: 0
CPU temperature ACPI: 94
[... 112 lines cut ...]
[ 1191.728106] legion PNP0C09:00: legion_laptop loaded for this device
[ 2804.066647] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:8
[ 2804.066651] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[ 2804.066894] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:8
[ 2804.066897] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[ 2804.067281] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:7
[ 2804.067283] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[ 2804.067702] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:7
[ 2804.067705] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[ 2804.068438] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[ 2804.068441] legion_laptop: Error for WMI method call to get brightness
[ 2804.068442] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[ 2804.068444] legion_laptop: Error for WMI method call to get brightness
[ 2804.068445] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[ 2804.068446] legion_laptop: Error for WMI method call to get brightness
[ 2804.068517] legion_laptop: Unexpected value in MINIFANCURVE register:0
[ 2804.075799] legion_laptop: Size of object: 48
[ 2804.075807] legion_laptop: ACPI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:5
[ 2804.075810] legion_laptop fan table wmi buffer00000000e6651b36: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[... cut ...] ......
[ 2804.075816] legion_laptop fan table wmi buffer000000008ee8fcc5: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2804.075819] legion_laptop fan table wmi buffer00000000fd9daa0b: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2804.075821] legion_laptop fan table wmi buffer00000000eea50ed3: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2804.075823] legion_laptop fan table wmi buffer000000005eadaa2e: 00 00 00 00 00 00 00 00
```

johnfanv2, 2023-06-24:
Log output look promising. I guess fan control might work.

Please upload the github, e.g. as a zip file, so they will be stored for a longer period combined with this repo. You can do that by drag/dropping them into the text field.

lilongueti, 2023-06-25:
i've updated the comment with the zip file, as for which features are the most useful for me, its just the fan curve really, it's sub woofer has never worked under linux (i've tried with hdajackretask), but i'm guessing this isn't related to this chip and outside the scope of this repo

MrDuartePT, 2023-10-20:
@lilongueti the fan curve is working on your legion?

MrDuartePT, 2023-10-20:
@lilongueti you can test with the latest commit if features work?

closed 2026-09-11 by github-actions[bot]