johnfanv2/LenovoLegionLinux #73: [BUG] GUI Power Options aren't shown properly

XenHat, 2023-06-23:
**Problem Descripition**
It would seem that the "Power Options" GPU and CPU values are swapped around in the GUI:

[image]

**Model and Debug Info**

Distribution: Arch Linux
Model name: Lenovo Legion 5 Pro 16ach6h (Model 82JQ)
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3060
Keyboard backlight: RGB (firmware control, no iCUE required)
Light in lid or logo: Yes
Light at IO-Ports at back: No (?)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
$ sudo dmidecode -t system | grep -v UUID | grep -v "Serial Number"
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JQ
        Version: Legion 5 Pro 16ACH6H
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JQ_BU_idea_FM_Legion 5 Pro 16ACH6H
        Family: Legion 5 Pro 16ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
$ sudo dmidecode -t system | grep -v UUID | grep -v "Serial Number"
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JQ
        Version: Legion 5 Pro 16ACH6H
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JQ_BU_idea_FM_Legion 5 Pro 16ACH6H
        Family: Legion 5 Pro 16ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 45
CPU temperature EC error: 0
CPU temperature EC: 45
CPU temperature ACPI error: 0
CPU temperature ACPI: 45
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 45
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1954
1 fanspeed EC error: 0
1 fanspeed EC: 1954
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1900
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1900
2 fanspeed error: 0
2 fanspeed: 1942
2 fanspeed EC error: 0
2 fanspeed EC: 1942
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1900
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1900
powermode access method: 3
powermode error: 0
powermode: 255
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 255
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: false
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 1
EC fan curve points size: 10
Current fan curve in hardware (embedded controller):
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 48	 0	 60	 0	 41
1700	 1900	 2	 2	 45	 54	 56	 60	 39	 44
1900	 2000	 2	 2	 51	 58	 56	 60	 42	 50
2200	 2100	 2	 2	 55	 62	 56	 60	 46	 127
2300	 2400	 2	 2	 59	 71	 56	 60	 127	 127
2600	 2700	 2	 2	 68	 76	 56	 65	 127	 127
2900	 3000	 2	 2	 72	 81	 61	 69	 127	 127
3500	 3500	 2	 2	 78	 90	 66	 77	 127	 127
3800	 3800	 2	 2	 87	 90	 72	 79	 127	 127
0	 0	 2	 2	 95	 127	 80	 127	 127	 127
=====================

```

johnfanv2, 2023-06-23:
Do you mean that e.g. 80W in the line "CPU long term power limit" should be a limit of the GPU?

Could you test the following: Change a power limit of the CPU in custom mode in the GUI and check if this changes some value of the CPU read by the tool RyzenAdj (https://github.com/FlyGoat/RyzenAdj)?

XenHat, 2023-06-23:
Before setting a custom value:

```
 sudo ./ryzenadj --info    
CPU Family: Cezanne
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x12, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU BIOS Interface Version: 18
Version: v0.13.0 
init_table
SMU_SERVICE REQ_ID:0x6
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x400005, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x66
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xcced7000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xfd, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
PM Table Version: 400005
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    25.000 | stapm-limit        |
| STAPM VALUE         |     9.740 |                    |
| PPT LIMIT FAST      |    48.000 | fast-limit         |
| PPT VALUE FAST      |     8.732 |                    |
| PPT LIMIT SLOW      |    37.500 | slow-limit         |
| PPT VALUE SLOW      |     9.257 |                    |
| StapmTimeConst      |    50.000 | stapm-time         |
| SlowPPTTimeConst    |     5.000 | slow-time          |
| PPT LIMIT APU       |    54.000 | apu-slow-limit     |
| PPT VALUE APU       |     9.257 |                    |
| TDC LIMIT VDD       |    33.000 | vrm-current        |
| TDC VALUE VDD       |     3.233 |                    |
| TDC LIMIT SOC       |    15.000 | vrmsoc-current     |
| TDC VALUE SOC       |     2.716 |                    |
| EDC LIMIT VDD       |    90.000 | vrmmax-current     |
| EDC VALUE VDD       |    90.000 |                    |
| EDC LIMIT SOC       |    20.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |    90.000 | tctl-temp          |
| THM VALUE CORE      |    49.000 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    50.000 | power-saving /     |
| CCLK BUSY VALUE     |    34.295 | max-performance    |

```
[image]

After:
```
sudo ./ryzenadj --info
CPU Family: Cezanne
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x12, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU BIOS Interface Version: 18
Version: v0.13.0 
init_table
SMU_SERVICE REQ_ID:0x6
[... 79 lines cut ...]
| PPT LIMIT FAST      |    48.000 | fast-limit         |
| PPT VALUE FAST      |    16.154 |                    |
| PPT LIMIT SLOW      |   100.001 | slow-limit         |
| PPT VALUE SLOW      |    11.271 |                    |
| StapmTimeConst      |    50.000 | stapm-time         |
| SlowPPTTimeConst    |     5.000 | slow-time          |
| PPT LIMIT APU       |    54.000 | apu-slow-limit     |
| PPT VALUE APU       |    11.271 |                    |
| TDC LIMIT VDD       |    33.000 | vrm-current        |
| TDC VALUE VDD       |     4.204 |                    |
| TDC LIMIT SOC       |    15.000 | vrmsoc-current     |
| TDC VALUE SOC       |     2.872 |                    |
| EDC LIMIT VDD       |    90.000 | vrmmax-current     |
| EDC VALUE VDD       |    65.676 |                    |
| EDC LIMIT SOC       |    20.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |    90.000 | tctl-temp          |
| THM VALUE CORE      |    50.268 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    50.000 | power-saving /     |

```

MrDuartePT, 2023-10-19:
For my testing the values are not swap and if you put in the `cpu short term power limit` a value not supported by the cpu, ryzenadj will change the stock value.

closed 2023-10-19 by MrDuartePT