johnfanv2/LenovoLegionLinux #74: Lenovo Ideapad is supported?

JuanH88, 2023-06-24:
I know it's not very appropriate to refer to this, sorry.  But I am wondering if this tool would work on other models of lenovo laptops, I am thinking of trying it on my ideapad gaming 3i

johnfanv2, 2023-06-24:
There was another Ideapad laptop where most of the features worked. The driver definitely has to be configured and a little bit adapted. We can try but I am not sure if all features will work. Which features are the most useful for you? What is the model year of the laptop?

To this end, please provide the following information:

**Model and Debug Info**

Distribution: ??? (including version, e.g. Ubuntu 24.04)
Model name: ???
CPU model: ??? (e.g. AMD Ryzen 7 5800H)
GPU model: ??? (e.g NVIDIA RTX 3700)
Keyboard backlight: ??? (RGB, single color with off/on, single color with off/medium/bright)
Light in lid or logo: ??? (yes/no)
Light at IO-Ports at back: ??? (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
???
```

Output of `sudo dmidecode -t bios`:
```text
???
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
???
```

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. If you do not have Windows skip it. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```

JuanH88, 2023-06-24:
Distribution: ArchLinux
Model name: Ideapad Gaming 3i 15IMH05
CPU model: i5 10th gen, i5 10300H
GPU model: NVIDIA GTX 1650
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: No
Light at IO-Ports at back: No

johnfanv2, 2023-06-24:
Apologies for the incorrect phrasing. Please also provide the the output of the the other commands listed above.

JuanH88, 2023-06-24:
sudo dmidecode -t system:

Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 81Y4
	Version: IdeaPad Gaming 3 15IMH05
	Serial Number: PF330NVS
	UUID: a1e0bee1-dafe-11eb-80f0-902e165d4b8d
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_81Y4_BU_idea_FM_IdeaPad Gaming 3 15IMH05
	Family: IdeaPad Gaming 3 15IMH05

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
---------------------------------------------------------------
sudo dmidecode -t bios:
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: EGCN36WW
	Release Date: 02/14/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 11 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.36
	Firmware Revision: 1.36

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
-------------------------------------------------------------------------------------
sudo cat /sys/kernel/debug/legion/fancurve:
cat: /sys/kernel/debug/legion/fancurve: No such file or directory

johnfanv2, 2023-06-24:
Sorry for the imprecise phrasing, again. :smiling_face_with_tear: 

Please also provide the the **files** of the the other steps ("Inspecting WMI entries", ... ).

closed 2026-09-11 by github-actions[bot]