johnfanv2/LenovoLegionLinux #76: Legion Slim 5 16APH8 Support

brett-owen, 2023-06-27:
Model name: Lenovo Legion Slim 5 16APH8
CPU model: AMD Ryzen 7 7840HS
GPU model: NVIDIA RTX 4060 Mobile
Keyboard backlight: RGB
OS: Archlinux

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82Y9
        Version: Legion Slim 5 16APH8
        Serial Number: ******
        UUID: ******
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82Y9_BU_idea_FM_Legion Slim 5 16APH8
        Family: Legion Slim 5 16APH8

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002A, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: M3CN31WW
        Release Date: 04/21/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.31
        Firmware Revision: 1.31

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/11876493/fwts_wmi.log)
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/11876486/acpi_re.tar.gz)
[wmi_windows.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/11876494/wmi_windows.txt)

Output of `sudo make forcereloadmodule`:
```text
legion_laptop: legion_laptop starts loading
legion PNP0C09:00: legion_laptop platform driver probing
legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9;  DMI_BIOS_VERSION:M3CN31WW
legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN31WW
legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
legion PNP0C09:00: legion_laptop not loaded for this device
legion: probe of PNP0C09:00 failed with error -12
legion_laptop: legion_laptop exit
legion_laptop: legion_laptop starts loading
legion PNP0C09:00: legion_laptop platform driver probing
legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN31WW
legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN31WW
legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
legion PNP0C09:00: legion_laptop is forced to load.
legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
legion PNP0C09:00: Using configuration for system: GKCN
legion PNP0C09:00: Could not evaluate ACPI _STA
legion PNP0C09:00: Could not init ACPI access
legion_laptop: Unloading legion shared
legion_laptop: Unloading legion shared done
legion PNP0C09:00: legion_laptop not loaded for this device
legion: probe of PNP0C09:00 failed with error -5
legion_laptop: legion_laptop exit
legion_laptop: legion_laptop starts loading
legion PNP0C09:00: legion_laptop platform driver probing
legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN31WW
legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN31WW
legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
legion PNP0C09:00: legion_laptop is forced to load.
legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
legion PNP0C09:00: Using configuration for system: GKCN
legion PNP0C09:00: Could not evaluate ACPI _STA
legion PNP0C09:00: Could not init ACPI access
legion_laptop: Unloading legion shared
legion_laptop: Unloading legion shared done
legion PNP0C09:00: legion_laptop not loaded for this device
```

Thanks, let me know if you need any more information to get this model working.

johnfanv2, 2023-07-16:
@brett-owen  Sorry, for the late response. Currently, there are lots of issues and a also work on lots of other stuff.

Is this a 2023 model?

I added your model/BIOS to the kernel module. Please get the new version, recompile, and reload the kernel module (no need to force anymore). 

Then, please test the following features. We have to try a bit to find the correct configuration.

# Temperatures and Fan Speeds
Please read the fan speeds and temperatures (see README) and confirm that they are sensible and change accordingly, e.g. when you put load onto the CPU the CPU temp and the fan speeds should increase, but GPU temp should stay roughly the same.

# Powermode

We will start by confirming that power mode change work **and to get me more debug info**. Please change the power mode with Fn+Q and provide the output of the following commands in **each** mode. 

```bash
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve
```

Then please also check if the power mode is read correctly in the OS (see README). Then please check if you can set the power mode with the GUI (see README) and the LED changes accordingly.

Tests the following features with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software. 

Please provide the log output from the GUI from the log tab for all the tests.

# Fn Lock
- Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
- Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn 

# Win Key
- Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
- Disable Win Key: Win Key has no function anymore

# Touchpad
- Enable Touchpad: Touchpad works
- Disable Touchpad: Touchpad does not work

# Battery Conservation
- Enabled: charging stops if above 60 % (or 80% depending on model)
- Disabled: charging works up to 100%
Note: Sometimes it takes a 10-20 seconds before it takes effect

# Rapid Charging
- Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
- Disabled: Charging is slower
Note: Sometimes it takes a 10-20 seconds before it takes effect .

# Charge Output Always On
- Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
- Disabled: USB outputs are off when laptop is off

# Display Overdrive
- Enabled: display somehow faster
- Disabled: display not faster
Just check if checkbox can be checked or not if you cannot verify that it has an effect.

# Keyboard Backlight

In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:

```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```

# Fan curve
First, please give again the output of 
```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```
Second, please use the GUI and enter custom mode. Then change the fan curve (currently only the first row for fan 1), write it to hardware, and read it back. Please report back if it works and provide the log output from the GUI and `sudo dmesg | tail -n 70`.

th4tkh13m, 2023-10-12:
@johnfanv2 I can help you continue the tests. Currently, my OS is Ubuntu 22.04.
Most options of LenovoLegionLinux are greyed out for me, and I want to help with the project. Please let me know if this should be separated into a new issue or if I can do something to help.

[image]

MrDuartePT, 2023-10-19:
@brett-owen pls report back if the driver works

Farcasut, 2023-10-26:
@MrDuartePT I have the same model and the error still persists with the current main. I get these messages:
[Thu Oct 26 12:47:12 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.GP18.SATA], AE_NOT_FOUND (20230331/dswload2-162)
[Thu Oct 26 12:47:12 2023] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20230331/psobject-220)
[Thu Oct 26 12:47:12 2023] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[Thu Oct 26 12:47:12 2023] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.GPP6.WLAN._S0W], AE_ALREADY_EXISTS (20230331/dswload2-326)
[Thu Oct 26 12:47:12 2023] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-220)
[Thu Oct 26 12:47:17 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.GPP0.PEGP.GPS.NVD1], AE_NOT_FOUND (20230331/psargs-330)
[Thu Oct 26 12:47:17 2023] ACPI Error: Aborting method \_SB.PCI0.GPP0.PEGP.GPS due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Thu Oct 26 12:47:17 2023] ACPI Error: Aborting method \_SB.PCI0.GPP0.PEGP._DSM due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Thu Oct 26 12:57:48 2023] legion: probe of PNP0C09:00 failed with error -5

MrDuartePT, 2023-10-26:
The acpi error is nothing to do with the legion driver. About the error try make forcereloadmodule.

MrDuartePT, 2023-10-26:
Also give me the dmidecode output

Farcasut, 2023-10-26:
@MrDuartePT  sorry for the late response
demidecode:
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.
Table at 0x9169C000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: M3CN31WW
	Release Date: 04/21/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.31
	Firmware Revision: 1.31

fancurve:

EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228186128
temperature access method: 5
CPU temperature error: 0
CPU temperature: 45
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 45
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 45
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 42
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 42
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 1
WMI light Y logo/lid: 1
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp

MrDuartePT, 2023-10-26:
Also give me the dmesg only of the legion driver after forcereloadmodule what to check one thing.

MrDuartePT, 2023-10-26:
Also the hexdump for the 3 power modes:
hexdump -c /sys/kernel/debug/legion/ecmemory

Farcasut, 2023-10-26:
@MrDuartePT 
```
[    0.023366] PM: hibernation: Registered nosave memory: [mem 0xa0000000-0xfed7ffff]
[    0.023367] PM: hibernation: Registered nosave memory: [mem 0xfed80000-0xfed80fff]
[    0.023367] PM: hibernation: Registered nosave memory: [mem 0xfed81000-0xffffffff]
[    0.023368] [mem 0xa0000000-0xfed7ffff] available for PCI devices
[    0.023369] Booting paravirtualized kernel on bare hardware
[    0.023370] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370452778343963 ns
[    0.026295] setup_percpu: NR_CPUS:320 nr_cpumask_bits:16 nr_cpu_ids:16 nr_node_ids:1
[    0.026468] percpu: Embedded 64 pages/cpu s225280 r8192 d28672 u262144
[    0.026472] pcpu-alloc: s225280 r8192 d28672 u262144 alloc=1*2097152
[    0.026473] pcpu-alloc: [0] 00 01 02 03 04 05 06 07 [0] 08 09 10 11 12 13 14 15 
[    0.026492] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-linux root=/dev/mapper/volgroup0-lv_root rw loglevel=3 quiet
[    0.026527] Unknown kernel command line parameters "BOOT_IMAGE=/boot/vmlinuz-linux", will be passed to user space.
[    0.027809] Dentry cache hash table entries: 4194306 (order: 13, 33554432 bytes, linear)
[    0.028359] Inode-cache hash table entries: 2097152 (order: 12, 16777216 bytes, linear)
[    0.028430] Fallback order for Node 0: 0 
[    0.028435] Built 1 zonelists, mobility grouping on.  Total pages: 8048859
[    0.028435] Policy zone: Normal
[    0.028504] mem auto-init: stack:all(zero), heap alloc:on, heap free:off
[    0.028549] software IO TLB: area num 16.
[    0.056753] Memory: 31876588K/32707124K available (16384K kernel code, 2123K rwdata, 12956K rodata, 3392K init, 3904K bss, 830276K reserved, 0K cma-reserved)
[    0.056895] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=16, Nodes=1
[    0.056913] ftrace: allocating 48466 entries in 190 pages
[    0.063277] ftrace: allocated 190 pages with 6 groups
[    0.063310] Dynamic Preempt: full
[    0.063332] rcu: Preemptible hierarchical RCU implementation.
[    0.063333] rcu: 	RCU restricting CPUs from NR_CPUS=320 to nr_cpu_ids=16.
[    0.063333] rcu: 	RCU priority boosting: priority 1 delay 500 ms.
[    0.063334] 	Trampoline variant of Tasks RCU enabled.
[    0.063334] 	Rude variant of Tasks RCU enabled.
[    0.063334] 	Tracing variant of Tasks RCU enabled.
[    0.063335] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.063335] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=16
[    0.064723] NR_IRQS: 20736, nr_irqs: 1096, preallocated irqs: 16
[    0.064902] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.065035] kfence: initialized - using 2097152 bytes for 255 objects at 0x(____ptrval____)-0x(____ptrval____)
[    0.065058] Console: colour dummy device 80x25
[    0.065059] printk: console [tty0] enabled
[    0.065085] ACPI: Core revision 20230331
[    0.065300] clocksource: hpet: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 133484873504 ns
[    0.065319] APIC: Switch to symmetric I/O mode setup
[    0.065832] AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR0, rdevid:160
[    0.065833] AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR1, rdevid:160
[    0.065834] AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR2, rdevid:160
[    0.065834] AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR3, rdevid:160
[    0.065835] AMD-Vi: Using global IVHD EFR:0x246577efa2254afa, EFR2:0x0
[    0.066361] x2apic: IRQ remapping doesn't support X2APIC mode
[    0.066364] Switched APIC routing to physical flat.
[    0.067134] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=-1 pin2=-1
[    0.081990] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x6d58962be3c, max_idle_ns: 881590460742 ns
[    0.081993] Calibrating delay loop (skipped), value calculated using timer frequency.. 7588.96 BogoMIPS (lpj=12643136)
[    0.082011] x86/cpu: User Mode Instruction Prevention (UMIP) activated
[    0.082468] LVT offset 1 assigned for vector 0xf9
[    0.082631] LVT offset 2 assigned for vector 0xf4
[    0.082669] process: using mwait in idle threads
[    0.082670] Last level iTLB entries: 4KB 512, 2MB 512, 4MB 256
[    0.082671] Last level dTLB entries: 4KB 3072, 2MB 3072, 4MB 1536, 1GB 0
[    0.082673] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.082675] Spectre V2 : Mitigation: Enhanced / earutomatic IBRS
[    0.082675] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
[    0.082677] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
[    0.082677] Spectre V2 : User space: Mitigation: STIBP always-on protection
[    0.082679] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl
[    0.082679] Speculative Return Stack Overflow: IBPB-extending microcode not applied!
[    0.082680] Speculative Return Stack Overflow: WARNING: See https://kernel.org/doc/html/latest/admin-guide/hw-vuln/srso.html for mitigation options.
[    0.082680] Speculative Return Stack Overflow: Mitigation: safe RET, no microcode
[    0.082685] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.082686] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.082687] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[... 1821 lines cut ...]
00000a0  \0  \0  \0   H   @  \0 300  \0   .  \0  \0  \b  \0   *  \0  \0
00000b0  \0   0  \0  \0   1  \0  \0  \0  \0  \0  \0 017  \0   z  \0  \0
[... cut ...]
0000220   Z 245  \0  \0  \0  \0  \0  \0  \0  \0  \0   Y 261  \0  \0  \0
0000230  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000240 001   #  \0  \0  \0  \0  \0     004  \0 002  \0  \0  \0  \0  \0
0000250  \0  \0  \0   @  \0  \0 377  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000260  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000270 347 333  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000280  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
*
0000300
````

MrDuartePT, 2023-10-26:
Sorry I give the wrong command edit with this and pls clean up some the dmesg:

`sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve`

Farcasut, 2023-10-26:
@MrDuartePT 
```
00000000  00 00 00 8c 00 00 00 00  00 00 80 10 00 00 00 00  |................|
00000010  00 00 2e 2b 2b 2f 30 2c  20 20 03 00 40 62 00 40  |...++/0,  ..@b.@|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 03 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00  |................|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  00 00 00 48 40 00 00 00  29 00 00 08 00 2a 00 00  |...H@...)....*..|
000000b0  00 35 00 00 31 00 00 00  00 00 00 0f 00 7a 00 00  |.5..1........z..|
000000c0  00 01 83 0d 8d 0a f3 3a  50 3c 40 1f e6 20 29 22  |.......:P<@.. )"|
000000d0  77 f3 e4 f5 56 0e e6 20  4a 03 51 11 00 00 00 00  |w...V.. J.Q.....|
000000e0  00 00 00 c0 0e b9 0e c0  0e bb 0e 00 00 00 00 00  |................|
000000f0  00 00 00 00 00 00 c0 44  0a 0d 00 ff ff 00 00 00  |.......D........|
00000100  00 00 00 00 00 00 05 00  4d 00 37 5a 15 50 5a 1d  |........M.7Z.PZ.|
00000110  50 5a 1d 50 5a 1d 50 5a  1d 50 5a 1d 50 5e 23 52  |PZ.PZ.PZ.PZ.P^#R|
00000120  69 23 5e 6c 23 00 00 00  00 00 00 00 00 00 00 4d  |i#^l#..........M|
00000130  00 37 5a 15 50 5a 1d 50  5a 1d 50 5a 1d 50 5a 1d  |.7Z.PZ.PZ.PZ.PZ.|
00000140  50 5a 1d 50 5e 23 52 69  23 5e 6c 23 00 00 00 00  |PZ.P^#Ri#^l#....|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 01 00  00 00 00 00 00 00 00 00  |................|
00000210  02 0c 0e 11 15 15 15 15  15 2a 00 00 00 00 00 00  |.........*......|
00000220  5a a5 00 00 00 00 00 00  00 00 00 59 b1 00 00 00  |Z..........Y....|
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000240  01 23 00 00 00 00 00 20  04 00 02 00 00 00 00 00  |.#..... ........|
00000250  00 00 00 40 00 00 ff 00  00 00 00 00 00 00 00 00  |...@............|
00000260  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000270  e7 db 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000300  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000600
```
second
```
00000000  00 00 00 8c 00 00 00 00  00 00 80 10 00 00 00 00  |................|
00000010  00 00 2f 2c 2c 31 31 2d  20 20 03 00 40 62 00 40  |../,,11-  ..@b.@|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 03 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00  |................|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  00 00 00 48 40 00 00 00  32 00 00 08 00 2a 00 00  |...H@...2....*..|
000000b0  00 33 00 00 32 00 00 00  00 00 00 0f 00 7a 00 00  |.3..2........z..|
000000c0  00 01 60 0d 8d 0a 3f 3b  50 3c 40 1f e6 20 29 22  |..`...?;P<@.. )"|
000000d0  97 f8 d8 f5 56 0e e6 20  4a 03 51 11 00 00 00 00  |....V.. J.Q.....|
000000e0  00 00 00 cb 0e c0 0e cc  0e cb 0e 00 00 00 00 00  |................|
000000f0  00 00 00 00 00 00 c0 44  01 0a 00 ff ff 00 00 00  |.......D........|
00000100  01 65 1b 01 11 0b 25 00  4d 00 37 5a 15 50 5a 1d  |.e....%.M.7Z.PZ.|
00000110  50 5a 1d 50 5a 1d 50 5a  1d 50 5a 1d 50 5e 23 52  |PZ.PZ.PZ.PZ.P^#R|
00000120  69 23 5e 6c 23 11 11 01  65 1b 01 11 0b 20 00 4d  |i#^l#...e.... .M|
00000130  00 37 5a 15 50 5a 1d 50  5a 1d 50 5a 1d 50 5a 1d  |.7Z.PZ.PZ.PZ.PZ.|
00000140  50 5a 1d 50 5e 23 52 69  23 5e 6c 23 11 11 00 00  |PZ.P^#Ri#^l#....|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 01 00  00 00 00 00 00 00 00 00  |................|
00000210  02 0c 0e 11 15 15 15 15  15 2a 00 00 00 00 00 00  |.........*......|
00000220  5a a5 00 00 00 00 00 00  00 00 00 59 b1 00 00 00  |Z..........Y....|
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000240  01 23 00 00 00 00 00 20  04 00 02 00 00 00 00 00  |.#..... ........|
00000250  00 00 00 40 00 00 ff 00  00 00 00 00 00 00 00 00  |...@............|
00000260  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000270  e7 db 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
```

Farcasut, 2023-10-26:
Also when I use the sensor command the fans are off(0 rpm)

arashsm79, 2023-12-14:
I'm having problems using 0.0.9-prerelease as well.

```
Linux 6.6.6 NixOS-23-11 SMP PREEMPT_DYNAMIC  x86_64 GNU/Linux
```

```
$ sudo dmidecode -s system-version
Legion Slim 5 16APH8
```

```
$ journalctl --boot
...
kernel: legion_laptop: legion_laptop starts loading
kernel: legion PNP0C09:00: legion_laptop platform driver probing
kernel: legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN>
kernel: legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN36WW
kernel: legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
kernel: legion PNP0C09:00: Using configuration for system: M3CN
kernel: legion PNP0C09:00: Could not evaluate ACPI _STA
kernel: legion PNP0C09:00: Could not init ACPI access: -5
kernel: legion_laptop: Unloading legion shared
kernel: legion_laptop: Unloading legion shared done
kernel: legion PNP0C09:00: legion_laptop not loaded for this device
kernel: legion: probe of PNP0C09:00 failed with error -5
...
```

MrDuartePT, 2023-12-14:
The problem with the fancurve is that hexdump don’t give a identification of what values are temps and fan speed. 
One user reverse engineer the ec for the legion go if the fan control is present and it the same it would be possible to make gen 8 support.

mcarans, 2024-03-11:
@MrDuartePT What are the steps involved to reverse engineer the EC for gen 8 support (I have a 14APH8)?

MrDuartePT, 2024-03-11:
> @MrDuartePT What are the steps involved to reverse engineer the EC for gen 8 support (I have a 14APH8)?

I scratching this ideia is very time consuming and probably not worth the hassle since it can change from model to model.

If you want to help the better approach is to find the acpi_call for the wmi implemented by Lenovo.
More info here and how to get the dumps:

https://github.com/hhd-dev/hwinfo/tree/master/devices/Legion_Pro_5_16IRX9

https://github.com/hhd-dev/hwinfo/blob/master/devices/legion_go/readme.md

mcarans, 2024-03-11:
@MrDuartePT Should I make PR against hwinfo with the dumps for my machine (removing serial number and UUID)?

mcarans, 2024-03-29:
@MrDuartePT I was wondering what you want me to do with the dumps (for Legion Slim 5 14 14APH8) you described previously. Are they still useful to you? Should I make a PR against hwinfo?

MrDuartePT, 2024-03-29:
> @MrDuartePT I was wondering what you want me to do with the dumps (for Legion Slim 5 14 14APH8) you described previously. Are they still useful to you? Should I make a PR against hwinfo?

Sorry did notice your reply. You can do it if hwinfo dosent have your device, also help to know if something changes between the slim and regular legion

mcarans, 2024-03-29:
@MrDuartePT I have made the PR against hwinfo: https://github.com/hhd-dev/hwinfo/pull/17

jul1u5, 2024-06-03:
I also have Legion Slim 5 14 14APH8. I see the PR https://github.com/hhd-dev/hwinfo/pull/17 is now merged. What else is needed to support this model? And can I help in any way?

MrDuartePT, 2024-06-03:
> I also have Legion Slim 5 14 14APH8. I see the PR [hhd-dev/hwinfo#17](https://github.com/hhd-dev/hwinfo/pull/17) is now merged. What else is needed to support this model? And can I help in any way?

That is not support for any driver of some sort that is only hw logs, that will help supporting new laptop. The [Legion GO PR](https://github.com/johnfanv2/LenovoLegionLinux/pull/185) will help support new devices. After we finalize that will process to add all the gen 8 and maybe gen 9 devices.

Also fan control on newer model (basically what is mainly broken on new model), might only work on custom mode (like LLT)

MrDuartePT, 2024-06-03:
Also someone pls do all the test in here (besides the fan curve): https://github.com/johnfanv2/LenovoLegionLinux/issues/76#issuecomment-1637196666

To be sure what is working or not

jul1u5, 2024-06-04:
I could do the tests for Legion Slim 5 14APH8, but right now I get this message:
```
[    6.809805] legion PNP0C09:00: Module not usable for this laptop because it is not in allowlist. Notify the maintainer if you want to add your device or force load with param force.
[    6.809807] legion PNP0C09:00: legion_laptop not loaded for this device
```

Could you add 14APH8 to the allowlist? I provided some info of my model in https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-1846454704 if that's helpful.

MrDuartePT, 2024-06-04:
> I could do the tests for Legion Slim 5 14APH8, but right now I get this message:
> 
> ```
> [    6.809805] legion PNP0C09:00: Module not usable for this laptop because it is not in allowlist. Notify the maintainer if you want to add your device or force load with param force.
> [    6.809807] legion PNP0C09:00: legion_laptop not loaded for this device
> ```
> 
> Could you add 14APH8 to the allowlist? I provided some info of my model in [#46 (comment)](https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-1846454704) if that's helpful.

clone the repo and run in the `kernel_module` folder: `sudo make forcereloadmodule`, to force the module to load

jul1u5, 2024-06-06:
Thanks, I see it wasn't loading because I didn't have the "force=1" parameter for the kernel module.

I can confirm that the following options work in the GUI:

- [x] **Power mode/platform profile**
- [x] **Fn Lock**
- [x] **Win Key**
- [x] **Touchpad**
- [x] **Battery Conservation**
- [x] **Rapid Charging**

I will test the **Charge Output Always On** option later today.

**Display Overdrive** doesn't work: the checkbox cannot be checked.

Changing **Keyboard Backlight** modifies the `WMI backlight 3 state` field in `/sys/kernel/debug/legion/fancurve` as follows:
- Off: `0`
- Low brightness: `1`
- High brightness: `2`

(`sudo dmesg` doesn't output anything when changing the backlight)

jul1u5, 2024-06-06:
Also, each **power mode** set the following in `/sys/kernel/debug/legion/fancurve`

## Quiet
```
powermode: 1
powermode EC: 2
powermode WMI: 1
```

## Balanced
```
powermode: 2
powermode EC: 0
powermode WMI: 2
```

## Performance
```
powermode: 3
powermode EC: 1
powermode WMI: 3
```

The following options also get changed but their value depends on the CPU/GPU usage:
```
1 fanspeed ACPI
1 fanspeed WMI3
2 fanspeed ACPI
2 fanspeed WMI3
```

MrDuartePT, 2024-06-07:
Thanks @jul1u5 I gonna look in to it.

ryanabx, 2024-09-11:
> The https://github.com/johnfanv2/LenovoLegionLinux/pull/185 will help support new devices. After we finalize that will process to add all the gen 8 and maybe gen 9 devices.

I have a 16APH8, anything I can do to help investigate getting the module to run? I haven't had much luck yet even getting the module to compile, but I'd like to help. Running Fedora Silverblue 40. I don't know how different the 16APH8 is from the 14APH8 but I will run the tests mentioned [here](https://github.com/johnfanv2/LenovoLegionLinux/issues/76#issuecomment-1637196666). Doesn't look like there's a lot of activity on the PR itself, and I'm a rust dev, or else I'd try to provide code help too.

eXWoLL, 2024-10-04:
Any update on the Slim 5 14APH8? I've tried TLP, PowerTop, Auto-cpufreq, even System76-power and none is giving me anything remotely close to the Windows Toolkit/Vantage :(

st0nie, 2026-09-14:
The Slim 5 16APH8 (BIOS M3CN) is in the allowlist (M3CN entry, plus the 82Y9-qualified 0x8227 variant from #503). Please retest with the current release and reopen with dmesg output if it still fails.

closed 2026-09-14 by st0nie

mcarans, 2026-09-14:
@st0nie Does this also fix Legion Slim 5 14APH8 (BIOS version MACN38WW, product 82Y5)?

vladbejenaru, 2026-10-04:
@mcarans it seems not. I also just discovered that my GPU (Legion Slim 5 14APH8 / MACN38WW / 82Y5) is limited at 55W  as it is not on the allow list. This is limiting the performance :( 
In Windows in performance mode hwinfo64 shows the GPU at over 100W when required. This is not achieved in Linux (running on EndeavourOS)