johnfanv2/LenovoLegionLinux #77: [BUG] fancurve-set fails when plug-off and switch to performance

st0nie, 2023-06-27:
**Problem Descripition**
When I try to switch to performance mode with power-profiles-daemon while unplugged, legion-linux.service fails to run

[image]

We should prevent users from using performance-mode with the power unplugged
Is there any way to hide the option of performance-mode in powerprofilesdaemon when unplugged?

st0nie, 2023-06-27:
CC @MrDuartePT

MrDuartePT, 2023-06-27:
@st0nie you are using a nvdia driver version that dosent support the power manager anymore (nvidia disable) i recomend edit the .env file (inside .local/legion_linux or /etc/legion_linux) and disable NVIDIA_GPU=1
Also I thing power-profile-daemon put the gpu to sleep or disable on battery

st0nie, 2023-06-27:
> @st0nie you are using a nvidia driver version that dosent support the power manager anymore (nvidia disable it on update) i recomend edit the .env file (inside .local/legion_linux or /etc/legion_linux) and disable NVIDIA_GPU=1

I see, but that's another issue.

> The performance mode on my system is not available on battery that is not the default behavior to any system that why the service fail. Didn’t exist any performance profile on battery

I mean power-profiles-daemon, not legion fn+q  function. Are you sure that you cannot switch to performance-mode via  power-profiles-daemon?

MrDuartePT, 2023-06-27:
No you can’t is limitation of the hardware power-profile-daemon profile are not the same that legion hardware profiles if i not 
mistaken.

st0nie, 2023-06-27:
> No you can’t is limitation of the hardware power-profile-daemon profile are not the same that legion hardware profiles if i not mistaken.

maybe we can provide an option for user to force switch to balanced-mode after user switching to performance-mode by mistakes

MrDuartePT, 2023-06-27:
Sorry I really confuse I try to change to performance on battery even via the gui and is impossible at least in my system and I using the latest LLL commit

st0nie, 2023-06-28:
> Sorry I really confuse I try to change to performance on battery even via the gui and is impossible at least in my system and I using the latest LLL commit. But I don’t use power profile daemon and really I don’t know how we can add the option to hide that but we can create a acpi rule if is unplugged and in performance change to balance

I use kde desktop environment, its power management feature provides power-profiles-daemon support, with this I can get my power-profile to sync with the system's fn+q, but if I unplug the power, the power-profile still contains the performance mode, but the fn+q provides it does not. This can lead to unintended results. I think we should run powerprofilesctl set balanced when /sys/firmware/acpi/platform_profile is changed and the system is on battery power. i think it would be better to use systemd's user service and .path

st0nie, 2023-06-28:
> Sorry I really confuse I try to change to performance on battery even via the gui and is impossible at least in my system and I using the latest LLL commit. But I don’t use power profile daemon and [...]

I use kde desktop environment, its power management feature provides power-profiles-daemon support, with this I can get my power-profile to sync with the system's fn+q, but if I unplug the power, the power-profile still contains the performance mode, but the fn+q provides it does not. This can lead to unintended results. I think we should run powerprofilesctl set balanced when /sys/firmware/acpi/platform_profile is changed and the system is on battery power. i think it would be better to use systemd's user service and .path

st0nie, 2023-06-28:
seems have been resolved
https://github.com/johnfanv2/LenovoLegionLinux/commit/615fbbf2c1d58bf39fd7b5441640104ed7b63485

closed 2023-06-28 by st0nie

MrDuartePT, 2023-06-28:
It not solve because I don’t update the script let me do that

MrDuartePT, 2023-06-28:
Now it fix 10fcf3d5169eec4102b0db0603dde857a9229a91

st0nie, 2023-06-28:
> Now it fix [10fcf3d](https://github.com/johnfanv2/LenovoLegionLinux/commit/10fcf3d5169eec4102b0db0603dde857a9229a91) If you have smart-live-rebuild on gentoo just run to update the package

ok ,thanks