johnfanv2/LenovoLegionLinux #79: Lenovo Legion 5 15ARH05 Support

shreshthUnderscore, 2023-07-03:
**Problem Descripition**
Proof of the software working with the model

**Model and Debug Info**

Distribution: Ubuntu 22.04.2 LTS
Model name: Lenovo Legion 5 15ARH05
CPU model: AMD Ryzen 5 4600H
GPU model: NVIDIA GTX 1650
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82B5
	Version: Lenovo Legion 5 15ARH05
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82B5_BU_idea_FM_Legion 5 15ARH05
	Family: Legion 5 15ARH05

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: EUCN37WW
	Release Date: 04/14/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 15488 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.37
	Firmware Revision: 1.37

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 66
CPU temperature EC error: 0
CPU temperature EC: 66
CPU temperature ACPI error: 0
CPU temperature ACPI: 66
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 52
GPU temperature EC error: 0
GPU temperature EC: 52
GPU temperature ACPI error: 0
GPU temperature ACPI: 52
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2566
1 fanspeed EC error: 0
1 fanspeed EC: 2566
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2500
1 fanspeed WMI error: 0
1 fanspeed WMI: 2500
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2500
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 2500
2 fanspeed error: 0
2 fanspeed: 2695
2 fanspeed EC error: 0
2 fanspeed EC: 2695
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2600
2 fanspeed WMI error: 0
2 fanspeed WMI: 2600
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 2600
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 2600
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 17
powermode ACPI error: -5
powermode ACPI: 82676736
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 3
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 59	 0	 54	 0	 38
1700	 1600	 2	 2	 55	 59	 51	 54	 35	 42
2000	 1900	 2	 2	 55	 59	 51	 54	 39	 127
2100	 2100	 2	 2	 55	 72	 51	 54	 39	 127
2300	 2300	 2	 2	 67	 76	 51	 54	 39	 127
2600	 2600	 2	 2	 73	 82	 51	 54	 39	 127
2900	 2900	 2	 2	 76	 87	 51	 58	 39	 127
3500	 3500	 2	 2	 83	 92	 55	 66	 39	 127
3800	 3800	 2	 2	 88	 96	 63	 70	 39	 127
4400	 4400	 2	 2	 93	 127	 67	 127	 39	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

johnfanv2, 2023-07-16:
@shreshthUnderscore Thanks alot for the report. This really helps. Is there anything that does not work?

MrDuartePT, 2023-10-19:
@shreshthUnderscore since you didn't report back and everything seems to be working a will close the issue.
Any problems pls reopen.

closed 2023-10-19 by MrDuartePT