johnfanv2/LenovoLegionLinux #80: Get the max speed fans can reach

wznmickey, 2023-07-04:
**Is your feature request related to a problem? Please describe.**
When I setting the fan speed, I do not know what would be the max speed my machine enbale.

**Describe the solution you'd like**
Add a command that can output the max fan speed.

**Describe alternatives you've considered**
Be able to warn if the speed I set is greater than the max

**Model and Debug Info**

Distribution: Manjaro(including version, e.g. Ubuntu 24.04)
Model name: Legion 5 Pro
CPU model: i9 12900 H (e.g. AMD Ryzen 7 5800H)
GPU model: 3060 (e.g NVIDIA RTX 3700)
Keyboard backlight: single color with off/medium/bright  (RGB, single color with off/on, single color with off/medium/bright)
Light in lid or logo: yes (yes/no)
Light at IO-Ports at back: no (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82RF
        Version: Legion 5 Pro 16IAH7H
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: 
        Family: Legion 5 Pro 16IAH7H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: J2CN49WW
        Release Date: 11/25/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.49
        Firmware Revision: 1.49

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 47
CPU temperature EC error: 0
CPU temperature EC: 47
CPU temperature ACPI error: 0
CPU temperature ACPI: 47
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 47
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 38
GPU temperature EC error: 0
GPU temperature EC: 38
GPU temperature ACPI error: 0
GPU temperature ACPI: 38
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 38
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 3517
1 fanspeed EC error: 0
1 fanspeed EC: 3517
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 3500
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 3500
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 3523
2 fanspeed EC error: 0
2 fanspeed EC: 3523
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 3500
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 3500
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 5
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       5       0       40      0       64      0       37
1700     1600    5       5       40      41      59      64      33      42
1800     1800    5       5       41      42      59      64      38      48
3500     3500    5       5       42      43      59      64      43      45
3500     3500    2       2       43      45      59      64      45      50
3500     3500    2       2       45      50      59      64      50      55
3500     3500    2       2       50      60      59      64      55      60
3000     3100    2       2       60      74      59      69      60      75
3400     3500    2       2       70      78      64      75      75      80
3700     3700    2       2       74      84      71      81      80      127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp

```

johnfanv2, 2023-07-16:
Do you know if you can read the maximal possible speed in Windows in Legion Toolkit or Vantage?

wznmickey, 2023-07-17:
No because I have not used this machine in Windows.

wznmickey, 2023-08-01:
I can set the fan speed as max and find the speed from sensors. But I am not sure whether this speed is the max and could contain for a long time.

MrDuartePT, 2023-10-19:
the max speed of the fan that could be sustain is that value rouded to lower that max.
I will give a exemple when you enable maximum fan speed in the gui, to be easy to understand:
```
sensors:
legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           5208 RPM
Fan 2:           5146 RPM
```
In my case the max fan speed is 5200 but you can set to 5100 just to be safe.
Reopen if you have any questions

closed 2023-10-19 by MrDuartePT

wznmickey, 2023-10-19:
自动回复：已收到，谢谢。