johnfanv2/LenovoLegionLinux #82: [BUG] CPU and GPU overclock marked with red

MegPredator, 2023-07-08:
**Problem Descripition**
CPU and GPU overclock are marked in RED, the main page also has the last option of setting the fans to maxiumum(the last option) also marked in RED

**Model and Debug Info**

Distribution: Archlinux (Linux-zen 6.4.2-zen1-1-zen)
Model name: 15ARH05H
CPU model: Ryzen 4600h
GPU model: NVIDIA RTX 2060
Keyboard backlight: single color(white) with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82B1
        Version: Lenovo Legion 5 15ARH05H
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82B1_BU_idea_FM_Legion 5 15ARH05H
        Family: Legion 5 15ARH05H

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected

Output of `sudo dmidecode -t bios`:
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: FSCN26WW
        Release Date: 09/09/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 15488 kB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.26
        Firmware Revision: 1.26

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 41
CPU temperature EC error: 0
CPU temperature EC: 41
CPU temperature ACPI error: 0
CPU temperature ACPI: 41
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 38
GPU temperature EC error: 0
GPU temperature EC: 38
GPU temperature ACPI error: 0
GPU temperature ACPI: 38
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1776
1 fanspeed EC error: 0
1 fanspeed EC: 1776
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 1700
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 1700
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 1685
2 fanspeed EC error: 0
2 fanspeed EC: 1685
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1600
2 fanspeed WMI error: 0
2 fanspeed WMI: 1600
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 1600
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 1600
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 1
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       7       0       72      0       54      0       38
1700     1600    5       7       65      72      51      54      35      42
2000     1900    5       7       65      72      51      54      39      49
2100     2100    5       7       65      72      51      54      45      127
2300     2300    5       7       65      76      51      54      45      127
2600     2600    5       7       73      80      51      54      45      127
2900     2900    5       7       76      87      51      65      45      127
3600     3600    5       7       83      89      61      76      45      127
3800     3800    5       7       86      93      71      79      45      127
4400     4400    5       7       89      127     76      127     45      127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

MrDuartePT, 2023-10-18:
The option is in red since only gen 8 support that features. In the same page of the gui the option are set at 0 just mean your laptop dosent support that feature

MrDuartePT, 2023-10-18:
well is not really bug since in not supported on the hardware (at ec level) I think i need to see with @johnfanv2 instead of showing red or a '0' value just show 'Not Supported*'

* The ec not support this feature you can use another tool to do this

This a kinda a message we can add to the gui

MrDuartePT, 2023-10-18:
Verify if Lenovo Legion Toolkit supported

MrDuartePT, 2023-10-19:
Since is not supported for older model i think i can close this.
Reopen if you have any question

closed 2023-10-19 by MrDuartePT

MrDuartePT, 2023-10-19:
Its probably because you model dont supported