johnfanv2/LenovoLegionLinux #83: [BUG] Noisy ACPI messages in dmesg

camberkenpas, 2023-07-11:
**Problem Descripition**
This happens in dmesg. After first booting, it doesn't happen. But if you put the laptop to sleep and resume, it will start. Or if you don't put it to sleep, it will eventually start on its own. These messages occur about every second. Perhaps this issue goes beyond the scope of this project?
```text
[  587.416970] No Local Variables are initialized for Method [SNTM]

[  587.416971] No Arguments are initialized for method [SNTM]

[  587.416972] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[  587.416975] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[  588.424882] ACPI Error: No handler for Region [RTCM] (0000000007a2f8f3) [SystemCMOS] (20230331/evregion-130)
[  588.424889] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
```

**Model and Debug Info**

Distribution: Kubuntu 23.04
Model name: Lenovo Legion 7i Gen 8 16IRX8H
CPU model: i9-13900HX
GPU model: Nvidia RTX 4090
Keyboard backlight: RGB Multiple colors
Light in lid or logo: No!
Light at IO-Ports at back: No

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82WQ
        Version: Legion Pro 7 16IRX8H
        Serial Number: PF4ERV89
        UUID: 946c879a-5315-4f41-9116-745d2214140e
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
        Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: KWCN36WW
        Release Date: 04/28/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.36
        Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

johnfanv2, 2023-07-11:
Please try to uninstall and unload the legion kernel module. Then do a restart and check if this problem persists. I guess this is a problem in the ACPI firmware in the hardware without any effects for controlling the features related to this software.

camberkenpas, 2023-07-11:
Yep, it's still an issue. Any idea on how I might approach this problem? I suspect you're right that this is a problem with the ACPI firmware. I've tried talking to Lenovo about this. No response yet, but I suspect it may not get anywhere.

```text
[Tue Jul 11 08:57:38 2023] No Local Variables are initialized for Method [SNTM]

[Tue Jul 11 08:57:38 2023] No Arguments are initialized for method [SNTM]

[Tue Jul 11 08:57:38 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[Tue Jul 11 08:57:38 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[Tue Jul 11 08:57:39 2023] ACPI Error: No handler for Region [RTCM] (0000000054c58dd4) [SystemCMOS] (20230331/evregion-130)
[Tue Jul 11 08:57:39 2023] ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)

[Tue Jul 11 08:57:39 2023] No Local Variables are initialized for Method [SNTM]

[Tue Jul 11 08:57:39 2023] No Arguments are initialized for method [SNTM]

[Tue Jul 11 08:57:39 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0.SNTM due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
[Tue Jul 11 08:57:39 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q77 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
```

MrDuartePT, 2023-10-19:
This acpi error are normal is not nothing to do with kernel module
I also have does is something related to the lenovo legion bios, since the legion bios not include the linux mode like the thinkpads.
The acpi error not cause any problem not bother with them.
I gonna close this since is not related to the kernel module

closed 2023-10-19 by MrDuartePT