johnfanv2/LenovoLegionLinux #84: Legion 5 Pro 16ARX8 Support

FranciscoRibeiro03, 2023-07-12:
**Problem Descripition**
`legion-laptop` module loads. EC Chip ID is not 8227 (it's 5507) and all values on the fan curve are 0.

When under stress, the CPU temps rise directly to 100ºC (which I assume is not what is actually happening, I doubt that the CPU is going from ~60ºC to 100ºC in 5 seconds and then back to ~60ºC when I stop the stress test)
Fan speeds look fine
GPU temps look fine, being roughly the same as shown on nvidia-smi
[image]

**Model and Debug Info**

Distribution: Arch Linux with Kernel 6.4.3
Model name: Legion 5 Pro Gen 8 2023 (16ARX8)
CPU model: AMD Ryzen 7 7745HX
GPU model: NVIDIA RTX 4060
Keyboard backlight: RGB with 4 zones
Light in lid or logo: No
Light at IO-Ports at back: No

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WM
	Version: Legion Pro 5 16ARX8
	Serial Number: [redacted]
	UUID: [redacted]
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WM_BU_idea_FM_Legion Pro 5 16ARX8
	Family: Legion Pro 5 16ARX8

Handle 0x0017, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: LPCN42WW
	Release Date: 06/02/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.42
	Firmware Revision: 1.42

Handle 0x0018, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 5
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 52
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 52
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 42
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 42
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2100
1 fanspeed EC error: 0
1 fanspeed EC: 48
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2100
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2100
2 fanspeed error: 0
2 fanspeed: 2200
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2200
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2200
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 16
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100	 0	 0	 0	 0	 0	 0	 0	 0	 0
200	 0	 0	 0	 0	 0	 0	 0	 0	 0
300	 0	 0	 0	 0	 0	 0	 0	 0	 0
400	 0	 0	 0	 0	 0	 0	 0	 0	 0
500	 0	 0	 0	 0	 0	 0	 0	 0	 0
600	 0	 0	 0	 0	 0	 0	 0	 0	 0
700	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100	 0	 0	 0	 0	 0	 0	 0	 0	 0
200	 0	 0	 0	 0	 0	 0	 0	 0	 0
300	 0	 0	 0	 0	 0	 0	 0	 0	 0
400	 0	 0	 0	 0	 0	 0	 0	 0	 0
500	 0	 0	 0	 0	 0	 0	 0	 0	 0
600	 0	 0	 0	 0	 0	 0	 0	 0	 0
700	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
```

johnfanv2, 2023-07-13:
For the fancurve: On newer models there is not yet support to customnize all the details of the fancurve, so it currently uses a new WMI method by lenovo. I guess 1000 = 100% of speed. You can try changing the fancurve in the GUI and report back if it has an effect.

For temperature: My assumption is that the CPU temperature readings might be correct and unfortunately the cooling is not good on your particular model :cry: .  
The temperature comes from an "official" WMI method and I can see from the log that it coincides with the temperature by direct ACPI access (52 is above log). But it might still be wrong.
You can try to compare it with the similar temperature sensor `k10temp-pci-00c3/tcl` (or similar). I would suggest using the GUI tool psensor, which can graph the different temperatures so it easy to compare it. @MrDuartePT Do you know another source to get CPU temperature readings on AMD processor maybe even for each core? What is the easiest way to get it?

MrDuartePT, 2023-10-19:
@johnfanv2 you can't get directly from the thermal zone since legion only expose the cooling_devices (/sys/class/thermal/).

but i found via hwmon:
hwmon2 - its the nvme
hwmon1 - its the apu but the temperature is only two degress less
sys/class/hwmon/hwmon1/temp1_input - cpu (deviated 2º degress of what lenovo driver gives)
hwmon2 - its the power adapter (ADP0)
hwmon4 - its the battery adapter (BAT0)

This is what the driver creates (hwmon3)
/sys/class/hwmon/hwmon3/temp1_input - cpu temperature
/sys/class/hwmon/hwmon3/temp2_input - gpu temperature
/sys/class/hwmon/hwmon3/temp3_input - ic temperature

jgirardet, 2024-01-28:
Is there something which can be done to help to  resolve this  fan curve not supported ?

MrDuartePT, 2024-01-28:
> Is there something which can be done to help to resolve this fan curve not supported ?

Basically the EC method dosent work on gen 8 legion and also try to see with @hhd-dev if the method for legion go also work on gen 8 legion.

https://github.com/hhd-dev/hwinfo/tree/master/devices/legion_go#retrieve-the-fan-curve

gandy92, 2024-03-16:
My Legion is also of model 16ARX8 and the same chip is reported by ```cat /sys/../fancurve```. 
I'm running Ubuntu 23.10, where I installed the acpi-call-dkms module. Following the instructions on how to retrieve the fancurve from above however, I get the following result:
```
echo '\_SB.GZFD.WMAB 0 0x05 0x0000' | sudo tee /proc/acpi/call; sudo cat /proc/acpi/call
\_SB.GZFD.WMAB 0 0x05 0x0000
Error: AE_NOT_FOUND
```
Is there anything I'm missing or that I could do differently?

MrDuartePT, 2024-03-16:
> My Legion is also of model 16ARX8 and the same chip is reported by `cat /sys/../fancurve`. I'm running Ubuntu 23.10, where I installed the acpi-call-dkms module. Following the instructions on how to [...]

Well AE_NOT_FOUND is very strange, mosty probably what should occur is giving a wrong fan table (since the fan id is not the same as the legion go)

gandy92, 2024-03-21:
Thanks for pointing this out - I dumped the distro package and compiled from nix-community/acpi_call (hope that's the right one?) with the original buffer size (4096) and increased buffer size (8192) but the result is still the same.

closed 2026-09-11 by github-actions[bot]