johnfanv2/LenovoLegionLinux #85: [New Model] Y7000p (8th gen Intel)

sakgoyal, 2023-07-15:
**Problem Description**
Add Support for the legion Y7000p (gtx 1060)

**Model and Debug Info**
Model name: Legion Y7000p-1060
Fedora 38 w/ KDE (Official Nobara Distro using plasma on x11)

CPU model: 8750h
GPU model: gtx1060-6gb
Keyboard backlight: white (off, med, bright)
Light in lid or logo: Big white Y logo
Light at IO-Ports at back: No (except small red power charging indicator)
<details> 
  <summary>Output of <code>sudo dmidecode -t system</code>: </summary>

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 81LF
        Version: Lenovo Legion Y7000P-1060
        Serial Number: [redacted]
        UUID: d56f1dfa-b1a5-11e8-b5f5-e86a640e2769
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_81LF_BU_idea_FM_Legion Y7000P-1060
        Family: Legion Y7000P-1060

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0024, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0031, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

```
</details>

<details>
<summary>Output of <code>sudo dmidecode -t bios</code>: </summary>

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: LENOVO
        Version: 9VCN22WW
        Release Date: 11/30/2020
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 10 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.22
        Firmware Revision: 1.22

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```
</details>

- Fn+L does nothing
- `/sys/kernel/debug/legion` directory does not exist 
- `/sys/class/leds/platform\:\:ylogo/` directory does not exist

Log file: [fwts.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12045257/fwts.log)

[image]

johnfanv2, 2023-07-16:
@sakgoyal This seems to be an older model. Do you know the approximate original year of this model. Nevertheless, I just I added your model/BIOS to the kernel module. Please get the new version, recompile, and reload the kernel module (no need to force anymore). 

Then, please test the following features. We have to try a bit to find the correct configuration.

# Temperatures and Fan Speeds
Please read the fan speeds and temperatures (see README) and confirm that they are sensible and change accordingly, e.g. when you put load onto the CPU the CPU temp and the fan speeds should increase, but GPU temp should stay roughly the same.

# Powermode

We will start by confirming that power mode change work **and to get me more debug info**. Please change the power mode with Fn+Q and provide the output of the following commands in **each** mode. 

```bash
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve
```

Then please also check if the power mode is read correctly in the OS (see README). Then please check if you can set the power mode with the GUI (see README) and the LED changes accordingly.

Tests the following features with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software. 

Please provide the log output from the GUI from the log tab for all the tests.

# Fn Lock
- Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
- Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn 

# Win Key
- Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
- Disable Win Key: Win Key has no function anymore

# Touchpad
- Enable Touchpad: Touchpad works
- Disable Touchpad: Touchpad does not work

# Battery Conservation
- Enabled: charging stops if above 60 % (or 80% depending on model)
- Disabled: charging works up to 100%
Note: Sometimes it takes a 10-20 seconds before it takes effect

# Rapid Charging
- Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
- Disabled: Charging is slower
Note: Sometimes it takes a 10-20 seconds before it takes effect .

# Charge Output Always On
- Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
- Disabled: USB outputs are off when laptop is off

# Display Overdrive
- Enabled: display somehow faster
- Disabled: display not faster
Just check if checkbox can be checked or not if you cannot verify that it has an effect.

# Keyboard Backlight

In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:

```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```

# Fan curve
First, please give again the output of 
```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```
Second, please use the GUI and enter custom mode. Do you have a custom mode? Then change the fan curve (currently only the first row for fan 1), write it to hardware, and read it back. Please report back if it works and provide the log output from the GUI and `sudo dmesg | tail -n 70`.

sakgoyal, 2023-07-16:
Its a late 2018 model. It came out sometime around December 2018 +/- a few months. 
when I try to compile the kernel module, I get an error. I did `make clean` as well but it still happened. 
I was able to do just the `make` but when I did `sudo make reloadmodule`, it failed. 
here is the log:  [log.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12065287/log.txt)

the thing that confuses me the most is that I am able to turn on conservation mode by changing the value in 
`/sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode` even though my laptop is clearly not an ideapad model. is this older model just fundamentally different to how it works now?

sakgoyal, 2023-07-16:
I spent some time searching through the arch wikis and looking for ideapad instead of legion. and I found this table here:
[Ideapad gaming 3](https://wiki.archlinux.org/title/Lenovo_IdeaPad_Gaming_3#System_Performance_Mode)
> System Performance Mode
> There are 3 modes available: Intelligent Cooling, Extreme Performance and Battery Saving. To switch between them, press Fn+Q on the IdeaPad's internal keyboard.

I seem to only have 2 modes. auto, and full fan speed. I dont know if it affects the CPU/GPU performance or not. 
[image]
This table also makes me believe fan control is not possible.

johnfanv2, 2023-07-16:
Thank for the log. Your model has a different embedded controller ID. I changed it. It should load now.

The kernel also loads the ideapad driver becaue the Legion and Ideapad models - at least older ones - share some functionality including the "conservation mode" feature.

sakgoyal, 2023-07-17:
There is no LED for the fan status. 
The only LEDs are the caps lock, num lock, power button, big Y logo on the lid, power status led on the right side next to the usb). 

> The kernel also loads the ideapad driver becaue the Legion and Ideapad models - at least older ones - share some functionality including the "conservation mode" feature.

thats really interesting. I guess it makes sense but thats really confusing. 

This time the compile worked. `make` worked as well as `make reloadmodule`. dmsg shows that it was loaded, and the `/sys/kernel/debug/legion` directory also exists now
here are all the logs I recorded. 

`sudo make reloadmodule > log.txt` output:  [log.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12066007/log.txt)
`sudo dmesg > log2.txt` output:  [log2.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12066008/log2.txt)
(default auto fan) `cat /sys/kernel/debug/legion/fancurve > log3.txt` output:  [log3.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12066011/log3.txt)
( max fan speed ) `cat /sys/kernel/debug/legion/fancurve > log4.txt` output: [log4.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12066010/log4.txt)

Thank you for spending so much time for such an old model. I truly appreciate your help. This is a really nice project.

johnfanv2, 2023-07-17:
I updated the code again. It seems like the fancurve is stored at the same places as other old Ideapad laptops. Try to read the fancurve again. Also try to modify it.

sakgoyal, 2023-07-17:
`sudo make reloadmodule > log.txt` output: [log.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074051/log.txt)
`sudo dmesg > log2.txt` output: [log2.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074052/log2.txt)
`sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C > log5.txt` output: [log5.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074058/log5.txt)

### default auto fan
`cat /sys/kernel/debug/legion/fancurve > log3.txt` output: [log3.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074053/log3.txt)
`sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C > log4.txt` output: [log4.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074056/log4.txt)

### max fan speed
`cat /sys/kernel/debug/legion/fancurve > log3_2.txt` output: [log3_2.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074054/log3_2.txt)
`sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C > log4_2.txt` output: [log4_2.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12074057/log4_2.txt)

johnfanv2, 2023-07-18:
Try changing any values (fan speed, temps, ... ) of the fan curve, e.g. by the GUI or from the console (see README). Then give the output of dmesg.

closed 2026-09-11 by github-actions[bot]