johnfanv2/LenovoLegionLinux #86: Lenovo Legion Y9000X IAH7 (2022) Support

YisuiDenghua, 2023-07-16:
Model name: Lenovo Legion Y9000X IAH7
CPU model: 12th Gen Intel i7-12700H (20) @ 4.600GHz
GPU model: NVIDIA GeForce RTX 3060 Mobile / Max-Q
Keyboard backlight: single color with off/medium/bright
System: GNU/Linux x86_64, NixOS 

By default legion-laptop module will not loaded in this model:

```
[    2.584254] legion_laptop starts loading
[    2.584824] legion PNP0C09:00: legion_laptop platform driver probing
[    2.585834] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82TF; DMI_BIOS_VERSION:JYCN34WW
[    2.585839] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[    2.585841] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[    2.585842] legion PNP0C09:00: legion_laptop not loaded for this device
[    2.586469] legion: probe of PNP0C09:00 failed with error -12

```

output of `sudo dmidecode -t system`

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x001F, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: J6H1:1-X Boot with Default; J8H1:1-X BIOS RECOVERY

Handle 0x001B, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82TF
	Version: Legion Y9000X IAH7

	Wake-up Type: Other
	SKU Number: LENOVO_MT_82TF_BU_idea_FM_Legion Y9000X IAH7
	Family: Legion Y9000X IAH7

Handle 0x0060, DMI type 15, 31 bytes
System Event Log
	Area Length: 98 bytes
	Header Start Offset: 0x0000
	Header Length: 16 bytes
	Data Start Offset: 0x0010
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x00F0
	Status: Valid, Not Full
	Change Token: 0x00000005
	Header Format: Type 1
	Supported Log Type Descriptors: 4
	Descriptor 1: POST error
	Data Format 1: POST results bitmap
	Descriptor 2: PCI system error
	Data Format 2: None
	Descriptor 3: System reconfigured
	Data Format 3: None
	Descriptor 4: Log area reset/cleared
	Data Format 4: None

```

Output of `dmidecode -t bios` :

```
[... 147 lines cut ...]
ACPI: SSDT 0x0000000000000000 002A83 (v02 LENOVO PtidDevc 00001000 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT11
Input file SSDT11, Length 0x1F1A (7962) bytes
ACPI: SSDT 0x0000000000000000 001F1A (v02 LENOVO Ther_Rvp 00001000 INTL 20200717)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
External object resolution file           SSDT10
Input file SSDT10, Length 0x6C21 (27681) bytes
ACPI: SSDT 0x0000000000000000 006C21 (v02 LENOVO AdlP_2MR 00001000 INTL 20200717)
Pass 1 parse of [SSDT]
Firmware Error (ACPI): Failure creating named object [\_SB.PC00.PEG2.PEGP._S0W], AE_ALREADY_EXISTS (20220331/dswload-495)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20220331/psobject-372)
Could not parse external ACPI tables, AE_ALREADY_EXISTS

```

[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12046737/acpi_re.zip)

## Gathering WMI info in Windows

[newfile.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12046792/newfile.txt)

_Originally posted by @YisuiDenghua in https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-1635321621_

johnfanv2, 2023-07-16:
@YisuiDenghua 

Is Y9000X the original or Chinese name?

I added your model/BIOS to the kernel module. Please get the new version, recompile, and reload the kernel module (no need to force anymore). 

Then, please test the following features. We have to try a bit to find the correct configuration.

# Temperatures and Fan Speeds
Please read the fan speeds and temperatures (see README) and confirm that they are sensible and change accordingly, e.g. when you put load onto the CPU the CPU temp and the fan speeds should increase, but GPU temp should stay roughly the same.

# Powermode

We will start by confirming that power mode change work **and to get me more debug info**. Please change the power mode with Fn+Q and provide the output of the following commands in **each** mode. 

```bash
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/fancurve
```

Then please also check if the power mode is read correctly in the OS (see README). Then please check if you can set the power mode with the GUI (see README) and the LED changes accordingly.

Tests the following features with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software. 

Please provide the log output from the GUI from the log tab for all the tests.

# Fn Lock
- Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
- Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn 

# Win Key
- Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
- Disable Win Key: Win Key has no function anymore

# Touchpad
- Enable Touchpad: Touchpad works
- Disable Touchpad: Touchpad does not work

# Battery Conservation
- Enabled: charging stops if above 60 % (or 80% depending on model)
- Disabled: charging works up to 100%
Note: Sometimes it takes a 10-20 seconds before it takes effect

# Rapid Charging
- Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
- Disabled: Charging is slower
Note: Sometimes it takes a 10-20 seconds before it takes effect .

# Charge Output Always On
- Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
- Disabled: USB outputs are off when laptop is off

# Display Overdrive
- Enabled: display somehow faster
- Disabled: display not faster
Just check if checkbox can be checked or not if you cannot verify that it has an effect.

# Keyboard Backlight

In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:

```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```

# Fan curve
First, please give again the output of 
```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 30
```
Second, please use the GUI and enter custom mode. Do you have a custom mode (e.g. in Legion Toolkit in Windows)? Then change the fan curve (currently only the first row for fan 1), write it to hardware, and read it back. Please report back if it works and provide the log output from the GUI and `sudo dmesg | tail -n 70`.

YisuiDenghua, 2023-07-17:
> @YisuiDenghua
> 
> Is Y9000X the original or Chinese name?
> 
> I added your model/BIOS to the kernel module. Please get the new version, recompile, and reload the kernel module (no need to force anymore).
> 
> Then, please test the following features. We have to try a bit to find the correct configuration.
> # Temperatures and Fan Speeds
> 
> Please read the fan speeds and temperatures (see README) and confirm that they are sensible and change accordingly, e.g. when you put load onto the CPU the CPU temp and the fan speeds should increase, but GPU temp should stay roughly the same.
> # Powermode
> 
> We will start by confirming that power mode change work **and to get me more debug info**. Please change the power mode with Fn+Q and provide the output of the following commands in **each** mode.
> 
> ```shell
> sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
> sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
> sudo cat /sys/kernel/debug/legion/fancurve
> ```
> 
> Then please also check if the power mode is read correctly in the OS (see README). Then please check if you can set the power mode with the GUI (see README) and the LED changes accordingly.
> 
> Tests the following features with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software.
> 
> Please provide the log output from the GUI from the log tab for all the tests.
> # Fn Lock
> 
>     * Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
> 
>     * Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn
> 
> 
> # Win Key
> 
>     * Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
> 
>     * Disable Win Key: Win Key has no function anymore
> 
> 
> # Touchpad
> 
>     * Enable Touchpad: Touchpad works
> 
>     * Disable Touchpad: Touchpad does not work
> 
> 
> # Battery Conservation
> 
>     * Enabled: charging stops if above 60 % (or 80% depending on model)
> 
>     * Disabled: charging works up to 100%
>       Note: Sometimes it takes a 10-20 seconds before it takes effect
> 
> 
> # Rapid Charging
> 
>     * Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
> 
>     * Disabled: Charging is slower
>       Note: Sometimes it takes a 10-20 seconds before it takes effect .
> 
> 
> # Charge Output Always On
> 
>     * Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
> 
>     * Disabled: USB outputs are off when laptop is off
> 
> 
> # Display Overdrive
> 
>     * Enabled: display somehow faster
> 
>     * Disabled: display not faster
>       Just check if checkbox can be checked or not if you cannot verify that it has an effect.
> 
> 
> # Keyboard Backlight
> 
> In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:
> 
> ```shell
> sudo cat /sys/kernel/debug/legion/fancurve
> sudo dmesg | tail -n 30
> ```
> 
> # Fan curve
> 
> First, please give again the output of
> 
> ```shell
> sudo cat /sys/kernel/debug/legion/fancurve
> sudo dmesg | tail -n 30
> ```
> 
> Second, please use the GUI and enter custom mode. Do you have a custom mode (e.g. in Legion Toolkit in Windows)? Then change the fan curve (currently only the first row for fan 1), write it to hardware, and read it back. Please report back if it works and provide the log output from the GUI and `sudo dmesg | tail -n 70`.

Sorry, I don't know how to use the new version in my system.

johnfanv2, 2023-07-17:
I updated the code. You just have to download the code from github again. Then, compile/build it again (`make`), and load the module (`make reloadmodule`). See the README (https://github.com/johnfanv2/LenovoLegionLinux) for detailed instructions.

YisuiDenghua, 2023-07-17:
> I updated the code. You just have to download the code from github again. Then, compile/build it again (`make`), and load the module (`make reloadmodule`). See the README (https://github.com/johnfanv2/LenovoLegionLinux) for detailed instructions.

Since NixOS doesn't follow FHS, `make` is not usable on it. I'm still trying to package the new version of the driver. But I'm not familiar with packaging a kernel module.

johnfanv2, 2023-07-17:
Maybe you could install make like here https://stackoverflow.com/questions/74649163/how-to-install-the-make-command-in-nixos or here https://www.reddit.com/r/NixOS/comments/s71md4/installing_make/. It is enough to have the files in your home folder and compile it there. There is no need to install the new driver but `insmod` must be possible. It is used inside the makefile.

YisuiDenghua, 2023-07-18:
@johnfanv2 

Thanks. Tested with `make` , no errors found.

Y9000X is it's original name.

**Conclusion : "Charge Output Always On", "Fan curve"**

## Temperatures and Fan Speeds

I tested CPU with `stress --cpu 4 --timeout 60`, then used `psensor` to monitor the temperature. 

CPU temp raised, then I heard the sound of the fan. But on `psensor` , the speeds of the fan didn't raise.

And then I test with the Lenovo Legion Linux GUI, it shows that the speeds of the fan raised.

GPU temp  stay roughly the same.

[image]

[image]

the 2nd image shows the temp and the fan's speeds when I'm putting load onto the CPU.

##Powermode

These are outputs of `sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C`, `sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C` and `sudo cat /sys/kernel/debug/legion/fancurve`

[ballance.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12079817/ballance.txt)

[performance.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12079820/performance.txt)

[quiet.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12079818/quiet.txt)

Then I used the GUI tools to change power mode, it works fine and the LED changed its colour. I also tested it with `powerprofilesctl list`, no problems found.

## Fn Lock

I used the GUI tools to change Fn Lock, it works as expected.

## Win Key

Works. 

## Touchpad

Works.

## Battery Conservation

Works.

## Rapid Charging

Works.

**## Charge Output Always On

Doesn't work.**

Log:

```
INFO:legion_linux.legion:Click on UI Charge Output from USB always on element for AlwaysOnUSBChargingFeature
Error occured 
Traceback (most recent call last):
  File "/home/yisui/LenovoLegionLinux/python/legion_linux/legion_linux/legion_gui.py", line 185, in update_feature_from_view
    self.feature.set(gui_value)
  File "/home/yisui/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 452, in set
    raise NotImplementedError()
[... 109 lines cut ...]
[38151.567197] legion_laptop: Set fancurve
[38151.628454] legion_laptop: Set fancurve
[38151.689694] legion_laptop: Ignoring invalid fancurve value 0 for attribute 9 at point 7
[38151.719726] legion_laptop: Ignoring invalid fancurve value 0 for attribute 10 at point 7
[38151.749809] legion_laptop: Set fancurve
[38151.811010] legion_laptop: Set fancurve
[38151.872210] legion_laptop: Set fancurve
[38151.933414] legion_laptop: Set fancurve
[38151.994804] legion_laptop: Set fancurve
[38152.056182] legion_laptop: Set fancurve
[38152.117437] legion_laptop: Set fancurve
[38152.178684] legion_laptop: Set fancurve
[38152.239918] legion_laptop: Ignoring invalid fancurve value 0 for attribute 9 at point 8
[38152.270010] legion_laptop: Ignoring invalid fancurve value 0 for attribute 10 at point 8
[38152.300060] legion_laptop: Set fancurve
[38152.361306] legion_laptop: Set fancurve
[38152.422512] legion_laptop: Set fancurve
[38152.483717] legion_laptop: Set fancurve
[38152.544998] legion_laptop: Set fancurve
[38152.606278] legion_laptop: Set fancurve
[38152.667540] legion_laptop: Set fancurve
[38152.728813] legion_laptop: Set fancurve
[38152.790004] legion_laptop: Ignoring invalid fancurve value 0 for attribute 9 at point 9
[38152.820121] legion_laptop: Ignoring invalid fancurve value 0 for attribute 10 at point 9
```

mxdh, 2023-08-31:
> 

IMO, Y9000X is the Chinese name.

When I enter the serial number of my laptop on pcsupport.lenovo.com, I get this link: [,](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/legion-series/legion-s7-16iah7/82tf/82tf000ccd) so I think its international name is Legion S7 16IAH7 or Legion Slim 7i Gen 7.

MrDuartePT, 2023-10-20:
> @johnfanv2
>
> Thanks. Tested with `make` , no errors found.
>
> Y9000X is it's original name.
>
> **Conclusion : "Charge Output Always On", "Fan curve"**
> ## Temperatures and Fan Speeds
>
> I tested CPU with [...]

It look you introduce some zeros in the values try read from hardware after fn+q and change for exemple on value, record to hadware and verify the value apply

Xiue233, 2024-08-27:
For this device, I have uploaded all my info in both linux and windows. [Testing new features on 2022 models [looking for testers] #2263193416](https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-2263193416)

I hope it is helpful. Until today, this module doesn't work properly on my Y9000X.

@johnfanv2 Thank you so much.

closed 2026-09-11 by github-actions[bot]