johnfanv2/LenovoLegionLinux #87: Add Support: 2022 Lenovo Yoga 7 - 82UF - 16IAH7 (BIOS J1CN38WW)

tubiakou, 2023-07-17:
## Model Summary
- Model Name: `82UF Yoga 7 16IAH7 (2022)`
- BIOS: ` J1CN38WW`
- CPU Model:    `12th Gen Intel i7-12700H (20) @ 4.600GHz`
- GPU Model:    `GPU 0:  Intel Alder Lake-P integrated Graphics   GPU 1: Intel DG2 Arc A370M`
- Keyboard Backlight:   `Single Colour (white) with off/low/high brightness`
- OS:  `Fedora Linux 38 (Thirty Eight) x86_64`
- Kernel: `6.4.3-0.rc2.250.vanilla.fc38.x86_64`

---

## Build Log with Errors
Any guidance on how to get the `sudo make forcereloadmodule` to complete successfully would be greatly appreciated.
Note that the build was done against a 6.4.3 RC2 Kernel.  However similiar build issues occurred against a Fedora stock 6.3.12 kernel.

```
$ make
make -C /lib/modules/6.4.3-0.rc2.250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.3-0.rc2.250.vanilla.fc38.x86_64'
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.3-0.rc2.250.vanilla.fc38.x86_64'
 

$ sudo make forcereloadmodule
rmmod legion-laptop.ko || true
insmod legion-laptop.ko force=1
dmesg --ctime

   <whole bunch of dmesg output>

[Tue Jul 11 01:50:03 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:03 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:03 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:03 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:03 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:03 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:04 2023] ACPI BIOS Error (bug):
 Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:04 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:04 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:05 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:05 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:05 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:07 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:07 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:07 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:08 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:08 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:08 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:09 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:09 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:09 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:10 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:10 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:10 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:10 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:10 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:10 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:12 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:12 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:12 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:13 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Tue Jul 11 01:50:13 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[Tue Jul 11 01:50:13 2023] ideapad_acpi VPC2004:00: Unknown event: 12
[Tue Jul 11 01:50:14 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[... 133 lines cut ...]
Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```
---

## Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```
---
## Uploaded Files
[acpi_dsl.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010367/acpi_dsl.zip)
[fwts_wmi.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010368/fwts_wmi.zip)
[windows_wmi.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12010369/windows_wmi.zip)

johnfanv2, 2023-07-17:
Interestingly it fails in the step `Creating platform profile failed`. It is a step that should be independent of the model. I added more debug output to the code. Please get the new code, compile it, and try to force-load it again and give the end of the dmesg output again (like above).

tubiakou, 2023-07-17:
Extremely quick response - thank you very much.

## Build Log (against commit 9d91706a1587d235e5a477d1e35d54e1f7ba88be)
```bash
$ cd /var/data/src/LenovoLegionLinux/

$ git pull
remote: Enumerating objects: 62, done.
remote: Counting objects: 100% (62/62), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 54 (delta 39), reused 40 (delta 25), pack-reused 0
Unpacking objects: 100% (54/54), 5.36 KiB | 152.00 KiB/s, done.
From https://github.com/johnfanv2/LenovoLegionLinux
   4f8b030..9d91706  main       -> origin/main
Updating 4f8b030..9d91706
Fast-forward
 kernel_module/legion-laptop.c                  | 182 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------
 python/legion_linux/legion_linux/legion.py     |  58 +++++++++++++++++++++++++++++++++++++--------------------
 python/legion_linux/legion_linux/legion_gui.py |  69 +++++++++++++++++++++++++++++++++++++-------------------------------
 3 files changed, 227 insertions(+), 82 deletions(-)

$ cd kernel_module/
$ make clean
make -C /lib/modules/6.4.4-0.rc1.250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module clean
make[1]: Entering directory '/usr/src/kernels/6.4.4-0.rc1.250.vanilla.fc38.x86_64'
  CLEAN   /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
make[1]: Leaving directory '/usr/src/kernels/6.4.4-0.rc1.250.vanilla.fc38.x86_64'

$ make
make -C /lib/modules/6.4.4-0.rc1.250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.4-0.rc1.250.vanilla.fc38.x86_64'
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.4-0.rc1.250.vanilla.fc38.x86_64'

$ sudo make forcereloadmodule
rmmod legion-laptop.ko || true
insmod legion-laptop.ko force=1
dmesg --ctime

                (bunch of prior info)
                
[Mon Jul 17 17:41:35 2023] legion_laptop: legion_laptop exit
[Mon Jul 17 17:41:35 2023] legion_laptop: legion_laptop starts loading
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UF; DMI_BIOS_VERSION:J1CN38WW
[Mon Jul 17 17:41:35 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UF; DMI_BIOS_VERSION:J1CN38WW
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: legion_laptop is forced to load.
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: Using configuration for system: GKCN
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: ACPI CFG: 2081284372
[Mon Jul 17 17:41:35 2023] legion_laptop: Creating RAM access to embedded controller
[Mon Jul 17 17:41:35 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x0000000043cff4c9
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: Read embedded controller ID 0x8227
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: Creating debugfs inteface
[Mon Jul 17 17:41:35 2023] legion_laptop: Creating sysfs inteface
[Mon Jul 17 17:41:35 2023] legion_laptop: Creating hwmon interface
[Mon Jul 17 17:41:35 2023] legion_laptop: Creating platform profile support
[Mon Jul 17 17:41:35 2023] legion PNP0C09:00: Creating platform profile failed: -17
[Mon Jul 17 17:41:35 2023] legion_laptop: Unloading legion hwon
[Mon Jul 17 17:41:35 2023] legion_laptop: Unloading legion hwon done
[Mon Jul 17 17:41:35 2023] legion_laptop: Unloading legion sysfs
[Mon Jul 17 17:41:35 2023] legion_laptop: Unloading legion sysfs done
[Mon Jul 17 17:41:35 2023] legion_laptop: Unloading legion dubugfs
[Mon Jul 17 17:41:35 2023] legion_laptop: Unloading legion dubugfs done
[... 20 lines cut ...]
[    1.614224] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input2
[    2.602768] nvme 0000:01:00.0: platform quirk: setting simple suspend
[    5.783044] input: PC Speaker as /devices/platform/pcspkr/input/input30
[    5.831583] input: Lenovo Yoga Tablet Mode Control switch as /devices/platform/PNP0C14:01/wmi_bus/wmi_bus-PNP0C14:01/06129D99-6083-4164-81AD-F092F9D773A6/input/input31
[    5.956367] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    6.090353] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[15563.130967] legion PNP0C09:00: legion_laptop platform driver probing
[15563.131591] legion_laptop: Creating platform profile support
[15563.131592] legion PNP0C09:00: Creating platform profile failed
[15760.566008] legion PNP0C09:00: legion_laptop platform driver probing
[15760.566650] legion_laptop: Creating platform profile support
[15760.566651] legion PNP0C09:00: Creating platform profile failed: -17
$
```
---

## Output of `sudo dmesg | grep profile`
```bash
$ sudo dmesg | grep profile
[15563.131591] legion_laptop: Creating platform profile support
[15563.131592] legion PNP0C09:00: Creating platform profile failed
[15760.566650] legion_laptop: Creating platform profile support
[15760.566651] legion PNP0C09:00: Creating platform profile failed: -17
$
```

tubiakou, 2023-08-05:
The above-mentioned failures continue to occur with the latest commit, 2e47eaf. Can you please suggest any troubleshooting ideas or tricks that might reveal what is going on here?

johnfanv2, 2023-08-05:
We have to find out why there is already a platform profile support loaded. In the meantime, get the newest code and load it with

```bash
sudo make reloadmodule_disableplatformprofile 
```
or set the kernel parameter legion_laptop.enable_platformprofile=0. This skips loading the platform profile but loads all other features. So you can do everything except reading and writing the power mode from software.

What is the output of ?
```bash
cat /sys/firmware/acpi/platform_profile_choices
cat /sys/firmware/acpi/platform_profile
```
Can you find out by what module or kernel feature `/sys/firmware/acpi/platform_profile` is created?

tubiakou, 2023-08-06:
Grabbed commit 2412105 and rebuilt against it.  Found that doing `sudo make reloadmodule_disableplatformprofile` had similar errors to before, e.g. _"**Module not useable for this laptop because it is not in allowlist.   ...   legion_laptop not loaded for this device.**_

So I hacked the makefile to include a new target `forcereloadmodule_disableplatformprofile` that simply included the `force=1` option.  With that, my make process looked like the following:

## Output of `make`
```bash
[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $ make
make -C /lib/modules/6.4.8-250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.8-250.vanilla.fc38.x86_64'
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.8-250.vanilla.fc38.x86_64'
[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $
```

## Output of `sudo make forcereloadmodule_disableplatformprofile`
[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $ sudo make forcereloadmodule_disableplatformprofile
rmmod legion-laptop.ko || true
insmod legion-laptop.ko enable_platformprofile=0 force=1
dmesg --ctime

    ... bunch of pre-compilation stuff...

[Sun Aug  6 19:22:34 2023] legion_laptop: legion_laptop exit
[Sun Aug  6 19:22:34 2023] legion_laptop: legion_laptop starts loading
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UF; DMI_BIOS_VERSION:J1CN38WW
[Sun Aug  6 19:22:34 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82UF; DMI_BIOS_VERSION:J1CN38WW
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: legion_laptop is forced to load.
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Using configuration for system: GKCN
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: ACPI CFG: 2081284372
[Sun Aug  6 19:22:34 2023] legion_laptop: Creating RAM access to embedded controller
[Sun Aug  6 19:22:34 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x00000000bb4358ed
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Read embedded controller ID 0x8227
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Creating debugfs inteface
[Sun Aug  6 19:22:34 2023] legion_laptop: Creating sysfs inteface
[Sun Aug  6 19:22:34 2023] legion_laptop: Creating hwmon interface
[Sun Aug  6 19:22:34 2023] legion_laptop: Creating platform profile support
[Sun Aug  6 19:22:34 2023] legion_laptop: Skipping creating platform profile support because enable_platformprofile is false
[Sun Aug  6 19:22:34 2023] legion_laptop: Init WMI driver support
[Sun Aug  6 19:22:34 2023] legion_laptop: Init keyboard backlight LED driver
[Sun Aug  6 19:22:34 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[Sun Aug  6 19:22:34 2023] legion_laptop: Error for WMI method call to get brightness
[Sun Aug  6 19:22:34 2023] legion_laptop: Error reading keyboard brighntess
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[Sun Aug  6 19:22:34 2023] legion_laptop: Init Y-Logo LED driver
[Sun Aug  6 19:22:34 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[Sun Aug  6 19:22:34 2023] legion_laptop: Error for WMI method call to get brightness
[Sun Aug  6 19:22:34 2023] legion_laptop: Error reading brighntess for light: 3
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Init Y-Logo LED driver failed. Skipping ...
[Sun Aug  6 19:22:34 2023] legion_laptop: Init IO-Port LED driver
[Sun Aug  6 19:22:34 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[Sun Aug  6 19:22:34 2023] legion_laptop: Error for WMI method call to get brightness
[Sun Aug  6 19:22:34 2023] legion_laptop: Error reading brighntess for light: 5
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[Sun Aug  6 19:22:34 2023] legion PNP0C09:00: legion_laptop loaded for this device
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here'
[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $

## Output of `cat/sys/firmware/acpi/...`

```bash
[srcrw@maxime:~] $ cat /sys/firmware/acpi/platform_profile_choices
low-power balanced performance
[srcrw@maxime:~] $ cat /sys/firmware/acpi/platform_profile
balanced
[srcrw@maxime:~] $
```

[... cut ...] to ensure things like **zcfan** and **thinkfan** were no longer present.

## Initial Testing
I have started in on the _**Initial Usage Testing**_ section of the readme.  I do seem to be able to get current fanspeeds etc. from `/sys/kernel/debug/legion/fancurve` so that's a huge improvement.  I also seem able to adjust current fan-speeds with things like `echo 1500 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm2_auto_point2_pwm` - again a very welcome improvement.

johnfanv2, 2023-08-07:
Could you give the output of `lsmod`? You could try temporarily unloading the module `ideapad_laptop` for testing. You will lose some features like setting conservation mode of the battery but it will be just for testing. It might enable the platform_profile on some models and with some kernel parameters (allow_v4_dytc).

tubiakou, 2023-08-07:
Please see below the output of `lsmod` _**before**_ unloading `idea_laptop`.  At this point, I can confirm the presence of `/sys/firmware/acpi/platform_profile*`.

After unloading the module with `sudo modprobe -r ideapad_laptop`, it no longer appears in `lsmod`.  Also, `/sys/firmware/acpi/platform_profile*` are now missing.

## Output of `lsmod`
```bash
Module                  Size  Used by
legion_laptop         147456  0
uinput                 20480  1
rfcomm                102400  4
snd_seq_dummy          12288  0
snd_hrtimer            12288  1
tun                    69632  2
nf_tables             356352  0
nfnetlink              20480  1 nf_tables
qrtr                   57344  4
bnep                   36864  2
sunrpc                827392  1
binfmt_misc            28672  1
snd_ctl_led            24576  0
snd_soc_skl_hda_dsp    24576  5
snd_soc_intel_hda_dsp_common    16384  1 snd_soc_skl_hda_dsp
snd_soc_hdac_hdmi      45056  1 snd_soc_skl_hda_dsp
snd_sof_probes         24576  0
snd_hda_codec_hdmi     94208  1
snd_hda_codec_realtek   192512  1
snd_hda_codec_generic   110592  1 snd_hda_codec_realtek
ledtrig_audio          12288  2 snd_ctl_led,snd_hda_codec_generic
snd_soc_dmic           12288  1
snd_sof_pci_intel_tgl    12288  0
iwlmvm                688128  0
snd_sof_intel_hda_common   229376  1 snd_sof_pci_intel_tgl
soundwire_intel        61440  1 snd_sof_intel_hda_common
soundwire_cadence      45056  1 soundwire_intel
snd_sof_intel_hda_mlink    36864  1 snd_sof_intel_hda_common
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
snd_sof_pci            24576  2 snd_sof_intel_hda_common,snd_sof_pci_intel_tgl
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_common
snd_sof               409600  4 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_probes,snd_sof_intel_hda
mac80211             1527808  1 iwlmvm
snd_sof_utils          16384  1 snd_sof
snd_soc_hdac_hda       28672  1 snd_sof_intel_hda_common
intel_uncore_frequency    12288  0
snd_hda_ext_core       36864  5 snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda_mlink,snd_sof_intel_hda
intel_uncore_frequency_common    12288  1 intel_uncore_frequency
intel_tcc_cooling      12288  0
snd_soc_acpi_intel_match    81920  2 snd_sof_intel_hda_common,snd_sof_pci_intel_tgl
x86_pkg_temp_thermal    16384  0
intel_powerclamp       20480  0
snd_soc_acpi           12288  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_common
soundwire_generic_allocation    12288  1 soundwire_intel
soundwire_bus         135168  3 soundwire_intel,soundwire_generic_allocation,soundwire_cadence
snd_soc_core          438272  8 soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_probes,snd_soc_dmic,snd_soc_skl_hda_dsp
coretemp               16384  0
kvm_intel             454656  0
snd_compress           28672  2 snd_soc_core,snd_sof_probes
libarc4                12288  1 mac80211
ac97_bus               12288  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_hda_intel          61440  0
vfat                   20480  1
fat                   102400  1 vfat
snd_intel_dspcfg       32768  3 snd_hda_intel,snd_sof,snd_sof_intel_hda_common
snd_intel_sdw_acpi     16384  2 snd_sof_intel_hda_common,snd_intel_dspcfg
snd_hda_codec         212992  8 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek,snd_soc_intel_hda_dsp_common,snd_soc_hdac_hda,snd_sof_intel_hda,snd_soc_skl_hda_dsp
kvm                  1318912  1 kvm_intel
snd_hda_core          139264  11 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_hda_codec_realtek,snd_soc_intel_hda_dsp_common,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda
snd_hwdep              20480  1 snd_hda_codec
snd_seq               106496  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
[... 112 lines cut ...]
dm_multipath           45056  0
```

## Output of make
Following the rmmod above, I reattempted to clean and make the module again, Different output - unknown symbol in module:
```bash
[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $ make
make -C /lib/modules/6.4.8-250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.8-250.vanilla.fc38.x86_64'
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.8-250.vanilla.fc38.x86_64'

[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $ sudo make forcereloadmodule
rmmod legion-laptop.ko || true
rmmod: ERROR: Module legion_laptop is not currently loaded
insmod legion-laptop.ko force=1
insmod: ERROR: could not insert module legion-laptop.ko: Unknown symbol in module
make: *** [Makefile:47: forcereloadmodule] Error 1
[srcrw@maxime:/var/data/src/LenovoLegionLinux/kernel_module] $
```

johnfanv2, 2023-08-15:
Yes, you should now be able to load the module without forcing it but still with disabling platform profile, i.e. with `make reloadmodule_disableplatformprofile`. I have not tried yet replicating the symbol error you get when you disable ideapad_laptop.

I guess that the best solution would be to manage powermode/platform_profile with legion_laptop and disable it ideapad_laptop. You can maybe try to disable it by setting the kernel parameter allow_v4_dytc in the ideapad_laptop: https://github.com/torvalds/linux/blob/91aa6c412d7f85e48aead7b00a7d9e91f5cf5863/drivers/platform/x86/ideapad-laptop.c#L135

If this does not work, I could export the power mode as another sysfs attribute that is used be the GUI.

If you just use ideapad_laptop without legion_laptop, does the platform_profile file changes if you change it with Fn+Q?

johnfanv2, 2023-08-15:
I tried to troubleshoot the "Unknown symbol in module" error. I made all functions static. Please try to get this error with the new code and provide the tail output of `dmesg` after you get this error. This should show the unknown symbol

tubiakou, 2023-08-17:
Thanks for the continued support - very much appreciated.

I updated the build environment slightly to:

Kernel:  **_Linux  6.4.11-250.vanilla.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC_**
LenovoLegionLinux:  commit f0d6d2a4fc9877f935dbb03eb1fc1b4392b8c428 (current as of 2023-08-17 12:00EDT)

No longer seeing `Unknown symbol`.  However I still need to force the module.  See build-logs below.

### With `ideapad_laptop` loaded
Most Fn-Function keycombos do their function, but Fn-Q does nothing (at least, `/sys/firmware/acpi/platform_profile` shows no change).                                                                  

### With `ideapad_laptop` unloaded:
Fn-Q apparently does nothing (no way to tell).

### With `ideapad_laptop` unloaded & `legion_laptop loaded`:
Unable to tell because of the following:
```bash
ls -l /sys/firmware/acpi/plat*
-rw-r--r-- 1 root root 4096 Aug 17 12:12 /sys/firmware/acpi/platform_profile
-r--r--r-- 1 root root 4096 Aug 17 12:12 /sys/firmware/acpi/platform_profile_choices
  
$ cat /sys/firmware/acpi/platform_profile_choices
quiet balanced balanced-performance performance

$ cat /sys/firmware/acpi/platform_profile
cat: /sys/firmware/acpi/platform_profile: Invalid argument
```

## Output of `make clean && make`
```bash
$ make clean && make
make -C /lib/modules/6.4.11-250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module clean
make[1]: Entering directory '/usr/src/kernels/6.4.11-250.vanilla.fc38.x86_64'
  CLEAN   /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
make[1]: Leaving directory '/usr/src/kernels/6.4.11-250.vanilla.fc38.x86_64'
make -C /lib/modules/6.4.11-250.vanilla.fc38.x86_64/build M=/var/data/src/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.11-250.vanilla.fc38.x86_64'
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /var/data/src/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /var/data/src/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.11-250.vanilla.fc38.x86_64'
$
```

## Output of `make reloadmodule_disableplatformprofile`                                                            
```bash
$ sudo make reloadmodule_disableplatformprofile
rmmod legion-laptop.ko || true
insmod legion-laptop.ko enable_platformprofile=0
dmesg --ctime
[Thu Aug 17 12:06:28 2023] legion_laptop: Set powermode
[Thu Aug 17 12:06:29 2023] legion_laptop: Set powermode
[Thu Aug 17 12:06:29 2023] legion_laptop: Unexpected power mode value ignored: 0
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion hwon
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion hwon done
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion sysfs
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion sysfs done
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion dubugfs
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion dubugfs done
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion ecram
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion ecram done
[Thu Aug 17 12:06:29 2023] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x00000000108c694e
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion shared
[Thu Aug 17 12:06:29 2023] legion_laptop: Unloading legion shared done
[Thu Aug 17 12:06:29 2023] legion_laptop: Legion platform unloaded
[Thu Aug 17 12:06:29 2023] legion_laptop: legion_laptop exit
[... 35 lines cut ...]
[Thu Aug 17 12:08:30 2023] legion_laptop: Creating hwmon interface
[Thu Aug 17 12:08:30 2023] legion_laptop: Creating platform profile support
[Thu Aug 17 12:08:30 2023] legion_laptop: Skipping creating platform profile support because enable_platformprofile is false
[Thu Aug 17 12:08:30 2023] legion_laptop: Init WMI driver support
[Thu Aug 17 12:08:30 2023] legion_laptop: Init keyboard backlight LED driver
[Thu Aug 17 12:08:30 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[Thu Aug 17 12:08:30 2023] legion_laptop: Error for WMI method call to get brightness
[Thu Aug 17 12:08:30 2023] legion_laptop: Error reading keyboard brighntess
[Thu Aug 17 12:08:30 2023] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[Thu Aug 17 12:08:30 2023] legion_laptop: Init Y-Logo LED driver
[Thu Aug 17 12:08:30 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[Thu Aug 17 12:08:30 2023] legion_laptop: Error for WMI method call to get brightness
[Thu Aug 17 12:08:30 2023] legion_laptop: Error reading brighntess for light: 3
[Thu Aug 17 12:08:30 2023] legion PNP0C09:00: Init Y-Logo LED driver failed. Skipping ...
[Thu Aug 17 12:08:30 2023] legion_laptop: Init IO-Port LED driver
[Thu Aug 17 12:08:30 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[Thu Aug 17 12:08:30 2023] legion_laptop: Error for WMI method call to get brightness
[Thu Aug 17 12:08:30 2023] legion_laptop: Error reading brighntess for light: 5
[Thu Aug 17 12:08:30 2023] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[Thu Aug 17 12:08:30 2023] legion PNP0C09:00: legion_laptop loaded for this device
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here'   
$ 
```

johnfanv2, 2023-08-17:
Could you please change with Fn+Q into different modes (LED color changes)? Then in each mode please output:
```bash
sudo cat /sys/firmware/acpi/platform_profile
sudo cat /sys/kernel/debug/legion/fancurve
sudo dmesg | tail -n 400
```

johnfanv2, 2023-08-17:
Please also provide the output of
```bash
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
```

tubiakou, 2023-08-18:
Not sure if this is expected behaviour or not, so I'll describe it before showing several lengthy dmesg spews:

Doing `sudo make forcereloadmodule_disableplatformprofile` with `ideapad_laptop` still present loads the legion module.  Fn-Q keypresses are detected, but do not change `/sys/firmware/acpi/platform_profile` contents - they remain `balanced`.  However dmesg registers the following output, once for each Fn-Q keypress:
```bash
[Fri Aug 18 00:07:23 2023] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[Fri Aug 18 00:07:23 2023] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
```
Unloading `ideapad_laptop` causes `/sys/firmware/acpi/platform_profile` to disappear, but Fn-Q keypresses continue to emit the above 2 lines to dmesg.

After rebooting the laptop, if I first unload `ideapad_laptop` before doing `sudo make forcereloadmodule_disableplatformprofile`, the insmod fails with:
```bash
$ sudo `make forcereloadmodule_disableplatformprofile`
rmmod legion-laptop.ko || true
rmmod: ERROR: Module legion_laptop is not currently loaded
insmod legion-laptop.ko enable_platformprofile=0 force=1
insmod: ERROR: could not insert module legion-laptop.ko: Unknown symbol in module
make: *** [Makefile:67: forcereloadmodule_disableplatformprofile] Error 1
```
This also happens even if I first clean/remake the `kernel_module` directory.

---

## Output after `sudo make forcereloadmodule_disableplatformprofile` (with ideapad_laptop module still loaded)
### Note: This output happened with `ideapad_laptop` loaded, and was after pressing Fn-Q 2 times
```bash
$ sudo cat /sys/firmware/acpi/platform_profile; sudo cat /sys/kernel/debug/legion/fancurve; sudo dmesg | tail -n 400
balanced
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081284372
temperature access method: 1
CPU temperature error: 0
CPU temperature: 49
CPU temperature EC error: 0
CPU temperature EC: 49
CPU temperature ACPI error: 0
CPU temperature ACPI: 49
CPU temperature WMI error: -14
CPU temperature WMI: 49
CPU temperature WMI2 error: -14
CPU temperature WMI2: 49
CPU temperature WMI3 error: -14
CPU temperature WMI3: 49
GPU temperature error: 0
GPU temperature: 29
GPU temperature EC error: 0
GPU temperature EC: 29
GPU temperature ACPI error: 0
GPU temperature ACPI: 29
GPU temperature WMI error: -14
GPU temperature WMI: 29
GPU temperature WMI2 error: -14
GPU temperature WMI2: 29
GPU temperature WMI3 error: -14
GPU temperature WMI3: 29
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2681
1 fanspeed EC error: 0
1 fanspeed EC: 2681
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2600
1 fanspeed WMI error: -14
1 fanspeed WMI: 2600
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2600
1 fanspeed WMI3 error: -14
[... 1020 lines cut ...]
[   96.922414] legion_laptop: Unloading legion dubugfs
[   96.922427] legion_laptop: Unloading legion dubugfs done
[   96.922428] legion_laptop: Unloading legion ecram
[   96.922429] legion_laptop: Unloading legion ecram done
[   96.922429] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x00000000e5ef5850
[   96.922435] legion_laptop: Unloading legion shared
[   96.922436] legion_laptop: Unloading legion shared done
[   96.922437] legion PNP0C09:00: legion_laptop not loaded for this device
[   96.922767] legion: probe of PNP0C09:00 failed with error -17
[  344.197567] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  344.197635] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
[  345.294720] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.EC0._Q44.WM00], AE_NOT_FOUND (20230331/psargs-330)
[  345.294759] ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q44 due to previous error (AE_NOT_FOUND) (20230331/psparse-529)
$
```

## Output of EC memory hexdumps
```bash
$ sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
cat: /sys/kernel/debug/legion/ecmemory: No such file or directory

$ sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
cat: /sys/kernel/debug/legion/ecmemoryram: No such file or directory
$
```

johnfanv2, 2023-08-18:
The driver was intended for Legion laptops and not Yoga laptops. Your ACPI dissassembly (zip files) shows that the ACPI firmware and the WMI interface is different. However, the output of `/sys/kernel/debug/legion/fancurve` and `ecmemory` shows that the embedded controller is (almost) the same as in the Legion laptops and that reading temperatures, fanspeeds, powermode (but not outputting via platform_profile) already works. Hopefully, these are the most important ones. What other features are important for you? 

I am confident we can make it work after solving the problems with ideapad_laptop. I have think about it a bit. In the meantime, you try finding the reason for `Unknown symbol in module`. To this end, try to the error. Immediately, after you see the error
```
insmod: ERROR: could not insert module legion-laptop.ko: Unknown symbol in module
```
run `sudo dmesg`. Hopefully, in the last lines there should be details about what symbol is unknow. Please provide these lines.

tubiakou, 2023-08-18:
Speaking purely for myself, the only thing I really need is configurable  fan-control functionality.  Getting it to a point where it takes effect automatically and reliably at boot-time is icing on the cake.  The rest of the driver's functionality are quite nice-to-haves, but not critical.  The only Fn keys I really need are the volume/mute controls, I don't really care about backlight controls, and I don't frequently need to toggle power-modes - this  laptop isn't used for gaming.

Mainly, I want to ensure I'm not blowing out internal components with things like `make -j$(nproc)`.  However, if time and inclination permit you to evolve the Yoga support for this driver, I would very, very-happily serve as a testing-platform.  In the meantime I will try and educate myself about `ideapad_laptop` and the effects that its presence/absence has, specifically wrt the `unknown symbol` documented above.

Thanks very much for your efforts thus far in getting the driver to load and manage the fans!

closed 2026-09-11 by github-actions[bot]