johnfanv2/LenovoLegionLinux #88: Add support: Legion 5 Pro 16ITH6

aligoldenhat, 2023-07-27:
**Model and Debug Info**

Distribution: Fedora Linux 38 (Workstation Edition) x86_64
Model name: 82JF Legion 5 Pro 16ITH6
CPU model: 11th Gen Intel i7-11800H (16) @ 4.600GHz 
GPU model: NVIDIA GeForce RTX 3050 Mobile 
Keyboard backlight: 4 zone rgb 
Light in lid or logo: yes 
Light at IO-Ports at back: no
bios: H1CN53WW

in "Build and Test Instruction" i get this error in log  sudo make reloadmodule
```text
$ make                              
make -C /lib/modules/6.4.4-200.fc38.x86_64/build M=/home/ali/code/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.4.4-200.fc38.x86_64'
make[1]: Leaving directory '/usr/src/kernels/6.4.4-200.fc38.x86_64'

$ sudo make reloadmodule 
rmmod legion-laptop.ko || true
insmod legion-laptop.ko
dmesg --ctime
[Thu Jul 27 09:00:57 2023] microcode: updated early: 0x42 -> 0x44, date = 2022-12-28
[Thu Jul 27 09:00:57 2023] Linux version 6.4.4-200.fc38.x86_64 (mockbuild@2aaadfeb41d049ad87820b2b55248688) (gcc (GCC) 13.1.1 20230614 (Red Hat 13.1.1-4), GNU ld version 2.39-9.fc38) #1 SMP PREEMPT_DYNAMIC Wed Jul 19 16:32:49 UTC 2023
[Thu Jul 27 09:00:57 2023] Command line: BOOT_IMAGE=(hd0,gpt2)/vmlinuz-6.4.4-200.fc38.x86_64 root=UUID=c1ea2c13-c8ab-4738-9579-f530f0e692d4 ro rootflags=subvol=root rd.driver.blacklist=nouveau modprobe.blacklist=nouveau rhgb quiet rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[Thu Jul 27 09:00:57 2023] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks

        <whole bunch of dmesg output>

[Thu Jul 27 10:32:03 2023] legion PNP0C09:00: Read embedded controller ID 0x8227
[Thu Jul 27 10:32:03 2023] legion PNP0C09:00: Creating debugfs inteface
[Thu Jul 27 10:32:03 2023] legion_laptop: Creating sysfs inteface
[Thu Jul 27 10:32:03 2023] legion_laptop: Creating hwmon interface
[Thu Jul 27 10:32:03 2023] legion_laptop: Creating platform profile support
[Thu Jul 27 10:32:03 2023] legion_laptop: Init WMI driver support
[Thu Jul 27 10:32:03 2023] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[Thu Jul 27 10:32:03 2023] legion_laptop: Init keyboard backlight LED driver
[Thu Jul 27 10:32:03 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Thu Jul 27 10:32:03 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Thu Jul 27 10:32:03 2023] legion_laptop: Init Y-Logo LED driver
[Thu Jul 27 10:32:03 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Thu Jul 27 10:32:03 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Thu Jul 27 10:32:03 2023] legion_laptop: Init IO-Port LED driver
[Thu Jul 27 10:32:03 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Thu Jul 27 10:32:03 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[Thu Jul 27 10:32:03 2023] legion_laptop: Error reading brighntess for light: 5
[Thu Jul 27 10:32:03 2023] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[Thu Jul 27 10:32:03 2023] legion PNP0C09:00: legion_laptop loaded for this device
[Thu Jul 27 10:40:34 2023] legion_laptop: Unloading legion WMI
[Thu Jul 27 10:40:34 2023] legion_laptop: Unloading legion WMI done
[Thu Jul 27 10:40:34 2023] legion_laptop: Unloading legion platform profile
[Thu Jul 27 10:40:34 2023] legion_laptop: Unloading legion platform profile done
[Thu Jul 27 10:40:34 2023] legion_laptop: Set powermode
[Thu Jul 27 10:40:34 2023] legion_laptop: Unexpected power mode value ignored: 0
[Thu Jul 27 10:40:35 2023] legion_laptop: Set powermode
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion hwon
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion hwon done
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion sysfs
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion sysfs done
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion dubugfs
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion dubugfs done
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion ecram
[Thu Jul 27 10:40:35 2023] legion_laptop: Unloading legion ecram done
[Thu Jul 27 10:40:35 2023] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x00000000f8f9c957
[... 326 lines cut ...]
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 3
EC fan curve points size: 9
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 5	 5	 0	 127	 0	 52	 0	 31
1800	 1700	 5	 5	 127	 127	 48	 52	 28	 35
1900	 1800	 5	 5	 127	 127	 48	 52	 32	 39
2100	 2100	 5	 5	 127	 127	 48	 52	 35	 127
2400	 2300	 2	 2	 127	 127	 48	 55	 127	 127
2600	 2600	 2	 2	 127	 127	 52	 60	 127	 127
2900	 2900	 2	 2	 127	 127	 56	 64	 127	 127
3500	 3500	 2	 2	 127	 127	 60	 68	 127	 127
3800	 3800	 2	 2	 127	 127	 63	 127	 127	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

johnfanv2, 2023-07-27:
It seems like it works mostly? What features are missing?

MrDuartePT, 2023-10-19:
@bagong @aligoldenhat since looks that driver is working i will close the issue.
If you have any features missing reopen.

closed 2023-10-19 by MrDuartePT