johnfanv2/LenovoLegionLinux #91: [BUG]

ToadOnbekend, 2023-08-01:
**Problem Descripition**
The program fails when I am doing the ```make``` command. I have installed the Linux headers as seen in the screenshot
[image]

**Model and Debug Info**

Distribution: Ubuntu 24.04
Model name: 16ARH7H
CPU model: AMD Ryzen 7 6800H
GPU model: NVIDIA RTX 3070
Keyboard backlight: RGB 4-zone
Light in lid or logo: NO
Light at IO-Ports at back: NO

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RG
	Version: Legion 5 Pro 16ARH7H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RG_BU_idea_FM_Legion 5 Pro 16ARH7H
	Family: Legion 5 Pro 16ARH7H

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JUCN58WW
	Release Date: 02/09/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.58
	Firmware Revision: 1.58

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 35
CPU temperature EC error: 0
CPU temperature EC: 35
CPU temperature ACPI error: 0
CPU temperature ACPI: 35
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 35
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 81721344
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
EC fanfullspeed error: false
EC fanfullspeed status: 0
EC fan curve current point id: 0
EC fan curve points size: 4
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 127	 0	 127	 0	 37
1600	 1600	 2	 2	 127	 127	 127	 127	 33	 41
1800	 1800	 2	 2	 127	 127	 127	 127	 38	 46
2000	 2000	 2	 2	 127	 127	 127	 127	 41	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

johnfanv2, 2023-08-01:
What error do you get when you run the `make` command?

ToadOnbekend, 2023-08-02:
This issue
[image]
It seems like my version of the compiler is newer.

johnfanv2, 2023-08-06:
I don't think this will be a problem if the compiler versions do not differ much. Obviosuly, it compiles and the module work as seen from the output of `/sys/kernel/debug/legion/fancurve`. Do you see any problems?

ToadOnbekend, 2023-08-06:
How do I do that? Not very new to linux.

johnfanv2, 2023-08-06:
If you are not very good a Linux that do not worry about the warning. It obviously works. Just test the functionality with the LegionLinux GUI and test that everything you need works.

If you really want to try it then, get the newest code and
1. Install docker: https://docs.docker.com/engine/install/ubuntu/ 
2. Modify the file kernel_module/Makfile and change the version gcc:12.2.0 to 11.3.0.
3. Then run
```bash
# Go to kernel_module folder
cd kernel_module
# Start container
sudo make moduleInContainer
# Run this inside container
make
```  
But it obviously seems to compile and load without problems, so it is not really needed except when you see any problems.

ToadOnbekend, 2023-08-06:
Nope after doing what you said it still didn't work. The docker installation is fine. I feel pretty stupid. 
Here is a screenshot to show you.
[image]

johnfanv2, 2023-08-06:
Sorry, for describing it not clearly :crying_cat_face: .
1. First, you have to get the newest code, e.g. do `git pull` inside the `LenovoLegionLinux` folder or delete it and clone/download it again.
2. Then you should see the the following lines in `kernel_module/Makfile`:
```Makefile
moduleInContainer:
	docker run -it --rm -v "$(shell pwd)":/kernel_module -w /kernel_module  -v /usr/src/:/usr/src/  -v /lib/modules/:/lib/modules gcc:12.2.0 bash
```
Change 12.2.0 to 11.3.0 (or whatever the version of gcc was used to compile your kernel). 
3. Then, follow the other commands above.

ToadOnbekend, 2023-08-07:
Still it doesn't read any further then the gcc:11.4.0 bash. So it reads that the kernel is 11 and not 11.4.0 as we want. See below
[image]

MrDuartePT, 2023-09-14:
In this situation I would recomend reinstall the latest kernel and gcc (available on the repo) and try to recompile the module again. GCC mitsmatch only happen if you update gcc after the kernel.
Also we now have a PPA available you can use that instead.

MrDuartePT, 2023-10-20:
Also i gonna close this issue since is inactive and this is not a module problem but a Ubuntu problem (it also will happen when compiled other module, since the gcc version not match with the ubuntu use to compile the kernel).

For further help try the Linux Channel in the Legion Discord server.

closed 2023-10-20 by MrDuartePT