johnfanv2/LenovoLegionLinux #92: support: IdeaPad Gaming 3 15ARH7 (2022)

cfrstr, 2023-08-06:
**Model and Debug Info**

Distribution: Fedora
Model name: IdeaPad Gaming 3 15ARH7
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3050TI Mobile
Keyboard backlight: White with off/med/high
Light in lid or logo: no
Light at IO-Ports at back: no only for power

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82SB
	Version: IdeaPad Gaming 3 15ARH7
	Serial Number: redacted
	UUID: redacted
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82SB_BU_idea_FM_IdeaPad Gaming 3 15ARH7
	Family: IdeaPad Gaming 3 15ARH7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002A, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output pf `sudo dmidecode -t bios`:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JNCN44WW(V2.04)
	Release Date: 04/10/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.44
	Firmware Revision: 1.44

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

Output of `sudo make forcereloadmodule`:
```
[Sun Aug  6 02:05:57 2023] legion_laptop: legion_laptop exit
[Sun Aug  6 02:05:57 2023] legion_laptop: legion_laptop starts loading
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82SB; DMI_BIOS_VERSION:JNCN44WW(V2.04)
[Sun Aug  6 02:05:57 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82SB; DMI_BIOS_VERSION:JNCN44WW(V2.04)
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: legion_laptop is forced to load.
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: Using configuration for system: GKCN
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: Could not evaluate ACPI _STA
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: Could not init ACPI access: -5
[Sun Aug  6 02:05:57 2023] legion_laptop: Unloading legion shared
[Sun Aug  6 02:05:57 2023] legion_laptop: Unloading legion shared done
[Sun Aug  6 02:05:57 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Sun Aug  6 02:05:57 2023] legion: probe of PNP0C09:00 failed with error -5
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here' 
```

WMI/ACPI ETC: https://github.com/cfrstr/15ARH7

johnfanv2, 2023-08-06:
I added your model. Please get the newest code again and try again. Then, please provide the following output for further debugging:
```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
```
and the dmesg output after loading the module (only the new stuff is needed after loading and not the complete log). I looked at the ACPI disassembly and WMI output. It does not provide all the features of the Legion output but we can try to get as most as possible.

cfrstr, 2023-08-06:
`sudo cat /sys/kernel/debug/legion/fancurve`

```
EC Chip ID: 0
EC Chip Version: 0
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228186128
temperature access method: 3
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: -14
GPU temperature WMI2: 35
GPU temperature WMI3 error: -14
GPU temperature WMI3: 35
fan speed access method: 3
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 0
ACPI rapidcharge error: 0
[... 136 lines cut ...]
[   87.575404] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:8
[   87.575406] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[   87.575501] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:7
[   87.575503] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[   87.575598] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:7
[   87.575600] legion_laptop: WMI evaluation error for: dc2a8805-3a8c-41ba-a6f7-092e0089cd3b:17
[   87.706737] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[   87.706758] legion_laptop: Error for WMI method call to get brightness
[   87.706761] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[   87.706762] legion_laptop: Error for WMI method call to get brightness
[   87.706763] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[   87.706764] legion_laptop: Error for WMI method call to get brightness
[   87.706792] legion_laptop: Unexpected value in MINIFANCURVE register:0
[   87.706838] legion_laptop: No access method for fancurve:0
[   87.706841] legion_laptop: Size of object: 48
[   87.706842] legion_laptop: ACPI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:5
[   87.706844] legion_laptop fan table wmi buffer000000008376faa3: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   87.706847] legion_laptop fan table wmi buffer000000005492b2c3: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   87.706849] legion_laptop fan table wmi buffer00000000db07de67: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   87.706851] legion_laptop fan table wmi buffer0000000076dbb3fc: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   87.706852] legion_laptop fan table wmi buffer0000000022679490: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   87.706853] legion_laptop fan table wmi buffer000000007d15d8af: 00 00 00 00 00 00 00 00                          ........
[  141.120742] hid-generic 0003:1038:1838.000A: hiddev97,hidraw4: USB HID v1.11 Device [SteelSeries SteelSeries Aerox 3 Wireless] on usb-0000:07:00.3-1/input3
[  201.119339] hid-generic 0003:1038:1838.000B: hiddev97,hidraw4: USB HID v1.11 Device [SteelSeries SteelSeries Aerox 3 Wireless] on usb-0000:07:00.3-1/input3
```

closed 2026-09-11 by github-actions[bot]