johnfanv2/LenovoLegionLinux #93: Legion Slim 5 16IRH8 Support

klmshrk, 2023-08-08:
Distribution: Ubuntu 23.04
Model name: Lenovo Legion Slim 5 16IRH8
CPU model: Intel Core i7-13700H
GPU model: NVIDIA RTX 4070
Keyboard backlight: single color

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82YA
	Version: Legion Slim 5 16IRH8
	Serial Number: ***
	UUID: ***
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82YA_BU_idea_FM_Legion Slim 5 16IRH8
	Family: Legion Slim 5 16IRH8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: M2CN25WW
	Release Date: 05/16/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.25
	Firmware Revision: 2.25

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12288851/fwts_wmi.log)
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12288892/acpi_re.zip)

klmshrk, 2023-08-16:
> I added your model to the driver. Please get the newest code, recompile, and try again.
> 
> Then, please test the following features. We have to try a bit to find the correct configuration.
> # Temperatures and Fan Speeds
> 
> Please read the fan speeds and temperatures (see README) and confirm that they are sensible and change accordingly, e.g. when you put load onto the CPU the CPU temp and the fan speeds should increase, but GPU temp should stay roughly the same.
> # Powermode
> 
> We will start by confirming that power mode change work **and to get me more debug info**. Please change the power mode with Fn+Q and provide the output of the following commands in **each** mode.
> 
> ```shell
> sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
> sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
> sudo cat /sys/kernel/debug/legion/fancurve
> ```
> 
> Then please also check if the power mode is read correctly in the OS (see README). Then please check if you can set the power mode with the GUI (see README) and the LED changes accordingly.
> 
> Tests the following features with the GUI. Please check that if works or otherwise give precise information about what does not work. Please try to check that reading and writing works with the software.
> 
> Please provide the log output from the GUI from the log tab for all the tests.
> # Fn Lock
> 
>     * Enable fn-lock: A LED might turn on. fn functions now work without pressing Fn
> 
>     * Disable fn-lock: A LED might turn on. fn functions now work only when pressing Fn
> 
> 
> # Win Key
> 
>     * Enable Win Key: Win Key should work, e.g. open start menu or trigger some other action
> 
>     * Disable Win Key: Win Key has no function anymore
> 
> 
> # Touchpad
> 
>     * Enable Touchpad: Touchpad works
> 
>     * Disable Touchpad: Touchpad does not work
> 
> 
> # Battery Conservation
> 
>     * Enabled: charging stops if above 60 % (or 80% depending on model)
> 
>     * Disabled: charging works up to 100%
>       Note: Sometimes it takes a 10-20 seconds before it takes effect
> 
> 
> # Rapid Charging
> 
>     * Enabled: Charging is faster; Check this by comparing the time until fully charged displayed in system information or by comparing charging current.
> 
>     * Disabled: Charging is slower
>       Note: Sometimes it takes a 10-20 seconds before it takes effect .
> 
> 
> # Charge Output Always On
> 
>     * Enabled: Some USB output is always powered even when laptop is off or sleeping; Test e.g. by trying to charge your phone.
> 
>     * Disabled: USB outputs are off when laptop is off
> 
> 
> # Display Overdrive
> 
>     * Enabled: display somehow faster
> 
>     * Disabled: display not faster
>       Just check if checkbox can be checked or not if you cannot verify that it has an effect.
> 
> 
> # Keyboard Backlight
> 
> In order to fix/configure the keyboard backlight and begin with setting fan curves, please put your keyboard backlight to off, medium, or bright. Provide the output of the following commands (in that order) in each backlight state:
> 
> ```shell
> sudo cat /sys/kernel/debug/legion/fancurve
> sudo dmesg | tail -n 30
> ```
> 
> # Fan curve
> 
> I guess this is a newer model, so we only can change the fan curve in custom mode for now. Use the GUI to go into custom mode and change the fan curve. I guess that in the newer models the value is from 0 to 1000 (maximal) and not in RPM. Please try to verify it. Please also try if changing the fan curve in custom mode really influences the fan curve.

legion PNP0C09:00: legion_laptop not loaded for this device. I redid the first step. idk what to do now

class1c-j, 2023-08-16:
Sorry if I shouldn't intervene in this thread, but I got the same model and same specs as @klmshrk but running on fedora kernel 6.4.10-200, tried installing with dkms and rebooted. my dmesg output includes 

```
[   17.388858] legion_laptop: legion_laptop starts loading
[   17.388943] legion PNP0C09:00: legion_laptop platform driver probing
[   17.388946] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[   17.388948] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[   17.388953] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[   17.388955] legion PNP0C09:00: Using configuration for system: M2CN
[   17.388965] legion PNP0C09:00: Could not evaluate ACPI _STA
[   17.388966] legion PNP0C09:00: Could not init ACPI access: -5
[   17.388968] legion_laptop: Unloading legion shared
[   17.388968] legion_laptop: Unloading legion shared done
[   17.388969] legion PNP0C09:00: legion_laptop not loaded for this device
[   17.389037] legion: probe of PNP0C09:00 failed with error -5
```

klmshrk, 2023-08-16:
@class1c-j I have the same output 
`[  495.942469] legion_laptop: legion_laptop starts loading
[  495.942562] legion PNP0C09:00: legion_laptop platform driver probing
[  495.942566] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[  495.942568] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[  495.942574] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[  495.942582] legion PNP0C09:00: Using configuration for system: M2CN
[  495.942600] legion PNP0C09:00: Could not evaluate ACPI _STA
[  495.942601] legion PNP0C09:00: Could not init ACPI access: -5
[  495.942603] legion_laptop: Unloading legion shared
[  495.942603] legion_laptop: Unloading legion shared done
[  495.942605] legion PNP0C09:00: legion_laptop not loaded for this device
[  495.942761] legion: probe of PNP0C09:00 failed with error -5
`

johnfanv2, 2023-08-16:
I updated the id of the embedded controller in the code. Please try with the new code again. This seems to be different from what I expected. Is this a 2023 model?

johnfanv2, 2023-08-16:
I expected a 2023 model. Maybe different models have slightly different firmware. Please try the new code.

klmshrk, 2023-08-16:
@johnfanv2 GUI doesn't start `sudo: python/legion_linux/legion_linux/legion_gui.py: command not found`

class1c-j, 2023-08-17:
@klmshrk did you forget the ./ and running from the repo root folder?

I tried running some of the tests and this is what I got.

# Temperatures and Fan Speeds
```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +43.0°C

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:              0 RPM
Fan 2:              0 RPM
CPU Temperature:  +56.0°C
GPU Temperature:  +49.0°C
IC Temperature:    +0.0°C

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

BAT1-acpi-0
Adapter: ACPI interface
in0:          15.49 V

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 24:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 25:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 26:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 27:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +51.0°C  (high = +100.0°C, crit = +100.0°C)

ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

nvme-pci-0600
Adapter: PCI adapter
Composite:    +46.9°C  (low  =  -0.1°C, high = +85.8°C)
                       (crit = +86.8°C)
Sensor 1:     +42.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)
```

Fan speeds change when performing heavy work.

```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +44.0°C

legion_hwmon-isa-0000
[... 819 lines cut ...]
2023-08-17 00:31:51,708 - INFO: Feature StrFileFeature reading: quiet balanced balanced-performance performance
2023-08-17 00:31:51,708 - INFO: Feature PlatformProfileFeature reading
2023-08-17 00:31:51,709 - INFO: Feature PlatformProfileFeature reading: balanced-performance
2023-08-17 00:32:45,819 - INFO: Click on UI Balanced Mode element for PlatformProfileFeature
2023-08-17 00:32:45,820 - INFO: Feature StrFileFeature reading
2023-08-17 00:32:45,820 - INFO: Feature StrFileFeature reading: quiet balanced balanced-performance performance
2023-08-17 00:32:45,821 - INFO: Feature PlatformProfileFeature writing: balanced
2023-08-17 00:32:46,024 - INFO: Feature StrFileFeature reading
2023-08-17 00:32:46,025 - INFO: Feature StrFileFeature reading: quiet balanced balanced-performance performance
2023-08-17 00:32:46,025 - INFO: Feature PlatformProfileFeature reading
2023-08-17 00:32:46,026 - INFO: Feature PlatformProfileFeature reading: balanced
2023-08-17 00:51:59,235 - INFO: Click on UI Rapid Charging element for RapidChargingFeature
2023-08-17 00:51:59,236 - INFO: Feature RapidChargingFeature writing: 0
2023-08-17 00:51:59,336 - INFO: Feature RapidChargingFeature reading
2023-08-17 00:51:59,338 - INFO: Feature RapidChargingFeature reading: 0
2023-08-17 00:51:59,438 - INFO: Feature BatteryConservation reading
2023-08-17 00:51:59,440 - INFO: Feature BatteryConservation reading: 0
2023-08-17 00:51:59,440 - INFO: Feature BatteryConservation reading
2023-08-17 00:51:59,441 - INFO: Feature BatteryConservation reading: 0
2023-08-17 00:51:59,441 - INFO: Feature RapidChargingFeature reading
2023-08-17 00:51:59,442 - INFO: Feature RapidChargingFeature reading: 0
2023-08-17 01:00:28,223 - INFO: Click on UI Touchpad Enabled (Lock or unlock touchpad; same as Fn + F10) element for TouchpadFeature
2023-08-17 01:00:28,224 - INFO: Feature TouchpadFeature writing: 0
2023-08-17 01:00:28,325 - INFO: Feature TouchpadFeature reading
2023-08-17 01:00:28,327 - INFO: Feature TouchpadFeature reading: 0

klmshrk, 2023-08-17:
yes, sorry. output of launching GUI 
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
INFO:legion_linux.legion:[    1.940156] legion_laptop: loading out-of-tree module taints kernel.
[    1.940197] legion_laptop: module verification failed: signature and/or required key missing - tainting kernel
[    1.941821] legion_laptop: legion_laptop starts loading
[    1.945410] legion PNP0C09:00: legion_laptop platform driver probing
[    1.945414] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[    1.945416] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[    1.945420] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[    1.945422] legion PNP0C09:00: Using configuration for system: M2CN
[    1.945423] legion_laptop: Creating RAM access to embedded controller
[    1.945438] legion_laptop: Succeffuly mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x0000000095fbc0ea
[    1.945681] legion PNP0C09:00: Read embedded controller ID 0x8227
[    1.945682] legion PNP0C09:00: Creating debugfs inteface
[    1.945690] legion_laptop: Creating sysfs inteface
[    1.946395] legion_laptop: Creating hwmon interface
[    1.949112] legion_laptop: Creating platform profile support
[    1.949115] legion_laptop: Init WMI driver support
[    1.949150] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[    1.949167] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[    1.949179] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[    1.949185] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[    1.949190] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[    1.949196] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[    1.949201] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[    1.949215] legion_wmi 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294: Register after probing for WMI.
[    1.949226] legion_laptop: Init keyboard backlight LED driver
[    1.949295] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    1.949349] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    1.949350] legion_laptop: Init Y-Logo LED driver
[    1.949370] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    1.949397] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    1.949398] legion_laptop: Init IO-Port LED driver
[    1.949416] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    1.949444] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    1.949444] legion PNP0C09:00: legion_laptop loaded for this device
[    5.670945] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[    8.530597] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[  178.968290] legion_laptop: Unloading legion WMI
[  178.968404] legion_laptop: Unloading legion WMI done
[  178.968405] legion_laptop: Unloading legion platform profile
[  178.968409] legion_laptop: Unloading legion platform profile done
[  178.968933] legion_laptop: Set powermode
[  178.968934] legion_laptop: Unexpected power mode value ignored: 0
[  180.457130] legion_laptop: Set powermode
[  180.458717] legion_laptop: Unloading legion hwon
[  180.458773] legion_laptop: Unloading legion hwon done
[  180.458774] legion_laptop: Unloading legion sysfs
[  180.458783] legion_laptop: Unloading legion sysfs done
[  180.458784] legion_laptop: Unloading legion dubugfs
[  180.458790] legion_laptop: Unloading legion dubugfs done
[  180.458791] legion_laptop: Unloading legion ecram
[  180.458792] legion_laptop: Unloading legion ecram done
[  180.458792] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x0000000095fbc0ea
[  180.458802] legion_laptop: Unloading legion shared
[  180.458802] legion_laptop: Unloading legion shared done
[  180.458803] legion_laptop: Legion platform unloaded
[  181.963319] legion_laptop: legion_laptop exit
[  182.013031] legion_laptop: legion_laptop starts loading
[  182.013174] legion PNP0C09:00: legion_laptop platform driver probing
[  182.013180] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[  182.013184] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82YA; DMI_BIOS_VERSION:M2CN25WW
[  182.013193] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[  182.013197] legion PNP0C09:00: Using configuration for system: M2CN
[  182.013200] legion_laptop: Creating RAM access to embedded controller
[  182.013237] legion_laptop: Succeffuly mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x0000000095fbc0ea
[  182.013801] legion PNP0C09:00: Read embedded controller ID 0x8227
[  182.013804] legion PNP0C09:00: Creating debugfs inteface
[  182.013836] legion_laptop: Creating sysfs inteface
[... 68 lines cut ...]
[... cut ...]
WARNING:legion_linux.legion:Feature LegionGUIAutostart exist not. exits: 0
INFO:legion_linux.legion:Feature NVIDIAGPUIsRunning with pattern /sys/bus/pci/devices/0000:01:00.0/power/runtime_status and path /sys/bus/pci/devices/0000:01:00.0/power/runtime_status
INFO:legion_linux.legion:Feature automatic_close setting to 0
INFO:legion_linux.legion:Feature close_to_tray setting to 0
INFO:legion_linux.legion:Feature open_closed_to_tray setting to 0
INFO:legion_linux.legion:No settings file exists.
INFO:legion_linux.legion:Feature LockFanController reading
```

johnfanv2, 2023-08-17:
Sorry for the bug. While I fixed other stuff I introduced a bug that makes it hang. Please get the newest code, recompile, and reload the module (or reinstall if you use dkms). If reloading does not work or hangs, please do a restart.

klmshrk, 2023-08-17:
@johnfanv2 sudo cat /sys/kernel/debug/legion/fancurve output

```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4245487636
temperature access method: 5
CPU temperature error: 0
CPU temperature: 61
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 61
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 61
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 35
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1500
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1500
2 fanspeed error: 0
2 fanspeed: 1500
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1500
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1500
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 230
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
[... 1262 lines cut ...]
[ 2311.962611] legion_laptop fan table wmi buffer00000000edbeaa11: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962614] legion_laptop fan table wmi buffer00000000279b1dce: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962615] legion_laptop fan table wmi buffer000000003d46b3df: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962617] legion_laptop fan table wmi buffer000000007543399b: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962619] legion_laptop fan table wmi buffer00000000cba35008: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962620] legion_laptop fan table wmi buffer0000000095282c51: 00 00 00 00 00 00 00 00                          ........
[ 2311.962623] legion_laptop: Size of object: 48
[ 2311.962636] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [FID0] at bit offset/length 0/8 exceeds size of target Buffer (0 bits) (20221020/dsopcode-198)

[ 2311.962644] No Local Variables are initialized for Method [WMAB]

[ 2311.962647] Initialized Arguments for Method [WMAB]:  (3 arguments defined for method invocation)
[ 2311.962648]   Arg0:   00000000e8048604 <Obj>           Integer 0000000000000000
[ 2311.962654]   Arg1:   000000002e78c2d7 <Obj>           Integer 0000000000000005
[ 2311.962659]   Arg2:   00000000d8f8a27a <Obj>           Buffer(0)

[ 2311.962666] ACPI Error: Aborting method \_SB.GZFD.WMAB due to previous error (AE_AML_BUFFER_LIMIT) (20221020/psparse-529)
[ 2311.962674] legion_laptop: ACPI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:5
[ 2311.962676] legion_laptop fan table wmi buffer00000000dfe3ba37: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962678] legion_laptop fan table wmi buffer000000004c8727d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962680] legion_laptop fan table wmi buffer0000000082ca11f6: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962681] legion_laptop fan table wmi buffer0000000023eb0365: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962683] legion_laptop fan table wmi buffer00000000806f3520: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2311.962684] legion_laptop fan table wmi buffer000000008f219c03: 00 00 00 00 00 00 00 00                          ........
```

class1c-j, 2023-08-17:
Yeah, now it doesn't hang, but it still doesn't seem to be able to read the fancurve.
[image]

GUI Error:

```
INFO:legion_linux.legion:Feature CPUOverclock reading
ERROR:legion_linux.legion:Feature CPUOverclock reading error [Errno 22] Invalid argument
ERROR:legion_linux.legion:[  886.185900] legion_laptop fan table wmi buffer00000000500ccbd7: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185901] legion_laptop fan table wmi buffer00000000a6bd1d42: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185902] legion_laptop fan table wmi buffer000000002f4429b8: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185903] legion_laptop fan table wmi buffer00000000d82cb37c: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185903] legion_laptop fan table wmi buffer000000001759e9f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185904] legion_laptop fan table wmi buffer000000003b9c65a0: 00 00 00 00 00 00 00 00                          ........
[  886.185905] legion_laptop: Size of object: 48
[  886.185908] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [FID0] at bit offset/length 0/8 exceeds size of target Buffer (0 bits) (20230331/dsopcode-198)
[  886.185910] ACPI Error: Aborting method \_SB.GZFD.WMAB due to previous error (AE_AML_BUFFER_LIMIT) (20230331/psparse-529)
[  886.185912] legion_laptop: ACPI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:5
[  886.185913] legion_laptop fan table wmi buffer00000000d82cb37c: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185913] legion_laptop fan table wmi buffer000000001759e9f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185914] legion_laptop fan table wmi buffer000000003b9c65a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185915] legion_laptop fan table wmi buffer00000000b076a8aa: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185915] legion_laptop fan table wmi buffer0000000004ee4cf7: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  886.185916] legion_laptop fan table wmi buffer00000000573eb7aa: 00 00 00 00 00 00 00 00                          ........
[ 1016.492592] legion_laptop: get_simple_wmi_attributewith raw value: 0
[ 1016.493100] legion_laptop: get_simple_wmi_attributewith raw value: 0
[ 1016.493238] legion_laptop: get_simple_wmi_attributewith raw value: 0
[ 1016.502075] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:1: expected type 3 but got 1; expected length 16 but got 0;

Error occured [Errno 22] Invalid argument
Traceback (most recent call last):
  File "/home/rm/Downloads/repos/LenovoLegionLinux/python/legion_linux/legion_linux/./legion_gui.py", line 205, in update_view_from_feature
    feature_value = self.feature.get()
                    ^^^^^^^^^^^^^^^^^^
  File "/home/rm/Downloads/repos/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 297, in get
    invalue = self._read_file_int(self.filename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rm/Downloads/repos/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 231, in _read_file_int
    return int(self._read_file_str(file_path))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rm/Downloads/repos/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 228, in _read_file_str
    raise err
  File "/home/rm/Downloads/repos/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 222, in _read_file_str
    out = str(filepointer.read()).strip()
              ^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument
```

bagong, 2023-08-17:
Are you using "sudo"? I only saw dark theme used when I had forgotten sudo.

class1c-j, 2023-08-17:
> Are you using "sudo"? I only saw dark theme used when I had forgotten sudo.
> […](#)

@bagong It doesn't make a difference, as you can see from @klmshrk's comment, the fan curve is empty when running `sudo cat  /sys/kernel/debug/legion/fancurve`. Here's the GUI with sudo:

[image]

klmshrk, 2023-08-25:
@johnfanv2 
Battery Conservation - works
Rapid Charging - works
fan curve still 0

Antoine-TH0MAS, 2024-01-17:
I'm planning to buy a legion slim 5 I7 13700HX rtx 4060. I'm only interested in having the 3 different thermals modes. What actually works?  Do you have any recommendations for setting up linux?

class1c-j, 2024-01-17:
> I'm planning to buy a legion slim 5 I7 13700HX rtx 4060. I'm only interested in having the 3 different thermals modes. What actually works? Do you have any recommendations for setting up linux?

I have the same model but with rtx4070. Right now everything works as it should except for the fan control as discussed in this thread.

class1c-j, 2024-01-17:
> Thank you for responding so quickly. So the the thermal modes only change the voltage? What's controlling the fans?
> 
> Sorry if my question is dumb, this is my first gaming laptop.

They do control the fans between the 3 presets. The issue with this module and gen 8 laptops is related to the fancurve, so you cannot edit how your fans behave beyond those three presets (quiet, balanced, performance), that is you can't adjust how fast they go for a given temperature range.
As I only use Linux for programming, I can't talk about gaming but it has not failed me yet with training deep learning models on the GPU with the balanced preset still keeping it cool to the touch.

bulend-karadag, 2024-10-01:
I have legion slim 5i I7 13700HX rtx 4050, which is almost the same of the one in this thread. I am also going through the same problem of the fan curve. Is there any news/development to solve this problem? 
PS: Linux users definitely need such efforts. You got my star.

FrostByghte, 2024-12-23:
I have the  Legion Slim 7 16IRH8, same everything as stated. The only thing not working is the fan modifications as discussed in this thread.

closed 2026-09-11 by github-actions[bot]