johnfanv2/LenovoLegionLinux #94: Legion Y740 (MTM:80UF009HV) support

Vhriz, 2023-08-08:
[Tue Aug  8 21:31:36 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Tue Aug  8 21:31:36 2023] legion: probe of PNP0C09:00 failed with error -5
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here' 

**Model and Debug Info**

Distribution: Linux Mint 21.2
Model name: Legion Y740-15IRH 
CPU model: Intel© Core™ i7-9750H
GPU model: NVIDIA Corporation TU116M [GeForce GTX 1660 Ti Mobile]
Keyboard backlight: RGB
Light in lid or logo: Yes
Light at IO-Ports at back: ??? Yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 81UF
	Version: Lenovo Legion Y740-15IRH
	Serial Number: 
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_81UF_BU_idea_FM_Legion Y740-15IRH
	Family: Legion Y740-15IRH

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0024, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0031, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.0.1 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: BVCN16WW(V1.12)
	Release Date: 06/21/2021
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 10 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.16
	Firmware Revision: 1.16

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
[???](cat: /sys/kernel/debug/legion/fancurve: No such file or directory)
```

johnfanv2, 2023-08-14:
@Vhriz Please get the newest version of the code and try again. I added your model to the code.

Vhriz, 2023-08-15:
Hello,
I'm unsure if its user failure or I do something wrong, but it still fails.
`[Tue Aug 15 13:16:09 2023] legion PNP0C09:00: Using configuration for system: BVCN
[Tue Aug 15 13:16:09 2023] legion PNP0C09:00: Could not evaluate ACPI _STA
[Tue Aug 15 13:16:09 2023] legion PNP0C09:00: Could not init ACPI access: -5
[Tue Aug 15 13:16:09 2023] legion_laptop: Unloading legion shared
[Tue Aug 15 13:16:09 2023] legion_laptop: Unloading legion shared done
[Tue Aug 15 13:16:09 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Tue Aug 15 13:16:09 2023] legion: probe of PNP0C09:00 failed with error -5
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here' `

johnfanv2, 2023-08-15:
@Vhriz It was a "problem" in the code. I disabled a ACPI check for your model that is not really needed. Hopefully, it will load. Please try to do the steps again from my last comment.

Would you please also give me the files and outputs of the steps described here https://github.com/johnfanv2/LenovoLegionLinux/issues/46 ? I will use it to configure and adapt the driver to your model's firmware. So it is needed after you can load it successfully.

Vhriz, 2023-08-15:
All right,
Now I got that output.
`[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Using configuration for system: BVCN
[Tue Aug 15 13:33:54 2023] legion_laptop: Creating RAM access to embedded controller
[Tue Aug 15 13:33:54 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x000000006f61682e
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Read embedded controller ID 0xffff
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Expected EC chip id 0x8226 but read 0xffff
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion ecram
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion ecram done
[Tue Aug 15 13:33:54 2023] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x000000006f61682e
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion shared
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion shared done
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Tue Aug 15 13:33:54 2023] legion: probe of PNP0C09:00 failed with error -12
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the li
[acpi_re_fwts_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/12343708/acpi_re_fwts_re.tar.gz)
d (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here' 
`

Vhriz, 2023-08-15:
Now I'm getting a different error at least.

`[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81UF; DMI_BIOS_VERSION:BVCN16WW(V1.12)
[Tue Aug 15 13:33:54 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81UF; DMI_BIOS_VERSION:BVCN16WW(V1.12)
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Using configuration for system: BVCN
[Tue Aug 15 13:33:54 2023] legion_laptop: Creating RAM access to embedded controller
[Tue Aug 15 13:33:54 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x000000006f61682e
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Read embedded controller ID 0xffff
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: Expected EC chip id 0x8226 but read 0xffff
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion ecram
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion ecram done
[Tue Aug 15 13:33:54 2023] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x000000006f61682e
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion shared
[Tue Aug 15 13:33:54 2023] legion_laptop: Unloading legion shared done
[Tue Aug 15 13:33:54 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Tue Aug 15 13:33:54 2023] legion: probe of PNP0C09:00 failed with error -12
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light in the lid (Y-logo) or at the IO-ports (all Legion 7), please help testing controlling it 'here' 
`

[acpi_re_fwts_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/12343713/acpi_re_fwts_re.tar.gz)

johnfanv2, 2023-08-15:
The embedded controller seems to be different. I disabled the check for the ID of it. Try again with the newest code.

Vhriz, 2023-08-15:
All RIght,
It started and I did the tests.
GUI shows everything at max (255), load from HW fails.
ERROR:legion_linux.legion:[Errno 1] Operation not permitted

`sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228710420
temperature access method: 1
CPU temperature error: 0
CPU temperature: 255
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: 0
CPU temperature WMI: 56
CPU temperature WMI2 error: -14
CPU temperature WMI2: 56
CPU temperature WMI3 error: -14
CPU temperature WMI3: 56
GPU temperature error: 0
GPU temperature: 255
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: 0
GPU temperature WMI: 53
GPU temperature WMI2 error: -14
GPU temperature WMI2: 53
GPU temperature WMI3 error: -14
GPU temperature WMI3: 53
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 65535
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: 0
1 fanspeed WMI: 1400
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 1400
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 1400
2 fanspeed error: 0
2 fanspeed: 65535
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: 0
2 fanspeed WMI: 1400
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 1400
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 1400
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: -1073458176
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
EC fanfullspeed error: error
EC fanfullspeed status: -1
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +46.0°C  

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +54.0°C  

legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           65535 RPM
Fan 2:           65535 RPM
CPU Temperature: +255.0°C  
GPU Temperature: +255.0°C  
IC Temperature:  +255.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +52.0°C  (high = +100.0°C, crit = +100.0°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:          12.52 V  

sudo su
# As root enter
# 2. point, 1. fan
echo 1500 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point2_pwm
# 2. point, 2.fan
echo 1600 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm2_auto_point2_pwm
# 3. point, 1. fan
echo 1700 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point3_pwm
# 3. point, 2.fan
echo 1800 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm2_auto_point3_pwm

# Read the current fancurve and check if changes were made
cat /sys/kernel/debug/legion/fancurve
[sudo] password for thevhriz:          
root@Vhriz-Linux:/home/thevhriz# 

`

johnfanv2, 2023-08-15:
Get the newest code (again) and try reading temperatures and fan speeds again. To fix setting the fan curve, please give me the output of the following with the newest code:

```bash
sudo cat /sys/kernel/debug/legion/fancurve
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
```

johnfanv2, 2023-08-15:
I don't see any memory address where the fan curve might be stored (usually 8 or 10 (might be 3) bytes  which are strictly increasing. I guess I cannot add fan curve settings to your model. Do you see any other stuff that should be fixed?

st0nie, 2026-09-14:
The Y740 (BIOS BVCN) is in the allowlist. Please retest with the current release and reopen with dmesg output if it still fails.

psai87, 2026-09-14:
new dmesg output
```shell
780.220185] ACPI Error: Needed [Integer/String/Buffer], found [Reference] 00000000ab76ae52 (20260408/exresop-388)
[ 8780.220196] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [And] (20260408/dswexec-433)
[ 8780.220203] ACPI Error: Aborting method \_SB.PCI0.LPCB.EC0._Q20 due to previous error (AE_AML_OPERAND_TYPE) (20260408/psparse-543)
[ 8799.600913] ACPI Error: Needed [Integer/String/Buffer], found [Reference] 0000000097f2edb8 (20260408/exresop-388)
[ 8799.600922] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [And] (20260408/dswexec-433)
[ 8799.600929] ACPI Error: Aborting method \_SB.PCI0.LPCB.EC0._Q20 due to previous error (AE_AML_OPERAND_TYPE) (20260408/psparse-543)
[13069.739704] legion_laptop: Loading legion_laptop
[13069.739780] legion legion: legion_laptop platform driver probing
[13069.739783] legion legion: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81UF; DMI_BIOS_VERSION:BVCN16WW(V1.12)
[13069.739784] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81UF; DMI_BIOS_VERSION:BVCN16WW(V1.12)
[13069.739789] legion legion: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[13069.739791] legion legion: Using configuration for system: BVCN
[13069.740065] legion legion: Using ACPI device PNP0C09:00 for EC methods
[13069.740066] legion legion: Skipping ACPI _STA check
[13069.740067] legion_laptop: Creating RAM access to embedded controller
[13069.740079] legion_laptop: Successfully mapped embedded controller: 0xfc7e0800 (in RAM)/0x0 (in EC) to virtual 0x00000000e31e2803
[13069.740471] legion legion: Read embedded controller ID 0xffff
[13069.740472] legion legion: Skipped checking embedded controller id
[13069.740473] legion legion: Creating debugfs interface
[13069.740487] legion_laptop: Creating sysfs interface
[13069.740607] legion_laptop: Creating hwmon interface
[13069.740638] legion_laptop: Creating platform profile support
[13069.740672] legion_laptop: Init WMI driver support
[13069.740713] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8-17: Register after probing for WMI.
[13069.740731] legion_wmi D062906B-12D4-4510-999D-4831EE80E985-18: Register after probing for WMI.
[13069.740744] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8-19: Register after probing for WMI.
[13069.740756] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59-20: Register after probing for WMI.
[13069.740773] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1-21: Register after probing for WMI.
[13069.740785] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F-22: Register after probing for WMI.
[13069.740805] legion_laptop: Init keyboard backlight LED driver
[13069.740807] legion_laptop: Error for WMI method call to get brightness
[13069.740808] legion_laptop: Error reading keyboard brightness
[13069.740808] legion legion: Failed to init keyboard backlight LED driver. Skipping ...
[13069.740809] legion_laptop: Init Y-Logo LED driver
[13069.740811] legion_laptop: Error for WMI method call to get brightness
[13069.740812] legion_laptop: Error reading brightness for light: 3
[13069.740813] legion legion: Failed to init Y-Logo LED driver. Skipping ...
[13069.740813] legion_laptop: Init IO-Port LED driver
[13069.740815] legion_laptop: Error for WMI method call to get brightness
[13069.740816] legion_laptop: Error reading brightness for light: 5
[13069.740816] legion legion: Failed to init IO-Port LED driver. Skipping ...
[13069.740817] legion legion: legion_laptop loaded for this device
[13231.545360] legion_laptop: Unloading legion WMI
[13231.545459] legion_laptop: Unloading legion WMI done
[13231.545461] legion_laptop: Unloading legion platform profile
[13231.545461] legion_laptop: Unloading legion platform profile done
[13231.545724] legion_laptop: Set powermode
[13231.545725] legion_laptop: Unexpected power mode value ignored: 0
[13233.050935] legion_laptop: Set powermode
[13233.051207] legion_laptop: Unloading legion hwon
[13233.051279] legion_laptop: Unloading legion hwon done
[13233.051280] legion_laptop: Unloading legion sysfs
[13233.051307] legion_laptop: Unloading legion sysfs done
[13233.051308] legion_laptop: Unloading legion dubugfs
[13233.051316] legion_laptop: Unloading legion dubugfs done
[13233.051317] legion_laptop: Unloading legion ecram
[13233.051318] legion_laptop: Unloading legion ecram done
[13233.051319] legion_laptop: Unmapping embedded controller memory at 0xfc7e0800 (in RAM)/0x0 (in EC) at virtual 0x00000000e31e2803
[13233.051327] legion_laptop: Unloading legion shared
[13233.051328] legion_laptop: Unloading legion shared done
[13233.051328] legion_laptop: Legion platform unloaded
[13233.051401] legion_laptop: legion_laptop exit
[13233.078464] legion_laptop: Loading legion_laptop
[13233.078545] legion legion: legion_laptop platform driver probing
[13233.078548] legion legion: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81UF; DMI_BIOS_VERSION:BVCN16WW(V1.12)
[13233.078550] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 81UF; DMI_BIOS_VERSION:BVCN16WW(V1.12)
[13233.078554] legion legion: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[13233.078556] legion legion: Using configuration for system: BVCN
[... 13 lines cut ...]
[13233.079504] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8-19: Register after probing for WMI.
[... cut ...]
[13233.079611] legion legion: legion_laptop loaded for this device
[13283.433207] legion_laptop: Error for WMI method call to get brightness
[13283.433213] legion_laptop: Error for WMI method call to get brightness
[13283.433215] legion_laptop: Error for WMI method call to get brightness
[13283.433392] legion_laptop: Unexpected value in MINIFANCURVE register: 255
[13283.433568] legion_laptop: Unexpected value in lockfanspeed register: 255
[13283.433744] legion_laptop: Unexpected value in maximumfanspeed register: 255

```

st0nie, 2026-09-14:
Thanks for the dmesg — reopening since you're engaged. Reading it carefully:

**What works on your Y740 now:** the module loads, sensors/fan RPM and platform profile come up. The keyboard/Y-Logo/IO-Port LED errors are expected — on the Y740 those lights are Corsair iCUE devices, not WMI-driven, so no software via this driver can control them (use OpenRGB/ckb-next instead).

**The `_Q20` ACPI errors are a firmware bug** in your DSDT (an `And` with a Reference operand), triggered by EC query events — they appear without this driver too and aren't caused by it.

**Fan control:** on BVCN the fan curve is deliberately disabled (`access_method_fancurve = ACCESS_METHOD_NO_ACCESS`) — the Y740's EC register layout for the fan curve was never validated, and writing unvalidated EC registers is exactly how machines get bricked fans. You do have `fan_fullspeed` and fan RPM monitoring.

If you want to pursue real fan-curve support: post a decompiled DSDT (`sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat && iasl -d dsdt.dat`) and `sudo cat /sys/kernel/debug/legion/ecmemory`, and we can check whether the standard EC offsets apply to the 0x8226 EC before enabling anything.

psai87, 2026-09-14:
dsdat info
cannot upload the files in git, so attached in the gdrive link
https://drive.google.com/drive/folders/1-DyvOFSiuzRdF4do1huCjHN1Qev7P_Et?usp=drive_link
THe drive link has all 3 files.

and ecmemory it shows as 
```shell
sai@sai-kde:~/Software/lenovo$ sudo cat /sys/kernel/debug/legion/ecmemory
������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������sai@sai-kde:~/Software/lenovo$ sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000300

```

st0nie, 2026-09-14:
Analyzed your DSDT (dsdt.dsl from the Drive link) — thanks for providing it. Definitive answer for the Y740:

**Fan curve control is not possible on this model, by hardware/firmware design:**

1. Both your EC dumps (`ecmemory` and ecmem.mem) are 0xFF from start to end — the EC does not answer the Super-I/O port reads on this machine, so EC-register fan table access is out.
2. The DSDT's EC memory window (ECMM at 0xFC7E0800 — matches the driver's mapping) only defines the classic laptop fields: SMBus, adapter/thermal status bits, and FAN1/FAN2 as single **enable bits** — there is no 10-point fan table structure anywhere, and no WMI fan method (no WMAB/`Fan_Set_Table` equivalent) either. The firmware simply doesn't expose a writable fan curve to the OS.

So `model_bvcn`'s current configuration (fan curve disabled, fan speed monitoring + `fan_fullspeed` via WMI) is already the maximum this machine can do safely.

**What does work for your goal of better thermals:**
- The BIOS "Active Policy" fan settings you found — they run at firmware level, independent of the OS, and keep working on Linux.
- `fan_fullspeed` on demand: `echo 1 | sudo tee /sys/bus/platform/drivers/legion/legion/fan_fullspeed` before a heavy load.
- Power limits (platform profile quiet/balanced) to reduce heat generation at the source.

Closing this now since the question is definitively answered — feel free to reopen if you find evidence of a fan-table interface on this firmware (e.g. a Windows tool that demonstrably writes per-point curves).

closed 2026-09-14 by st0nie

psai87, 2026-09-14:
Thank you for the support.

I am not able to run the below cmd, getting error No such device

```c
echo 1 | sudo tee /sys/bus/platform/drivers/legion/legion/fan_fullspeed
echo 1 | sudo tee /sys/devices/platform/legion/fan_fullspeed

```

st0nie, 2026-09-14:
That error is on me — I have to correct my earlier statement here. "No such device" (`ENODEV`) tells us exactly what is going on:

`model_bvcn` configures `fan_fullspeed` to use the Lenovo `FAN_METHOD` WMI data block (GUID `92549549-4bde-4f06-ac04-ce8bf898dbaa`). Before evaluating the WMI method, the driver checks that the firmware actually exposes that GUID, and if it doesn't, the write fails with `ENODEV` — which is the error `tee` prints for you. Your dmesg from earlier confirms it: the only WMI devices on your machine are the six event GUIDs (`BFD42481`×2, `D062906B`, `BC72A435`, `10AFC6D9`, `D320289E`). There is no `92549549-...` data block, and also no `dc2a8805-...` "Other Method" block (that one has a `FAN_FULLSPEED` feature we could have used as a fallback). So unlike what I wrote before, `fan_fullspeed` is **not** available on the Y740 — sorry for the wrong pointer.

You can double-check with:

```bash
ls /sys/bus/wmi/devices/
```

`92549549-4bde-4f06-ac04-ce8bf898dbaa` will be absent.

So on this machine the fan can only be influenced through the BIOS fan settings (Active Policy etc.), which run in firmware and work regardless of the OS. I've opened #574 to set `access_method_fanfullspeed = ACCESS_METHOD_NO_ACCESS` for `model_bvcn`, so the dead attribute is hidden instead of accepting writes that cannot succeed.