johnfanv2/LenovoLegionLinux #95: Testing fan control and more on Lenovo Yoga laptops [looking for testers]

johnfanv2, 2023-08-20:
This is a issue to test the driver for Lenovo Yoga laptops. If you have Lenovo Yoga laptop and want to help testing then please provide the following information. It will be used to add fan control and more for Lenovo Yoga laptops. It must be tested **on multiple different** models.

**Model and Debug Info**

Model name: ???
CPU model: ??? (e.g. AMD Ryzen 7 5800H)
GPU model: ??? (e.g NVIDIA RTX 3700)
Keyboard backlight: ??? (RGB, single color with off/on, single color with off/medium/bright)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
???
```

Output of `sudo dmidecode -t bios`:
```text
???
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
???
```

The following information would be useful.

## Inspecting WMI entries
```bash
# Install required tools
sudo apt install fwts

# Create folder for all the new files
mkdir fwts_re
cd fwts_re
sudo fwts wmi - > fwts_wmi.log
```
Then upload files.

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

Then upload files.

## Gathering WMI info in Windows
If you have Windows, then the following information are very very useful to make use of the ACPI methods. Open powershell as admin in Windows and run the following script. It will list all available lenovo WMI methods. Copy output to a file and upload.
```powershell
$wmi_classes = Get-WmiObject -Namespace 'ROOT/WMI' -List -Class "*LENOVO*"
foreach ($wmi_class in $wmi_classes){
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "########################################"
  Write-Host "Name:" $wmi_class.Name
  Write-Host "Class Name:" $wmi_class.Name 
  Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
  Write-Host "Description:" $wmi_class.Methods.Count
  Write-Host "Methods:"
  foreach ($method in $wmi_class.Methods){
    Write-Host "Name:" $method.Name
    Write-Host "WmiMethodId:" $method.Qualifiers["WmiMethodId"].Value
    Write-Host "Class Name:" $wmi_class.Name 
    Write-Host "Class GUID:" $wmi_class.Qualifiers["guid"].Value
    Write-Host "Description:" $method.Qualifiers["Description"].Value
    Write-Host "Implemented:" $method.Qualifiers["Implemented"].Value
    Write-Host ""
  }
  Write-Host ""
}
```

benoe77, 2023-08-22:
Hi, trying to participate:

Model name: Lenovo Yoga Slim 7-14ARE05
CPU model: AMD Ryzen 7 4700U
GPU model: AMD Radeon RX Vega 7
Keyboard backlight: single color with off/medium/bright)

sudo dmidecode -t system
`# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82A2
	Version: Lenovo Yoga Slim 7-14ARE05
	Serial Number:
	UUID:
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82A2_BU_idea_FM_IDEAPAD
	Family: IDEAPAD

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
`

sudo dmidecode -t bios
`# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: DMCN43WW
	Release Date: 09/16/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.43
	Firmware Revision: 1.29

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
`

sudo cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: no file or directory of this name

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12411514/fwts_wmi.log)
[DSDT.dsl.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/12411522/DSDT.dsl.zip)

johnfanv2, 2023-08-22:
It seems like the kernel module was not loaded. What is the output after you have run these commands
```bash
# Go to folder kernel_module in the directory LenovoLegionLinux
cd LenovoLegionLinux/kernel_module
# Compile
make
# Load but disable platform_profile since there might be conflict between ideapad driver and legion driver for now (will be fixed in future)
# This is the only difference compared to the instructions in the README
# This output is particularly interesting
sudo make forcereloadmodule_disableplatformprofile
sudo cat /sys/kernel/debug/legion/fancurve
```

benoe77, 2023-08-22:
> 

I haven't run these commands yet... :)

benoe77, 2023-08-22:
make:

```
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-26-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  CC [M]  /home/user/sources/LenovoLegionLinux/kernel_module/legion-laptop.o
  MODPOST /home/user/sources/LenovoLegionLinux/kernel_module/Module.symvers
  CC [M]  /home/user/sources/LenovoLegionLinux/kernel_module/legion-laptop.mod.o
  LD [M]  /home/user/sources/LenovoLegionLinux/kernel_module/legion-laptop.ko
  BTF [M] /home/user/sources/LenovoLegionLinux/kernel_module/legion-laptop.ko
Skipping BTF generation for /home/user/sources/LenovoLegionLinux/kernel_module/legion-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-26-generic'
``` 

:(

johnfanv2, 2023-08-22:
No error at all, just a insignificant warning. Feel free to proceed to the next step - everything is on track.

benoe77, 2023-08-23:
After I run 
sudo make forcereloadmodule_disableplatformprofile I get a long dmesg output.

```
sudo cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: No file or directory
```

please find WMI info from Windows attached.
[YogaSlim7_WMI.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12421287/YogaSlim7_WMI.txt)

johnfanv2, 2023-08-23:
> and a long dmesg output.
Sorry, for not mentioning it. The bottom of this output, e.g. last 100 lines, would be helpful since it is the log output of the kernel module. Currently, it looks like the kernel module was not loaded due to some error that we'll find in this log.

benoe77, 2023-08-23:
dmesg:

```
[sze aug 23 18:10:37 2023] legion_laptop: legion_laptop starts loading
[sze aug 23 18:10:37 2023] legion PNP0C09:00: legion_laptop platform driver probing
[sze aug 23 18:10:37 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82A2; DMI_BIOS_VERSION:DMCN43WW
[sze aug 23 18:10:37 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82A2; DMI_BIOS_VERSION:DMCN43WW
[sze aug 23 18:10:37 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[sze aug 23 18:10:37 2023] legion PNP0C09:00: legion_laptop is forced to load.
[sze aug 23 18:10:37 2023] legion PNP0C09:00: legion_laptop is forced to load and would otherwise be not loaded
[sze aug 23 18:10:37 2023] legion PNP0C09:00: Using configuration for system: GKCN
[sze aug 23 18:10:37 2023] legion PNP0C09:00: ACPI CFG: 4227858436
[sze aug 23 18:10:37 2023] legion_laptop: Creating RAM access to embedded controller
[sze aug 23 18:10:37 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x000000008fcc5d01
[sze aug 23 18:10:37 2023] legion PNP0C09:00: Read embedded controller ID 0xffff
[sze aug 23 18:10:37 2023] legion PNP0C09:00: Expected EC chip id 0x8227 but read 0xffff
[sze aug 23 18:10:37 2023] legion_laptop: Unloading legion ecram
[sze aug 23 18:10:37 2023] legion_laptop: Unloading legion ecram done
[sze aug 23 18:10:37 2023] legion_laptop: Unmapping embedded controller memory at 0xfe00d400 (in RAM)/0x0 (in EC) at virtual 0x000000008fcc5d01
[sze aug 23 18:10:37 2023] legion_laptop: Unloading legion shared
[sze aug 23 18:10:37 2023] legion_laptop: Unloading legion shared done
[sze aug 23 18:10:37 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[sze aug 23 18:10:37 2023] legion: probe of PNP0C09:00 failed with error -12
```

johnfanv2, 2023-08-24:
I updated the code. Please download/clone the newest code, compile again (`make`), try to load the module again (`sudo make forcereloadmodule_disableplatformprofile`), and try to read statistics (`sudo cat /sys/kernel/debug/legion/fancurve`).

benoe77, 2023-08-24:
sudo cat /sys/kernel/debug/legion/fancurve

```
EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4227858436
temperature access method: 1
CPU temperature error: 0
CPU temperature: 255
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: -14
CPU temperature WMI: 255
CPU temperature WMI2 error: -14
CPU temperature WMI2: 255
CPU temperature WMI3 error: -14
CPU temperature WMI3: 255
GPU temperature error: 0
GPU temperature: 255
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: -14
GPU temperature WMI: 255
GPU temperature WMI2 error: -14
GPU temperature WMI2: 255
GPU temperature WMI3 error: -14
GPU temperature WMI3: 255
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 65535
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: -14
1 fanspeed WMI: 65535
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 65535
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 65535
2 fanspeed error: 0
2 fanspeed: 65535
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: -14
2 fanspeed WMI: 65535
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 65535
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 65535
powermode access method: 3
powermode error: -14
powermode: 67305474
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: -2147200000
powermode WMI error: -14
powermode WMI: -2147200000
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: -22
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

[YogaSlim7_dmesg.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12431270/YogaSlim7_dmesg.txt)

johnfanv2, 2023-08-24:
Thanks for the output. It seems like it does not work right now and we have to adapt the driver a bit. Could you upload the files from the step "Disassembling ACPI tables"? Then I adapt the driver code and we can try again.

benoe77, 2023-08-24:
Uploaded DSDT.dsl above, isn't that enough?

johnfanv2, 2023-08-24:
Sorry, I missed it. After you edited you comment the link was not clickable anymore.

benoe77, 2023-08-25:
dmesg: 
```
[p aug 25 10:21:18 2023] legion_laptop: legion_laptop starts loading
[p aug 25 10:21:18 2023] legion PNP0C09:00: legion_laptop platform driver probing
[p aug 25 10:21:18 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82A2; DMI_BIOS_VERSION:DMCN43WW
[p aug 25 10:21:18 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82A2; DMI_BIOS_VERSION:DMCN43WW
[p aug 25 10:21:18 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[p aug 25 10:21:18 2023] legion PNP0C09:00: legion_laptop is forced to load.
[p aug 25 10:21:18 2023] legion PNP0C09:00: Using configuration for system: DMCN
[p aug 25 10:21:18 2023] legion PNP0C09:00: Skipping ACPI _STA check
[p aug 25 10:21:18 2023] legion_laptop: Creating RAM access to embedded controller
[p aug 25 10:21:18 2023] legion_laptop: Succeffuly mapped embedded controller: 0xfe700d00 (in RAM)/0x0 (in EC) to virtual 0x000000008ecdca97
[p aug 25 10:21:18 2023] legion PNP0C09:00: Read embedded controller ID 0xffff
[p aug 25 10:21:18 2023] legion PNP0C09:00: Skipped checking embedded controller id
[p aug 25 10:21:18 2023] legion PNP0C09:00: Creating debugfs inteface
[p aug 25 10:21:18 2023] legion_laptop: Creating sysfs inteface
[p aug 25 10:21:18 2023] legion_laptop: Creating hwmon interface
[p aug 25 10:21:18 2023] legion_laptop: Creating platform profile support
[p aug 25 10:21:18 2023] legion_laptop: Skipping creating platform profile support because enable_platformprofile is false
[p aug 25 10:21:18 2023] legion_laptop: Init WMI driver support
[p aug 25 10:21:18 2023] legion_laptop: Init keyboard backlight LED driver
[p aug 25 10:21:18 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[p aug 25 10:21:18 2023] legion_laptop: Error for WMI method call to get brightness
[p aug 25 10:21:18 2023] legion_laptop: Error reading keyboard brighntess
[p aug 25 10:21:18 2023] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[p aug 25 10:21:18 2023] legion_laptop: Init Y-Logo LED driver
[p aug 25 10:21:18 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[p aug 25 10:21:18 2023] legion_laptop: Error for WMI method call to get brightness
[p aug 25 10:21:18 2023] legion_laptop: Error reading brighntess for light: 3
[p aug 25 10:21:18 2023] legion PNP0C09:00: Init Y-Logo LED driver failed. Skipping ...
[p aug 25 10:21:18 2023] legion_laptop: Init IO-Port LED driver
[p aug 25 10:21:18 2023] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[p aug 25 10:21:18 2023] legion_laptop: Error for WMI method call to get brightness
[p aug 25 10:21:18 2023] legion_laptop: Error reading brighntess for light: 5
[p aug 25 10:21:18 2023] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[p aug 25 10:21:18 2023] legion PNP0C09:00: legion_laptop loaded for this device
```

sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
```
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000300
```

sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
```
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000080  04 c8 90 01 2e f5 d0 02  00 04 b1 04 00 2c 41 06  |.............,A.|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  0d d0 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000d0  00 01 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000600
```

johnfanv2, 2023-08-26:
Sorry, for all the effort. It seems like this specific is not compatible at all. The driver works on other Yoga models, but apparently not on this one as far as I can see.

benoe77, 2023-08-28:
:(